AllStarLink/app_rpt #336: rpt.conf "connpgm" and "discpgm" not working

Allan-N, 2024-07-01:
As reported on Community :

_One thing I ran into on the otherwise very smooth setup is the connpgm (and discpgm) programs. However I put these options into the rpt.conf file, template section or general part I cannot seem to get this to work. Even just a simple touch of a file in /tmp doesn’t seem to get executed. Have I missed something in the extensive (again kudo’s) documentation about changes to this option for ASL3?_

Sure enough, I put together a simple pair of shell scripts that logged it's arguments, updated my "rpt.conf" file to exec these scripts, restarted asterisk, connected/disconnected, and the scripts were not executed.  At first glance, it "looks" like we are not parsing the variables.

Docs @ https://wiki.allstarlink.org/wiki/Conpgm_%26_discpgm_-_Run_a_script_or_program_on_node_connect_or_disconnect

KB4MDD, 2024-07-01:
Did the cli show any errors?  (or enable debugging on app_rpt and see what is reported)

Allan-N, 2024-07-01:
I did not enable debug but no errors are reported.  But, looking at the sources I don't see where we even look for the variable in the config.  Maybe just an oversight when the code was updated?

KB4MDD, 2024-07-01:
I have these settings on my node:

connpgm = /usr/local/sbin/smlogger 1     ; Default: Disabled.
discpgm = /usr/local/sbin/smlogger 0 ; Default: Disabled.

My node is logging information.  (My asterisk instance is running as root)

Allan-N, 2024-07-01:
Are you certain that `smlogger` is logging on your system?

From the top of the app_rpt sources, exec :
```
find . -type f -exec grep "connpgm" {} /dev/null \;
```

KB4MDD, 2024-07-01:
Yes - I double checked.

 find . -type f -exec grep "connpgm" {} /dev/null \;
./apps/app_rpt.c:       if (!myrpt->p.connpgm)
./apps/app_rpt.c:       i = strlen(them) + strlen(myrpt->p.connpgm) + +100;
./apps/app_rpt.c:       sprintf(a, "%s %s %s &", myrpt->p.connpgm, myrpt->name, them);

https://github.com/search?q=repo%3AAllStarLink%2Fapp_rpt%20connpgm&type=code

Allan-N, 2024-07-01:
Right.  That's the code that would exec the programs.  What's missing is code that sets "p.connpgm".

KB4MDD, 2024-07-01:
connpgm was missing.  I see that discpgm was present - that is probably why my log file was being udpated.

Pull request being issued.

jxmx, 2024-07-06:
This will be included in app_rpt 3.0.1

PR #337 (2024-07-06): 'connpgm' is not being loaded from rpt.config
The connpgm setting is not being loaded from rpt.conf.

This closes #336.
commit 5db69c69 (2024-07-06): 'connpgm' is not being loaded from rpt.config (#337)

The connpgm setting is not being loaded from rpt.conf.

This closes #336.
closed 2024-07-06 by jxmx