AllStarLink/app_rpt #338: With outstreamcmd= set, shutdown/restart of Asterisk is not reliable

jxmx, 2024-07-02:
With the "broadcastify" setup as documented with `outstreamcmd`, shutting down or restarting asterisk with `systemctl` is not reliable. It's either very slow or fails. Asterisk/app_rpt should be killing off any spawned processes at shutdown.

jxmx, 2024-07-07:
The solution here is probably not to change app_rpt, but to make a monolithic item for app_rpt to kill on shutdown.

InterLinked1, 2024-07-08:
> but to make a monolithic item for app_rpt to kill on shutdown.

What do you really mean by this, exactly?

> The solution here is probably not to change app_rpt, 

All Asterisk modules are expected to clean up anything they started. If `app_rpt` is not fully doing so when requested to unload, then it is a bug in my opinion.

jxmx, 2024-07-08:
Could be. I haven't really spent time debugging yet. What APPEARS to be happening is that Asterisk/app_rpt is killing off the parent bash/sh process but then lame is re-parenting it's PID in a way that systemd is detecting as being part of the Asterisk group of PIDs but not to Asterisk/app_rpt. I was just thinking that a quick-and-easy fix might be to make sure that the Broadcastify process is a single-shot application that talks over a socket or something to the item that actually doing the lame/ezstream process.

pcpackrat, 2024-09-22:
I noticed that when using outstreamcmd the CPU pegs at 100% when astres.sh is issued.  I stopped using it for now.  

Also the huge delay is quite normal for broadcastify style streaming.  Even when using a local streaming server.

commit af9593ca (2025-01-12): app_rpt: Tighten logic for stopping outstreamcmd.

* Add warnings if outstreamcmd fails to start.
* Fix handling if fork() fails.
* Use ast_safe_fork() instead of fork().
* Close all file descriptors in child before exec.
* Refactor kill(outstreamcmd) into helper function.
* If outstreamcmd fails to execute, don't continously try to restart
  it in a loop. Instead, log an error and give up.

Resolves: #338
commit ea1fe44c (2025-01-15): app_rpt: Tighten logic for stopping outstreamcmd.

* Add warnings if outstreamcmd fails to start.
* Fix handling if fork() fails.
* Use ast_safe_fork() instead of fork().
* Close all file descriptors in child before exec.
* Refactor kill(outstreamcmd) into helper function.
* If outstreamcmd fails to execute, don't continously try to restart
  it in a loop. Instead, log an error and give up.

Resolves: #338
commit f0bf1aad (2025-01-15): app_rpt: Tighten logic for stopping outstreamcmd.

* Add warnings if outstreamcmd fails to start.
* Fix handling if fork() fails.
* Use ast_safe_fork() instead of fork().
* Close all file descriptors in child before exec.
* Refactor kill(outstreamcmd) into helper function.
* If outstreamcmd fails to execute, don't continously try to restart
  it in a loop. Instead, log an error and give up.

Resolves: #338
PR #452 (2025-01-17): app_rpt: Tighten logic for stopping outstreamcmd.
* Add warnings if outstreamcmd fails to start.
* Fix handling if fork() fails.
* Use ast_safe_fork() instead of fork().
* Close all file descriptors in child before exec.
* Refactor kill(outstreamcmd) into helper function.
* If outstreamcmd fails to execute, don't continously try to restart it in a loop. Instead, log an error and give up.

Resolves: #338
closed 2025-01-17 by Allan-N