AllStarLink/app_rpt #344: chan_simpleusb: Replace time() with clock_getting()

jxmx, 2024-07-03:
Based on https://github.com/AllStarLink/ASL3/issues/76, the boot-time clock skew seems to cause chan_simpleusb to lock the PTT open until a restart of asterisk. Delaying asterisk start on boot until NTP has synchronized seems to resolve the issue but that is not a reliable fix. Within the code, comparisons are being made using `time(2)` when the likely resolution is to use `clock_gettime(2)` with a `CLOCK_MONOTONIC` type. That way, "wallclock" time changes in the OS don't affect the USB channel driver.

Allan-N, 2024-07-03:
My suggestion would be to add an rpt_time() function that behaves like time() but is based off of clock_gettime().  Then search/replace ALL calls to time() with rpt_time().

mpannen1979, 2024-07-04:
I have a pi3 and a pi4 with USB interfaces that seem to be affected by this--there is USB HID error messages.

I am trying the 'wait for time sync' change to the asterisk.service file in the meantime

PR #346 (2024-07-06): chan_simpleusb/chan_usbradio: Use monotonic clock
This changes chan_simpleusb and chan_usbradio to use the monotonic clock.  This will address timing issues when NTP updates the internal clock.

This closes #344
commit 34681d36 (2024-07-06): chan_simpleusb/chan_usbradio: Use monotonic clock (#346)

This changes chan_simpleusb and chan_usbradio to use the monotonic clock.  This will address timing issues when NTP updates the internal clock.

This closes #344
closed 2024-07-06 by jxmx