AllStarLink/app_rpt #345: ASL 2.0.0 Beta 6 to ASL 3.0 Upgrade Question

NU5D, 2024-07-04:
First and foremost THANK YOU All for the hard work and diligence that went into and continues with ASL 3. GOD SPEED.

My question.

I have a working repeater/node. The particulars - GE Mastr III - C-Media CM108 sound card - using a Wyse Dell 3040 (8gb) running ASL 2.0.0 beta 6 and all is well. USB Radio - the dsp handles squelch and ctcss encode and decode. I have discriminator (vol/sq hi) for receive, LOCAL PTT and HSD (GE’s High Speed Data - direct modulation input used with EDACS, DMR, etc) and GND.

Using the CLI Setup on 2.0.0 for USB radio there was plenty of digital pot range for my sound card config. I honestly cannot remember if I had 20 db boost enabled or not - but whichever, there was plenty of range for both send and receive. Also this is a test bed repeater - not at a site so I can tinker at will. I am using an HP8921 to look at the output and generate the input signals.

Now for the meat and potatoes - I installed Deb 12 latest and greatest on the Wyse Dell 3040 - Net Install - web server and ssh - had to add net-tools and followed the add repo / install recommendations - really nice setup.

I no longer have sufficient range in the Transmit direction - the #6 TX Voice is at 999 to get just at +/- 3 kHz transmitter deviation and the CTCSS is at 200 for +/- 750 ctcss (88.5)… and works just great !!! My question / concern - having the TX level set at max and the TX boost seems to make no difference -

This is not a gripe or complaint - just an observation and wondering if I may be missing something in my setup. Again Great Work and thanks again,  steve nu5d

rpt.conf and usbradio.conf - the site would not allow attaching .conf files..

rpt dot conf

;*********************************************
;********   Template-tized rpt.conf   ********
;*********************************************
; Note to editors: set tabs to 4 space characters. No wrap to keep comments neat.
; vim: tabstop=4
;
; Radio Repeater configuration file for use with app_rpt ASL3
;

[general]
node_lookup_method = both           ;method used to lookup nodes
                                    ;both = dns lookup first, followed by external file (default)
                                    ;dns = dns lookup only
                                    ;file = external file lookup only

[nodes]
; If you are using automatic update for allstar link nodes, and you probably are,
; no allstar remote nodes should be defined here. Only place a definition
; for your local nodes, and private (not on allstar directory) nodes here.

; Sample nodes on this server. These are required. Use IP 127.0.0.1
; IAX port number if not the default 4569.
;1998 = radio@127.0.0.1/1998,NONE

; Sample public or private node behind your router.
;1001 = radio@192.168.1.8:4568/1001,NONE

; Sample private node on the internet.
;1000 = radio@8.8.8.8:4569/1000,NONE

559827 = radio@127.0.0.1/559827,NONE

[node-main](!)
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
;;;;; Template for all your nodes ;;;;;
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
; Set the defaults for your node(s) here.
; Add your nodes below the line that says
; Add you nodes below.

;;; rxchannel ;;;
; Must also be enabled in modules.conf
; Enable the selected channel driver in modules.conf !!!
; Rx/Tx audio/signalling channel. Choose ONLY 1 per node stanza.
; rxchannel = dahdi/pseudo          ; No radio (hub)
; rxchannel = SimpleUSB/1999        ; SimpleUSB
; rxchannel = Radio/1999            ; USBRadio (DSP)
; rxchannel = Voter/1999            ; RTCM device
; rxchannel = Pi/1                  ; Raspberry Pi PiTA
; rxchannel = Dahdi/1               ; PCI Quad card
; rxchannel = Beagle/1              ; BeagleBoard
; rxchannel = USRP/127.0.0.1:34001:32001    ;GNU Radio interface USRP
[... 765 lines cut ...]
;       place settings that are different than the template.
;
; Note: the device string is automatically found when the
;       USB setting "devstr=" is empty.
;
; Note: the interface "tune" settings will be added to the
;       per-node settings (below).

[559827](node-main)

txboost = yes

;;;;; ASL3 Tune settings ;;;;;
devstr = 1-1:1.0
rxmixerset = 875
txmixaset = 999
txmixbset = 990
rxvoiceadj = 1.000000
rxctcssadj = 0.908583
txctcssadj = 200
rxsquelchadj = 851

These are the new Version 3 files - I will have to do some digging to get the version 2.0.0 beta 6 files.

steve nu5d

tsawyer, 2024-07-04:
@KB4MDD This might be a question you can address. Please take a look.

KB4MDD, 2024-07-04:
I will take a look.

NU5D, 2024-07-04:
I made a mistake.  Please forgive my error - This morning I set up a 2nd Wyse Dell 3040 - with ASL 2.0.0 Beta 6, and I had the very same results.  There is nothing wrong with Rev 3.  Rev 2.0.0 required the same settings.   TX 990 for almost 3 kHz Transmitter deviation.  There is no issue.  Thank all for the very prompt response and again I am very sorry for the false alarm.  73, Steve NU5D..

KB4MDD, 2024-07-04:
Closing no action needed.

closed 2024-07-04 by KB4MDD