AllStarLink/app_rpt #347: Cannot connect to EchoLink *ECHOTEST node (9999)

Allan-N, 2024-07-06:
Attempts to connect to the EchoLink *ECHOTEST* (9999) are failing with a **Node not found in database** error.

```
wa3wco-shari*CLI> rpt fun 491302 *33009999
    -- Call for node el0/009999 on echolink/el0-4, failed. Node not found in database.
    -- Hungup 'DAHDI/pseudo-775952880'
    -- Hungup 'DAHDI/pseudo-827054318'
       > Received OK from Echolink server nasouth.echolink.org.
       > Received OK from Echolink server naeast.echolink.org.
       > Directory completed downloading(partial,compressed), 961 records.
    -- Hungup 'DAHDI/pseudo-855280375'
    -- <DAHDI/pseudo-237703529> Playing 'rpt/node.gsm' (language 'en')
    -- <DAHDI/pseudo-237703529> Playing 'digits/3.ulaw' (language 'en')
    -- <DAHDI/pseudo-237703529> Playing 'digits/0.ulaw' (language 'en')
    -- <DAHDI/pseudo-237703529> Playing 'digits/0.ulaw' (language 'en')
    -- <DAHDI/pseudo-237703529> Playing 'digits/9.ulaw' (language 'en')
    -- <DAHDI/pseudo-237703529> Playing 'digits/9.ulaw' (language 'en')
    -- <DAHDI/pseudo-237703529> Playing 'digits/9.ulaw' (language 'en')
    -- <DAHDI/pseudo-237703529> Playing 'digits/9.ulaw' (language 'en')
    -- <DAHDI/pseudo-237703529> Playing 'rpt/connection_failed.gsm' (language 'en')
    -- Hungup 'DAHDI/pseudo-237703529'
```

This works :
```
wa3wco-shari*CLI> echolink dbget n 9999
9999|*ECHOTEST*|54.176.61.77
```

This does NOT work :
```
wa3wco-shari*CLI> echolink dbget n 009999
Error: Entry for 009999 not found!
Command 'echolink dbget n 009999' failed.
```

and :
```
wa3wco@wa3wco-shari:~ $ sudo asterisk -x 'echolink dbdump' > /tmp/echodb
wa3wco@wa3wco-shari:~ $ grep 9999 /tmp/echodb
899993|SK6RIC-R|70.34.202.109
9999|*ECHOTEST*|54.176.61.77
999927|IZ8WMQ-L|128.235.36.228
999973|W8LGX-L|50.41.31.177
```

Most importantly, connecting to a 6-digit node (e.g. *ARRLNEWS*, #826983) works !

Allan-N, 2024-07-06:
This will be included in app_rpt 3.0.1

closed 2024-07-06 by Allan-N

k2cb, 2025-03-10:
I am still experiencing this issue with a new install, using app_rpt 3.2.0

Connected to Asterisk 20.11.0+asl3-3.2.0-2.deb12 currently running on node61653 (pid = 769)
node61653*CLI> echolink dbget n 021005
Error: Entry for 021005 not found!
Command 'echolink dbget n 021005' failed.
node61653*CLI> echolink dbget n 330926
330926|WA2JWR-L|71.58.32.179
node61653*CLI> rpt fun 61653 *33021005
    -- Hungup 'DAHDI/pseudo-410977062'
    -- Hungup 'DAHDI/pseudo-932922742'
    -- Hungup 'DAHDI/pseudo-105772099'
    -- <DAHDI/pseudo-1968087063> Playing 'rpt/node.gsm' (language 'en')
    -- <DAHDI/pseudo-1968087063> Playing 'digits/3.ulaw' (language 'en')
    -- <DAHDI/pseudo-1968087063> Playing 'digits/0.ulaw' (language 'en')
    -- <DAHDI/pseudo-1968087063> Playing 'digits/2.ulaw' (language 'en')
    -- <DAHDI/pseudo-1968087063> Playing 'digits/1.ulaw' (language 'en')
    -- <DAHDI/pseudo-1968087063> Playing 'digits/0.ulaw' (language 'en')
    -- <DAHDI/pseudo-1968087063> Playing 'digits/0.ulaw' (language 'en')
    -- <DAHDI/pseudo-1968087063> Playing 'digits/5.ulaw' (language 'en')
    -- <DAHDI/pseudo-1968087063> Playing 'rpt/connection_failed.gsm' (language 'en')
    -- Hungup 'DAHDI/pseudo-1968087063'
node61653*CLI>

Any suggestions?

Allan-N, 2025-03-10:
Use `echolink dbget n 21005` (without the leading zeros)

PR #348 (2024-07-07): chan_echolink: Remove leadings zeros from node number
This changes chan_echolink to remove leading zeros in the node number prior to validation in the internal database.

app_rpt requires node numbers to be 6 digits so we will receive some numbers padded with leading zeros.

This closes #347
commit d7ffe113 (2024-07-07): chan_echolink: Remove leadings zeros from node number (#348)

This changes chan_echolink to remove leading zeros in the node number prior to validation in the internal database.

app_rpt requires node numbers to be 6 digits so we will receive some numbers padded with leading zeros.

This closes #347