AllStarLink/app_rpt #359: Add C108AH variant to app_rpt

W0REW, 2024-07-15:
I have been using AllStarLink for over two years on our clubs repeater.  I am now exploring updating to ASL3.   In the process of building the repeater I compiled ASL2 on Bullseye and added a define and code changes to use a  CMedia Unitek Y-247A to build radio-less nodes using an Mpow 071 usb headset.  I have built several of these for club members.

I would like to make similar modifications to ASL3 so I can build a test network but haven't figured out how to compile modified code. Is there a way for me to compile the new allstarlink software containing code changes from scratch?   The changes seem to be simple (famous last words) in ASL3, but were more involved in ASL2.  

./include/asterisk/res_usbradio.h:55:               #define C108AH_MPOW_ID  	0x0014
./channels/chan_usbradio.c:1267:			if (o->devtype == C108AH_PRODUCT_ID || o->devtype == C108AH_MPOW_ID) {
./channels/chan_simpleusb.c:1214:		if (o->devtype == C108AH_PRODUCT_ID || o->devtype == C108AH_MPOW_ID) {
 
If there is no way for me to do it would you consider an enhancement request.  

Thanks,
Roy Wood
W0REW

InterLinked1, 2024-07-15:
I'm assuming you installed from source rather than packages. You should certainly be able to compile from source, are you running into an issue when trying to do so?

If you already have an installation from source and have made a change to certain Asterisk modules, you just be able to do `make && make install`.

Allan-N, 2024-07-15:
Two questions and one bit of information :
1. Can you paste the output from the `lsusb` command when you have one of the Y-247A dongles plugged in (just to confirm both the USB vendorID and productID).
2. You write that you are "exploring" using the Y-247A dongles.  Were you successful?
3. Information on building ASL3's asterisk+app_rpt from source is available at https://allstarlink.github.io/developers/source-install/

W0REW, 2024-07-15:
1)
w0ef@pi4-asl3head:~/workspace/appt_rpt_w0rew $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 0d8c:0014 C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A)
Bus 001 Device 004: ID 413c:2514 Dell Computer Corp. Dell Universal Receiver
Bus 001 Device 003: ID 0d8c:0014 C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A)
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

One Unitek device is the MPow headset, the other is a UGreen sound dongle which is used on a radio-less node to monitor the repeater.

2)
I have the devices working on ASL2.  The Mpow headset is connected to a Raspberry Pi configured as a SimpleUsb node.  Pressing the minus key on the headset control keys the node.  I am "exploring" moving our nodes to ASL3.

3)
I see where I was confused.  I installed ASL3 using apt, then cloned the software from GitHub.  I missed the part where the change to the master branch was commented out in rpt_install.sh.  My excuse this week is I just had minor surgery.

Thanks for the help...
Roy

Allan-N, 2024-07-15:
Earlier, you noted changes to :
- ./include/asterisk/res_usbradio.h
- ./channels/chan_usbradio.c
- ./channels/chan_simpleusb.c

In ASL3, don't forget :
- ./res/res_usbradio.c

Since you are still "exploring" validating the audio interfaces on ASL3 and because you're willing to modify/build from source to test things out I'd suggest that we let you continue your efforts.  Once all is working and you're happy then please let us know (here) of any changes you needed to make and we'll look to pull them in.  Also, please let us know if you have any questions/issues.

qoph1, 2024-10-13:
I have tested with the changes suggested by W0REW, as well as res/res_usbradio.c as pointed out by Allan-N. I also added the product ID to asl-find-sound to auto populate devstr. Seems to be working. 

Add define for the Y247A product ID 0x0014. 
include/asterisk/res_usbradio.h:54: #define Y247A_PRODUCT_ID  	0x0014

channels/chan_usbradio.c:1249: if (o->devtype == C108AH_PRODUCT_ID || o->devtype == Y247A_PRODUCT_ID) {

channels/chan_simpleusb.c:1195: if (o->devtype == C108AH_PRODUCT_ID || o->devtype ==Y247A_PRODUCT_ID) {

res/res_usbradio.c:322:if ((dev->descriptor.idVendor
				 == C108_VENDOR_ID) &&
				(((dev->descriptor.idProduct & 0xfffc) == C108_PRODUCT_ID) ||
				 (dev->descriptor.idProduct == C108B_PRODUCT_ID) ||
				 (dev->descriptor.idProduct == Y247A_PRODUCT_ID) ||

/usr/bin/asl-find-sound:29:elif [ $((16#$IDPROD          )) -eq $((16#0014)) ]; then	# Y-247A_PRODUCT_ID
			:

commit 34fd2e83 (2024-11-27): res_usbradio: Add ability to specify allowed vid:pid

This updates res_usbradio to allow users to specify USB device descriptor(s) that can be used for interface devices. There are compatible chips that are not supported in the code. This will allow users to easily add the compatible chip(s).

This adds res_usbradio.conf for configuration. A sample configuration file was added.

Reworked the supported/known USB devices lookup.  The supported devices are now in an array.

This closes #359.
commit 57fca49e (2024-12-06): res_usbradio: Add ability to specify allowed vid:pid

This updates res_usbradio to allow users to specify USB device descriptor(s) that can be used for interface devices. There are compatible chips that are not supported in the code. This will allow users to easily add the compatible chip(s).

This adds res_usbradio.conf for configuration. A sample configuration file was added.

Reworked the supported/known USB devices lookup. The supported devices are now in an array.

This closes #359.
PR #428 (2024-12-16): res_usbradio: Add ability to specify allowed vid:pid
This updates res_usbradio to allow users to specify USB device descriptor(s) that can be used for interface devices. There are compatible chips that are not supported in the code. This will allow users to easily add the compatible chip(s).

This adds res_usbradio.conf for configuration. A sample configuration file was added.

Reworked the supported/known USB devices lookup.  The supported devices are now in an array.

This closes #359.
commit c048cb2e (2024-12-16): res_usbradio: Add ability to specify allowed vid:pid (#428)

This updates res_usbradio to allow users to specify USB device descriptor(s) that can be used for interface devices.  There are compatible chips that are not supported in the code.  This will allow users to easily add the compatible chip(s).

This adds res_usbradio.conf for configuration.  A sample configuration file was also added.

This closes #359.
closed 2024-12-16 by Allan-N