AllStarLink/app_rpt #360: Correct ASL DNS Node Address Resolution

tsawyer, 2024-07-15:
Change ASL to use core Asterisk DNS services to resolve ASL node numbers, remote base node numbers and future ASL services.

My limited understanding of how this done is to query the SVR record first to find the A record. Apparently TXT records are for debugging and should not be used for ASL address resolution. 

An example query of a remote base node is shown below. Also shown is a link to the ASL DNS specification and a blog post about PJSIP DNS.

https://wiki.allstarlink.org/wiki/DNS_Servers
https://www.asterisk.org/pjsip-dns-support/

```
./asl-node-lookup 53431

A (53431.nodes.allstarlink.org)
  Host 53431.nodes.allstarlink.org not found: 2(SERVFAIL)

A (53431.remotebase.nodes.allstarlink.org)
  75.138.7.130

SRV (_iax._udp.53431.nodes.allstarlink.org)
  11 10 4569 53431.remotebase.nodes.allstarlink.org.

TXT (53431.nodes.allstarlink.org)
  NN=53431
    RT=2024-07-15 02:50:10
    RB=1
    IP=75.138.7.130
    PIP=0
    PT=4569
    RH=register-central
```

Allan-N, 2024-07-15:
$0.02 : using the SRV record to get the hostname and IAX port makes sense, and will likely help with IPv6 support, but it does mean that each lookup will be 2+ DNS queries (one for the SRV record, one for the A record, etc).  Right now, we're only issuing a single TXT query and we get back all the info we need.

KB4MDD, 2024-07-15:
At the time that the DNS code was written, the remote base lookup was broken.  The only option forward was to use the TXT records.  You are correct, it will take two queries to achieve the same results.  It is unfortunate that the TXT records are not working.

I will get some changes together for testing.  It is unfortunate that the TXT records are not working.

Allan-N, 2024-07-15:
> It is unfortunate that the TXT records are not working.

I thought @ajpaul25 just fixed the TXT record issue

tsawyer, 2024-07-15:
I'm not sure he fixed the TXT records or something else, but according to him they are for debugging only.

https://en.wikipedia.org/wiki/TXT_record#:~:text=A%20TXT%20record%20(short%20for,center%2C%20or%20other%20accounting%20information.

@KB4MDD are you looking at https://www.asterisk.org/res_resolver_unbound/

KB4MDD, 2024-07-15:
I had already written the SRV lookup code, but it would not work due to a problem with remote bases.  Because of the problem, I switched to the TXT records.  The SRV lookup code has been added.  I left the original TXT lookup code and disabled it -- just in case it can be used in the future.

The SRV lookup method requires two DNS requests.  The TXT record only required one DNS request.

commit 187def7f (2024-07-15): app_rpt: Update to use DNS SRV record

This updates app_rpt to use DNS SRV records when looking up a node by DNS.

The original code used the TXT record which was labeled for debugging purposes. The original TXT lookup code remains and is switched off for now.

This closes #360
commit a164275d (2024-07-16): app_rpt: Update to use DNS SRV record

This updates app_rpt to use DNS SRV records when looking up a node by DNS.

The original code used the TXT record which was labeled for debugging purposes. The original TXT lookup code remains and is switched off for now.

This closes #360
commit 4c2fd1e1 (2024-07-16): app_rpt: Update to use DNS SRV record

This updates app_rpt to use DNS SRV records when looking up a node by DNS.

The original code used the TXT record which was labeled for debugging purposes. The original TXT lookup code remains and is switched off for now.

This closes #360
PR #361 (2024-07-16): app_rpt: Update to use DNS SRV record
This updates app_rpt to use DNS SRV records when looking up a node by DNS.

The original code used the TXT record which was labeled for debugging purposes.  The original TXT lookup code remains and is switched off for now.

This closes #360
closed 2024-07-16 by Allan-N