AllStarLink/app_rpt #364: Contribution documentation

JimZAH, 2024-07-21:
Firstly, thanks for this much needed release. I'm now running this at home before deploying on our repeater sites.

I need to make some changes to app_rpt before I can deploy, adding a keychunk filter for example. 

I've not had much success building this using the methods in the README.md. 

Do you have any documentation to assist with building for development?

tsawyer, 2024-07-21:
Take a look around here if you've not already https://allstarlink.github.io/developers/source-install/

Note: https://github.com/AllStarLink/app_rpt/issues/90 a PR would be really nice.

JimZAH, 2024-07-21:
@tsawyer thank you. 
I'm not sure how I missed this. These steps look like the ones I tried yesterday. I'll give it another go. 

Thanks for the PR link, I developed a UK style keychunk for the old Allstar. It never got merged sadly but has been working well for a long time. 

If I can get the dev environment working, I'll work on a keychunk filter. It's likely to be a UK style keychunk but I can't see a reason we couldn't have an option to switch between the two different types.

JimZAH, 2024-07-21:
@tsawyer Thank you for your guidance.

closed 2024-07-21 by JimZAH