AllStarLink/app_rpt #367: Allow faster morse

JimZAH, 2024-07-21:
There is mention in the rpt.conf file of faster morse speed:
```
[morse]
;;;;; Mores code ;;;;;
; Morse code parameters, these are common to all repeaters.
speed = 25                          ; Approximate speed in WPM
frequency = 800                     ; Morse Telemetry Frequency
amplitude = 4096                    ; Morse Telemetry Amplitude
idfrequency = 1750                  ; Morse ID Frequency
idamplitude = 2048                  ; Morse ID Amplitude

;[morse]
;speed = 35                         ; Not leagle in the USA
;frequency = 800                    ; Morse Telemetry Frequency *Changed
;amplitude = 4096                   ; Morse Telemetry Amplitude
;idfrequency = 750                  ; Morse ID Frequency        *Changed
;idamplitude = 512                  ; Morse ID Amplitude        *Changed
```

However the max limit is set to 20wpm.

tsawyer, 2024-07-21:
Duplicate of #368

closed 2024-07-21 by tsawyer

Allan-N, 2024-08-02:
The fix for this issue has been incorporated into ASL3-Asterisk 20.9.1+asl3-3.0.4-1