AllStarLink/app_rpt #371: POCSAG and usbradio makes PTT stuck ON

eliggett, 2024-07-24:
Greetings, 

On my system I am using the usbradio channel driver with a simplex radio node. Not realizing that usbradio doesn't have POCSAG support, I issued the following test command, which caused the radio to hang keyed until asterisk was killed. 

simplex900*CLI> rpt page 617081 600 123 A "test 123 de W6EL"
[2024-07-23 17:41:54.442] ERROR[288797]: chan_usbradio.c:1851 usbradio_text: Channel 617081: Cannot parse usbradio cmd: PAGE 600 123 A  test 123 de W6EL
simplex900*CLI> radio unkey
simplex900*CLI>

There are really three requests here. 

1: Don't route POCSAG to chan_usbradio since it does not support paging yet. This can possibly be prevented [here, in rpt_cli.c](https://github.com/AllStarLink/app_rpt/blob/83061dbe00bfd057e238b586e506dbf4dd679c16/apps/app_rpt/rpt_cli.c#L869). 

2: Modify the error handling from chan_usbradio so that if it receives a text command that it doesn't understand, something makes sure the channel driver isn't keyed up. I don't really understand why it keys up to begin with but this is just my own unfamiliarity with the code. Adding this level of fault protection may end up helping other use cases too. Ie, if we don't understand the text command, we report this, and, if something told us to key up, that thing is told to stop keying. 

3: Eventually, add POCSAG to chan_usbradio, because it's so cool. With my 900 MHz node, I am tempted to get some of those inexpensive pagers and try it out. 

Thank you all for your hard work on ASL, I love it!!

--E
de W6EL

KB4MDD, 2024-07-29:
I have confirmed this issue on my install.

```
k4mdddevelop*CLI> rpt page 2231 600 123 T test
[2024-07-29 09:41:41.706] DEBUG[88692]: channel.c:4799 ast_sendtext_data: Thread LWP 88692 is blocking 'Radio/usb_2231', already blocked by thread LWP 88688 in procedure ast_waitfor_nandfds
[2024-07-29 09:41:41.706] ERROR[88692]: chan_usbradio.c:1830 usbradio_text: Channel usb_2231: Cannot parse usbradio text: PAGE 600 123 T  test
```
My device was also hung in transmit.

KB4MDD, 2024-08-06:
This is caused by the incomplete implementation of paging in chan_usbradio.  chan_usbradio is not sending an end of page text message to drop out of transmit.  

I will remove chan_usbradio from paging until this can be fully implemented.

KB4MDD, 2024-11-25:
This has been resolved.

closed 2024-11-25 by KB4MDD