AllStarLink/app_rpt #382: monitor (record/archive) function not working as it used to

kenalker, 2024-08-10:
**Describe the bug**
I am using the “monitor” function to record repeater audio.  We have one node that mixes audio from three RTCMs plus anyone attaching to the node via Allstar.  Audio is recorded to a WAV file (date_and_time_stamp.WAV as filename) each time someone keys up until they unkey (one file per key-up), as expected.  With the “previous” (very old) version of Allstar, every file contained audio, as would be expected.  With ASL3 I’m finding that most, but not all, files are 60 bytes in length and contain no audio. The only files that are a sensible size and have audio appear to be from people who are connected to the node via Allstar.  Also, every now and then we get a file that is recorded from someone coming in from an RTCM that sounds like some kind of techo music (it’s very musically interesting, actually - maybe an incorrect CODEC or something), otherwise its just a 60 byte file with no audio content.

I reported this problem on the forum [here](https://community.allstarlink.org/t/monitor-record-function-not-working-properly-in-asl3/21510), but there isn't anything there that isn't in this report.  Also, the same issue is posted [here](https://community.allstarlink.org/t/audio-archiving-still-the-same-in-asl3/21331/4) which includes a sample file from Andriy_Grin of the "techno music" I referenced above, if you want to hear something fun and interesting (but no intelligible audio).

**To Reproduce**
Steps to reproduce the behavior:
1) Add the archivedir command to your node stanza in rpt.conf.  Ensure directory where wave files will be written has proper permissions (ASL3 is not setting permissions to directory correctly when directory is created; I'll report this separately).  Here is a snippet of rpt.com:
`[1900](node-main)`<br>`
archivedir = /var/spool/asterisk/monitor
`
2) Key up your repeater and speak.
3) Play back the WAV file created in the /var/spool/asterisk/monitor/##### directory (where ##### is your actual node number)

**Expected behavior**
I expect to have a WAV (*.wav) file recorded that I can then play back just as would happen in the older version of ASL.

**Software versions (listed in asl-menu, option 4)**
OS: Debian GNU/Linux 12 (bookworm)
OS Kernel: 6.1.0-23-amd64
Asterisk: 20.8.1+asl3-3.0.2-1.deb12
ASL [app_rpt] : 3.0.1
Packages:
allmon3               1.2.1-2
asl3                  3.2.0-1.deb
asl3-asterisk         2:20.8.1+asl3-3.0.2-1.deb12
asl3-asterisk-config  2:20.8.1+asl3-3.0.2-1.deb12
asl3-asterisk-dev     2:20.8.1+asl3-3.0.2-1.deb12
asl3-asterisk-doc     2:20.8.1+asl3-3.0.2-1.deb12
Our previous system that worked flawlessly was many years old and I am unsure how to determine the ASL version but Asterisk was "Asterisk 1537 built by root @ repeater on a i686 running Linux on 2016-09-03 21:38:23 UT".

**Have you run a software update and rebooted?**
Yes.

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
Beelink SER5 Mini PC with AMD Ryzen 5 5560U 6 core (up to 4.0GHz) 16GB DDR4 RAM, 500GB NVMe M.2 SSD, RX Vega7

wd5m, 2025-01-26:
> I just bumped into [this](https://www.asterisk.org/monitor-is-deprecated-farewell/) article that states, "The Monitor dialplan application and module in Asterisk ... has since been replaced by the MixMonitor dialplan application and module and will be removed as of Asterisk 21". How does this apply to ASL3? Should I be using mixmonitor.so instead of monitor.so?

If "monitor" is going away, I hope the "archivedir" method can be updated to use mixmonitor, or something suitable. The reasons for moving to mixmonitor, described in the linked posting, seem prudent. In the meantime, I implemented a method for ASL3 to record audio using mixmonitor commands with a script that gets called from a node event when the number of connections changes from zero (0).  So, while this alternate script method is possible, it is not nearly as simple as using "archivedir".  The "archivedir" method also logs connection activity info in a txt file, which I have found quite useful over time.

David WD5M

mkmer, 2025-01-27:
Our local repeater also has been using this function for years, and your note made me wonder if ours was broken too.
I can report our wav files sound fine and are of varying sizes.  Since there were a LOT of things to workout when moving to ASL3, maybe something is missing in your config?  I'm not sure the best way to compare configs (or what one may matter in this case).
Our codec section looks like this (maybe the first place to start?):
```
load => codec_adpcm.so              ; Adaptive Differential PCM Coder/Decoder
load => codec_alaw.so               ; A-law Coder/Decoder
load => codec_a_mu.so               ; A-law and Mulaw direct Coder/Decoder
noload => codec_dahdi.so            ; Generic DAHDI Transcoder Codec Translato
load => codec_g726.so               ; ITU G.726-32kbps G726 Transcoder
load => codec_gsm.so                ; GSM Coder/Decoder
load => codec_ulaw.so               ; mu-Law Coder/Decoder
noload => codec_ilbc.so             ; http://en.wikipedia.org/wiki/Internet_Low_Bitrate_Codec
```

wd5m, 2025-01-27:
> Our local repeater also has been using this function for years, and your note made me wonder if ours was broken too. I can report our wav files sound fine and are of varying sizes. Since there were [...]

Are you using simpleusb? This issue may be related to using the simpleusb radio interface. My limited testing seems to get good recordings on an ASL3 HUB VSP VM node without chan_simpleusb loaded. On the other hand, my Raspberry PI4 ASL3 node, using simpleusb, exhibits the symptoms described for this issue.  Audio from the radio is not recorded and the described warning message is logged.

Using mixmonitor records without issues on both nodes.

David WD5M

kenalker, 2025-01-27:
> If "monitor" is going away, I hope the "archivedir" method can be updated to use mixmonitor, or something suitable. The reasons for moving to mixmonitor, described in the linked posting, seem [...]

David - can you share your script and the call method?

kenalker, 2025-01-27:
> Our local repeater also has been using this function for years, and your note made me wonder if ours was broken too. I can report our wav files sound fine and are of varying sizes. Since there were [...]

Thank you for your feedback!  My codec setup (in modules.conf) is identical to yours, so that may not be the issue.  What exact version of ASL3 are you using?  What channel drivers do you have enabled?  Are you using the "archivedir" command to record?  Do you have just a single node?  How does data flow between your repeater and the ASL3 box (ie. what hardware is in between)? 
 Perhaps shoot me full copies of your configs via email, and I can compare?

kenalker, 2025-01-27:
> Are you using simpleusb? This issue may be related to using the simpleusb radio interface. My limited testing seems to get good recordings on an ASL3 HUB VSP VM node without chan_simpleusb loaded. [...]

I am using chan_voter since I have RTCM's at each repeater site.  I had chan_simpleusb enabled because we used to have a URIx at one site, but we removed the URIx a while back so I unloaded chan_simpleusb (via modules.conf) and restarted Asterisk but that did not fix the problem described in this ticket.

wd5m, 2025-01-27:
> David - can you share your script and the call method?

I call the script via the event segment(s) in rpt.conf.  On my systems running multiple nodes, each node has its own event segment. Here's an example event segment. This is almost a copy of an example from the old Wiki for events.

```
[events614320]
/etc/asterisk/custom/mixmonitor.sh 0 614320 = s|f|RPT_NUMLINKS
/etc/asterisk/custom/mixmonitor.sh 1 614320 = s|t|RPT_NUMLINKS
```

My script is below.  If you don't need the mp3 conversion, that part could be removed and just rename the wav file.  As noted in the script, I'm not sure my choice of using "IAX2" as the channel  argument for mixmonitor is the best choice. I haven't found a quick easy way to get or provide the actual connecting channel. This seems to be working for my use though.

```
#!/bin/bash
# ASL3 broke rpt archivedir so we use mixmonitor.
# https://github.com/AllStarLink/app_rpt/issues/382
# app_mixmonitor must be loaded in modules.conf
# This mixmonitor.sh script starts/stops recording when the number of connections (RPT_NUMLINKS)
# transitions above zero (0) or back to zero.
# There may be an way to do this in extensions.conf too.
#
# I place this script in the /etc/asterisk/custom/ directory with asterisk ownership
# and execute permissions.
#
# These applications are also required:
#   lame
#   soxi
#   libsox-fmt-mp3
#   bc
#
# To run from [events] section in rpt.conf...
#
# /etc/asterisk/custom/mixmonitor.sh 0 <yournodenumber> = s|f|RPT_NUMLINKS
# /etc/asterisk/custom/mixmonitor.sh 1 <yournodenumber> = s|t|RPT_NUMLINKS
#
# The first argument indicates:
#   0 (zero) when RPT_NUMBLINKS changs to 0 (zer) connections, which
#   results in stopping recording, renaming the file and converting to mp3.
#
#   1 (one) when RPT_NUMLINKS changes from from 0 connections, which
#   resutls in starting the recording for the node.
#
# The second argument is the local node number to be recorded.
#
# Be sure to initially create the directory /var/www/html/archive when installing
# this script with ownership of asterisk.
#   sudo mkdir /var/www/html/archive
#   sudo chown asterisk:asterisk /var/www/html/archive
#
# This script will create a subdirectory for the node number specified, similar to archivedir.
#
# 20241122 - D McAnally WD5M
# 20250127 - D McAnally WD5M Add more comments and modify for multi node system.
PGM=`basename $0`
export TERM=dumb
[[ -z "${2}" ]] && exit 100     # Missing node number argument.
NODE="${2}"
ASLRECORDPATH="/var/www/html/archive/${NODE}"
/usr/bin/mkdir -p "${ASLRECORDPATH}"    # make sure the node subdirectory exists.
ASLRECORD="${ASLRECORDPATH}/.aslrecord.wav"

declare -i DEBUG=0      # enables logging if set to 1
LOGFILE="/tmp/${PGM}.log"
function log() {
        DATE=`date '+%a %b %d %T %Z %Y'`
        echo "${DATE} ${@}" >> ${LOGFILE}
        echo "${DATE} ${@}"
}

aslrecordstop() {
        # The aslrecordstop function stops mixmonitor and converts the audio to mp3.
        [ ${DEBUG} -eq 1 ] && log "aslrecordstop ${1} starting"
        [[ -z ${1} ]] && return 100
        [[ ! -f ${1} ]] && return 100
        FLEN=`/usr/bin/soxi -D ${1}`
        FLENI=$(echo "${FLEN}/1" | /usr/bin/bc)
        # ignore audio recording less than 5 seconds.
        if [[ ${FLENI} -le 5 ]] ; then
                [ ${DEBUG} -eq 1 ] && log "${1} audio length ${FLENI} is less than 5 seconds"
                /usr/bin/rm -f ${1}
                return
        fi
        DIR="${1%/*}"
        NEWFN=`/usr/bin/stat -c %w "${1}" | \
                /usr/bin/awk -F '.' '{print $1}' | \
                /usr/bin/tr -d '-' | \
                /usr/bin/tr -d ':' | \
                /usr/bin/tr -d ' '`
        /usr/bin/mv -f ${1} ${DIR}/${NEWFN}.wav
        /usr/bin/lame --quiet -b 16 --resample 8 -a "${DIR}/${NEWFN}.wav" "${DIR}/${NEWFN}.mp3"
        if [[ "${?}" == "0" ]] ; then
                /usr/bin/touch --reference=${DIR}/${NEWFN}.wav ${DIR}/${NEWFN}.mp3
                /usr/bin/rm -f ${DIR}/${NEWFN}.wav
        fi
        [ ${DEBUG} -eq 1 ] && log "aslrecordstop ${1} finished"
}
if [ "${1}" == "0" ]; then
        # No more connections on the node. Turn off audio archiving/monitoring.
        [ ${DEBUG} -eq 1 ] && log "stopping mixmonitor ${2}"
        # Note: Using IAX2 as the channel is probably sketchy for multi node systems.
        # Also, tlb and echolink may use different channel names. e.g. tlb versus IAX2
        # Maybe there is a way to find and use the actual connecting channel name?
        /usr/sbin/asterisk -rx "mixmonitor stop IAX2"
        [ ${DEBUG} -eq 1 ] && log "mixmonitor stop rc is ${?}"
        aslrecordstop ${ASLRECORD}
elif [ "${1}" == "1" ] ; then
        # Someone connected to the node. Turn on audio archiving/monitoring.
        [ ${DEBUG} -eq 1 ] && log "starting mixmonitor ${2}"
        # Note: Using IAX2 as the channel is probably sketchy for multi node systems
        # Also, tlb and echolink may use different channel names. e.g. tlb versus IAX2
        # Maybe there is a way to find and use the actual connecting channel name?
        /usr/sbin/asterisk -rx "mixmonitor start IAX2 ${ASLRECORD}"
        [ ${DEBUG} -eq 1 ] && log "mixmonitor start rc is ${?}"
fi
exit 0
```

kenalker, 2025-01-27:
I just completed a few hours of investigating and found that app_rpt.c is sending the "mixmonitor start ..." command to Asterisk (via the ast_cli_command routine found in ast_cli.h).  This answers the question/concern about "monitor" vs. "mixmonitor" above.  It appears it is using the newer version (mixmonitor), so that is put to rest.

Interestingly, I can execute "mixmonitor start voter filename.wav49" from the Asterisk CLI (then "mixmonitor stop voter") and it records just fine.  So, the problem doesn't seem to lie within the mixmonitor code.  At this point I'm wondering if when app_rpt.c assembles the "mixmonitor start ..." command to send to Asterisk it is somehow getting the channel name wrong.  I can't seem to find any way to turn on the proper debugging (at the proper level) to expose what is being sent from app_rpt to Asterisk.  I may have to resort to adding hooks (printf's or something) to the code and recompiling, but I'm trying to figure out a way to debug this without having to go that far.

The 60-byte (only) files I have been plagued with for months haven't appeared once during my testing; I'm only getting the strange sounding "techno music" files; no explanation on this.

As an aside, I have another slightly newer installation of ASL3 on another node entirely (different repeater system) and I tried the archivedir command on it this morning and it records silence rather than the "techno music" described above (except that the ID and courtesy tone that the ASL3 code generates are recording just fine).

If anyone has any thoughts or comments on above, please interject!

mkmer, 2025-01-27:
> > Our local repeater also has been using this function for years, and your note made me wonder if ours was broken too. I can report our wav files sound fine and are of varying sizes. Since there were a LOT of things to workout when moving to ASL3, maybe something is missing in your config? I'm not sure the best way to compare configs (or what one may matter in this case). Our codec section looks like this (maybe the first place to start?):
> > ```
> > load => codec_adpcm.so              ; Adaptive Differential PCM Coder/Decoder
> > load => codec_alaw.so               ; A-law Coder/Decoder
> > load => codec_a_mu.so               ; A-law and Mulaw direct Coder/Decoder
> > noload => codec_dahdi.so            ; Generic DAHDI Transcoder Codec Translato
> > load => codec_g726.so               ; ITU G.726-32kbps G726 Transcoder
> > load => codec_gsm.so                ; GSM Coder/Decoder
> > load => codec_ulaw.so               ; mu-Law Coder/Decoder
> > noload => codec_ilbc.so             ; http://en.wikipedia.org/wiki/Internet_Low_Bitrate_Codec
> > ```
> 
> Thank you for your feedback!
> 
> My codec setup (in modules.conf) is identical to yours, so that may not be the issue. What exact version of ASL3 are you using? What channel driver modules do you have enabled in modules.conf? Are you using the "archivedir" command to record? Do you have just a single node? How does data flow between your repeater and the ASL3 box (ie. what hardware is in between; RTCM, URIx, etc.)? Perhaps shoot me full copies of your configs via email, and I can compare?

Version: Asterisk 20.11.0+asl3-3.2.0-2.deb12 
I have 2 nodes -> 
 1 -> repeater using RTCM's and Voter (chan_dahadi, chan_echolink, chan_iax2, chan_voter are enabled)
 2 -> "radio less node" using simpleusb (chan_dahadi, chan_iax2, and chan_simpleusb are enabled)

Both have `archivedir=<blah>` set in rpt.conf 

Both are recording wav files in the archivedir.  
Time to put on the detective hat :)

kenalker, 2025-01-28:
> Version: Asterisk 20.11.0+asl3-3.2.0-2.deb12 I have 2 nodes -> 1 -> repeater using RTCM's and Voter (chan_dahadi, chan_echolink, chan_iax2, chan_voter are enabled) 2 -> "radio less node" using [...]

I just upgraded to the latest ASL3 and it is the same as yours.  I also enabled chan_echolink just so my channel drivers would match yours (even though this likely won't fix anything).  Result: neither of these changed anything.  That said, this pushed me to thinking this really does have something to do with config differences rather than an issue I should be debugging at the code level (at least just yet).

So now I'm curious about your OS and OS Kernel versions, etc.  Can you send me a cut-and-paste of the entire page from your asl-menu Main Menu item #4 "Show System Version Numbers"?  [focusing solely on your node being used with the repeater employing the RTCM]

Also, there are ways to change the codec being used with the RTCM (if I'm reading comments in voter.conf correctly).  Can you send me the contents of your voter.conf?

Can you shoot over the line where you call archivedir?

Are you positive that when you key up your repeater that uses the RTCM that the file produced on your ASL3 box is truly listenable and includes audio traversing the RTCM itself?  Can you send me an example file you've recorded via that ASL3 box JUST IN CASE it has something to do with the audio client that I'm using to listen to play the files (wouldn't that be crazy)?

Thanks so much!!

kenalker, 2025-01-28:
> Are you using simpleusb? This issue may be related to using the simpleusb radio interface. My limited testing seems to get good recordings on an ASL3 HUB VSP VM node without chan_simpleusb loaded. [...]

David - Can you send me a cut-and-paste of the entire page from your asl-menu Main Menu item number 4 "Show System Version Numbers"?  [Also - thanks for the script!!]

mkmer, 2025-01-28:
Let's see if I can get everything you asked :)
OS/Kernel: `Linux Moline 6.1.0-30-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.124-1 (2025-01-12) x86_64 GNU/Linux`
Kernels .29 and .30 have issues with USB sound - hopefully that's not your problem.  I haven't noticed a problem on our RTCM systems.

Voter CONFIG: (I don't think it's anything "special"
```
[general]
port = 667                      ; specifies the UDP port to use for incoming connections
buflen = 480                    ; specifies the receive buffer length in milliseconds.
                                ; This parameter should be set to the maximum expected network latency,
                                ; plus a little padding (100 milliseconds of padding is a good amount).
                                ; The default is 480 milliseconds, the minimum is 320 milliseconds.
                                ; Buffer length may be specified on a per-stanza and per-client basis, see below.

password = xxx                 ; password common to all clients (Main Menu Item 6 - Host Password, on RTCM/VOTER)
utos = n                        ; Turn on IP TOS for Ubiquiti (ToS is enable by default on the RTCM/VOTER) from the host TO the clients

[29972]                         ; define the 1999 instance stanza
Moline = xxxx,transmit        ; master,transmit,adpcm,nulaw,nodeemp,noplfilter,buflen=value,gpsid[=value],prio=value
Lakeview =xxxx,transmit         ;
Ada = xxxx,transmit        ; master - this client is the Master Timing source (the RTCM client that is on the same LAN as the Asterisk server.)
                                ; There can only be 1 Master Timing source per ENTIRE Asterisk server.
                                ;
                                ; secret - password unique to this client
                                ; transmit - this client is intended to have transmit audio sent to it and will have a transmitter connected to it.
                                ; adpcm - this client is to be sent audio in 8000 samples/sec, 4-bit, IMA ADPCM format, rather than default G.711 Mulaw (aka ulaw).
                                ; nulaw - this client is to be sent audio in 4000 samples/second Mulaw (Nulaw, as we call it), rather than the standard 8000 samples/second.
                                ; nodeemp - this client is not to perform de-emphasis of the receiver audio  (This is only to be used with non-voting clients). Switches
                                ; the hardware de-emphasis filter OUT on the client
                                ; noplfilter - this client is to not to perform hardware pl filtering of the audio. Switches the hardware PL filter OUT on the client.
                                ; buflen=value - receive buffer length override for this client only.
                                ; gpsid[=value] - This specifies a gps identity to associate with the specified client (as referred in the /etc/asterisk/gps.conf file).
                                ; prio=value - define a specific priority for this client when voting. Lower numbers are higher priority. Default is 0 if nothing specified.

plfilter = n                    ; use DSP IIR 6 pole High pass filter, 300 Hz corner with 0.5 db ripple (note clients filter PL by default already) 

[20250128060515.zip](https://github.com/user-attachments/files/18578763/20250128060515.zip)

Mode 3 setting in RTCM/VOTER

thresholds = 255,110=5          ;
```

rpt.conf at archive dir:
```
[29972]
archivedir=/disk2/nodeaudio

statpost_program=/usr/bin/wget,-q,--timeout=15,--tries=1,--output-document=/dev/null
statpost_url=http://stats.allstarlink.org/uhandler.php
...
```

mkmer, 2025-01-28:
Zipped up audio from today (via RTCM)
[20250128060515.zip](https://github.com/user-attachments/files/18578770/20250128060515.zip)

kenalker, 2025-02-08:
@mkmer Can you tell me exactly what brand/model computer you are using for the repeater with the RTCM on it?  Also, what CPU manufacturer/model # does the computer have?

kenalker, 2025-02-08:
> Are you using simpleusb? This issue may be related to using the simpleusb radio interface. My limited testing seems to get good recordings on an ASL3 HUB VSP VM node without chan_simpleusb loaded. [...]

@wd5m I have some questions for you:

1) When you state "Audio from the radio is not recorded" are you getting a 60-byte file, no file at all, or a recorded file with silence, or a recorded file that sounds like techno-music rather than human speech?
2) What CPU is being using on the VM?
3) Is the repeater that is connected to the ASL3 HUB VPS VM using an RTCM?
4) When you state that "Using mixmonitor records with issues..." are issuing the mixmonitor command from the Asterisk CLI, or via some other mechanism?  Can you tell me exactly what the full mixmonitor command is that you are issuing that works?
5) Can you send me a cut-and-paste of the entire page from your asl-menu Main Menu item number 4 "Show System Version Numbers"? (for your ASL3 HUB VSP VM)

wd5m, 2025-02-08:
>     1. When you state "Audio from the radio is not recorded" are you getting a 60-byte file, no file at all, or a recorded file with silence, or a recorded file that sounds like techno-music rather than human speech?

I get a recorded file that sounds like techno-music from the radio. Normal voice from a linked Allstar node.
 
>     2. What CPU is being using on the VM?

 cat /proc/cpuinfo
processor       : 0
vendor_id       : AuthenticAMD
cpu family      : 23
model           : 49
model name      : AMD EPYC-Rome Processor
(...)

>     3. Is the repeater that is connected to the ASL3 HUB VPS VM using an RTCM?

No
 
>     4. When you state that "Using mixmonitor records with issues..." are issuing the mixmonitor command from the Asterisk CLI, or via some other mechanism?  Can you tell me exactly what the full mixmonitor command is that you are issuing that works?
 
I am using mixmonitor without issues. I originally used a script that issued mixmonitor commands via asterisk -rx. I have since switched to using extensions.conf with calls to mixmonitor upon connection from nodes. For example:
`        same => n(connect),MixMonitor(/var/www/html/archive/${EXTEN}/${DTNOW}.wav,a,/etc/asterisk/custom/wav2mp3.sh /var/www/html/archive/${EXTEN}/${DTNOW})
`

>     5. Can you send me a cut-and-paste of the entire page from your asl-menu Main Menu item number 4 "Show System Version Numbers"? (for your ASL3 HUB VSP VM)

```
AllStarLink [ASL] Version Info
                                                              
 OS            : Debian GNU/Linux 12 (bookworm)
 OS Kernel     : 6.1.0-30-amd64

 Asterisk      : 20.11.0+asl3-3.2.0-2.deb12
 ASL [app_rpt] : 3.2.0

 Installed ASL packages :

   Package                         Version
   ==============================  ==============================
   allmon3                         1.4.2-1.deb12
   asl3                            3.7.1-1.deb
   asl3-asterisk                   2:20.11.0+asl3-3.2.0-2.deb12
   asl3-asterisk-config            2:20.11.0+asl3-3.2.0-2.deb12
   asl3-asterisk-modules           2:20.11.0+asl3-3.2.0-2.deb12
   asl3-menu                       1.11-1.deb12

```

mkmer, 2025-02-10:
> [@mkmer](https://github.com/mkmer) Can you tell me exactly what brand/model computer you are using for the repeater with the RTCM on it? Also, what CPU manufacturer/model # does the computer have?

Architecture:             x86_64
  CPU op-mode(s):         32-bit, 64-bit
  Address sizes:          36 bits physical, 48 bits virtual
  Byte Order:             Little Endian
CPU(s):                   2
  On-line CPU(s) list:    0,1
Vendor ID:                GenuineIntel
  Model name:             Intel(R) Pentium(R) CPU G2120 @ 3.10GHz
    CPU family:           6
    Model:                58
    Thread(s) per core:   1
    Core(s) per socket:   2
    Socket(s):            1
    Stepping:             9
    CPU(s) scaling MHz:   60%
    CPU max MHz:          3100.0000
    CPU min MHz:          1600.0000
    BogoMIPS:             6185.99

I'm not sure on the brand.  May need to go to the hill do check that :(

Allan-N, 2025-02-10:
@wd5m>I get a recorded file that sounds like techno-music from the radio. Normal voice from a linked Allstar node.

wd5m, 2025-02-11:
> Can you attach one of your techno-music files?

Attached

> Also, what OS / app / tool / utility are you using to play back the recording? And have you specified any sound formatting options?

I'm using VLC or Audacity on Windows 11 for playback.  No custom settings. I have tried resampling and playback speed variations with no joy.  I have seen 60 byte recording files if I use "core reload" to reload rpt.conf  to enable archivedir. I got the attached audio file after restarting asterisk (astres.sh) .

The soxi tool on Linux returns this info on this file.
```
 soxi 20250210225125.WAV

Input File     : '20250210225125.WAV'
Channels       : 1
Sample Rate    : 8000
Precision      : 16-bit
Duration       : 00:00:12.16 = 97280 samples ~ 912 CDDA sectors
File Size      : 19.8k
Bit Rate       : 13.0k
Sample Encoding: GSM
```

soxi on the 60 byte file returned this.
```
Input File     : '20250210223632.WAV'
Channels       : 1
Sample Rate    : 8000
Precision      : 16-bit
Sample Encoding: GSM
```
[20250210225125.WAV.gz](https://github.com/user-attachments/files/18745431/20250210225125.WAV.gz)

kenalker, 2025-02-11:
@Allan-N I'm attaching two of my "techno-music" sounding files.  I discovered that if I play the files at 4x or 8x speed (load into Audacity then type <shift-ctrl-p> and set Clip Speed to 400% or 800%) they actually start resembling human voices.

I have created a spread sheet of repeaters that use archivedir that are recording successfully and those that are not (compiled from this ticket and from a couple of other systems).  I have entered in the spread sheet for each system the versions of ASL3, Asterisk, and apr_rpt as well as processor type, port number being used (667 or 1667), and any outstanding differences in config files, etc.  I saw from the data that everyone with Intel processors was having success with archivedir and those with AMD processors were not.  I then built a new Intel based ASL3 system over the weekend and it is recording techno-music so the processor theory ended up being a coincidence (probably not enough systems in the spread sheet yet to be statistically sound).

The next thing I noticed is that everyone using port 667 are having success and those who are using 1667 are not (and yes, those using 667 **are** using ASL3 and not ASL2; I guess they implemented the "trick" to be able to use the low port number).  I then figured that perhaps the old port was hardcoded somewhere in the code base so the call to mixmonitor() from apt_rpt was recording a "stream" from the wrong port.

Another theory was that the wrong channel was being called in the mixmonitor() call from apt_rpt.

However, discovering that when the WAV file is sped up and is actually a recording of the proper conversation invalidates all three of these theories (well, at least the latter two).  Finally, I felt I had to have some way to debug the code and lacking single-step ability I need to put hooks in to write messages to the console so I know what arguments are being passed to mixmonitor(), etc.  I installed ASL3 from source and start putting in hooks to see what is going on, but, alas, my RTCM is having problems holding a connection to my source-code-built machine and I'm stuck (per my forum post for help).

As an aside, I just tried to load @wd5m's WAV file into Audacity and Audacity tells me it doesn't recognize the file which is very odd since it recognized mine.  I wonder why our WAV files would be any different.  @wd5m, perhaps you can try slowing yours down as described above and see if it becomes more intelligible, as did mine.

wd5m, 2025-02-11:
On Tue, Feb 11, 2025 at 1:25 AM kenalker ***@***.***> wrote:

> (...)
>
> As an aside, I just tried to load @wd5m <https://github.com/wd5m>'s WAV
> file into Audacity and Audacity tells me it doesn't recognize the file
> which is very odd since it recognized mine. I [...]

Be sure you decompress/extract my file. It was attached as a gzipped (.gz)
file.

David

> Message ID: ***@***.***>
>

kenalker, 2025-02-11:
I did unzip it but Audacity didn't like it, nonetheless.  I suspected the same issue, so I unzipped using anther decompression utility and had the same experience.

kenalker, 2025-02-11:
> On Tue, Feb 11, 2025 at 1:25 AM kenalker ***@***.***> wrote:
>  (...)
>
>  As an aside, I just tried to load [@wd5m](https://github.com/wd5m) <https://github.com/wd5m>'s WAV
>  file into Audacity and [...]

I did unzip it but Audacity didn't like it, nonetheless. I suspected the same issue, so I unzipped using anther decompression utility and had the same experience. (Curious to know if you speed yours up if it becomes intelligible).

kenalker, 2025-02-11:
I added archivedir to another system last week and it is recording silence other than the courtesy tone and the ID, which are recorded clear as day.  So, the human speech portion is dead quiet (file size is as expected for said recording which indicates it is recording silence).  Of all the data I've been collecting, this is the first to exhibit this behavior.  If anyone else has this happen, please let me know.

kenalker, 2025-02-11:
The 60-byte file issue has disappeared for me.  I'm suspecting it is a different problem than the techno-music problem, so maybe it should be in its own trouble ticket.  I don't now why, but it hasn't happened on my system in weeks now.  That said, I did a little research on it and it appears to be an issue that was plaguing Asterisk users in the past; not specific to ASL3, but, perhaps, the mixmonitor.c code itself.  Here are some links I stumbled upon that may (or may not) be useful:

https://community.asterisk.org/t/asterisk-recorded-files-all-60-bytes/37465/9
https://asterisk-users.digium.narkive.com/k7MzanGa/asterisk-mixmonitor-starts-recording-44-bytes-file
https://community.asterisk.org/t/how-to-set-sample-rate-and-encoding-of-audios-generated-by-monitor/75374

When I issue "mixmonitor start voter/12345 testing.wav49 a" (12345 was replaced by my node number) from the Asterisk CLI and then IMMEDIATELY look at the file created, it is 60 bytes and it stays that way for a while.  Upon issuing "mixmonitor stop voter/12345" the file jumped to 17740 bytes.  So, PERHAPS when 60-byte files are created/left behind the "mixmonitor stop" is not being issued by app_rpt.c.

bobwwj555, 2025-02-19:
My node will not record at all running the latest updated ASL3 as of post, using the archive = ....  I have many nodes on each server. I put the archive in under each node under different file names. Not sure whats wrong, hope you guys figure this out.  Tried only one node archive too, reboot still will not "archive" any file. W4BWW

kenalker, 2025-02-19:
> My node will not record at all running the latest updated ASL3 as of post, using the archive = .... I have many nodes on each server. I put the archive in under each node under different file names. [...]

Was it recording before, or is this the first time you have tried it?

Also, when you say it will not record, are files being written at all, or are they blank, or are they being recorded but not listenable?

bobwwj555, 2025-02-19:
First time tried, Files will not write.

kenalker, 2025-02-19:
There was a permissions problem in an earlier version.  If you upgraded from an earlier version, perhaps your directory has the wrong permissions on it.  Please see https://github.com/AllStarLink/ASL3/issues/90.  Once you are creating files, please come back and let us know if they are playable or garbled.  I'd like to know either way.  Good luck.

bobwwj555, 2025-02-19:
Thank you! , I'm trying.. Changed the permissions, I then rebooted looked, I only did one node, (have many on the industrial PC on the repeater site). Was not the problem on mine, still not writing files. Tried various things, I will wait and attempt to study more. I will spin a new ASL3 testnode up on the prox server to play with more vs my live system with few live repeaters and links connected. I want the HUB node moved and running better too on it. Then maybe I can offer more useful info then. I have a system if someone finds issues and want to test check things. PM me W4BWW.net Again thanks, Ill post if I find more useful info more specific to this.

w4msi, 2025-03-17:
> I call the script via the event segment(s) in rpt.conf. On my systems running multiple nodes, each node has its own event segment. Here's an example event segment. This is almost a copy of an [...]

I've tested your script and it seems to work fine for me on my node...however I'd like to also record local RX on the node as well. Currently this only records audio coming from other connected nodes and no audio is recorded when anyone keys directly into my node/repeater. With the old archive function it recorded both sides. Is there a way to make your script include local audio as well and not ONLY the audio coming from other nodes?

wd5m, 2025-03-17:
> I've tested your script and it seems to work fine for me on my node...however I'd like to also record local RX on the node as well. Currently this only records audio coming from other connected nodes and no audio is recorded when anyone keys directly into my node/repeater. With the old archive function it recorded both sides. Is there a way to make your script include local audio as well and not ONLY the audio coming from other nodes? Of course, the script also requires lame and soxi to be installed.

I decided that script was overly complicated. I replaced it with a couple of lines in extensions.conf to call MixMonitor from there, and a script to convert the MixMonitor wav file recording to an mp3 file.  Here is the wav2mp3 script, with comments showing the lines I add to extensions.conf in selected segments. 

The wav2mp3 script is only required to create mp3 files. MixMonitor will create and append to the wav file, even if the wav2mp3 script command is not specified.  I am calling MixMonitor like this from the [radio-secure], [iaxrpt] and [allstar-public] segments of extensions.conf. Exact syntax for these lines may differ somewhat in each segment of extensions.conf. I use the "STRFTIM" feature to set a date string for the filename. That may require loading func_strings.so in modules.conf.

```
#!/bin/bash
# /etc/asterisk/custom/wav2mp3.sh
# Convert mixmonitor wav files to mp3
# This script was created to be run from an ASL3 Allstar extension.conf segment(s)
# as a MixMonitor "command" script that runs when MixMonitor stops recording.
# Call with the path/filename as an argument, without the file extension.
# e.g. /etc/asterisk/custom/wav2mp3.sh /var/www/html/archive/12345/20250203
# These lines are an example of use in Allstar extensions.conf, 
# placed before the call to Rpt(...
#        ; Set a variable to date-time to use for MixMonitor filename.
#        ; "load => func_strings.so" may be required in modules.conf.
#        same => n,Set(DTNOW=${STRFTIME(${EPOCH},,%Y%m%d)})
#        same => n,MixMonitor(/var/www/html/archive/${EXTEN}/${DTNOW}.wav,a,/etc/asterisk/custom/wav2mp3.sh /var/www/html/archive/${EXTEN}/${DTNOW})
#
# Note, you should create a separate method to archive or remove older wav/mp3 files over time.
# 20250203 WD5M
#
# exit if input file does not exist
[[ -f "${1}.wav" ]] || exit 100
# Exit if length of input file is less than 5 seconds
FLEN=`/usr/bin/soxi -D ${1}.wav`
FLENI=$(echo "${FLEN}/1" | /usr/bin/bc)
[[ ${FLENI} -le 5 ]] && exit
# convert the file to mp3 and set mp3 file timestamp same as wav file
/usr/bin/lame --quiet -b 32 --resample 8 -a "${1}.wav" "${1}.mp3"
/usr/bin/touch --reference=${1}.wav ${1}.mp3
```

David, WD5M

w4msi, 2025-03-18:
@wd5m I gave that a try and I must have something wrong. I don't see any recordings at all from any mode (RF input into my node or from remotely connected nodes) Here is one of my stanzas from extensions.conf....do I have this right?

[allstar-public]

exten => s,1,Ringing
        same => n,Set(RESP=${CURL(https://register.allstarlink.org/cgi-bin/authwebphone.pl?${CALLERID(name)})})
        same => n,Set(NODENUM=${CALLERID(number)})
        same => n,GotoIf($["${RESP:0:1}" = "?"]?hangit)
        same => n,GotoIf($["${RESP:0:1}" = ""]?hangit)
        same => n,GotoIf($["${RESP:0:5}" != "OHYES"]?hangit)
        same => n,Set(CALLSIGN=${RESP:5})
        ;exten => s,n,GotoIf($[${DB_EXISTS(whitelist/${CALLSIGN})}]?:hangit)  ; Not in whitelist
        same => n,Wait(3)
        same => n,Playback(rpt/node,noanswer)
        same => n,Saydigits(${NODENUM})
        same => n,Set(CALLERID(name)=${CALLSIGN})
        same => n,Set(CALLERID(num)=0)

##################################################
        same => n,Set(DTNOW=${STRFTIME(${EPOCH},,%Y%m%d)})
        same => n,MixMonitor(/var/www/html/archive/${EXTEN}/${DTNOW}.wav,a,/etc/asterisk/local/wav2mp3.sh /var/www/html/archive/${EXTEN}/${DTNOW})

##################################################

        same => n,Rpt(${NODENUM}|X)
        same => n,Hangup
        same => n(hangit),Answer
        same => n,Wait(1)
        same => n,Hangup

#tryinclude custom/extensions.conf

wd5m, 2025-03-18:
The allstar-public segment uses "${NODENUM}" instead of "${EXTEN}" for the node number, so maybe this would be the correct syntax...
`same => n,MixMonitor(/var/www/html/archive/${NODENUM}/${DTNOW}.wav,a,/etc/asterisk/custom/wav2mp3.sh /var/www/html/archive/${NODENUM}/${DTNOW})`

That should place the recording file under the directory path "/var/www/html/archive/<youraslnodenbr>/"  Of course, you can specify any path as long as it exists. I use a variable for a directory name because I may have systems with multiple ASL nodes, and it's similar to the old archive method. If only a filename is specified there, I think the MixMonitor recording file gets placed in "/var/spool/asterisk/monitor/" or whatever spool path is specified in the asterisk.conf file.

Also, "###..." may not be a comment in extensions.conf.  Prefix those lines with a semicolon (;) if you leave them in the file. e.g.
; ##################################################

David, WD5M

w4msi, 2025-03-18:
@wd5m I corrected my "comments" by adding a semicolon in there. After playing around for a bit, I took a path I created earlier that your original script was recording to (so I know it's accessible and permissions are correct for asterisk to save in) and put that path in this script and then removed the "EXTEN / NODENUM" variable and put the complete path in there. Rebooted the node and then tested and still no recordings :/  And I do have the func_strings.so enabled as well as mixmonitor.so in my modules.conf file. I appreciate your help and hopefully we can figure this thing out.

wd5m, 2025-03-18:
> [@wd5m](https://github.com/wd5m) I corrected my "comments" by adding a semicolon in there. After playing around for a bit, I took a path I created earlier that your original script was recording to [...]

[w4msi](https://github.com/w4msi)
Check your asterisk console for errors or other helpful information.  Below is what one of our nodes shows on the asterisk -rvvv console. I've only included lines related to MixMonitor from the extensions file.  These lines are mixed with others.

These lines were shown for connecting and disconnecting using radio-secure...

`
    -- Executing [473547@radio-secure:5] Set("IAX2/216.147.122.14:9700-10378", "DTNOW=20250318") in new stack
    -- Executing [473547@radio-secure:14] MixMonitor("IAX2/216.147.122.14:9700-10378", "/var/www/html/archive/473547/20250318.wav,a,/etc/asterisk/custom/wav2mp3.sh /var/www/html/archive/473547/20250318") in new stack
   == Begin MixMonitor Recording IAX2/216.147.122.14:9700-10378

(after disconnect...)

  == MixMonitor close filestream (mixed)
  == Executing [/etc/asterisk/custom/wav2mp3.sh /var/www/html/archive/473547/20250318]
  == End MixMonitor Recording IAX2/216.147.122.14:9700-10378
`
 And these were shown using allstar-public...

`
    -- Executing [s@allstar-public:12] Set("IAX2/allstar-public-11618", "DTNOW=20250318") in new stack
    -- Executing [s@allstar-public:13] MixMonitor("IAX2/allstar-public-11618", "/var/www/html/archive/473547/20250318.wav,a,/etc/asterisk/custom/wav2mp3.sh /var/www/html/archive/473547/20250318") in new stack
  == Begin MixMonitor Recording IAX2/allstar-public-11618

(after disconnect...)

  == MixMonitor close filestream (mixed)
  == Executing [/etc/asterisk/custom/wav2mp3.sh /var/www/html/archive/473547/20250318]
  == End MixMonitor Recording IAX2/allstar-public-11618
`

David, WD5M

w4msi, 2025-03-18:
@wd5m 
I modified the IP addresses and nodenumbers for privacy but here's the output I get. Doesn't appear to call MixMonitor at all. Also, this is for an inbound connection. If I have an outbound connection it says call accepted and shows teh DAHDI/pseudo where it's announcing the node number it connected to. But that's it.

-- Accepting UNAUTHENTICATED call from 50.30.xxx.xxx:4569:
    --        > requested format = ulaw,
    --        > requested prefs = (ulaw|speex16|ilbc),
    --        > actual format = ulaw,
    --        > host prefs = (ulaw|adpcm|gsm),
    --        > priority = mine
    -- Executing ["MyNodeNumber"@radio-secure:1] NoOp("IAX2/50.30.xxx.xxx:4569-10465", "Connect from node: "MyOtherNode") in new stack
    -- Executing ["MyNodeNumber"@radio-secure:2] NoOp("IAX2/50.30.xxx.xxx:4569-10465", "Connect to: "MyNodeNumber"") in new stack
    -- Executing ["MyNodeNumber"@radio-secure:3] NoOp("IAX2/50.30.xxx.xxx:4569-10465", "The Channel IP is 50.30.xxx.xxx") in new stack
    -- Executing ["MyNodeNumber"@radio-secure:4] GotoIf("IAX2/50.30.xxx.xxx:4569-10465", "0?connect") in new stack
    -- Executing ["MyNodeNumber"@radio-secure:5] GotoIf("IAX2/50.30.xxx.xxx:4569-10465", "1?denylist") in new stack
    -- Goto (radio-secure,"MyNodeNumber",9)
    -- Executing ["MyNodeNumber"@radio-secure:9] GotoIf("IAX2/50.30.xxx.xxx:4569-10465", "0?:connect") in new stack
    -- Goto (radio-secure,"MyNodeNumber",14)
    -- Executing ["MyNodeNumber"@radio-secure:14] Rpt("IAX2/50.30.xxx.xxx:4569-10465", ""MyNodeNumber"") in new stack

wd5m, 2025-03-18:
[w4msi](https://github.com/w4msi) Have the MixMonitor changes been added to the radio-secure segment of extensions.conf?

David, WD5M

w4msi, 2025-03-19:
@wd5m 
Here is the radio-secure stanza of my extensions.conf

[radio-secure]
;exten => ${NODE},1,rpt(${EXTEN})
exten => _XXXX!,1,NoOp(Connect from node: ${CALLERID(num)})

        same => n,NoOp(Connect to: ${EXTEN})
        ;same => n,NoOp(The IAXPEER is ${IAXPEER(CURRENTCHANNEL)})
        same => n,NoOp(The Channel IP is ${CHANNEL(peerip)})

        ;If channel IP eq localhost go to connect
        same => n,GotoIf($["${CHANNEL(peerip)}" = "127.0.0.1"]?connect)

        ;If allowlist/extension dosen't exist check the denylist
        same => n,GotoIf($[${DB_KEYCOUNT(allowlist/${EXTEN})} = 0]?denylist) ;goto denylist
        ;If allowlist/extension/callerid exists go to connect
        same => n,GotoIf(${DB_EXISTS(allowlist/${EXTEN}/${CALLERID(num)})}?connect) ;goto connect
        same => n,NoOp(${EXTEN} not in allowlist, Hangup)
        same => n,Hangup

        ;If denylist/extension/callerID doesn't exist goto connect
        same => n(denylist),GotoIf(${DB_EXISTS(denylist/${EXTEN}/${CALLERID(num)})}?:connect) ;don't goto connect
        same => n,NoOp(${EXTEN} is in denylist, Hangup)
        same => n,Hangup

;#####################################################################
        same => n,Set(DTNOW=${STRFTIME(${EPOCH},,%Y%m%d)})
        same => n,MixMonitor(/etc/asterisk/local/AUDIO/"MyNodeNumber"/${DTNOW}.wav,a,/etc/asterisk/local/wav2mp3.sh /etc/asterisk/local/AUDIO/"MyNodeNumber"/${DTNOW})
;#####################################################################

        same => n(connect),rpt(${EXTEN})
        same => n,Hangup

wd5m, 2025-03-20:
The original issue with archivedir may be resolved in the recent ASL3 update. I recommend trying archivedir again, which is what I am doing now.

David, WD5M

w4msi, 2025-03-20:
@wd5m Just updated, rebooted, enabled archive in rpt.conf and it seems to be working. Haven't tested it extensively but yes the garbled techno audio seems to now be a thing of the past. Guess I should have a waited a couple more days before trying to make it work haha. Anyway, thank you David for your help and thank you ASL devs for fixing this. 

Dusty, W4MSI

kenalker, 2025-03-20:
I just upgraded to apt_rpt 3.3.0 and it solved the problem on my primary ASL3 install.  YAY!  I am VERY excited.  If I'm reading the Commit properly, it appears that adding string length protection to all sscanf() calls may be what solved this (I'm guessing the mixmonitor call was corrupt).  After spending dozens of hours on this, I'm in hopes that the dev(s) who fixed this will contact me as I'd really like to understand how this solved the problem, and why it only showed up in SOME installs.  (Also, was this bug manifesting itself in other ways throughout the code)?

I am wondering if anyone has figured out if the files we've been recording that sound like "techno-music" are recoverable.  If I speed some of the files up by 4x (and in some cases it must be 8x) they actually become somewhat intelligible which leads me to think they should be salvageable, but I've not figured out how.  Perhaps understanding what the bug was doing to the mixmonitor call would shed light on how these WAV files were being recorded, and thus, help determine what to do to "fix/alter" them.

As an aside, my second install for another repeater system is still recording silence from the RTCM instead of audio (except that the ID and courtesy tone that the ASL3 code generates are recording just fine).  On this install audio from attached Allstar nodes is being recorded just fine.  Perhaps this is a one-off and I just need to start from scratch, but if anyone else is having this issue, please chime in.

Thank you **SO MUCH** to the dev or devs who fixed this!

kenalker, 2025-03-20:
> The original issue with archivedir may be resolved in the recent ASL3 update. I recommend trying archivedir again, which is what I am doing now.
> 
> David, WD5M

How did you become aware of this so quickly after the ASL3 update?

kenalker, 2025-03-20:
My ASL3 install that was recording silence instead of audio from the RTCM was set to duplex mode 3 so audio wasn't making it to the ASL3 box.  Problem solved.  It is amazing how the brain clears after a more prominent problem is popped off the stack.

Allan-N, 2025-03-20:
>I just upgraded to apt_rpt 3.3.0 and it solved the problem on my primary ASL3 install. YAY! I am VERY excited. If I'm reading the Commit properly, it appears that adding string length protection to all sscanf() calls may be what solved this (I'm guessing the mixmonitor call was corrupt).

The sscanf() change you may have noticed, https://github.com/AllStarLink/app_rpt/pull/542, is targeted for our next set of packages.  It's not likely that those changes would have any effect on the archiving as they are to help ensure that configuration and CLI commands don't overrun internal buffers.

The biggest "archivedir" changes were to correct the directory/file permissions.  Those changes, https://github.com/AllStarLink/app_rpt/pull/342 and  https://github.com/AllStarLink/app_rpt/pull/379, went in last year.

As for the most recent set of packages, we did include 2 smaller changes, https://github.com/AllStarLink/app_rpt/pull/492 and https://github.com/AllStarLink/app_rpt/pull/507.  I'm not convinced that either of these "fixed" anything.

wd5m, 2025-03-20:
The only reason I tried "archivedir" again was because I recalled seeing the changes Allan-N mentioned above. Also, this update took Asterisk to version 22. It never hurts to recheck an issue after a major update/upgrade like this. It's only been about a day, with little activity on my systems. If the issue doesn't resurface, I'm okay with whatever fixed it. Kudos to the development team(s). I know, from following the activity in GitHub, they putting in a LOT of effort and time to fix bugs and improve Allstarlink. Thank you!

David, WD5M

Allan-N, 2025-05-10:
>Changed to duplex mode 2 recording is working.

This duplex=3 issue should be fixed with [Issue #600](https://github.com/AllStarLink/app_rpt/issues/600) /  [PR #603](https://github.com/AllStarLink/app_rpt/pull/603) that will be in our "next" release.

wd5m, 2025-06-25:
This issue, recording with archivedir, seems to still be happening with chan_tlb audio after the last ASL3 update. The Mixmonitor command will record audio, but includes silence during periods without transmissions.

> ********** AllStarLink [ASL] Version Info **********
> 
> OS            : Debian GNU/Linux 12 (bookworm)
> OS Kernel     : 6.1.0-37-amd64
> 
> Asterisk      : 22.4.1+asl3-3.5.2-1.deb12
> ASL [app_rpt] : 3.5.2
> 
> Installed ASL packages :
> 
>   Package                         Version
>   ==============================  ==============================
>   asl3                            3.8-1.deb
>   asl3-asterisk                   2:22.4.1+asl3-3.5.2-1.deb12
>   asl3-asterisk-config            2:22.4.1+asl3-3.5.2-1.deb12
>   asl3-asterisk-modules           2:22.4.1+asl3-3.5.2-1.deb12
>   asl3-menu                       1.14-1.deb12
>   asl3-update-nodelist            1.5.1-1.deb12
>   dahdi                           1:3.1.0-2
>   dahdi-dkms                      1:3.4.0-6+asl
>   dahdi-linux                     1:3.4.0-6+asl
>

wd5m, 2025-07-02:
> > This issue, recording with archivedir, seems to still be happening with chan_tlb audio after the last ASL3 update. The Mixmonitor command will record audio, but includes silence during periods without transmissions.
> 
> [@wd5m](https://github.com/wd5m) - can you elaborate? I'm not following.

Testing again today, ASL3 is recording again from a TLB connection. I suppose this could be due to one of the recent ASL3 updates.  I installed this update this morning.

********** AllStarLink [ASL] Version Info **********

OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.1.0-37-amd64

Asterisk      : 22.4.1+asl3-3.5.4-1.deb12
ASL [app_rpt] : 3.5.4

Installed ASL packages :

  Package                         Version
  ==============================  ==============================
  allmon3                         1.5.1-1.deb12
  asl3                            3.8-1.deb
  asl3-asterisk                   2:22.4.1+asl3-3.5.4-1.deb12
  asl3-asterisk-config            2:22.4.1+asl3-3.5.4-1.deb12
  asl3-asterisk-modules           2:22.4.1+asl3-3.5.4-1.deb12
  asl3-menu                       1.14-1.deb12
  asl3-update-nodelist            1.5.1-1.deb12
  dahdi                           1:3.1.0-2
  dahdi-dkms                      1:3.4.0-6+asl
  dahdi-linux                     1:3.4.0-6+asl

David, WD5M

kenalker, 2025-08-13:
@Allan-N Well, we seem to be discussing four issues in this thread now:
1) recordings that sound like "technomusic"
2) recordings that result in 60-byte files with no audio
3) silent recordings when duplex=3
4) silence issue reported [above](https://github.com/AllStarLink/app_rpt/issues/382#issuecomment-3005452008) by @wd5m

No.1 and No.2 were the two primary issues being reported.
No.1 I believe to be solved.  I sure wish we knew HOW.  It may always be a mystery.
No.2 has been opened in its own ticket #600 
No.3 has been opened in its own ticket #714
No.4 I'm not sure on this one because I'm not understanding the issue.  Is it the same as I reported (ie. continue here) or is it new (ie. open new bug report)?

I'd say this could be closed, pending No.4 (if that's why you are asking).

I hope this answers your question.