AllStarLink/app_rpt #383: Autopatch issue - Cannot start tone on DAHDI/pseudo

kyle470, 2024-08-10:
Issue when trying to use Autopatch. Seems similar to the dreaded "cannot start dialtone" issue on previous versions:
rpt.conf entry was taken from an earlier version of ASL with the quiet=1 workaround:

[2024-07-24 10:01:18.087] WARNING[17064]: app_rpt/rpt_bridging.c:555 rpt_play_to                                                                             ne: Cannot start tone on DAHDI/pseudo-481952349
[2024-07-24 10:01:18.088] WARNING[17065][C-0000001d]: pbx.c:2928 pbx_extension_h                                                                             elper: No application 'Wait,1' for extension (icom-ctl, *01, 1)
[2024-07-24 10:01:18.098] WARNING[17064]: app_rpt.c:1352 rpt_call:  has no PBX?
[2024-07-24 10:01:18.118] WARNING[17064]: app_rpt/rpt_bridging.c:555 rpt_play_to                                                                             ne: Cannot start tone on DAHDI/pseudo-481952349

rpt.conf entry:
69=autopatchup,context=icom-ctl,noct=1,farenddisconnect=1,dialtime=90000,quiet=1

kyle470, 2024-12-12:
Just upgraded one of my boxes to 20.10.0+asl3-3.1.0-1.deb12 and the issue persists:

3040-3*CLI> rpt fun 2211 *6500
[2024-12-11 21:04:56.367] WARNING[303958]: app_rpt/rpt_bridging.c:567 rpt_play_tone: Cannot start tone on DAHDI/pseudo-645073531
    -- Hungup 'DAHDI/pseudo-802033485'
    -- Hungup 'DAHDI/pseudo-645073531'

I found this article and made sure to load all the modules etc. Still no luck
https://allstarlink.github.io/adv-topics/autopatch/?h=autopatch

InterLinked1, 2025-04-05:
> Issue when trying to use Autopatch. Seems similar to the dreaded "cannot start dialtone" issue on previous versions: rpt.conf entry was taken from an earlier version of ASL with the quiet=1 [...]

What actually happens when this occurs? Is anything not working apart from the warning appearing?

> [2024-07-24 10:01:18.088] WARNING[17065][C-0000001d]: pbx.c:2928 pbx_extension_h elper: N application 'Wait,1' for extension (icom-ctl, *01, 1) 

The issue for this specific warning is your dialplan is invalid. Probably need Wait(1) instead of Wait,1

> [2024-07-24 10:01:18.098] WARNING[17064]: app_rpt.c:1352 rpt_call: has no PBX?
> [2024-07-24 10:01:18.118] WARNING[17064]: app_rpt/rpt_bridging.c:555 rpt_play_to ne: Cannot start tone on DAHDI/pseudo-481952349

mkmer, 2025-04-14:
Solved https://github.com/AllStarLink/ASL3/issues/142

closed 2025-04-14 by mkmer