AllStarLink/app_rpt #384: ASL3 ID is unreliable

kyle470, 2024-08-10:
**Describe the bug**
ID will often begin but not fully send ID audio. PTT is asserted but the ID is not always fully sent out. This occurs both with standard CWID as well as a custom ID audio file I have used for many years.

Manually sending the audio works reliably:
rpt playback 2212 filename

Just updated to latest ASL3 today and issue persists.

**Have you run a software update and rebooted?**
 - YES 

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - :Raspberry Pi4b

**Additional context**
Pi4b does have two RF CM108 nodes on it, but I have not noticed any correlation of this ID issue to having multiple RF nodes. This ID reliability issue happens on both RF nodes, even if one has been idle for hours.

KB4MDD, 2024-08-10:
Are you using chan_simpleusb or chan_usbradio?
Are you using PL on transmit?

kyle470, 2024-08-10:
chan_usbradio. Not using app_rpt for PL.

Since creating this ticket, I have spun up several ASL3 boxes using Dell
Wyse 3040 units, have not yet noticed this issue on that hardware. May be
an issue with multiple RF nodes on a Raspberry Pi 4 as in past versions of
ASL?

jxmx, 2024-12-08:
The does not appear to be a repeatable or wide-spread problem. Please re-open if there is a diagnosable problem here. Cleanup GitHub.

closed 2024-12-08 by jxmx