AllStarLink/app_rpt #387: Issue building with ASL3 Pi image

eliggett, 2024-08-20:
Greetings, 

I was hoping to do some debug and possibly fix an issue I ran into with app_rpt, but I an running into an issue building app_rpt. 

The host is a Pi4 with 8 GB ram, running the ASL3 image. 

I followed the steps as outlined here: 
https://github.com/AllStarLink/app_rpt/blob/master/README.md

I did add the "-u asterisk" to the phreaknet arguments: 

```phreaknet install -d -b -v 20 -u asterisk```

```
DAHDI already present but install not forced, skipping...
Proceeding to install Asterisk 20 (~major 20)...
asterisk-20-current.tar.gz  100%[========================================>]  26.98M  9.54MB/s    in 2.8s
Asterisk MMmmPP: 200902
Installing production version asterisk-20.9.2...
```
I got through the "phreaknet" script, with the only additional step being to install git prior to running it, not a big deal. 

However, when I run rpt_install.sh, I received the following errors: 
```
root@simplex900:/usr/src# ./rpt_install.sh
Already up to date.
Updating ourself
Already up to date.
Script is now the latest version
asterisk-20.9.2/
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libusb-dev is already the newest version (2:0.1.12-32).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
... additional adding modules...
Adding module utils/pi-tune-menu.c
Adding module utils/radio-tune-menu.c
Adding module utils/simpleusb-tune-menu.c
4
CC="cc" CXX="g++" LD="" AR="" RANLIB="" CFLAGS="" LDFLAGS="" make -C menuselect CONFIGURE_SILENT="--silent" makeopts
make[1]: Entering directory '/usr/src/asterisk-20.9.2/menuselect'
make[1]: 'makeopts' is up to date.
make[1]: Leaving directory '/usr/src/asterisk-20.9.2/menuselect'
Generating input for menuselect ...

***********************************************************
  The existing menuselect.makeopts file did not specify
  that 'app_rpt' should not be included.  However,
dependencies for this module were not found.

  Either run 'make menuselect' or remove the existing
  menuselect.makeopts file to resolve this issue.
***********************************************************
```

I attempted to follow the directions in the message, running ```make menuselect``` in the asterisk directory, but oddly, the same error results as above. If I remove the menuselect.makeopts file, then I can run ```make menuselect```, however, it does not allow me to select app_rpt (it is labeled with "XXX"). I believe this is due to some dependency being missing. The dependencies are listed for the module as:

```Depends on: tonezone(E), res_curl(M), curl(E), dahdi(E)```

The module res_curl is already selected. I am unsure how to satisfy the other dependencies. But I digress, I suspect the patch steps are intended to remedy the build issue. 

If someone can guide me in to getting this built, it would be much appreciated. I have a few bugs I'd like to trace out and having a working build system would help tremendously. 

Thanks, 

--E
de W6EL

KB4MDD, 2024-08-20:
I just built a rpi5 with those instructions with no problem.  I used the latest x64 os, but I have built on x32.

You can look at these steps to see if anything is different.

https://github.com/AllStarLink/ASL3/blob/5630803e7378a5ad1b964e921bf77356e49773ec/RPi-install.md

eliggett, 2024-08-20:
Can you share with me your Makefile, menuselect.makeopts, and menuselect.makedeps? 

I'll diff them with what I generated here and see if I can figure it out. 

Thanks very much, 

--E
de W6EL

Allan-N, 2024-08-20:
The ASL3 Manual also has a page on building from source :

https://github.com/AllStarLink/ASL3-Manual/blob/main/docs/developers/source-install.md

eliggett, 2024-08-20:
These are all basically the same directions. 

I think something is up with the ASL3 install specifically or the asterisk main repo has some change which is not compatible. I am tempted to see if I can specify the same asterisk repo commit that has worked for you, but how about we start with the Makefile, menuselect.makeopts, and menuselect.makedeps, and see what I can see. 

Thanks, 

--E
de W6EL

Allan-N, 2024-08-20:
I just wiped my RPi and will try a "by the instructions" build shortly.

eliggett, 2024-08-20:
@Allan-N the dedication is tremendous, thank you for trying that out. 

Will you specify ```-u asterisk``` with the phreaknet script? It complained when I did not.

KB4MDD, 2024-08-20:
I built two RPI5's last week from the repo using the phreaknet script.  There have been no changes in the repo for 3 weeks.

I will send a note to one of the other developers.  I will report back.

Allan-N, 2024-08-20:
And, of course, just minutes after I wrote that message (and before getting up from the chair) the compile finished.

Please let me know if adding the dependencies addressed your compile issues. If all is good for you then I'll work on updating the documentation.

eliggett, 2024-08-20:
Going to do some google here, but I guess the package is renamed? Yes apt is up to date here. 

```
sudo apt install git subversion libtimezone-dev
[sudo] password for eliggett:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libtimezone-dev
eliggett@simplex900:/usr/src/asterisk-20.9.2-untouched $ apt-cache search libtimezone
gir1.2-timezonemap-1.0 - typelib interface for libtimezonemap
libtimezonemap-data - GTK+3 timezone map widget - data files
libtimezonemap1 - GTK+3 timezone map widget
libtimezonemap1-dev - GTK+3 timezone map widget - development files
```

eliggett, 2024-08-20:
Yeah, I can't find this libtimezone-dev package or anything similar. Google isn't helping.

eliggett, 2024-08-20:
Or maybe you meant 'libtonezone-dev' ? Let's try that and see...

Allan-N, 2024-08-20:
Yes, that was a typo. The missing package was libtonezone-dev.

eliggett, 2024-08-20:
I removed /usr/src/asterisk-20.9.2 as well as app_rpt. 

Made sure to add the packages discussed, including the tone zone. 

Re-ran phreaknet.sh and then the rpt_install.sh script. 

Unfortunately basically the same error(s).

```
Adding module res/res_usbradio.exports.in
Adding module utils/pi-tune-menu.c
Adding module utils/radio-tune-menu.c
Adding module utils/simpleusb-tune-menu.c
4
Making apps...CC="cc" CXX="g++" LD="" AR="" RANLIB="" CFLAGS="" LDFLAGS="" make -C menuselect CONFIGURE_SILENT="--silent" makeopts
make[1]: Entering directory '/usr/src/asterisk-20.9.2/menuselect'
make[1]: 'makeopts' is up to date.
make[1]: Leaving directory '/usr/src/asterisk-20.9.2/menuselect'
Generating input for menuselect ...
menuselect/menuselect --check-deps menuselect.makeopts

***********************************************************
  The existing menuselect.makeopts file did not specify
  that 'app_rpt' should not be included.  However,
dependencies for this module were not found.

  Either run 'make menuselect' or remove the existing
  menuselect.makeopts file to resolve this issue.
***********************************************************

make: *** [Makefile:372: menuselect.makeopts] Error 255
Making channels...CC="cc" CXX="g++" LD="" AR="" RANLIB="" CFLAGS="" LDFLAGS="" make -C menuselect CONFIGURE_SILENT="--silent" makeopts
make[1]: Entering directory '/usr/src/asterisk-20.9.2/menuselect'
make[1]: 'makeopts' is up to date.
make[1]: Leaving directory '/usr/src/asterisk-20.9.2/menuselect'
menuselect/menuselect --check-deps menuselect.makeopts

***********************************************************
  The existing menuselect.makeopts file did not specify
  that 'app_rpt' should not be included.  However,
dependencies for this module were not found.

  Either run 'make menuselect' or remove the existing
  menuselect.makeopts file to resolve this issue.
***********************************************************

make: *** [Makefile:372: menuselect.makeopts] Error 255
Making res...CC="cc" CXX="g++" LD="" AR="" RANLIB="" CFLAGS="" LDFLAGS="" make -C menuselect CONFIGURE_SILENT="--silent" makeopts
make[1]: Entering directory '/usr/src/asterisk-20.9.2/menuselect'
make[1]: 'makeopts' is up to date.
make[1]: Leaving directory '/usr/src/asterisk-20.9.2/menuselect'
menuselect/menuselect --check-deps menuselect.makeopts

***********************************************************
  The existing menuselect.makeopts file did not specify
  that 'app_rpt' should not be included.  However,
dependencies for this module were not found.

  Either run 'make menuselect' or remove the existing
  menuselect.makeopts file to resolve this issue.
***********************************************************

make: *** [Makefile:372: menuselect.makeopts] Error 255
Done making.Makg installCC="cc" CXX="g++" LD="" AR="" RANLIB="" CFLAGS="" LDFLAGS="" make -C menuselect CONFIGURE_SILENT="--silent" makeopts
make[1]: Entering directory '/usr/src/asterisk-20.9.2/menuselect'
make[1]: 'makeopts' is up to date.
make[1]: Leaving directory '/usr/src/asterisk-20.9.2/menuselect'
menuselect/menuselect --check-deps menuselect.makeopts

***********************************************************
  The existing menuselect.makeopts file did not specify
  that 'app_rpt' should not be included.  However,
dependencies for this module were not found.

  Either run 'make menuselect' or remove the existing
  menuselect.makeopts file to resolve this issue.
***********************************************************

make: *** [Makefile:372: menuselect.makeopts] Error 255
root@simplex900:/usr/src# 
```

eliggett, 2024-08-20:
I re-read the error message carefully. 
"...or remove the existing menuselect.makeopts file to resolve this issue."

I removed that file. And now it seems to be proceeding! 

This is merely a Pi, so I shall update you in 15 minutes :)

eliggett, 2024-08-20:
Ok, it does appear to build, but it does not place the resulting shared objects into their correct location upon ```make install```. Starting asterisk shows: 
```
[2024-08-20 15:29:29.203] NOTICE[359242] loader.c: Module 'res_adsi' has been loaded but may be removed in a future release.
[2024-08-20 15:29:29.237] ERROR[359242] loader.c: Error loading module 'app_rpt.so': /usr/lib/asterisk/modules/app_rpt.so: cannot open shared object file: No such file or directory
[2024-08-20 15:29:29.237] ERROR[359242] loader.c: Error loading module 'chan_dahdi.so': /usr/lib/asterisk/modules/chan_dahdi.so: cannot open shared object file: No such file or directory
[2024-08-20 15:29:29.237] ERROR[359242] loader.c: Error loading module 'chan_sip.so': /usr/lib/asterisk/modules/chan_sip.so: cannot open shared object file: No such file or directory
[2024-08-20 15:29:29.237] ERROR[359242] loader.c: Error loading module 'res_timing_dahdi.so': /usr/lib/asterisk/modules/res_timing_dahdi.so: cannot open shared object file: No such file or directory
[2024-08-20 15:29:29.237] ERROR[359242] loader.c: Error loading module 'res_rtp_asterisk.so', missing dependency: res_pjproject
```

I will go see if I can manually place these items. We are close!

eliggett, 2024-08-20:
Yeah, sadly these so files were not built. It's not that there was an error, they are simply not listed in any Makefile or menuselect.makeopts file. 

I really don't see where the app_rpt script would possibly be inserting these to the menuselect.makeopts file.

eliggett, 2024-08-20:
ok, I do see where the script patches the Makefile. And mine appears patched correctly. But for whatever reason, ```make apps``` is glossing over app_rpt. hmm.

InterLinked1, 2024-08-20:
> ok, I do see where the script patches the Makefile. And mine appears patched correctly. But for whatever reason, `make apps` is glossing over app_rpt. hmm.

It's probably skipping `app_rpt` since it thinks the dependencies aren't met. You should be able to confirm this for certain be manually running menuselect and seeing if `app_rpt` is enabled in "Apps". My guess is it's not.

The dependencies here are referring to the dependencies in the `<depends>` xmldocs at the top of `app_rpt.c`. Do you have `res_curl` enabled for the build and you already have DAHDI present?

eliggett, 2024-08-20:
You are correct that menuselect will not let me add app_rpt. It is listed as "XXX". 

```res_curl``` is enabled. And I believe I have DAHDI already "present" per this line from the phreaknet.sh output:
```
DAHDI already present but install not forced, skipping...
```

Allan-N, 2024-08-20:
Q? did you start with the ASL3 Pi Appliance image? or were/are you starting from a clean Debian/Raspbian image without ASL3 already installed?  Note: my compile/build was using the former.

eliggett, 2024-08-20:
Starting from an ASL3 Pi image

Allan-N, 2024-08-20:
Q? @InterLinked1 : are there other packages that should be installed before using the phreaknet command to build?

InterLinked1, 2024-08-20:
> Q? @InterLinked1 : are there other packages that should be installed before using the phreaknet command to build?

I don't *think* so but I don't remember 100% off the top of my head right now.

Here's an easy, hacky way to determine what exactly is causing the issue if you have a few minutes to tinker:

Go to `app_rpt.c` and find the `<depend>` xml entries at the top of the file. One by one, eliminate each line (only one at a time) until menuselect will let you build app_rpt (run `make menuselect` and see if the XXX gives way to a toggleable checkbox). If you can identify that, that would be a big clue as to what the underlying issue is.

eliggett, 2024-08-20:
oh I like this hack, I will try :) Thank you!

eliggett, 2024-08-20:
...and the winner is tonezone. Removing that item causes menuselect to be happy. 

But I do have tonezone installed. Hmm!

```
root@simplex900:/usr/src/asterisk-20.9.2# dpkg --list | grep tonezone
ii  libtonezone-dev:arm64                1:3.1.0-2                               arm64        tonezone library (development)
ii  libtonezone2.0:arm64                 1:3.1.0-2                               arm64        tonezone library (runtime)
```

Allan-N, 2024-08-20:
And that's the package/dependency that was added after the initial "phreaknet" build.  I wonder if the dependencies had already been reviewed/established.

eliggett, 2024-08-20:
I wonder if it will build now? Let's see. Maybe it's just not detecting libtonezone2.0?

eliggett, 2024-08-20:
Not quite. Needs DAHDI headers and also includes a nice warning about tonezone being a char* and not an int. 

```
app_rpt.c: In function ‘rpt_call’:
app_rpt.c:1205:22: warning: format ‘%d’ expects argument of type ‘int’, but argument 6 has type ‘const char  ’ [-Wformat=]
 1205 |         ast_debug(1, "tzone: %d, patchquiet: %d, patchexten[0]: %d\n",
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1206 |                         myrpt->p.tonezone,
      |                         ~~~~~~~~~~~~~~~~~
      |                                 |
      |                                 const char *
app_rpt/rpt_bridging.c:29:10: fatal error: dahdi/user.h: No such file or directory
   29 | #include <dahdi/user.h>
      |          ^~~~~~~~~~~~~~
```

Allan-N, 2024-08-20:
I "think" you can check the dependencies with :
```
grep HAVE_ config.status | grep -e DAHDI -e CURL -e TONEZONE
```

eliggett, 2024-08-20:
And the grep output: 

```root@simplex900:/usr/src/asterisk-20.9.2# grep HAVE_ config.status | grep -e DAHDI -e CURL -e TONEZONE
D["HAVE_CURL"]=" 1"
```

Should I apt-install:
1. dahdi-dkms
2. dahdi-source
3. ???

InterLinked1, 2024-08-20:
> Not quite. Needs DAHDI headers and also includes a nice warning about tonezone being a char* and not an int.
>
> ```
> app_rpt.c: In function ‘rpt_call’:
> app_rpt.c:1205:22: warning: format ‘%d’ expects [...]

This is probably the real underlying issue. You should have DAHDI headers if it was installed properly. I'm guessing before chan_dahdi also had XXX in menuselect.

When you ran PhreakScript, did you pass both -d and -f, to force it to install DAHDI?

eliggett, 2024-08-20:
I did not use -f, I just followed the README blindly. 

With -f, which dahdi package is ultimately installed?

Allan-N, 2024-08-21:
From my "/usr/src" directory I removed the asterisk and dahdi directories/files (to start clean).  I then exec'd `phreaknet install -d -b -v 20 -u asterisk`. Then, after I saw "Menuselect build configuration successfully completed" message in the output I checked the config.status file content.  I now have :
```
D["HAVE_CURL"]=" 1"
D["HAVE_DAHDI"]=" 1"
...
D["HAVE_TONEZONE"]=" 1"
```
Still compiling ... :-)

eliggett, 2024-08-21:
It looks like ```-f -d``` will cause DAHDI to be built using the asterisk source: 
```
		menuselect/menuselect --enable chan_dahdi --enable app_meetme --enable app_flash menuselect.makeopts
```

Is that acceptable? Will it break the dkms one that ASL3 already has?

eliggett, 2024-08-21:
Do you think I could copy the headers from /usr/src/dahdi-3.3.0/include/dahdi (where they seem to be right now) to /usr/include/dahdi?

InterLinked1, 2024-08-21:
> It looks like `-f -d` will cause DAHDI to be built using the asterisk source:
> 
> ```
> 		menuselect/menuselect --enable chan_dahdi --enable app_meetme --enable app_flash menuselect.makeopts
> ```
> 
> Is that acceptable? Will it break the dkms one that ASL3 already has?

It looks like you're referring to `chan_dahdi`, which is not the same thing as DAHDI. DAHDI is the kernel module, `chan_dahdi` is the Asterisk channel driver for analog/digital TDM cards. There's no harm in building `chan_dahdi` whether you use it or not.

eliggett, 2024-08-21:
It's funny, all I have in /usr/include/dahdi is tonezone.h. 

Maybe I will symlink the other headers in there just for now. I agree that this should not be how it's done though. 

It looks like the DAHDI Makefile is supposed to handle this with the "make install-include" step, perhaps this just wasn't run? 
```
install-include:
	for hdr in $(INST_HEADERS); do \
		install -D -m 644 include/dahdi/$$hdr $(DESTDIR)/usr/include/dahdi/$$hdr; \
	done
	@rm -rf $(DESTDIR)/usr/include/zaptel
```

InterLinked1, 2024-08-21:
> It's funny, all I have in /usr/include/dahdi is tonezone.h.
>
> Maybe I will symlink the other headers in there just for now. I agree that this should not be how it's done though.
>
> It looks like the [...]

If you used `phreaknet install -d -f`, that all should be taken care of automatically, unless it failed to finish installing.

If you post the full log output from running that, I can take a look and comment further.

encbar5, 2024-08-21:
I think I've run into something like this. Sounds like you are trying to run a phreaknet install on top of a debian system (ASL3?) that has the debian dahdi dkms package installed. These don't play nicely together. If you are doing phreaknet, I recommend that you uninstall libtonzone and any debian packages related to dahdi, and let the phreaknet script build from source.

eliggett, 2024-08-21:
Copied headers as the DHADI Makefile would have done into /usr/include/dahdi directory. 

Was able to build app_rpt.so. 

However, it cannot be used due to an undefined symbol issue: 
```
ERROR[396061] loader.c: Error loading module 'app_rpt.so': /usr/lib/asterisk/modules/app_rpt.so: undefined symbol: tone_zone_set_zone
```

@encbar5 I don't see a way to install tonezone using the phreaknet.sh script. In any case, I do have it installed.

encbar5, 2024-08-21:
tonezone is installed as part of the dahdi build when using phreaknet.

eliggett, 2024-08-21:
Well, chan_dahdi will not compile. It's not clear to me if I actually need it, but anyway, here is the output: 
```
   [CC] chan_dahdi.c -> chan_dahdi.o
chan_dahdi.c: In function ‘dahdievent_to_analogevent’:
chan_dahdi.c:2669:14: error: ‘DAHDI_EVENT_PULSE’ undeclared (first use in this function); did you mean ‘DAHDI_EVENT_NONE’?
 2669 |         case DAHDI_EVENT_PULSE:
      |              ^~~~~~~~~~~~~~~~~
      |              DAHDI_EVENT_NONE
chan_dahdi.c:2669:14: note: each undeclared identifier is reported only once for each function it appears in
chan_dahdi.c:2672:14: error: ‘DAHDI_EVENT_PULSE_BREAK’ undeclared (first use in this function); did you mean ‘ANALOG_EVENT_PULSE_BREAK’?
 2672 |         case DAHDI_EVENT_PULSE_BREAK:
      |              ^~~~~~~~~~~~~~~~~~~~~~~
      |              ANALOG_EVENT_PULSE_BREAK
chan_dahdi.c: In function ‘dahdi_handle_event’:
chan_dahdi.c:7907:14: error: ‘DAHDI_EVENT_PULSE’ undeclared (first use in this function); did you mean ‘DAHDI_EVENT_NONE’?
 7907 |         case DAHDI_EVENT_PULSE:
      |              ^~~~~~~~~~~~~~~~~
      |              DAHDI_EVENT_NONE
chan_dahdi.c:7908:14: error: ‘DAHDI_EVENT_PULSE_BREAK’ undeclared (first use in this function); did you mean ‘ANALOG_EVENT_PULSE_BREAK’?
 7908 |         case DAHDI_EVENT_PULSE_BREAK:
      |              ^~~~~~~~~~~~~~~~~~~~~~~
      |              ANALOG_EVENT_PULSE_BREAK
chan_dahdi.c:7792:18: warning: unused variable ‘x’ [-Wunused-variable]
 7792 |         int res, x;
      |                  ^
make[1]: *** [/usr/src/asterisk-20.9.2/Makefile.rules:165: chan_dahdi.o] Error 1
```

The first missing identifier, ```DAHDI_EVENT_PULSE``` I cannot find anywhere. I searched all over github and I just can't find where it would be defined. 

https://github.com/search?q=+DAHDI_EVENT_PULSE&type=code

Anyway, I did find that I can add ```--with-tonezone``` to the configure arguments, and this should cause it to notice that I have this library and include it within the compile. Working through it.

eliggett, 2024-08-21:
Tried configuring --with-tonezone

The error is still here. app_rpt compiles just fine now, but it will not load into asterisk. 
```
[2024-08-20 21:47:57.034] ERROR[437871] loader.c: Error loading module 'app_rpt.so': /usr/lib/asterisk/modules/app_rpt.so: undefined symbol: tone_zone_set_zone
```
This symbol is, in fact, defined in the library: 
```
# readelf -Ws --dyn-syms /usr/lib/aarch64-linux-gnu/libtonezone.so | grep tone_zone_set_zone
    26: 0000000000001a24   160 FUNC    GLOBAL DEFAULT   11 tone_zone_set_zone
```

Interestingly though, the app_rpt.so itself does not seem to be looking for that library. Maybe the loader.c is supposed to figure it out? 

```
ldd /usr/lib/asterisk/modules/app_rpt.so | grep libtonezone
(not found)
```

Anyway, short of removing those calls, I am a bit stuck right now.

InterLinked1, 2024-08-21:
> Well, chan_dahdi will not compile. It's not clear to me if I actually need it, but anyway, here is the output:
>
> ```
>    [CC] chan_dahdi.c -> chan_dahdi.o
> chan_dahdi.c: In function [...]

These are out of tree patches in PhreakScript: https://github.com/search?q=repo%3AInterLinked1%2Fphreakscript%20DAHDI_EVENT_PULSE&type=code

As I said, you don't need chan_dahdi unless you are planning on using telephony cards, e.g. you have analog phones you want to connect to your Asterisk system, a PRI, etc.

However, the fact that you are getting this error suggests to me that you did *NOT* install DAHDI via PhreakScript, but rather it is picking up on the package. I think your problem is mixing and matching the two, pick one or the other and see if that works. If you are aiming to compile from source, I would advise uninstalling *all* DAHDI/Asterisk-related packages and following the instructions to compile from source. The two scripts (phreaknet.sh and rpt_install.sh) should take care of everything for you, and there shouldn't be any errors or issues with compiling or loading anything. If there are, then there's a mismatch somewhere.

InterLinked1, 2024-08-21:
To be more specific, I would recommend at least the flags `phreaknet install -d -b -v 20 -f`.

I think the -f is needed since the mechanism PhreakScript uses to detect if DAHDI is already present I think is just checking if /etc/dahdi exists, something very primitive like that. If it was created due to the package, it will never compile DAHDI without the -f option. But the best bet is probably to make sure it's fully uninstalled from the package and then do a compile from source.

eliggett, 2024-08-21:
Also, @InterLinked1 , do you recommend ```-u asterisk``` for the phreaknet script? Any reason to not do so?

eliggett, 2024-08-21:
Hi Allan, 

You did not specify the asterisk user to phreaknet, correct? (```-u asterisk```)

And you left DAHDI installed as it comes from ASL3, right? 

Thanks, 

--E

Allan-N, 2024-08-21:
> You did not specify the asterisk user to phreaknet, correct? (-u asterisk)

Actually, I started off specifying "-u asterisk" but the asterisk process was spitting out a bunch of what looked like permission-related messages with my SimpleUSB node configuration.  In the interest of getting to a quick resolution I dropped the "-u" option.

> And you left DAHDI installed as it comes from ASL3, right?

Yes.  Here, I think the "-f" option forced DAHDI to be compiled from source vs. using just the installed bits.

eliggett, 2024-08-21:
ok! We are cooking with gas now! I was able to compile! 

Systemd seems completely confused about if asterisk is running or not. I will reboot and see if I can figure it out. 

The rpt_install.sh script may have a typo, I received an error in the sound installer section (my line numbers won't match as I added some things to the script for my troubleshooting):

```
Checking for sounds...RPT sounds don't exist yet, adding them now...
/usr/src/rpt_install.sh: 184: cd: can't cd to rpt
# Here are the lines: 
183: mkdir /var/lib/asterisk/sounds/en/rpt
184: 	cd rpt
```

And I have this repeating error in the asterisk log, but I believe this is related to the confusion systemd is having. I'll remove the unused modules from the config later. 

```
[2024-08-21 12:12:54.248] Asterisk 20.9.2 built by root @ simplex900 on a aarch64 running Linux on 2024-08-21 18:51:44 UTC
[2024-08-21 12:12:54.285] NOTICE[556975] loader.c: 74 modules will be loaded.
[2024-08-21 12:12:54.334] NOTICE[556975] cdr.c: CDR simple logging enabled.
[2024-08-21 12:12:54.336] NOTICE[556975] dnsmgr.c: Managed DNS entries will be refreshed every 300 seconds.
[2024-08-21 12:12:54.358] NOTICE[556975] indications.c: Default country for indication tones: us
[2024-08-21 12:12:54.358] NOTICE[556975] indications.c: Setting default indication country to 'us'
[2024-08-21 12:12:54.443] NOTICE[556975] res_smdi.c: No SMDI interfaces are available to listen on, not starting SMDI listener.
[2024-08-21 12:12:54.452] WARNING[556975] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2024-08-21 12:12:54.452] WARNING[556975] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2024-08-21 12:12:54.452] NOTICE[556975] chan_dahdi.c: Ignoring any changes to 'userbase' (on reload) at line 23.
[2024-08-21 12:12:54.452] NOTICE[556975] chan_dahdi.c: Ignoring any changes to 'vmsecret' (on reload) at line 31.
[2024-08-21 12:12:54.452] NOTICE[556975] chan_dahdi.c: Ignoring any changes to 'hassip' (on reload) at line 35.
[2024-08-21 12:12:54.452] NOTICE[556975] chan_dahdi.c: Ignoring any changes to 'hasiax' (on reload) at line 39.
[2024-08-21 12:12:54.452] NOTICE[556975] chan_dahdi.c: Ignoring any changes to 'hasmanager' (on reload) at line 47.
[2024-08-21 12:12:54.452] WARNING[556975] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2024-08-21 12:12:54.452] WARNING[556975] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2024-08-21 12:12:54.729] NOTICE[556975] ./xpmr/xpmr.c: xpmr rxlpf: 0
[2024-08-21 12:12:54.729] NOTICE[556975] ./xpmr/xpmr.c: xpmr rxhpf: 1
[2024-08-21 12:12:54.729] NOTICE[556975] ./xpmr/xpmr.c: xpmr txlpf: 1
[2024-08-21 12:12:54.729] NOTICE[556975] ./xpmr/xpmr.c: xpmr txhpf: 1
[2024-08-21 12:12:54.768] NOTICE[556975] loader.c: Module 'res_adsi' has been loaded but may be removed in a future release.
[2024-08-21 12:12:54.806] ERROR[556975] loader.c: Error loading module 'chan_sip.so': /usr/lib/asterisk/modules/chan_sip.so: cannot open shared object file: No such file or directory
[2024-08-21 12:12:54.806] ERROR[556975] loader.c: Error loading module 'res_rtp_asterisk.so', missing dependency: res_pjproject
[2024-08-21 12:12:55.043] NOTICE[557018] app_rpt.c: Normal Repeater Init  617081
[2024-08-21 12:12:55.046] NOTICE[557021] chan_usbradio.c: Channel 617081: Set option TONE VERIFY, mode: OFF(0).
[2024-08-21 12:12:55.078] ERROR[557022] chan_usbradio.c: Channel 617081: Is not able to detach the USB device
[2024-08-21 12:12:55.604] ERROR[557022] chan_usbradio.c: Channel 617081: Is not able to detach the USB device
[2024-08-21 12:12:56.132] ERROR[557022] chan_usbradio.c: Channel 617081: Is not able to detach the USB device
[2024-08-21 12:12:56.660] ERROR[557022] chan_usbradio.c: Channel 617081: Is not able to detach the USB device
[2024-08-21 12:12:57.188] ERROR[557022] chan_usbradio.c: Channel 617081: Is not able to detach the USB device
[2024-08-21 12:12:57.716] ERROR[557022] chan_usbradio.c: Channel 617081: Is not able to detach the USB device
```

Let's see if a reboot does the trick. 

--E
de W6EL

eliggett, 2024-08-21:
A reboot did not remedy the issue. 

```journalctl -u asterisk```:
```
Aug 21 12:25:23 simplex900 systemd[1]: Stopped asterisk.service - Asterisk PBX.
Aug 21 12:25:23 simplex900 systemd[1]: asterisk.service: Consumed 5.830s CPU time.
Aug 21 12:25:23 simplex900 systemd[1]: Starting asterisk.service - Asterisk PBX...
Aug 21 12:26:12 simplex900 systemd[1]: asterisk.service: Failed with result 'protocol'.
Aug 21 12:26:12 simplex900 systemd[1]: Failed to start asterisk.service - Asterisk PBX.
Aug 21 12:26:12 simplex900 systemd[1]: asterisk.service: Consumed 1.700s CPU time.
Aug 21 12:26:17 simplex900 systemd[1]: asterisk.service: Scheduled restart job, restart counter is at 4.
Aug 21 12:26:17 simplex900 systemd[1]: Stopped asterisk.service - Asterisk PBX.
Aug 21 12:26:17 simplex900 systemd[1]: asterisk.service: Consumed 1.700s CPU time.
Aug 21 12:26:17 simplex900 systemd[1]: Starting asterisk.service - Asterisk PBX...
Aug 21 12:27:48 simplex900 systemd[1]: asterisk.service: start operation timed out. Terminating.
Aug 21 12:27:48 simplex900 systemd[1]: asterisk.service: Failed with result 'timeout'.
Aug 21 12:27:48 simplex900 systemd[1]: Failed to start asterisk.service - Asterisk PBX.
Aug 21 12:27:48 simplex900 systemd[1]: asterisk.service: Consumed 1.805s CPU time.
Aug 21 12:27:53 simplex900 systemd[1]: asterisk.service: Scheduled restart job, restart counter is at 5.
Aug 21 12:27:53 simplex900 systemd[1]: Stopped asterisk.service - Asterisk PBX.
Aug 21 12:27:53 simplex900 systemd[1]: asterisk.service: Consumed 1.805s CPU time.
Aug 21 12:27:53 simplex900 systemd[1]: Starting asterisk.service - Asterisk PBX...
Aug 21 12:29:23 simplex900 systemd[1]: asterisk.service: start operation timed out. Terminating.
Aug 21 12:29:25 simplex900 systemd[1]: asterisk.service: Failed with result 'timeout'.
Aug 21 12:29:25 simplex900 systemd[1]: Failed to start asterisk.service - Asterisk PBX.
Aug 21 12:29:25 simplex900 systemd[1]: asterisk.service: Consumed 5.764s CPU time.
Aug 21 12:29:30 simplex900 systemd[1]: asterisk.service: Scheduled restart job, restart counter is at 6.
Aug 21 12:29:30 simplex900 systemd[1]: Stopped asterisk.service - Asterisk PBX.
Aug 21 12:29:30 simplex900 systemd[1]: asterisk.service: Consumed 5.764s CPU time.
Aug 21 12:29:30 simplex900 systemd[1]: Starting asterisk.service - Asterisk PBX...
```

ps aux | grep asterisk:
```
asterisk    1766 26.9  0.7 2205992 62840 ?       Ssl  12:29   0:01 /usr/sbin/asterisk -g -f -p -U asterisk
asterisk    1767  0.0  0.0   2060   896 ?        S    12:29   0:00 astcanary /var/run/asterisk/alt.asterisk.canary.tweet.tweet.tweet 1766
```

```service asterisk status```:
```
● asterisk.service - Asterisk PBX
     Loaded: loaded (/lib/systemd/system/asterisk.service; enabled; preset: enabled)
     Active: activating (start) since Wed 2024-08-21 12:31:06 PDT; 31s ago
       Docs: man:asterisk(8)
   Main PID: 1851 (asterisk)
      Tasks: 43 (limit: 8737)
        CPU: 2.878s
     CGroup: /system.slice/asterisk.service
             ├─1851 /usr/sbin/asterisk -g -f -p -U asterisk
             └─1852 astcanary /var/run/asterisk/alt.asterisk.canary.tweet.tweet.tweet 1851

Aug 21 12:31:06 simplex900 systemd[1]: Starting asterisk.service - Asterisk PBX...
```

```astres.sh```:
```
Asterisk is not running!
```

Any ideas?

KB4MDD, 2024-08-21:
I suspect some of these issues are associated with installing source on the asl3 pi image.  I did not have these issues with a default debian install.  The phreaknet install will create an init.d entry for asterisk.

https://github.com/AllStarLink/ASL3/blob/5630803e7378a5ad1b964e921bf77356e49773ec/RPi-install.md  These notes talk about removing that and creating systemd files.  (The systemd files may be there as a result of using the asl3 pi image.)

eliggett, 2024-08-21:
Indeed, there was an init.d file present. It is removed now. 

With that said though, asterisk will not fork to background, it is stuck trying to get the usbradio device over and over. 
```
[2024-08-21 12:54:10.389] ERROR[5451] chan_usbradio.c: Channel 617081: Is not able to detach the USB device
```
Suspect a device permissions issue now. 

The config file and asl-find-sound show the correct device. udev still has the 90-asl3.rules file, which appears unchanged. There are also two new files, xpp.rules and dahdi.rules.

KB4MDD, 2024-08-21:
What happens when you `systemctl stop asterisk` and then` sudo -s` and `run asterisk -cg`?

eliggett, 2024-08-21:
```
root@simplex900:/tmp# asterisk -cg
Asterisk 20.9.2, Copyright (C) 1999 - 2022, Sangoma Technologies Corporation and others.
Created by Mark Spencer <markster@digium.com>
Asterisk comes with ABSOLUTELY NO WARRANTY; type 'core show warranty' for details.
This is free software, with components licensed under the GNU General Public
License version 2 and other licenses; you are welcome to redistribute it under
certain conditions. Type 'core show license' for details.
=========================================================================
Running as user 'asterisk'
Running under group 'asterisk'
PBX UUID: 898c35ae-1c1e-4e17-8a66-3499d7598f07
[2024-08-21 13:26:23.962] NOTICE[2602]: loader.c:2559 load_modules: 71 modules will be loaded.
[ Initializing Custom Configuration Options ]
[2024-08-21 13:26:24.011] NOTICE[2602]: cdr.c:4568 cdr_toggle_runtime_options: CDR simple logging enabled.
[2024-08-21 13:26:24.013] NOTICE[2602]: dnsmgr.c:493 do_reload: Managed DNS entries will be refreshed every 300 seconds.
[2024-08-21 13:26:24.036] NOTICE[2602]: indications.c:1100 load_indications: Default country for indication tones: us
[2024-08-21 13:26:24.036] NOTICE[2602]: indications.c:424 ast_set_indication_country: Setting default indication country to 'us'
[2024-08-21 13:26:24.130] NOTICE[2602]: res_smdi.c:1424 load_module: No SMDI interfaces are available to listen on, not starting SMDI listener.
[2024-08-21 13:26:24.138] WARNING[2602]: chan_dahdi.c:18822 process_dahdi: Only FXO signalled channels may belong to a call group
[2024-08-21 13:26:24.138] WARNING[2602]: chan_dahdi.c:18830 process_dahdi: Only FXO signalled channels may belong to a pickup group
[2024-08-21 13:26:24.139] NOTICE[2602]: chan_dahdi.c:19987 process_dahdi: Ignoring any changes to 'userbase' (on reload) at line 23.
[2024-08-21 13:26:24.139] NOTICE[2602]: chan_dahdi.c:19987 process_dahdi: Ignoring any changes to 'vmsecret' (on reload) at line 31.
[2024-08-21 13:26:24.139] NOTICE[2602]: chan_dahdi.c:19987 process_dahdi: Ignoring any changes to 'hassip' (on reload) at line 35.
[2024-08-21 13:26:24.139] NOTICE[2602]: chan_dahdi.c:19987 process_dahdi: Ignoring any changes to 'hasiax' (on reload) at line 39.
[2024-08-21 13:26:24.139] NOTICE[2602]: chan_dahdi.c:19987 process_dahdi: Ignoring any changes to 'hasmanager' (on reload) at line 47.
[2024-08-21 13:26:24.139] WARNING[2602]: chan_dahdi.c:18822 process_dahdi: Only FXO signalled channels may belong to a call group
[2024-08-21 13:26:24.139] WARNING[2602]: chan_dahdi.c:18830 process_dahdi: Only FXO signalled channels may belong to a pickup group
[2024-08-21 13:26:24.432] NOTICE[2602]: ./xpmr/xpmr.c:1774 createPmrChannel: xpmr rxlpf: 0
[2024-08-21 13:26:24.432] NOTICE[2602]: ./xpmr/xpmr.c:1775 createPmrChannel: xpmr rxhpf: 1
[2024-08-21 13:26:24.432] NOTICE[2602]: ./xpmr/xpmr.c:1776 createPmrChannel: xpmr txlpf: 1
ERROR: numrxcodes != numtxcodes
[2024-08-21 13:26:24.432] NOTICE[2602]: ./xpmr/xpmr.c:1777 createPmrChannel: xpmr txhpf: 1
[2024-08-21 13:26:24.473] NOTICE[2602]: loader.c:2659 load_modules: Module 'res_adsi' has been loaded but may be removed in a future release.
Asterisk Ready.
*CLI> [2024-08-21 13:26:24.736] NOTICE[2643]: app_rpt.c:5706 rpt_master: Normal Repeater Init  617081
ERROR: numrxcodes != numtxcodes
[2024-08-21 13:26:24.739] NOTICE[2647]: chan_usbradio.c:2520 usbradio_setoption: Channel 617081: Set option TONE VERIFY, mode: OFF(0).
ALSA lib confmisc.c:165:(snd_config_get_card) Cannot get card index for 1
ALSA lib confmisc.c:165:(snd_config_get_card) Cannot get card index for 1
ALSA lib confmisc.c:165:(snd_config_get_card) Cannot get card index for 1
[2024-08-21 13:26:24.855] ERROR[2648]: chan_usbradio.c:992 hidthread: Channel 617081: Is not able to detach the USB device
ALSA lib confmisc.c:165:(snd_config_get_card) Cannot get card index for 1
ALSA lib confmisc.c:165:(snd_config_get_card) Cannot get card index for 1
ALSA lib confmisc.c:165:(snd_config_get_card) Cannot get card index for 1
[2024-08-21 13:26:25.471] ERROR[2648]: chan_usbradio.c:992 hidthread: Channel 617081: Is not able to detach the USB device
ALSA lib confmisc.c:165:(snd_config_get_card) Cannot get card index for 1
ALSA lib confmisc.c:165:(snd_config_get_card) Cannot get card index for 1
```

eliggett, 2024-08-21:
I'll also add the following since it seems to be an issue with the sound card in some way shape or forum:
The card: 
```
# asl-find-sound
1-1.4:1.0	-->	0d8c:0013 C-Media Electronics Inc.
# lsusb | grep 0d8c
Bus 001 Device 003: ID 0d8c:0013 C-Media Electronics, Inc. USB Audio Device
```

usbradio.conf node stanza:
```
[617081](node-main)
rxboost = yes
; 900 MHz System
;;;;; ASL3 Tune settings ;;;;;
;devstr = 1-1.3:1.0
;devstr = 1-1.4.1:1.0
devstr = 1-1.4:1.0
rxmixerset = 50
txmixaset = 800
txmixbset = 500
rxvoiceadj = 1.125000
rxctcssadj = 0.500000
txctcssadj = 200
rxsquelchadj = 500
```

eliggett, 2024-08-22:
I thought I'd try a different USB port. I moved from the lower black USB port to the upper most blue USB port:
```
# asl-find-sound
1-1.1:1.0	-->	0d8c:0013 C-Media Electronics Inc.
```
And then I adjusted the usbradio.conf file  ```devstr``` to match. 
Here is the result: 
```
[2024-08-22 15:44:23.298] NOTICE[2962] app_rpt.c: Normal Repeater Init  617081
[2024-08-22 15:44:23.301] NOTICE[2966] chan_usbradio.c: Channel 617081: Set option TONE VERIFY, mode: OFF(0).
[2024-08-22 15:44:23.325] ERROR[2967] chan_usbradio.c: Channel 617081: Device string 1-1.1:1.0 was not found.
root@simplex900:/var/log/asterisk# asl-find-sound
1-1.1:1.0	-->	0d8c:0013 C-Media Electronics Inc.
```

I then removed the device entry from the usbradio.conf file, hoping the automatic device finder would do its job. Here is the result of that: 
```
[2024-08-22 15:50:12.837] NOTICE[3148] app_rpt.c: Normal Repeater Init  617081
[2024-08-22 15:50:12.840] NOTICE[3165] chan_usbradio.c: Channel 617081: Set option TONE VERIFY, mode: OFF(0).
[2024-08-22 15:50:12.865] ERROR[3166] chan_usbradio.c: Channel 617081: No USB devices are available for assignment.
```

I do note that the various scripts used here invoke a system-wide update (apt upgrade -y). I wonder if I have stumbled on to some kind of incompatible mix?

eliggett, 2024-08-22:
Results are effectively the same with simpleusb instead of usbradio:
```
[2024-08-22 16:02:23.173] NOTICE[3630]: chan_simpleusb.c:2527 simpleusb_setoption: Channel 617081: Set option TONE VERIFY, mode: OFF(0).
[2024-08-22 16:02:23.197] ERROR[3631]: chan_simpleusb.c:941 hidthread: Channel 617081: Device string 1-1.1:1.0 was not found.
```

KB4MDD, 2024-08-22:
Hopefully @InterLinked1 can comment.  I don't have a recommendation.  As noted previously, I loaded two systems last week and did not have any issues.  I have never attempted to compile from source on the asl3 pi image.

If you have a spare card, you might consider loading the debian image, followed by the instructions for compiling the source code.

eliggett, 2024-08-23:
I have found some things: 

1. There were some new udev files added with the asterisk install. I removed these files:
```
# group 1: 
/dev/udev/rules.d/xpp.rules
/dev/udev/rules.d/dahdi.rules

# group 2:
/lib/udev/rules.d/60-vtcore.rules
/lib/udev/rules.d/90-pi-bluetooth.rules
```

Removing the files in group 1 (with a reboot) had no effect. 

Removing the files in group 2 caused the USB paths (the ones that are in /dev/bus/usb/001) to match the paths from my other Pi node. Prior to this, they were off by one, which did not make any sense. Oddly, the devstr listed in ```udevadm info -q all -n /dev/usb/bus/001/00X | grep M: ``` would match between the two raspberry Pi machines I have. But the "X" would be different. Now, with the files in group 2 removed, the "X" is the same between the two machines. 

2. I got wild and did this: 
```chmod -R 777 /dev/usb/bus/001```
And the error message changed! The new message was: 
```
[2024-08-22 16:54:37.644] NOTICE[1717] app_rpt.c: Normal Repeater Init  617081
[2024-08-22 16:54:37.647] NOTICE[1721] chan_simpleusb.c: Channel 617081: Set option TONE VERIFY, mode: OFF(0).
[2024-08-22 16:54:37.722] ERROR[1722] chan_simpleusb.c: Channel 617081: Unable to open DSP device 1: Permission denied.
[2024-08-22 16:54:37.725] ERROR[1721] chan_simpleusb.c: Channel 617081: Unable to open DSP device 1: Permission denied.
[2024-08-22 16:54:37.744] ERROR[1721] chan_simpleusb.c: Channel 617081: Unable to open DSP device 1: Permission denied.
```

3. Realizing that this error has now moved from usb device to "dsp" device...
```
chmod 777 /dev/dsp*
```

4. And now asterisk is starting without errors:
```
[2024-08-22 17:05:51.901] NOTICE[2070] app_rpt.c: Normal Repeater Init  617081
[2024-08-22 17:05:51.904] NOTICE[2073] chan_simpleusb.c: Channel 617081: Set option TONE VERIFY, mode: OFF(0).
[2024-08-22 17:05:51.944] WARNING[2073] chan_simpleusb.c: Possibly stuck USB read channel. [617081]
[2024-08-22 17:05:51.970] WARNING[2073] chan_simpleusb.c: USB read channel [617081] was not stuck.
```
(That warning, I think, is somewhat normal, and it only shows up once.)

However, I cannot get asterisk to start using systemctl or service. The process will start, but something fails to fork to background. In the case of ```systemctl start asterisk```, asterisk starts, but the console hangs waiting for something, and in about 90 seconds, systemd kills off asterisk:
```
Job for asterisk.service failed because a timeout was exceeded.
See "systemctl status asterisk.service" and "journalctl -xeu asterisk.service" for details.
```
Maybe there is something wrong with the startup scripts and/or the canary? 

When it tries to start asterisk, I see these processes: 
```
asterisk    2125 10.4  0.7 2205720 63028 ?       Ssl  17:07   0:07 /usr/sbin/asterisk -g -f -p -U asterisk
asterisk    2126  0.0  0.0   2060   896 ?        S    17:07   0:00 astcanary /var/run/asterisk/alt.asterisk.canary.tweet.tweet.tweet 2125
```

I think I'm close. It looks like one (or possibly several) of the attempts made at installing asterisk did something to the startup and/or udev scripts.

InterLinked1, 2024-08-23:
> I do note that the various scripts used here invoke a system-wide update (apt upgrade -y). I wonder if I have stumbled on to some kind of incompatible mix?

`phreaknet install` does update/update the packages (update/upgrade, but not dist-upgrade). I haven't heard of that causing issues before but I suppose there could be something here we aren't aware of.

As far as the issues you are seeing with the USB channel drivers, I don't have any insight on that unfortunately.

> With that said though, asterisk will not fork to background, it is stuck trying to get the usbradio device over and over.

There is an issue here, but this also points out something else, if this is something that can possibly happen, maybe the opening of USB devices should happen in its own thread, instead of the main Asterisk thread (which calls load_module), to avoid blocking everything else and ultimately blocking startup. What do you think @KB4MDD ?

There is also a way in Asterisk to execute code when Asterisk is done loading OR as soon as the module loads as a callback. Here is an example: https://github.com/asterisk/asterisk/blob/30a936cd250099ea1b71553c2c678ba6daeff4d0/pbx/pbx_config.c#L1981

[40 more replies]

closed 2024-09-09 by Allan-N