AllStarLink/app_rpt #391: Load bridge_simple by default

Mason10198, 2024-08-26:
I have discovered that autopatch will not work properly in ASL3 until the following line is **manually** added to `modules.conf`:
```
load => bridge_simple.so
```
Currently, `bridge_simple.so` is not referenced at all in the `modules.conf` provided with ASL3 and results in the following error:
```
bridge.c:776 bridge_base_init: Bridge 19cd1d38-b099-4684-9397-babcc4a22322: Could not create class basic.  No technology to support it.
```

InterLinked1, 2025-01-11:
> Moving to the correct repo

This is only in this repo because `configs/rpt/modules.conf` is here, but it isn't a code change.

I personally think `configs/rpt` should be moved into a separate repository at some point, and only `configs/sample` should stay in this repo, since it belongs with the code.

Allan-N, 2025-01-29:
Our plans are to move ASL3 forward from Asterisk 20.x.x. to Asterisk 22.x.x.  As part of that move we need to update the `modules.conf` file.  My plan is to include all of the modules needed for SIP phones and Autopatch configurations.  The extra module references will be IN the file but commented out so no more copy/paste issues, just removing a bunch of ";"s.

closed 2025-02-24 by Allan-N