AllStarLink/app_rpt #392: DTMF linking cut off for > 4 digit node numbers (with Tailscale)

dbehnke, 2024-08-26:
**Describe the bug**
When I attempt to do *3 + >4 digit node number, linking does not work.. digits after 4 get cut off

**To Reproduce**
Steps to reproduce the behavior:
1. monitor log /var/log/asterisk/messages
2. dial *3 ######
3. notice in log only *3 #### get read. last ## don't get acknowledged
4.  repeat the steps with a 4 digit node number, such as 2560, link works.
5. using a macro such as *51 and configuring macro to *803##### or similar does not work.
6. using rpt cmd [mynode] ilink 6 ##### works. 

**Expected behavior**
I can dial node numbers > 4 digits

**sample log**
This is me trying to connect to 594950 using dtmf, notice only 5949 gets picked up

```
[2024-08-23 11:42:13.480] DTMF[1252] channel.c: DTMF begin '*' received on SimpleUSB/550464
[2024-08-23 11:42:13.480] DTMF[1252] channel.c: DTMF begin passthrough '*' on SimpleUSB/550464
[2024-08-23 11:42:13.928] DTMF[1252] channel.c: DTMF end '*' received on SimpleUSB/550464, duration 447 ms
[2024-08-23 11:42:13.928] DTMF[1252] channel.c: DTMF end accepted with begin '*' on SimpleUSB/550464
[2024-08-23 11:42:13.928] DTMF[1252] channel.c: DTMF end passthrough '*' on SimpleUSB/550464
[2024-08-23 11:42:14.398] DTMF[1252] channel.c: DTMF begin '3' received on SimpleUSB/550464
[2024-08-23 11:42:14.398] DTMF[1252] channel.c: DTMF begin passthrough '3' on SimpleUSB/550464
[2024-08-23 11:42:14.760] DTMF[1252] channel.c: DTMF end '3' received on SimpleUSB/550464, duration 362 ms
[2024-08-23 11:42:14.760] DTMF[1252] channel.c: DTMF end accepted with begin '3' on SimpleUSB/550464
[2024-08-23 11:42:14.760] DTMF[1252] channel.c: DTMF end passthrough '3' on SimpleUSB/550464
[2024-08-23 11:42:15.123] DTMF[1252] channel.c: DTMF begin '5' received on SimpleUSB/550464
[2024-08-23 11:42:15.123] DTMF[1252] channel.c: DTMF begin passthrough '5' on SimpleUSB/550464
[2024-08-23 11:42:15.422] DTMF[1252] channel.c: DTMF end '5' received on SimpleUSB/550464, duration 298 ms
[2024-08-23 11:42:15.422] DTMF[1252] channel.c: DTMF end accepted with begin '5' on SimpleUSB/550464
[2024-08-23 11:42:15.422] DTMF[1252] channel.c: DTMF end passthrough '5' on SimpleUSB/550464
[2024-08-23 11:42:15.806] DTMF[1252] channel.c: DTMF begin '9' received on SimpleUSB/550464
[2024-08-23 11:42:15.806] DTMF[1252] channel.c: DTMF begin passthrough '9' on SimpleUSB/550464
[2024-08-23 11:42:16.020] DTMF[1252] channel.c: DTMF end '9' received on SimpleUSB/550464, duration 213 ms
[2024-08-23 11:42:16.020] DTMF[1252] channel.c: DTMF end accepted with begin '9' on SimpleUSB/550464
[2024-08-23 11:42:16.020] DTMF[1252] channel.c: DTMF end passthrough '9' on SimpleUSB/550464
[2024-08-23 11:42:16.445] DTMF[1252] channel.c: DTMF begin '4' received on SimpleUSB/550464
[2024-08-23 11:42:16.445] DTMF[1252] channel.c: DTMF begin passthrough '4' on SimpleUSB/550464
[2024-08-23 11:42:16.701] DTMF[1252] channel.c: DTMF end '4' received on SimpleUSB/550464, duration 255 ms
[2024-08-23 11:42:16.701] DTMF[1252] channel.c: DTMF end accepted with begin '4' on SimpleUSB/550464
[2024-08-23 11:42:16.701] DTMF[1252] channel.c: DTMF end passthrough '4' on SimpleUSB/550464
[2024-08-23 11:42:17.043] DTMF[1252] channel.c: DTMF begin '9' received on SimpleUSB/550464
[2024-08-23 11:42:17.043] DTMF[1252] channel.c: DTMF begin passthrough '9' on SimpleUSB/550464
[2024-08-23 11:42:17.299] DTMF[1252] channel.c: DTMF end '9' received on SimpleUSB/550464, duration 255 ms
[2024-08-23 11:42:17.299] DTMF[1252] channel.c: DTMF end accepted with begin '9' on SimpleUSB/550464
[2024-08-23 11:42:17.299] DTMF[1252] channel.c: DTMF end passthrough '9' on SimpleUSB/550464
```

KB4MDD, 2024-08-26:
Here is some debugging information from my computer.  I was able to get the failure as soon as I started asterisk.  I will not that I enabled res_resolver_unbounded.so and set the nameserver to 1.1.1.1.

```
k4mdddevelop*CLI> rpt fun 2231 *353431
[2024-08-26 15:22:19.543] DEBUG[43943]: app_rpt/rpt_telemetry.c:381 cancel_pfxtone: cancel_pfxfone!!
[2024-08-26 15:22:19.543] DEBUG[43943]: app_rpt.c:1435 collect_function_digits: digits=3  source=0
[2024-08-26 15:22:19.543] DEBUG[43943]: app_rpt.c:1487 collect_function_digits: @@@@ action: ilink, param = 3
[2024-08-26 15:22:19.543] DEBUG[43943]: app_rpt.c:1493 collect_function_digits: @@@@ table index i = 3
[2024-08-26 15:22:19.543] DEBUG[43943]: app_rpt/rpt_functions.c:69 function_ilink: @@@@ ilink param = 3, digitbuf =
[2024-08-26 15:22:19.543] DEBUG[43943]: app_rpt.c:1505 collect_function_digits: rv=0
[2024-08-26 15:22:19.654] DEBUG[43943]: app_rpt/rpt_telemetry.c:381 cancel_pfxtone: cancel_pfxfone!!
[2024-08-26 15:22:19.654] DEBUG[43943]: app_rpt.c:1435 collect_function_digits: digits=35  source=0
[2024-08-26 15:22:19.654] DEBUG[43943]: app_rpt.c:1487 collect_function_digits: @@@@ action: ilink, param = 3
[2024-08-26 15:22:19.655] DEBUG[43943]: app_rpt.c:1493 collect_function_digits: @@@@ table index i = 3
[2024-08-26 15:22:19.655] DEBUG[43943]: app_rpt/rpt_functions.c:69 function_ilink: @@@@ ilink param = 3, digitbuf = 5
[2024-08-26 15:22:19.655] DEBUG[43943]: app_rpt/rpt_config.c:312 tlb_query_node_exists: chan_tlb not loaded.
[2024-08-26 15:22:19.655] NOTICE[43943]: app_rpt/rpt_config.c:624 node_lookup: Longest node 6.
[2024-08-26 15:22:19.655] DEBUG[43943]: app_rpt.c:1505 collect_function_digits: rv=0
[2024-08-26 15:22:19.765] DEBUG[43943]: app_rpt/rpt_telemetry.c:381 cancel_pfxtone: cancel_pfxfone!!
[2024-08-26 15:22:19.765] DEBUG[43943]: app_rpt.c:1435 collect_function_digits: digits=353  source=0
[2024-08-26 15:22:19.766] DEBUG[43943]: app_rpt.c:1487 collect_function_digits: @@@@ action: ilink, param = 3
[2024-08-26 15:22:19.766] DEBUG[43943]: app_rpt.c:1493 collect_function_digits: @@@@ table index i = 3
[2024-08-26 15:22:19.766] DEBUG[43943]: app_rpt/rpt_functions.c:69 function_ilink: @@@@ ilink param = 3, digitbuf = 53
[2024-08-26 15:22:19.766] DEBUG[43943]: app_rpt/rpt_config.c:312 tlb_query_node_exists: chan_tlb not loaded.
[2024-08-26 15:22:19.766] NOTICE[43943]: app_rpt/rpt_config.c:624 node_lookup: Longest node 6.
[2024-08-26 15:22:19.766] DEBUG[43943]: app_rpt.c:1505 collect_function_digits: rv=0
[2024-08-26 15:22:19.885] DEBUG[43943]: app_rpt/rpt_telemetry.c:381 cancel_pfxtone: cancel_pfxfone!!
[2024-08-26 15:22:19.885] DEBUG[43943]: app_rpt.c:1435 collect_function_digits: digits=3534  source=0
[2024-08-26 15:22:19.885] DEBUG[43943]: app_rpt.c:1487 collect_function_digits: @@@@ action: ilink, param = 3
[2024-08-26 15:22:19.885] DEBUG[43943]: app_rpt.c:1493 collect_function_digits: @@@@ table index i = 3
[2024-08-26 15:22:19.885] DEBUG[43943]: app_rpt/rpt_functions.c:69 function_ilink: @@@@ ilink param = 3, digitbuf = 534
[2024-08-26 15:22:19.886] DEBUG[43943]: app_rpt/rpt_config.c:312 tlb_query_node_exists: chan_tlb not loaded.
[2024-08-26 15:22:19.886] NOTICE[43943]: app_rpt/rpt_config.c:624 node_lookup: Longest node 6.
[2024-08-26 15:22:19.886] DEBUG[43943]: app_rpt.c:1505 collect_function_digits: rv=0
[2024-08-26 15:22:19.968]   == Manager 'admin' logged on from 127.0.0.1
[2024-08-26 15:22:19.969]   == Manager 'admin' logged on from 127.0.0.1
[2024-08-26 15:22:19.995] DEBUG[43943]: app_rpt/rpt_telemetry.c:381 cancel_pfxtone: cancel_pfxfone!!
[2024-08-26 15:22:19.995] DEBUG[43943]: app_rpt.c:1435 collect_function_digits: digits=35343  source=0
[2024-08-26 15:22:19.995] DEBUG[43943]: app_rpt.c:1487 collect_function_digits: @@@@ action: ilink, param = 3
[2024-08-26 15:22:19.996] DEBUG[43943]: app_rpt.c:1493 collect_function_digits: @@@@ table index i = 3
[2024-08-26 15:22:19.996] DEBUG[43943]: app_rpt/rpt_functions.c:69 function_ilink: @@@@ ilink param = 3, digitbuf = 5343
[2024-08-26 15:22:19.996] DEBUG[43943]: app_rpt/rpt_config.c:312 tlb_query_node_exists: chan_tlb not loaded.
[2024-08-26 15:22:19.996] DEBUG[43943]: app_rpt/rpt_config.c:396 node_lookup_bydns: Resolving DNS SRV records for: _iax._udp.5343.nodes.allstarlink.org
[2024-08-26 15:22:23.137] DEBUG[43943]: app_rpt/rpt_config.c:411 node_lookup_bydns: No SRV records returned for _iax._udp.5343.nodes.allstarlink.org
[2024-08-26 15:22:23.137] NOTICE[43943]: app_rpt/rpt_config.c:624 node_lookup: Longest node 6.
[2024-08-26 15:22:23.138] DEBUG[43943]: app_rpt.c:1505 collect_function_digits: rv=0
[2024-08-26 15:22:23.138] DEBUG[43943]: app_rpt/rpt_telemetry.c:381 cancel_pfxtone: cancel_pfxfone!!
```

KB4MDD, 2024-08-27:
The main problem is the time to look up DNS records.  I have confirmed it here.  

app_rpt does not process more that 4 digits due to the DTMF timer expiring.   The DTMF timer is hard coded for 3 seconds.  If the DNS lookup does not complete in 3 seconds, the DTMF timer expires and the DTMF buffer is reset.  At that point you see a `cancel_pfxfone!!` message and DTMF processing stops.

jxmx, 2024-08-27:
So, to be clear, there's a complete time allocation of 3s to receive all DTMF tones and then execute the complete round-trip of the DNS lookup? In general, the DNS lookup de facto timers are to allocate 4 seconds for DNS lookups (2s for a first lookup, 2s to do a second resolver lookup if the first fails/timesout).

KB4MDD, 2024-08-27:
I was not clear.  The DTMF timer is reset with each digit.

jxmx, 2024-08-27:
https://allstarlink.github.io/user-guide/known-issues-diffs/#dtmf-linking-dns-lookup-timeouts

jxmx, 2025-01-03:
With the fixed DNS services, this no longer appears to be a problem.

closed 2025-01-03 by jxmx