AllStarLink/app_rpt #395: app_rpt: AMI output consistency issues with rpt_manager.c

W6HBR, 2024-09-03:
I've been doing data captures from the AMI and found a couple of inconsistencies with a couple of the data items being reported.

First is for RPT_RXKEYED - there are two extra CRLF's added which cause the last line of the reported data to be orphaned into its own message as well as adding an extra unnecessary end-of-message terminator.  AMI messages are terminated with a double CRLF (\r\n\r\n).  In this case, the LastTxKeyedTime has an extra CRLF before and after the reported line. (see image)
[image]

Not sure where the extra CRLF's in RPT_RXKEYED are coming from as the rpt_manager_trigger function appears correct.  Possibly the date/time value being passed already has a CRLF at the end causing this.
[image]
 
 
The second issue is purely cosmetic, but also adds an extra unnecessary end-of-message terminator to the xstat message.
[image]

This extra CRLF can be found on line 333 of rpt_manager.c
[image]

W6HBR, 2024-09-03:
From some searching, it looks like the ctime() function adds the CRLF to the returned value.  If that is the case, then the extra CRLF on lines 27 & 28 should be removed.
[image]

Allan-N, 2024-09-15:
Do you have any code / scripts that could be used to show (reproduce) the output?

Allan-N, 2024-09-15:
And, yes, the ctime() strings ends in '\n'.  Easy enough to fix.

W6HBR, 2024-09-15:
I can capture the raw AMI traffic if that is what you're asking for.  That's how I was able to figure out where the output was incorrect.

Allan-N, 2024-09-15:
I was looking for some code, command, or script that I could run to see the output before making a change and confirming the expected/improved output afterwards.  Think "test code" :-)

W6HBR, 2024-09-16:
To my knowledge, the only way to see the AMI traffic is to authenticate to an AMI session and capture it.  If you don't have a utility to do so, I can send you one I created for just this purpose.  You can then copy the output to Notepad++ or other editor where you can see the embedded\hidden CRLF characters.

[image]

[image]

closed 2024-09-20 by Allan-N

Allan-N, 2024-09-22:
The fix for this issue has been incorporated into ASL3-Asterisk 20.9.3+asl3-3.0.5-1