AllStarLink/app_rpt #398: Autopatch keeps linked nodes keyed

eliggett, 2024-09-12:
Greetings, 

With the latest updates to app_rpt and dahdi-linux, autopatch on my simplex node is working smoothly with a functioning vox. 

However, if any nodes are linked during an autopatch call, the linked nodes remain  keyed for the entirety of the autopatch call. This is not only inconvenient, but it can tie up resources significantly depending on what you have connected, making it difficult for others (with half-duplex nodes) to disconnect from your active node with the autopatch call. 

It would be great if the same PTT action created by the autopatch vox could be sent to remote nodes so that they are not keyed the entire time. 

Thanks for your consideration, 

--E
de W6EL

jxmx, 2024-10-06:
@eliggett - Does the DAHDI 3.4.0 modules fix this issue for you?

eliggett, 2024-10-06:
Negative, still keying the other nodes full time.