AllStarLink/app_rpt #399: Simpleusb and Usbradio drivers are introducing digital clipping into all Tx audio

davidgsd, 2024-09-14:
Please see comments in https://github.com/AllStarLink/app_rpt/pull/396.  While implementing an Rx Audio Statistics feature I stumbled across a HUGE problem in the channel drivers.

The sections of code that I if 0'd out and added comments to are a major, major issue and bug, as they **will introduce** distortion in Tx audio on all ASL nodes that use any URI made in the past 10 years. I cannot stress enough how serious this is.

If incoming IAX audio happens to be less than 90%, which is usually the case for most users, you won't get distortion because these blocks of code are scaling by ~110%, which is only a 0.83dB level difference, but any IAX audio between 0dBFS and -0.83dBFS will be digitally clipped and distorted ie. sine waves squared off at the top and bottom (which has a very negative impact on audio quality and is very noticeable) on any and all nodes using ASL that have any URI which is not using the original 10+ year-old CM108 IC.

Once this digital clipping is introduced, it doesn't matter what any of your mixer settings are -- the Tx audio will be distorted. Let me know if anyone has any questions on this or if anything I have explained above is not clear.

Fixing this now will greatly improve audio quality while bringing Tx audio levels to exactly where they should be, which is difference of -0.83dB. Differences of less than 1dB are usually not even perceptible thus fixing this will not cause "low" audio.

davidgsd, 2024-09-14:
Also see this post for additional notes: https://community.allstarlink.org/t/asl-usb-audio-notes/21832/

davidgsd, 2024-09-14:
Below are the code excerpts in question:

chan_simpleusb.c:
```
					/* TBR - below appears to be an attempt to match levels to the original CM108
					 * IC which has been out of production for over 10 years. Scaling audio to 
					 * 109.375% will result in clipping! Any adjustments for CM1xxx gain differences
					 * should be made in the mixer settings, not in the audio stream itself.
					 */
					/* Adjust the audio level for CM119 A/B devices */
					if (o->devtype != C108_PRODUCT_ID) {
						register int v;

						sp = (short *) o->simpleusb_write_buf;
						for (i = 0; i < FRAME_SIZE; i++) {
							v = *sp;
							v += v >> 3;	/* add *.125 giving * 1.125 */
							v -= *sp >> 5;	/* subtract *.03125 giving * 1.09375 */
							if (v > 32765.0) {
								v = 32765.0;
							} else if (v < -32765.0) {
								v = -32765.0;
							}
							*sp++ = v;
						}
					}
```

chan_usbradio.c:

```
	/* TBR - below appears to be an attempt to match levels to the original CM108
	 * IC which has been out of production for over 10 years. Scaling all rx audio to 
	 * 80% would result in a 20% loss in dynamic range, added quantization noise, and 
	 * outgoing IAX audio reduced by 2dB. Any adjustments for CM1xxx IC gain differences
	 * should be made in the mixer settings, not in the audio stream itself.
	 */
	/* Decrease the audio level for CM119 A/B devices */
	if (o->devtype != C108_PRODUCT_ID) {
		register short *sp = (short *) (o->usbradio_read_buf + o->readpos);
		register float v;
		register int i;

		for (i = 0; i < res / 2; i++) {
			v = ((float) *sp) * 0.800;
			*sp++ = (int) v;
		}
	}
```

```
	/* TBR - below appears to be an attempt to match levels to the original CM108
	 * IC which has been out of production for over 10 years. Scaling audio to 110%
	 * will result in clipping! Any adjustments for CM1xxx IC gain differences
	 * should be made in the mixer settings, not in the audio stream itself.
	 */
	/* For the CM108 adjust the audio level */
	if (o->devtype != C108_PRODUCT_ID) {
		register short *sp = (short *) o->usbradio_write_buf;
		register float v;
		register int i;

		for (i = 0; i < sizeof(o->usbradio_write_buf) / 2; i++) {
			v = ((float) *sp) * 1.10;
			if (v > 32765.0) {
				v = 32765.0;
			} else if (v < -32765.0) {
				v = -32765.0;
			}
			*sp++ = (int) v;
		}
	}
```

Note that C108_PRODUCT_ID is not a general define that includes CM108AH, CM108B, or CM119.  It is in fact a specific product ID returned by these ICs. From res_usbradio.h:

```
#define C108_PRODUCT_ID  	0x000c
#define C108B_PRODUCT_ID  	0x0012
#define C108AH_PRODUCT_ID  	0x013c
#define N1KDO_PRODUCT_ID  	0x6a00
#define C119_PRODUCT_ID  	0x0008
#define C119A_PRODUCT_ID  	0x013a
#define C119B_PRODUCT_ID    0x0013
```

So in other words all Tx audio on all of the above ICs (excepting only the very old and long-unavailable CM108) will be scaled to 110%, and this will cause digital clipping.

davidgsd, 2024-09-14:
A few additional minor notes on this issue from emails where this was further discussed and a couple further thoughts:

1. I think what happened with this code was that when original CM108 went EOL and was replaced by CM119A/B ~10 years ago, someone noticed some minor (< 1dB) differences in audio levels and decided to just multiply all the raw audio samples to compensate (a very quick and dirty hack). But then the code was never updated over the subsequent 10 years and as several other CMxxx versions came out. There have been 6 versions of these ICs, and the gain differences between A to B were much more significant than the differences between original CM1xx (no suffix) and A. So we have this old scaling being done that introduces distortion, yet only makes a very small change in levels.

2. In addition there is inconsistent scaling done on Rx audio - Usbradio is reducing it by 2dB, but Simpleusb is not. Thus for the past 10 years all Usbradio outgoing IAX audio has been 2dB quieter, and people using Simpleusb driver are 2dB louder relative to those who are using Usbradio.

3. Multiplying raw audio samples is not a solution. Any audio or DSP expert would agree that such an approach is just plain wrong. If a DAC can only put out X Volts, software cannot make it put out 1.1 * X Volts - such an attempt will just clip the audio, which has very detrimental effects to audio quality.

4. There is a common misconception that a eg. 10% difference in (linear) signal voltage is a big difference. It's actually not at all, in fact it's pretty much imperceptible. The ear hears audio volume on a logarithmic scale, and levels have to be converted to log dB to know how much audible difference there actually is. Removing this 1.1x scaling hack will reduce Tx audio by 9.1%, which is -0.83 dB (20*log10(1/1.1)), and the reduction is actually a little less than that on average because the audio will no longer be clipped.

If you reduce the volume on any radio, stereo system, etc. by less than 1dB, you won't even be able to tell the difference.  Volume knobs cover a large range, typically from -96dB fully CCW to 0dB fully CW, in the context of 16-bit audio which is what most audio systems use and what Asterisk and the CM1xx ICs use. 1 decibel is in fact defined as the smallest change in level that is discernible by the human ear, and 0.8dB is less than that.

This can be easily confirmed by running the code with these hacks commented out, and switch between that and the current ASL code and I'll bet you won't be able to tell any difference in overall levels without actually looking at the audio in a wave editor or service monitor.  To put it in perspective, volume controls have a range of about 330° of rotation from off (-96dB) to max (0dB), which is about 330/96 = 3.4° rotation per dB, thus for example on a stereo system if you rotate the volume knob down by 0.8dB it's a change of only ~3°.  That's a very minor change in actual perceived volume. You should however be able to tell that a signal normalized to 0dBFS is no longer distorted and sounds better.

Introducing 10% clipping into a signal **is** a significantly detrimental because clipping very much affects the signal's spectral content, but bringing a signal down by 0.8 dB to where it should be is not a significant change.

5. Users should be aware that there will be an almost imperceptible 0.8 dB reduction in tx audio, but this is such a minor difference (the equivalent of changing the txmix setting by about 18 within the range of 0 - 999), that I do believe the best thing to do is just remove these hacks now and be done with it. But I can't speak for all repeater admins and there may be some who have set their audio levels so precisely that 0.8 dB of volume reduction might in fact be noticeable.

If this was a FOSS package for digital audio/music/video production or if it were in the main Asterisk code, an issue like this would definitely be fixed right away, with a simple note in the release notes to the effect "audio clipping issue fixed, note that you may see a 0.8 dB reduction in output audio volume."

6. I would also mention that I highly suspect that within any general Asterisk or VOIP or other digital audio -related software package, any code that introduces digital clipping would be considered a major violation of coding standards / best practices and would be immediately and fully removed once discovered.

davidgsd, 2024-10-02:
Just found another issue in the SimpleUSB channel driver that will introduce clipping into nearly all rx audio:

In chan_simpleusb.c::simpleusb_read()
```
	/* scale and clip values */
	if (o->rxvoiceadj > 1.0) {	
		register int i, x;
		register float f1;
		register int16_t *p = (int16_t *) f->data.ptr;

		for (i = 0; i < f->samples; i++) {
			f1 = (float) p[i] * o->rxvoiceadj;
			...
			}
	}
```

I had seen the rxvoiceadj parameter before but there seems to be no documentation on it and I had not yet had time  to look more closely at the code. Had some time to further investigate this today with some extra debug output log messages enabled and I figured out what it does.

With the above comment stating "scale and clip values", it only follows that the code would do exactly that, and surprisingly, no one seems to have noticed this or thought anything of it before. To be clear, audio should never be clipped, not in ASL, Asterisk, or **any** application for **any** reason. There is absolutely no possible benefit to clipping audio and doing so will result in reduced audio quality for all ASL users. The rx/tx mixer settings already enable users to adjust audio levels to within +/-1dB as needed. For ASL to try make additional adjustments to raw audio samples is unnecessary and will result only in quantization noise, clipping, inconsistent audio levels at various steps in the processing done in simpleusb_read(), and increased CPU use.

USBRadio does not appear to use the rxvoiceadj parameter to do any audio scaling. It references the parameter in a number of places, reading it and saving it to/from the conf files and optionally a CM1xx attached EEPROM, and printing the value in different places -- but from what I can tell the value is never used in any kind of calculation or processing of audio. xpmr.c has a copy of rxvoiceadj (pChan->prxVoiceAdjust) but this also is never actually used. Thus it seems pretty clear to me that all code in chan_usbradio.c and xmpr.c referencing these variables is dead code, that just makes it harder for everyone to understand what the various channel driver settings and variables actually do, and that really should be deleted.

In the case of SimpleUSB, rxvoiceadj is used in an attempt to make minor level adjustments "between" each of the CM1xx mixer level settings. The rxmixset parameter can range from 0-999, but the CM1xxA/AH ADC mixer has only 16 gain steps of 1.5dB each, and the CM1xxB ADC mixer has 36 steps of 1dB each. Thus when writing the rxmix gain setting to the CM1xx the value is divided by the number of steps and the remainder assigned to rxvoiceadj. But there's a few problems with that:
* The line that assigns this remainder to rxvoiceadj (last line in chan_simpleusb.c::mixer_write()) is using a value (1.187962) that corresponds to 1.496dB per mixer step, which is not correct for CM1xxB ICs.
* The scaling done at the end of chan_simpleusb.c::simpleusb_read() is done after all other processing thus will likely affect only outgoing IAX audio -- not echoed audio or measurements (ie. as shown by tune menu level meter). So when users are adjusting their levels this scaling provides no benefit and will only result in up to 1.5dB of clipping being introduced that users will have no awareness of if they are not actually monitoring their outgoing IAX audio.
* This is inconsistent with USBRadio which is not doing this scaling.
* Users have no way to know when this scaling is being done or what specific rxmix values will happen to divide evenly by 16 or 36. For my testing I added prints of the exact values sent and returned from the CM1xx and this could be helpful to add to the code but without that it's not at all clear to users of what the 0-999 numbers translate to.
* Some USB interfaces such as the AIOC emulate a CM1xx but don't actually have any internal mixers thus the rxmix setting has no effect in hardware yet results in arbitrary scaling of 0 - 1.5dB being done in SimpleUSB.

Removing this code will prevent clipping, reduce quantization noise and CPU use, make it easier and more consistent and predictable for users to set levels, remove a gross and egregious violation of best-practices as relate to any audio/voice/DSP application, and make ASL easier to understand and more consistent between channel drivers.

Allan-N, 2024-10-03:
On the one hand we should look at changing the 0-999 range to match that of the chipset (0-15, 0-35, etc).  But, that would be a confusing change to those who have only known 0-999.  There's also the issue of how to handle existing configurations.  I get that we don't want to clip, ever.  Just wondering how we can present a consistent scale that will work with the various chips.  Or, is this another case where we need a "legacy_config" variable that controls how each of the settings should be interpreted?

jxmx, 2024-10-06:
@davidgsd - Do you have PR for this?

davidgsd, 2024-10-06:
@jxmx  I could do a PR pretty quickly, it's really just a matter of commenting out or deleting these few sections of code that are doing the scaling and clipping.  Allan had mentioned he wasn't sure what the best way to roll this out would be, since some small changes in audio levels will result, and that some further discussion might be needed. To recap all the above: SimpleUSB now scales tx audio by 1.09375 (0.78dB) and rx audio by an essentially arbitrary number between 0-1.5dB, and USBRadio scales tx audio by 1.1 (0.83dB), and rx audio by 0.8 (-1.94dB).

With how inconsistent this all is, and as the level differences are relatively minor, retaining backwards compatibility by adding some sort of legacy config parameter I don't think makes sense. To me, any code that purposely and arbitrarily alters raw audio samples is just a very bad idea, that should be removed asap rather than a cfg option provided that preserves it in existing installs. Thus I would recommend removing this code and stating in the release notes something like the following: 

"Removed legacy audio scaling code in SimpleUSB and USBRadio channel drivers that would introduce digital clipping into Rx and Tx audio for all URIs using CM108AH, CM119A, and CM1xxB ICs. SimpleUSB was scaling Tx audio by 1.09375 (0.78dB), and Rx audio by 0-1.5dB, corresponding to the remainder of the rxmixerset parameter divided by the number of CM1xx ADC mixer steps (16 steps for CM1xxA/AH, 36 steps for the CM1xxB); USBRadio was scaling Tx audio by 1.1 (0.83dB), and Rx audio by 0.8 (-1.94dB). This change will result in minor audio level differences on all nodes using URIs with C-Media CM1xx(A/AH/B) ICs. In the case of SimpleUSB these level differences are less than 1dB for Tx audio, which is unlikely to be noticeable, and an average of 0.75dB for Rx audio which is also unlikely to be noticeable. In the case of USBRadio, the level difference for Tx audio is also less than 1dB which is unlikely to be noticeable, but is almost 2dB for Rx audio. For nodes using the USBRadio channel driver it is advised to check audio levels after upgrading, and it may be warranted to decrease the rxmixerset parameter value by 63 (1000/16) for CM1xxA/AH URIs (-1.5dB) and 56 (1000/36 * 2) for CM1xxB URIs (-2dB). Removed legacy rxvoiceadj parameter as this was not used in any USBRadio processing, and its use in SimpleUSB would result only in a slight increase Rx audio levels but also in digital clipping of Rx audio. In summary, any changes in audio levels should be less than 2dB in all cases, but after this update all users are advised to check their audio level settings using the tune menu utilities."

Let me know if anyone has any questions/feedback on the above and on how to best do the PR.

jxmx, 2024-10-06:
There are over 2000 nodes of ASL3 now. While I believe this change is a good change to make, I believe the responsible thing to do is wrap the changes in configuration variable. While for many people this change won't matter (much), this does have the potential to cause some problems with people's setup and forcing people to re-level audio is undesired in a minor upgrade. Insert obligatory ham reference to climbing up mountains to a repeater site here.

What I would propose is the creation of a variable for both channel drivers called `legacyaudioscaling`. For app_rpt v3, when `legacyaduioscaling` audio is undefined or "on" (or "true" or whatever), the old behavior is preserved. Edit the default configuration to set `legacyaudioscaling = off` for a going-forward behavior. Then, in a next major release ("v4"?) the default behavior is flipped around, or maybe then the code is just completely removed.

jxmx, 2024-10-06:
But, yes, we need a PR to build some development packages for testing.

davidgsd, 2024-10-06:
Sounds good, and yes it's very easy to add such a cfg to simpleusb/usbradio .conf, and anyone who reads the release notes and wants to have better audio quality can easily edit their .conf files. I should be able to submit a PR in the next day or two.

Allan-N, 2024-10-29:
Fixed by https://github.com/AllStarLink/app_rpt/pull/418

closed 2024-10-29 by Allan-N