AllStarLink/app_rpt #408: "tune" menu commands should accept "node" arg and pre-select interface settings

Allan-N, 2024-09-24:
On systems with a single node, trying to use the `simpleusb-tune-menu` (or `radio-tune-menu`) command to "tune" the audio interface is a simple matter.  This is because there is only one node/interface to configure.  The configuration process gets more complicated when you have more than one node because the menu commands and channel driver(s) don't know which node you want to display/update.  Without this pre-selection the user is presented with settings for "a" interface but not necessarily the interface they are hoping to view/update.

We can/should make this simpler.  I propose that we add a "-n <node>" argument to the tune commands that would be used to select the node/interface configuration being viewed/updated.

With that functionality in place we can update the `node-setup` menu to pass this new argument when tuning the interface for a specific node.

encbar5, 2024-09-24:
Is this going to be available from the Asterisk CLI as well?

Allan-N, 2024-09-24:
The `simpleusb-tune-menu` command provide selection "1" to select the USB device.  This uses a combination of the following CLI commands :

- susb active
- susb tune menu-support 1
- susb active <node>

My proposal would simply be exec'ing "susb active", checking if the currently selected device is what was requested with the command line argument, and changing the selected device as needed.

closed 2024-10-26 by Allan-N