AllStarLink/app_rpt #412: app_rpt: Be able to execute dialplan on nodes instead of "localplay"

InterLinked1, 2024-09-28:
The `rpt localplay` command allows for playing audio files onto a node. This is used by functionality such as: https://github.com/AllStarLink/ASL3/pull/101/files

More flexible would be the ability to execute arbitrary dialplan on a node in the same contexts (e.g. `localexec`), which could play audio or do other stuff. This would eliminate a lot of duplicated functionality in Asterisk in the ASL repos, such as constructing and playing the time, when there are Asterisk applications that already do the same things.

mkmer, 2025-03-07:
I'm thinking something like this could work (few options or all of them):

```
[command_process]
; ---- MJK test extension ---
exten=> _C1,1,rpt(${NODE4}|localplay|rpt/goodafternoon)
same => n, Hangup()

exten=> _C2,1,rpt(${NODE4}|playback|rpt/goodafternoon)
same => n, Hangup()

exten=> _C3,1,rpt(${NODE4}|cli|"rpt localplay ${NODE4} rpt/goodafternoon")
same => n, Hangup()
```

The most generic version would be `|cli|` -> one could put ANY command line string in there and execute it.

This is a "mockup" of a possible solution: https://github.com/AllStarLink/app_rpt/tree/Allow-rpt-xxx-from-dialplan

InterLinked1, 2025-03-07:
> I'm thinking something like this could work (few options or all of them):
>
> ```
> [command_process]
> ; ---- MJK test extension ---
> exten=> _C1,1,rpt(${NODE4}|localplay|rpt/goodafternoon)
> same => n, [...]

That's not really what I was talking about, I was referring to executing dialplan withing the repeater channel, and eliminating the use of special CLI commands for that purpose altogether. There wouldn't be any modification to the Rpt() application usage itself.