AllStarLink/app_rpt #415: ASL-3.0.5 Segmentation fault

PD0SBH, 2024-10-05:
From the community:

Raspberry Pi 3B (2x)
ASL3 Image on one system.
Raspberry Pi OS (Bookworm) with ASL 3 added via apt repo on the other.
CM108 Soundfob: ID 0d8c:0012 C-Media Electronics, Inc. USB Audio Device

[usbradio.conf.txt](https://github.com/user-attachments/files/17264046/usbradio.conf.txt)
[gdb-asterisk.txt](https://github.com/user-attachments/files/17264047/gdb-asterisk.txt)

Allan-N, 2024-10-05:
Moved to app_rpt.

KB4MDD, 2024-10-05:
Can you run `/var/lib/asterisk/scripts/ast_coredumper` on the core file and post the results?

davidgsd, 2024-10-05:
As Allen mentioned in an email looks like what happened here was when the RxAudioStats feature was added it was necessary to work around some old hacks in the code that scale the raw audio samples (see https://github.com/AllStarLink/app_rpt/issues/399). Initially I had 'if 0'd put that code because it made no sense to keep it there. I did test on both channel drivers at that point.  After submitting the PR however it was decided to leave the audio scaling hacks in place and open an the above issue report on that subject.  As a result I had to move a section of code to before that scaling was done in chan_usbradio.c. Looks I failed to then test that change with USBRadio. I see now that the readpos is advanced as Allen noted. The scaling code uses that pointer, thus attempting to write past the end of the buffer. So that is likely the cause of the issue, and I regret any inconvenience this has caused. LMK how this should be best be addressed. I can submit a PR or maybe there is a quicker way to move the readpos increment to below where it's used in the scaling code. There are several simple ways this could be fixed, first would be to change the following line in chan_usbradio.c:
		`register short *sp = (short *) (o->usbradio_read_buf + o->readpos);`
To:
		`register short *sp = (short *) (o->usbradio_read_buf + (o->readpos - res));`
The readpos variable is not read from anywhere else in the rest of the function.

PD0SBH, 2024-10-05:
Confirmed fixed, thanks.

closed 2024-10-05 by PD0SBH