AllStarLink/app_rpt #419: idrecording playback sporadic and tailmessage playback very rare

knobby4444, 2024-10-17:
**Describe the bug**
When using idrecording in ASL3 the playback is sporadic.  Sometimes the idrecording will play after someone keys the radio when the repeater had been idle for 10 minutes or more.  Sometimes the idrecording will not play at all for an entire day.  It will sometimes   I have not been able to identify and pattern yet.  But.. sometimes after I issue a command like "rpt playback 49520 hello" the hello message will play followed by the idrecording.  After that the idrecording may continue to play as expected for some time, and then stop again.  It's as though the idrecording playback got stuck and once I played the other file, it started again.  Yesterday idrecording was not playing at all in the early morning, then someone linked their node and keyed down, and the idrecording played and then continued to work when someone would keydown from the linked node, or if I key down on dvswitch connected to my node via iaxuser.  But if someone keyed the radio, idrecording would never play.  By this morning idrecording was once again not playing after a keydown from anywhere.  Then the person linked their node again and it played the idrecording.   Tailmessage seems much different, it almost never plays.  3 times I have done an rpt playback command and heard the tailmessage right after.  This morning I heard someone keydown over the linked node and heard the tailmessage play.  That's the only time I've heard tailmessage play as I would expect it.  Perhaps this issue should concentrate on the idrecording and leave tailmessage as another issue.

**To Reproduce**
Default install of ASL3, on debian or the appliance.  Setup the node like this:
[49520](node-main)
statpost_url = http://stats.allstarlink.org/uhandler
duplex = 1 
rxchannel = SimpleUSB/49520
;;;;;;;;;;;;;;;;;;; Your node settings here ;;;;;;;;;;;;;;;;;;;
;startup_macro = *8132000
hangtime = 10                     ; squelch tail hang time (in ms) (optional, default 5 seconds, 5000 ms)
althangtime = 40                  ; longer squelch tail
totime = 180000                     ; transmit time-out time (in ms) (optional, default 3 minutes 180000 ms)

;idrecording = /etc/asterisk/rpt/k5fdrepeater-hector              ; id recording or morse string
idrecording = /usr/share/asterisk/sounds/en/hello ; test sound
;idtalkover =                       ; Talkover ID (optional) default is none
idtime = 120000                     ; id interval time (in ms) (optional) Default 5 minutes (300000 ms)
politeid = 30000                    ; time in milliseconds before ID timer expires to try and ID in the tail. (optional, default 30000)

;tailmessagelist=/etc/asterisk/rpt/plinuse
tailmessagelist = /usr/share/asterisk/sounds/en/goodbye

**Expected behavior**
If the repeater system is idle and someone keys down via radio or direct iaxclient, after then unkey, the idrecording should play after the idtime expires.  That's what I would expect compared to other systems I have worked with, but I'm not certain that is how it is designed in ASL3.  When it is playing the idrecording, it seems to work this way except the idrecording plays after idtime minus politeid which is fine.

**Screenshots**
No screens

**Software versions (listed in asl-menu, option 4)**

            │ OS            : Debian GNU/Linux 12 (bookworm)                                                     
                 │ OS Kernel     : 6.6.47+rpt-rpi-v8                                                                  
                 │                                                                                                    
                 │ Asterisk      : 20.9.1+asl3-3.0.4-1.deb12                                                          
                 │ ASL [app_rpt] : 3.0.4                                                                              
                 │                                                                                                    
                 │ Installed ASL packages :                                                                           
                 │                                                                                                    
                 │   Package               Version                                                                    
                 │   ====================  ==============================                                             
                 │   allmon3               1.3.4-1.deb12                                                              
                 │   asl3                  3.4.0-1.deb                                                                
                 │   asl3-asterisk         2:20.9.1+asl3-3.0.4-1.deb12                                                
                 │   asl3-asterisk-config  2:20.9.1+asl3-3.0.4-1.deb12                                                
                 │   asl3-asterisk-module  2:20.9.1+asl3-3.0.4-1.deb12                                                
                 │   asl3-menu             1.8-1.deb12                                                                
                 │                                                      

**Have you run a software update and rebooted?**
I have installed a few times using debian, and this last test I'm using the raspi appliance.  Using pi3b+ and pi4b.  This last time I was careful to only make changes in rpt.conf in one spot, as shown above.  Rebooted, updated, restarted more than a few times.

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
Above

**Additional context**
I first noticed this problem using my own sound files to play.  But, I have tested using canned sound files from asterisk and it's the same result.  Someone mentioned ider_state in the stats and it flips between 1 and 2 and doesn't seem to have an effect on the playback.  I just switched back to canned sound files for testing and playback seemed to work fine.  Then switched back to custom files and things seem to be working well:

13:13:00  asterisk restart
13:13:37 idrecording plays, about the politeid time after restart
13:14:20 key dbswitch
13:15:55 idrecording plays about 1.5 minutes after keydown.  That's the idtime of 2 minutes, minus the politeid time of 30 seconds.  I think this is how it should work.  I restarted asterisk again and keyed dvswitch and it played the idrecording as soon as I unkeyed.  I haven't been able to see a pattern as to when it will id as soon as someone keys, or after the idtime.  But I think after the idtime would be correct.  In its current state it seems to id over someone keying dvswitch, then playing the idrecording again after the idtime.  But if someone keys the radio, it does not id over them, it only ids after the idtime.  In a few hours or tomorrow morning, I expect that it will not play the idrecording at all.  I will report back with more data...

Allan-N, 2024-10-17:
Moved to app_rpt

knobby4444, 2024-12-23:
The TL;DR on this one at this point is that when someone keys the radio in duplex=1 mode, idrecording never plays.  If I make a phone connection like dvswitch and key, it plays as expected.  Looking at the code I think it's something around line 5166 in app_rpt/apps/app_rpt.c.  A bit of a maze there and related code above.  This warning seems correct:

* WARNING:  YOU ARE NOW HEADED INTO ONE GIANT MAZE OF SWITCH STATEMENTS THAT DO MOST OF THE WORK FOR
  *           APP_RPT.  THE MAJORITY OF THIS IS VERY UNDOCUMENTED CODE AND CAN BE VERY HARD TO READ.
  *           IT IS ALSO PROBABLY THE MOST ERROR PRONE PART OF THE CODE, ESPECIALLY THE PORTIONS
  *           RELATED TO THREADED OPERATIONS.