AllStarLink/app_rpt #420: Large log files on /var/log/asterisk

du6jip, 2024-10-29:
The Logs on /var/log/asterisk are large files. Before I did the screenshot the previous one I deleted was 3gb. on my other server running the same version of asterisk was 14gb. I have to deleted them because  for some reason they are slowing the servers.

[image]

[2024-10-28 20:12:08.830] Asterisk 20.9.3+asl3-3.0.6-1.deb12 built by builder @ [allstarlink.org](http://allstarlink.org/) on a x86_64 running Linux on 2024-10-05 13:19:13 UTC
[2024-10-28 20:12:08.857] NOTICE[506725] loader.c: 72 modules will be loaded.
[2024-10-28 20:12:08.925] NOTICE[506725] cdr.c: CDR simple logging enabled.
[2024-10-28 20:12:08.927] NOTICE[506725] dnsmgr.c: Managed DNS entries will be refreshed every 300 seconds.
[2024-10-28 20:12:08.941] NOTICE[506725] indications.c: Default country for indication tones: us
[2024-10-28 20:12:08.941] NOTICE[506725] indications.c: Setting default indication country to ‘us’
[2024-10-28 20:12:09.013] NOTICE[506725] res_smdi.c: No SMDI interfaces are available to listen on, not starting SMDI listener.
[2024-10-28 20:12:09.021] WARNING[506725] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2024-10-28 20:12:09.021] WARNING[506725] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2024-10-28 20:12:09.021] NOTICE[506725] chan_dahdi.c: Ignoring any changes to ‘userbase’ (on reload) at line 23.
[2024-10-28 20:12:09.021] NOTICE[506725] chan_dahdi.c: Ignoring any changes to ‘vmsecret’ (on reload) at line 31.
[2024-10-28 20:12:09.021] NOTICE[506725] chan_dahdi.c: Ignoring any changes to ‘hassip’ (on reload) at line 35.
[2024-10-28 20:12:09.021] NOTICE[506725] chan_dahdi.c: Ignoring any changes to ‘hasiax’ (on reload) at line 39.
[2024-10-28 20:12:09.021] NOTICE[506725] chan_dahdi.c: Ignoring any changes to ‘hasmanager’ (on reload) at line 47.
[2024-10-28 20:12:09.021] WARNING[506725] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2024-10-28 20:12:09.021] WARNING[506725] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2024-10-28 20:12:09.107] ERROR[506773] chan_echolink.c: Cannot find database entry for IP address 44.31.195.218, Callsign 4H8WIO.
[2024-10-28 20:12:09.126] WARNING[506725] loader.c: Module ‘res_adsi’ has been loaded but may be removed in a future release.
[2024-10-28 20:12:09.299] NOTICE[506771] app_rpt.c: Normal Repeater Init 581290
[2024-10-28 20:12:09.299] NOTICE[506771] app_rpt.c: Normal Repeater Init 1000
[2024-10-28 20:12:09.300] NOTICE[506771] app_rpt.c: Normal Repeater Init 1200
[2024-10-28 20:12:09.301] NOTICE[506771] app_rpt.c: Normal Repeater Init 1300
**[2024-10-28 20:12:11.792] WARNING[506787][C-00000023] func_channel.c: Unknown or unavailable item requested: ‘peerip’
[2024-10-28 20:12:11.792] WARNING[506787][C-00000023] func_channel.c: Unknown or unavailable item requested: ‘peerip’
[2024-10-28 20:12:15.121] WARNING[506796][C-00000026] func_channel.c: Unknown or unavailable item requested: ‘peerip’
[2024-10-28 20:12:15.121] WARNING[506796][C-00000026] func_channel.c: Unknown or unavailable item requested: ‘peerip’**
[2024-10-28 20:12:55.516] WARNING[506758] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.516] WARNING[506759] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.516] WARNING[506760] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.516] WARNING[506761] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.516] WARNING[506762] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.516] WARNING[506763] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.516] WARNING[506764] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.516] WARNING[506765] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.516] WARNING[506766] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506757] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506758] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506759] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506760] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506761] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506762] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506763] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506764] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506765] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506766] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506757] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506758] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506759] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506760] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506761] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506762] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506763] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506764] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506765] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506766] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506757] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506758] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506759] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506760] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506761] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506762] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[2024-10-28 20:12:55.517] WARNING[506763] chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame
[... cut ...]
[2024-10-27 16:40:37.342] NOTICE[607] dnsmgr.c: dnssrv: host 'register.allstarlink.org' changed from 162.248.92.131:443 to 34.105.111.212:443
[2024-10-27 16:45:37.343] NOTICE[607] dnsmgr.c: dnssrv: host 'register.allstarlink.org' changed from 34.105.111.212:443 to 162.248.92.131:443
[2024-10-27 17:10:37.342] NOTICE[607] dnsmgr.c: dnssrv: host 'register.allstarlink.org' changed from 162.248.92.131:443 to 34.105.111.212:443
[2024-10-27 17:15:37.345] NOTICE[607] dnsmgr.c: dnssrv: host 'register.allstarlink.org' changed from 34.105.111.212:443 to 162.248.92.131:443

KB4MDD, 2024-10-29:
Can you post your `/etc/asterisk/iax.conf`, `/etc/asterisk/extensions.conf` and `/etc/asterisk/modules.conf`

Allan-N, 2024-10-29:
For the "[2024-10-27 16:24:28.051] NOTICE[447048] manager.c: 203.212.232.86 tried to authenticate with nonexistent user '<insert-AMI-login-name>'" messages, this looks like someone is trying to use AMI to interact with your asterisk server and trying various names.  Is the 203.212.232.86 IP address one that you know/control?

du6jip, 2024-10-29:
[extensions.txt](https://github.com/user-attachments/files/17557306/extensions.txt)
[iax.txt](https://github.com/user-attachments/files/17557307/iax.txt)
[modules.txt](https://github.com/user-attachments/files/17557308/modules.txt)

As far as 203.212.232.86, that's in India. The cloud is in Los Angeles, CA

The other server experiencing the same way too, although its a different ip address. I had to downgrade it back to ASL 2.

Allan-N, 2024-10-29:
For the `dnsmgr.c: dnssrv: host 'register.allstarlink.org' changed from 34.105.111.212:443 to 162.248.92.131:443` messages, they are harmless but a fix is on it's way [(asterisk PR #925)](https://github.com/asterisk/asterisk/pull/925).

Allan-N, 2024-10-29:
>As far as 203.212.232.86, that's in India. The cloud is in Los Angeles, CA

Did you intend to extend AMI access to your server (port 5038)?

KB4MDD, 2024-10-29:
The message "chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame" is occurring on a connection from allstar-public.  This connection is used by web transceivers.  

Do you have someone connecting to your node with dvswitch?

I ask this because this same problem was reported in this post [](https://community.allstarlink.org/t/asl3-warnings-and-high-cpu-when-dvswitch-mobile-connects-with-ulaw/21315)  The operator noted, if I change the DVSM account settings to use ADPCM, this warning is not logged, and CPU usage is normal on the ASL3 server.

du6jip, 2024-10-29:
Oh I forgot to mention all of the server that I manage, 533550,  581290, 516710, 533554 ang 576160 ( downgraded to ASL 2)  are all experiencing "attempt access"  coming from different ip addresses. All with large log files that if I dont delete them, the nodes/asterisk are restarting by it self. The becomes useless especially during Nets

KB4MDD, 2024-10-29:
Please check `manager.conf ` and see if you have the `bindaddr = 127.0.0.1`.  If you have it set to 0.0.0.0, you are allowing connections from outside of your network.

du6jip, 2024-10-29:
Yes, this Node is primarily for members with DVSwitch mobile app.

I have to ask them before changing it back to 127.0.0.1 on managers.conf. There might be some members who monitors this on their SUPERMON. 
I don't see this in ASL 2.0 though. An attempt to access.

KB4MDD, 2024-10-29:
The problem with `chan_iax2.c: No voice format and no peer format available on IAX2/allstar-public-5764, backlogging frame` appears to be associated with DVSwitch.  The last post at https://dvswitch.groups.io/g/Mobile/topic/107061137#msg3963 indicates that they are working on resolving that issue.

> A temporary workaround for this problem is to disable warnings in /etc/asterisk/logger.conf.
> console => notice,error,dtmf
> messages.log => notice,error,dtmf
> 
> This avoids the full log disk and pegged CPU from logging the warnings when DVSM connects to ASL3. 

> Also, changing the DVSM account settings to use ADPCM addresses the issue.

No changes to app_rpt are needed at this time.

KB4MDD, 2024-10-29:
The `failed to authenticate as` messages seem to be the result of having the server open to the public internet.  These messages may have not been logged in ASL2.

Allan-N, 2024-10-29:
@InterLinked1, we talked/slack'd about the `No voice format and no peer format available on IAX2` messages after the end-Sept "stress test".  At that time, you mentioned [Commit 00101d3](https://github.com/AllStarLink/app_rpt/commit/00101d3add13edc55b6a688439d51606d61dcd53).

InterLinked1, 2024-10-29:
While that resolves the user's immediate concern, this seems more like an XY problem to me. The problem here is not the log file filling up, it's the scenario in which the `No voice format and no peer format available on IAX2` log message is appearing, as this indicates that something has gone wrong codec-wise.

I am the one that added this to Asterisk, as it used to segfault in this scenario. For that reason, there is likely a serious underlying problem that should be addressed.

I'm guessing that there are no codecs negotiated at this point and it's trying to send some other kind of frame, but my recollection is fuzzy. Steps to accurately reproduce this would be helpful so we can do further analysis.

Allan-N, 2024-11-06:
To me, I would think that if we detect a codec mismatch we should log who the connection attempt was from, what they offered, and what we have available.  Then, because we are never going to be able to hear each other, the connection should simply be dropped.

KB4MDD, 2024-11-06:
Based on the research by one user from the links I included, the problem is in chan_iax2.c.  This would require a change to asterisk.

wd5m, 2024-11-06:
This looks like the same issue I [posted about in the forum in July](https://community.allstarlink.org/t/asl3-warnings-and-high-cpu-when-dvswitch-mobile-connects-with-ulaw/21315).  I found that DVSwitch Mobile (DVSM) appears not to always provide a CODEC "requested format = Unknown" to ASL3. This occurs when DVSM is configured with an account using a registered Allstar node number. See log message below when DVSM is set for the CODEC to be server selected or ULaw. 

> Connected to Asterisk 20.9.3+asl3-3.0.6-1.deb12 currently running on turbo (pid = 6553)
    -- Hungup 'IAX2/129.222.126.2:28506-3172'
    -- Accepting UNAUTHENTICATED call from 129.222.126.2:28506:
    --        > requested format = Unknown,
    --        > requested prefs = (slin|ulaw|adpcm|alaw),
    --        > actual format = ulaw,
    --        > host prefs = (ulaw|adpcm|gsm),
    --        > priority = mine

As previously noted, the ASL3 server then goes CPU bound and with continuous log warnings messages from chan_iax2 No voice format, until the PTT from DVSM, or DVSM disconnects.

If DVSM CODEC is set to ADPCM, or SLIN (with SLIN allowed in ASL3), the connection completes normally with the selected CODEC.

I have repeated these symptoms on both Raspbery PI and Debian 12 VPS VM standard ASL3 servers. 

In my cursory look at chan_iax2.c in ASL3, it seemed the problem might be in the "__get_from_jb" function, perhaps starting with this part of the code, and maybe "cleanup".

> 		if (!voicefmt) {
> 			/* Really shouldn't happen, but if it does, should be looked into */
> 			ast_log(LOG_WARNING, "No voice format and no peer format available on %s, backlogging frame\n", ast_channel_name(pvt->owner));
> 			goto cleanup; /* Don't crash if there's no voice format */
> 		}

I view this as a serious bug, as it would seem to expose ASL3 to a potential **denial of service** issue as well.

David McAnally, WD5M
WX5FWD SKYWARN Team

InterLinked1, 2024-11-06:
The "Requested format = Unknown" part seems suspicious, I don't remember off hand but that seems strange to me. There should be a voice format here but there is not... I agree that `chan_iax2` should handle this better, but that may also be symptomatic of another, unrelated issue. Do you have the iax2 configuration from both sides?

InterLinked1, 2024-11-07:
> I agree, there might be something further upstream. Perhaps there is Allstar code (apt_rpt?) that could catch and mitigate this symptom before asterisk / chan_iax2 sees it?
> 
> > Do you have the iax2 configuration from both sides?
> 
> The iax.conf configuration is the basic configuration from ASL3 install. The [radio] stanza is in play for the DVSwitch Mobile connection. ULaw, GSM and ADPCM CODECS are allowed, which I believe is the default ASL3 configuration. I also allowed slin for testing that CODEC with DVSwitch Mobile.
> 
> I have informed DVSwitch Mobile developers of this issue. I'm told they are working on it too. As noted earlier I have tested DVSwitch Mobile CODEC settings for "Server Select", as well as ULaw, ADPCM and SLIN (with allow slin in ASL3 iax.conf). Those are the only CODEC options DVSwitch Mobile provides in their node account setup. My current work around is to use Web Transceiver (WT) mode in DVSM. However, I can't force others to use WT mode. Hence the possibility someone could connect with DVSM and cause a denial of service issue due to asterisk going CPU bound.
> 
> In the meantime, I may try adding an event or other type of monitor script for the "Requested format = Unknown" asterisk log entry, and force a disconnect to mitigate the issue.
> 
> Regards, David McAnally, WD5M

Thanks David, whether app_rpt and/or chan_iax2, this is definitely not something that is okay, and hopefully we can get a fix so you don't need to add a workaround.

Are you able to enable IAX2 debug and reproduce the scenario for us? That might contain enough additional information.

`iax2 set debug on` will send the debug messages to the console.

wd5m, 2024-11-07:
> Are you able to enable IAX2 debug and reproduce the scenario for us? That might contain enough additional information.
> 
> `iax2 set debug on` will send the debug messages to the console.

Here's the result. 

> root@turbo:~# asterisk -rvvv
> Asterisk 20.9.3+asl3-3.0.6-1.deb12, Copyright (C) 1999 - 2022, Sangoma Technologies Corporation and others.
> Created by Mark Spencer <markster@digium.com>
> Asterisk comes with ABSOLUTELY NO WARRANTY; type 'core show warranty' for details.
> This is free software, with components licensed under the GNU General Public
> License version 2 and other licenses; you are welcome to redistribute it under
> certain conditions. Type 'core show license' for details.
> =========================================================================
> Connected to Asterisk 20.9.3+asl3-3.0.6-1.deb12 currently running on turbo (pid = 6553)
> turbo*CLI> iax2 set debug on
> IAX2 Debugging Enabled
> Rx-Frame Retry[ No] -- OSeqno: 000 ISeqno: 000 Type: IAX     Subclass: NEW
>     Timestamp: 00001ms  SCall: 11801  DCall: 00000 129.222.126.2:62262
>     CALLTOKEN       : Present
> 
> Tx-Frame Retry[ No] -- OSeqno: 000 ISeqno: 001 Type: IAX     Subclass: CTOKEN
>     Timestamp: 00001ms  SCall: 00001  DCall: 11801 129.222.126.2:62262
>     CALLTOKEN       : 51 bytes
> 
> Rx-Frame Retry[ No] -- OSeqno: 001 ISeqno: 001 Type: IAX     Subclass: ACK
>     Timestamp: 00001ms  SCall: 11801  DCall: 00001 129.222.126.2:62262
> Rx-Frame Retry[ No] -- OSeqno: 000 ISeqno: 000 Type: IAX     Subclass: NEW
>     Timestamp: 00000ms  SCall: 11802  DCall: 00000 129.222.126.2:62262
>     VERSION         : 2
>     CALLED NUMBER   : 614320
>     CALLING NUMBER  : 516224
>     CALLING NAME    : WD5M
>     USERNAME        : radio
>     FORMAT          : 108
>     CODEC_PREFS     : (slin|ulaw|adpcm|alaw)
>     CALLTOKEN       : 51 bytes
> 
> Tx-Frame Retry[-01] -- OSeqno: 000 ISeqno: 001 Type: IAX     Subclass: ACK
>     Timestamp: 00007ms  SCall: 05008  DCall: 11802 129.222.126.2:62262
>     -- Accepting UNAUTHENTICATED call from 129.222.126.2:62262:
>     --        > requested format = Unknown,
>     --        > requested prefs = (slin|ulaw|adpcm|alaw),
>     --        > actual format = ulaw,
>     --        > host prefs = (ulaw|adpcm|gsm),
>     --        > priority = mine
> Tx-Frame Retry[000] -- OSeqno: 000 ISeqno: 001 Type: IAX     Subclass: ACCEPT
>     Timestamp: 00010ms  SCall: 05008  DCall: 11802 129.222.126.2:62262
>     FORMAT          : 4
>     FORMAT2         : ulaw
> 
> Rx-Frame Retry[ No] -- OSeqno: 001 ISeqno: 001 Type: IAX     Subclass: ACK
>     Timestamp: 00010ms  SCall: 11802  DCall: 05008 129.222.126.2:62262
>     -- Executing [614320@radio-secure:1] NoOp("IAX2/129.222.126.2:62262-5008", "Connect from node: 516224") in new stack
>     -- Executing [614320@radio-secure:2] NoOp("IAX2/129.222.126.2:62262-5008", "Connect to: 614320") in new stack
>     -- Executing [614320@radio-secure:3] NoOp("IAX2/129.222.126.2:62262-5008", "The Channel IP is 129.222.126.2") in new stack
>     -- Executing [614320@radio-secure:4] GotoIf("IAX2/129.222.126.2:62262-5008", "0?connect") in new stack
>     -- Executing [614320@radio-secure:5] GotoIf("IAX2/129.222.126.2:62262-5008", "1?denylist") in new stack
>     -- Goto (radio-secure,614320,9)
>     -- Executing [614320@radio-secure:9] GotoIf("IAX2/129.222.126.2:62262-5008", "0?:connect") in new stack
>     -- Goto (radio-secure,614320,12)
>     -- Executing [614320@radio-secure:12] Rpt("IAX2/129.222.126.2:62262-5008", "614320") in new stack
> Tx-Frame Retry[000] -- OSeqno: 001 ISeqno: 001 Type: CONTROL Subclass: ANSWER
>     Timestamp: 00435ms  SCall: 05008  DCall: 11802 129.222.126.2:62262
> Tx-Frame Retry[000] -- OSeqno: 002 ISeqno: 001 Type: CONTROL Subclass: (255?)
>     Timestamp: 00438ms  SCall: 05008  DCall: 11802 129.222.126.2:62262
> Rx-Frame Retry[ No] -- OSeqno: 001 ISeqno: 002 Type: IAX     Subclass: ACK
>     Timestamp: 00435ms  SCall: 11802  DCall: 05008 129.222.126.2:62262
> Rx-Frame Retry[ No] -- OSeqno: 001 ISeqno: 003 Type: IAX     Subclass: ACK
>     Timestamp: 00438ms  SCall: 11802  DCall: 05008 129.222.126.2:62262
[... 23 lines cut ...]
>     Timestamp: 01037ms  SCall: 05008  DCall: 11802 129.222.126.2:62262
> [2024-11-07 11:31:11.261] WARNING[6591]: chan_iax2.c:4211 __get_from_jb: No voice format and no peer format available on IAX2/129.222.126.2:62262-5008, backlogging frame
> [2024-11-07 11:31:11.261] WARNING[6592]: chan_iax2.c:4211 __get_from_jb: No voice format and no peer format available on IAX2/129.222.126.2:62262-5008, backlogging frame
> [2024-11-07 11:31:11.261] WARNING[6593]: chan_iax2.c:4211 __get_from_jb: No voice format and no peer format available on IAX2/129.222.126.2:62262-5008, backlogging frame
> 
> (...repeats until keyed or disconnected...)
> 
> [2024-11-07 11:31:11.265] WARNING[6586]: chan_iax2.c:4211 __get_from_jb: No voice format and no peer format available on IAX2/129.222.126.2:62262-5008, backlogging frame
> Rx-Frame Retry[ No] -- OSeqno: 006 ISeqno: 006 Type: CONTROL Subclass: RDUNKEY
>     Timestamp: 03676ms  SCall: 11802  DCall: 05008 129.222.126.2:62262
> Tx-Frame Retry[-01] -- OSeqno: 006 ISeqno: 007 Type: IAX     Subclass: ACK
>     Timestamp: 03676ms  SCall: 05008  DCall: 11802 129.222.126.2:62262
> Rx-Frame Retry[ No] -- OSeqno: 007 ISeqno: 006 Type: IAX     Subclass: HANGUP
>     Timestamp: 08651ms  SCall: 11802  DCall: 05008 129.222.126.2:62262
>     CAUSE CODE      : 16
> 
> Tx-Frame Retry[-01] -- OSeqno: 006 ISeqno: 008 Type: IAX     Subclass: ACK
>     Timestamp: 08651ms  SCall: 05008  DCall: 11802 129.222.126.2:62262
>     -- Hungup 'IAX2/129.222.126.2:62262-5008'
>     -- Hungup 'DAHDI/pseudo-473624913'
>     -- Hungup 'DAHDI/pseudo-59191071'
> turbo*CLI> iax2 set debug off
> turbo*CLI>

David McAnally, WD5M

wd5m, 2024-11-09:
After comparing iax2 debug traces between "web transceiver" mode and standard ASL node account connections from DVSwitch Mobile, perhaps the "requested format = Unknown," indicator is NOT involved with this issue. Both connection methods show the "requested format = Unknown," indicator.  However, web transceiver mode does not result go CPU bound and generate the chan_iax2 warning loop. Note that asterisk does not go CPU bound until after rpt(...) in extensions.conf.

I turned off verbose messages in the asterisk console and only set iax2 debug on, to get a cleaner debug output.  One difference between WT and ASL connection methods was the following warning was NOT on the WT connection debug trace. I've included several preceding debug trace lines before the warning, in case those are of interest.

> Tx-Frame Retry[000] -- OSeqno: 000 ISeqno: 001 Type: IAX     Subclass: ACCEPT
>     Timestamp: 00010ms  SCall: 14317  DCall: 02255 129.222.126.2:61312
>     FORMAT          : 4
>     FORMAT2         : ulaw
> 
> Rx-Frame Retry[ No] -- OSeqno: 001 ISeqno: 001 Type: IAX     Subclass: ACK
>     Timestamp: 00010ms  SCall: 02255  DCall: 14317 129.222.126.2:61312
> Tx-Frame Retry[000] -- OSeqno: 001 ISeqno: 001 Type: CONTROL Subclass: ANSWER
>     Timestamp: 00013ms  SCall: 14317  DCall: 02255 129.222.126.2:61312
> Tx-Frame Retry[000] -- OSeqno: 002 ISeqno: 001 Type: CONTROL Subclass: (255?)
>     Timestamp: 00016ms  SCall: 14317  DCall: 02255 129.222.126.2:61312
> Rx-Frame Retry[ No] -- OSeqno: 001 ISeqno: 002 Type: IAX     Subclass: ACK
>     Timestamp: 00013ms  SCall: 02255  DCall: 14317 129.222.126.2:61312
> Tx-Frame Retry[000] -- OSeqno: 003 ISeqno: 001 Type: TEXT    Subclass: 0
>     Timestamp: 00668ms  SCall: 14317  DCall: 02255 129.222.126.2:61312
> Tx-Frame Retry[000] -- OSeqno: 004 ISeqno: 001 Type: TEXT    Subclass: 0
>     Timestamp: 00671ms  SCall: 14317  DCall: 02255 129.222.126.2:61312
> Rx-Frame Retry[ No] -- OSeqno: 001 ISeqno: 003 Type: IAX     Subclass: ACK
>     Timestamp: 00016ms  SCall: 02255  DCall: 14317 129.222.126.2:61312
> [2024-11-09 09:26:40.908] WARNING[193843]: app_rpt.c:3106 periodic_process_links: 0x5568abcc2820 newkeytimer expired on connected node, setting newkey from 2 to 0.

The above warning is followed by these debug trace messages, then the looping chan_iax2 warnings start.

> Tx-Frame Retry[000] -- OSeqno: 005 ISeqno: 001 Type: VOICE   Subclass: 4
>     Timestamp: 02200ms  SCall: 14317  DCall: 02255 129.222.126.2:61312
> Rx-Frame Retry[ No] -- OSeqno: 001 ISeqno: 004 Type: IAX     Subclass: ACK
>     Timestamp: 00668ms  SCall: 02255  DCall: 14317 129.222.126.2:61312
> Rx-Frame Retry[ No] -- OSeqno: 001 ISeqno: 005 Type: IAX     Subclass: ACK
>     Timestamp: 00671ms  SCall: 02255  DCall: 14317 129.222.126.2:61312
> Rx-Frame Retry[ No] -- OSeqno: 001 ISeqno: 006 Type: IAX     Subclass: ACK
>     Timestamp: 02200ms  SCall: 02255  DCall: 14317 129.222.126.2:61312
> Rx-Frame Retry[ No] -- OSeqno: 001 ISeqno: 006 Type: TEXT    Subclass: 0
>     Timestamp: 02228ms  SCall: 02255  DCall: 14317 129.222.126.2:61312
> Tx-Frame Retry[-01] -- OSeqno: 006 ISeqno: 002 Type: IAX     Subclass: ACK
>     Timestamp: 02228ms  SCall: 14317  DCall: 02255 129.222.126.2:61312
> Rx-Frame Retry[ No] -- OSeqno: 002 ISeqno: 006 Type: TEXT    Subclass: 0
>     Timestamp: 02228ms  SCall: 02255  DCall: 14317 129.222.126.2:61312
> Tx-Frame Retry[-01] -- OSeqno: 006 ISeqno: 003 Type: IAX     Subclass: ACK
>     Timestamp: 02228ms  SCall: 14317  DCall: 02255 129.222.126.2:61312
> [2024-11-09 09:26:41.006] WARNING[193832]: chan_iax2.c:4211 __get_from_jb: No voice format and no peer format available on IAX2/129.222.126.2:61312-14317, backlogging frame
> [2024-11-09 09:26:41.007] WARNING[193833]: chan_iax2.c:4211 __get_from_jb: No voice format and no peer format available on IAX2/129.222.126.2:61312-14317, backlogging frame

Does this point to a possible issue in apt_rpt?

David McAnally, WD5M

InterLinked1, 2024-11-09:
I think there are two issues here, once with chan_iax2, and one with app_rpt. Because the app_rpt issue is triggering the issue with chan_iax2, I'm going to focus on the chan_iax2 part first.

From my previous work on a related issue, I think the underlying issue has something to do with text frames being sent without any voice initially, my memory is a bit hazy but in the past that triggered this very issue with chan_iax2, only then it would cause a crash. Whether that's a legitimate thing to do or not, there is probably a better way to handle this. Once the high CPU is resolved, then it would make sense to look at the error in app_rpt as that shouldn't be happening either - but it makes more sense to look at the IAX2 issue first since this provides an ideal test scenario for that issue.

wd5m, 2024-11-13:
Something else that "might" be of interest. If I set "jitterbuffer = no" in iax.conf, asterisk does NOT go CPU bound when connecting from DVSM as an Allstar node number.   That's not something I would do even as a work around, but it might lead to useful info.

David McAnally WD5M

InterLinked1, 2024-11-13:
> Something else that "might" be of interest. If I set "jitterbuffer = no" in iax.conf, asterisk does NOT go CPU bound when connecting from DVSM as an Allstar node number. That's not something I would do even as a work around, but it might lead to useful info.

Thanks, yup, that code path is only used with the jitterbuffer enabled, so disabling it is a workaround in the meantime. The frames get backlogged into the jitterbuffer currently. Haven't had time to look at this further yet but hoping to take a look this weekend.

jxmx, 2024-12-06:
This seems to be cropping up more based on forum and Slack postings.

jxmx, 2024-12-06:
@InterLinked1 any ideas?

wd5m, 2025-01-01:
Just adding another example of this issue. I am able to initiate this failure by connecting from DVSM via WT Mode, then sending a function command via the keypad, e.g. *70 followed by send.  As described earlier, if I hit PTT the program loop stops. So this issue may not be limited to DVSM Allstar node account connections.

This test was with the latest ASL3 update on a Debian 12 Raspberry Pi 4.

> ********** AllStarLink [ASL] Version Info **********
> 
> OS            : Debian GNU/Linux 12 (bookworm)
> OS Kernel     : 6.6.62+rpt-rpi-v8
> 
> Asterisk      : 20.11.0+asl3-3.2.0-2.deb12
> ASL [app_rpt] : 3.2.0
> 
> Installed ASL packages :
> 
>   Package                         Version
>   ==============================  ==============================
>   allmon3                         1.4.2-1.deb12
>   asl3                            3.6.0-1.deb
>   asl3-asterisk                   2:20.11.0+asl3-3.2.0-2.deb12
>   asl3-asterisk-config            2:20.11.0+asl3-3.2.0-2.deb12
>   asl3-asterisk-modules           2:20.11.0+asl3-3.2.0-2.deb12
>   asl3-menu                       1.11-1.deb12
>   asl3-pi-appliance               1.8.1-1.deb12
>   asl3-update-nodelist            1.5.1-1.deb12
>   cockpit                         287.1-0+deb12u3
>   cockpit-bridge                  287.1-0+deb12u3
>   cockpit-networkmanager          287.1-0+deb12u3
>   cockpit-packagekit              287.1-0+deb12u3
>   cockpit-sosreport               287.1-0+deb12u3
>   cockpit-storaged                287.1-0+deb12u3
>   cockpit-system                  287.1-0+deb12u3
>   cockpit-wifimanager             1.1.1-1.deb12
>   cockpit-ws                      287.1-0+deb12u3
>   dahdi                           1:3.1.0-2
>   dahdi-dkms                      1:3.4.0-4+asl
>   dahdi-linux                     1:3.4.0-4+asl
>

Allan-N, 2025-01-01:
> Just adding another example of this issue. I am able to initiate this failure by connecting from DVSM via WT Mode, then sending a function command via the keypad, e.g. *70 followed by send. As [...]

What version of DVSM?  and what version of AndroidOS?
Any changes to default the DVSM settings?
Any unusual changes to your ASL configuration files?

wd5m, 2025-01-01:
> What version of DVSM? and what version of AndroidOS? 

DVSwitch Mobile Version 2.0.5 (198)

> Any changes to default the DVSM settings?

No.  

> Any unusual changes to your ASL configuration files?

No. This was a recent (new) ASL3 installation. After I noticed this issue happening in WT mode with function commands, I tried enabling/disabling various CODECs in ASL3 to see if any specific one affected this issue. DVSM is requesting SLIN, ULAW, ADPCM and ALAW. They all behaved the same.  

As noted earlier in this issue report, I can avoid the looping issue by creating a ASL node account entry in DVSM and setting its CODEC to ADPCM, or SLIN.  Configuring  a DVSM account entry for iaxrpt password authenticated access also exhibits the issue upon connecting to ASL3.

Previously, I had thought we could avoid this issue by asking our group of users to confine DVSM use to WT Mode. However, we want to use ASL function commands too,. This DVSM WT mode discovery added another wrinkle to our use guidance. 

David WD5M

Allan-N, 2025-01-01:
Thank you for the info.  Hopefully this will help us reproduce and resolve the issue.

InterLinked1, 2025-01-01:
> As noted earlier in this issue report, I can avoid the looping issue by creating a ASL node account entry in DVSM and setting its CODEC to ADPCM, or SLIN. Configuring a DVSM account entry for iaxrpt password authenticated access also exhibits the issue upon connecting to ASL3.

It makes sense that changing the codec would change things since one of the factors is there being no mutually agreed upon codecs.

I'm a bit surprised that though, it makes sense adpcm and slin work fine, but why not ulaw or alaw? Do both nodes not support that?

wd5m, 2025-01-01:
> I'm a bit surprised that though, it makes sense adpcm and slin work fine, but why not ulaw or alaw? Do both nodes not support that?

DVSM requests slin, ulaw, adpcm and alaw, if set to allow the server to select the CODEC. In DVSM account settings, I can only set for Server Selected, ulaw, adpcm and slin. I know that selecting ulaw in DVSM still exhibits the issue on ASL3. While selecting adpcm or slin does not.  On the other hand, only allowing adpcm, or slin, in ASL3 iax.conf configuration settings does not seem to prevent the issue. I suspect this might be related to how DVSM and ASL3 negotiate the connection. Since keying from the DVSM side seems to avoid or stop the issue, maybe ASL3 gets a bit more necessary information for the connection when DVSM transmits.

I don't find a setting for CODEC when WT mode is enabled in DVSM. In that case, DVSM requests slin, ulaw, adpcm or alaw from the server.

The ASL3 console shows one thing in common when the issue occurs, versus not occuring, with DVSM account CODEC settings.  When the issue occurs, using a DVSM account setting of "server selected" or "ulaw", the ASL3 console log indicates...

>    -- Accepting UNAUTHENTICATED call from 98.97.83.226:45657:
>     --        > requested format = Unknown,
>     --        > requested prefs = (slin|ulaw|adpcm|alaw),
>     --        > actual format = ulaw,
>     --        > host prefs = (slin|ulaw|adpcm|alaw|gsm),
>     --        > priority = mine

Note the requested format = Unknown.  When DVSM account CODEC is set to slin or adpcm, the requested format is slin or adpcm. 

When DVSM is set to use WT mode, the ASL3 console log shows...

>  Accepting AUTHENTICATED call from 98.97.83.226:45657:
>     --        > requested auth methods = (MD5),
>     --        > actual auth method = MD5,
>     --        > encrypted = no,
>     --        > requested format = Unknown,
>     --        > requested prefs = (slin|ulaw|adpcm|alaw),
>     --        > actual format = slin,
>     --        > host prefs = (slin|ulaw|adpcm|alaw|gsm),
>     --        > priority = mine

The "requested format = Unknown" is shown, however, the issue does not occur unless I send a function request immediately after connection, e.g. *70 send. If I transmit from DVSM first, the issue does not occur.

Note, I have added (allowed) slin and alaw in the ASL3 node for testing as shown in the console log above.

David

InterLinked1, 2025-01-10:
After some testing, this preliminary patch seems to resolve the issue - will be submitted upstream, but feedback welcome from the OPs with this issue:

```
diff --git a/channels/chan_iax2.c b/channels/chan_iax2.c
index a99e789de5..305efe914d 100644
--- a/channels/chan_iax2.c
+++ b/channels/chan_iax2.c
@@ -786,7 +786,8 @@ struct chan_iax2_pvt {
        unsigned short peercallno;
        /*! Negotiated format, this is only used to remember what format was
            chosen for an unauthenticated call so that the channel can get
-           created later using the right format */
+           created later using the right format. We also use it for
+        authenticated calls to check the format from __get_from_jb. */
        iax2_format chosenformat;
        /*! Peer selected format */
        iax2_format peerformat;
@@ -4205,10 +4206,25 @@ static void __get_from_jb(const void *p)
                         * In this case, fall back to using the format negotiated during call setup,
                         * so we don't stall the jitterbuffer completely. */
                        voicefmt = ast_format_compatibility_bitfield2format(pvt->peerformat);
+                       if (!voicefmt) {
+                               /* As a last resort, we can use pvt->chosenformat.
+                                * This is set when we receive a call (either authenticated or unauthenticated),
+                                * so even if we haven't received any voice frames yet, we can still use the
+                                * right format.
+                                *
+                                * If we have to do this, in most cases, we aren't even processing voice frames
+                                * anyways, it's likely a non-voice frame. In that case, the format doesn't
+                                * really matter so much, because we could just pass 20 to jb_get instead
+                                * of calling ast_format_get_default_ms. However, until jb_get returns,
+                                * we don't actually know what kind of frame it is for sure, so use
+                                * the right format just to be safe. */
+                               voicefmt = ast_format_compatibility_bitfield2format(pvt->chosenformat);
+                       }
                }
                if (!voicefmt) {
-                       /* Really shouldn't happen, but if it does, should be looked into */
-                       ast_log(LOG_WARNING, "No voice format and no peer format available on %s, backlogging frame\n", ast_channel_name(pvt->owner));
+                       /* Really shouldn't happen, but if it does, should be looked into.
+                        * This will cause a barrage of logs and result in an unusable channel. */
+                       ast_log(LOG_ERROR, "No voice, peer, or chosen format available on %s, backlogging frame\n", ast_channel_name(pvt->owner));
                        goto cleanup; /* Don't crash if there's no voice format */
                }
                ret = jb_get(pvt->jb, &frame, ms, ast_format_get_default_ms(voicefmt));
@@ -11557,6 +11573,11 @@ static int socket_process_helper(struct iax2_thread *thread)
                                                                                        VERBOSE_PREFIX_4,
                                                                                        using_prefs);

+                                                       /* Unlike unauthenticated calls, we don't need to store
+                                                        * the chosen format for channel creation.
+                                                        * However, this is helpful for __get_from_jb. */
+                                                       iaxs[fr->callno]->chosenformat = format;
+
                                                        ast_set_flag(&iaxs[fr->callno]->state, IAX_STATE_STARTED);
                                                        c = ast_iax2_new(fr->callno, AST_STATE_RING, format,
                                                                &iaxs[fr->callno]->rprefs, NULL, NULL, 1);
```

Allan-N, 2025-01-10:
>After some testing, this preliminary patch seems to resolve the issue - will be submitted upstream

and I will be working to incorporate the change into an ASL3 build

Allan-N, 2025-01-11:
@du6jip, @wd5m, and others watching :

We think we have a fix for this issue.  Please let me know if you are available/willing to help us test the fix.

wd5m, 2025-01-12:
I should be able to help test.

David McAnally
WD5M

closed 2025-01-16 by Allan-N