AllStarLink/app_rpt #421: PTT stuck on at boot

toyotahead, 2024-11-04:
I am in the process of upgrading my shack away from raspberry pi's. Consolidating all the individual rpi's functions onto a 8 core Intel(R) Atom(TM) CPU C2758 @ 2.40GHz CPU, 32GB ram, 512G SSD machine. The bare metal has Proxmox 8.2.7 on it. All of the various other functions (Allstar, APRS, APRSC, WEEWX, NOAAv2, OpenWebRX, ADS-B, and likely others I have forgotten at this moment) I have been able to get to work as a CT rather then as a VM except Allstar. So Allstar is the only VM and has a dedicated CM108 passed through to it.  Yes I agree, it is complete overkill....

**Describe the bug**
I am experiencing an issue with the PTT being stuck keyed upon first boot or whenever asterisk is restarted. To make it "behave" I went into Interface Tune CLI, and entered F to make the transmitter key up a few times with a test tone. (obviously the PTT is already keyed) Upon doing so the 3 tones can be heard, and once done the PTT remains off. From that point forward the PTT operates as per normal until the next reboot or restart of asterisk.

For a possible reference, a similar sounding but not exactly the same issue from the past: https://github.com/AllStarLink/ASL3/issues/76 

**To Reproduce**
Steps to reproduce the behaviour:
1. Warm or Cold boot the system (PTT is unkeyed at this point throughout BIOS and into the early stages of loading Debian12)
2. As the system boot nears the console prompt the PTT becomes keyed and remains keyed indefinitely.  The longest I have waited was 10min (radio was on ultra low power into a dummy load)
3. At this point if I enter the ASL-Menu and drill into the Interface Tuning CLI and initiate a F (Toggle PTT and Tone output several times) the radio will transmit 3x tones. The 1st tone is heard, PTT drops, PTT keys + 2nd tone, PTT drops, PTT keys + 3rd tone, PTT drops. At the conclusion of the 3rd tone the PTT remains off. 
4. PTT will now key up correctly from here forward because of local rf, internet traffic, telemetry, etc until the next reboot or restart of asterisk.

**Expected behaviour**
- Expected behaviour is that upon boot the PTT would remain in an unkeyed state until an active TX event initiates. 
- Expected behaviour is that upon restarting asterisk the PTT would remain in an unkeyed state until an active TX event initiates.

Software versions (listed in asl-menu, option 4)
OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.1.0-26-amd64
Asterisk      : 20.9.3+asl3-3.0.6-1.deb12
ASL [app_rpt] : 3.0.6
Installed ASL packages :
Package                         Version
 ==============================  
allmon3                         1.4.1-1.deb12
asl3                                3.5.0-1.deb
asl3-asterisk                   2:20.9.3+asl3-3.0.6-1.deb12
asl3-asterisk-config         2:20.9.3+asl3-3.0.6-1.deb12
asl3-asterisk-modules     2:20.9.3+asl3-3.0.6-1.deb12
asl3-menu                      1.9-1.deb12
dahdi                              1:3.1.0-2
dahdi-dkms                    1:3.4.0-4+asl
dahdi-linux                     1:3.4.0-4+asl

**Have you run a software update and rebooted?**
 - YES

**Platform**
  - A fresh Debian 12 Virtual Machine (non UEFI) running under a fresh Proxmox v8.2.7 
    (followed these instructions: https://allstarlink.github.io/user-guide/install/ )
  - CM108 usb is attached to the VM via hardware passthru
  - Radio Interface is a CM108 as a SimpleUSB device, Duplex 2.

**Additional context**
Using a CM108 to interface to the radios CDM750's. (In the past this adapter has worked flawlessly for years on these same radios while on a rpi 3b+ and the older allstar version) 
I am aware there are a few variants of CM108's out there but I am not sure how to identify this any more accurately as the pcb has been encapsulated in black expoxy and I can not read the writing on the chip.  

lsusb shows it as: 0d8c:000c C-Media Electronics, Inc. Audio Adapter

nano /sys/kernel/debug/usb/devices
T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=12   MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0d8c ProdID=000c Rev= 1.00
S:  Product=C-Media USB Headphone Set
C:* #Ifs= 4 Cfg#= 1 Atr=80 MxPwr=100mA
I:* If#= 0 Alt= 0 #EPs= 0 Cls=01(audio) Sub=01 Prot=00 Driver=snd-usb-audio
I:  If#= 1 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=00 Driver=snd-usb-audio
I:* If#= 1 Alt= 1 #EPs= 1 Cls=01(audio) Sub=02 Prot=00 Driver=snd-usb-audio
E:  Ad=01(O) Atr=09(Isoc) MxPS= 200 Ivl=1ms
I:  If#= 2 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=00 Driver=snd-usb-audio
I:* If#= 2 Alt= 1 #EPs= 1 Cls=01(audio) Sub=02 Prot=00 Driver=snd-usb-audio
E:  Ad=82(I) Atr=05(Isoc) MxPS= 100 Ivl=1ms
I:* If#= 3 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs
E:  Ad=83(I) Atr=03(Int.) MxPS=   4 Ivl=32ms

/var/log/asterisk/messages.log from the last boot up
`
[2024-11-04 13:48:55.291] Asterisk 20.9.3+asl3-3.0.6-1.deb12 built by builder @ allstarlink.org on a x86_64 running Linux on 2024-10-05 13:19:13 UTC
[... 193 lines cut ...]
;       place settings that are different than the template.
;
; Note: the device string is automatically found when the
;       USB setting "devstr=" is empty.
;
; Note: the interface "tune" settings will be added to the
;       per-node settings (below).

[53011](node-main)
preemphasis = yes
ctcssfrom = no

plfilter = no
invertptt = yes

;;;;; ASL3 Tune settings ;;;;;
devstr =
rxmixerset = 300
txmixaset = 700
txmixbset = 700
`

If there is any other information I have forgotten to attach please let me know. 

Thank you for your assistance.

encbar5, 2024-11-05:
@toyotahead The thread you linked to says that this was fixed in `3.0.11` but you are still running `3.0.6` from the logs above. What steps did you actually do to perform the software update?
@jxmx Are x86_64 packages always updated to the same version that the arch64 packages are?

jxmx, 2024-11-05:
Yes x86_64 and arm are always in sync

Allan-N, 2024-11-05:
The 3.0.11 reference was the Pi Appliance image version (allstar3-arm64-3.0.11.img.xz), not the app_rpt version (3.0.6)

... and, yes, we have too many version #'s

KB4MDD, 2024-11-05:
I need more debugging information.

Edit `/etc/asterisk/logger.conf`
Change 
`messages.log => notice,warning,error,dtmf`
to read
`messages.log => notice,warning,error,dtmf,debug,verbose`

Edit `/etc/asterisk/cli.conf `  
Add a line to the bottom that reads
`core set debug 7 chan_simpleusb.so`

Stop asterisk `systemctl stop asterisk`
Remove the current messages.log
`rm /var/log/asterisk/messages.log`

Start asterisk `systemctl start asterisk`

Attach the messages.log to this issue (provided that you are keyed up when you restart).

toyotahead, 2024-11-05:
> @toyotahead The thread you linked to says that this was fixed in `3.0.11` but you are still running `3.0.6` from the logs above. What steps did you actually do to perform the software update? @jxmx Are x86_64 packages always updated to the same version that the arch64 packages are?

I used standard apt.

sudo apt update
sudo apt upgrade

toyotahead, 2024-11-05:
> I need more debugging information.
>
> Edit `/etc/asterisk/logger.conf` Change `messages.log => notice,warning,error,dtmf` to read `messages.log => notice,warning,error,dtmf,debug,verbose`
>
> Edit [...]

[2024-11-05 16:38:01.938] NOTICE[2287] dnsmgr.c: Managed DNS entries will be refreshed every 300 seconds.
[2024-11-05 16:38:01.961] NOTICE[2287] indications.c: Default country for indication tones: us
[2024-11-05 16:38:01.961] NOTICE[2287] indications.c: Setting default indication country to 'us'
[2024-11-05 16:38:02.074] NOTICE[2287] res_smdi.c: No SMDI interfaces are available to listen on, not starting SMDI listener.
[2024-11-05 16:38:02.083] WARNING[2287] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2024-11-05 16:38:02.083] WARNING[2287] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2024-11-05 16:38:02.083] NOTICE[2287] chan_dahdi.c: Ignoring any changes to 'userbase' (on reload) at line 23.
[2024-11-05 16:38:02.083] NOTICE[2287] chan_dahdi.c: Ignoring any changes to 'vmsecret' (on reload) at line 31.
[2024-11-05 16:38:02.083] NOTICE[2287] chan_dahdi.c: Ignoring any changes to 'hassip' (on reload) at line 35.
[2024-11-05 16:38:02.083] NOTICE[2287] chan_dahdi.c: Ignoring any changes to 'hasiax' (on reload) at line 39.
[2024-11-05 16:38:02.083] NOTICE[2287] chan_dahdi.c: Ignoring any changes to 'hasmanager' (on reload) at line 47.
[2024-11-05 16:38:02.083] WARNING[2287] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2024-11-05 16:38:02.083] WARNING[2287] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2024-11-05 16:38:02.568] WARNING[2287] loader.c: Module 'res_adsi' has been loaded but may be removed in a future release.
[2024-11-05 16:38:02.606] WARNING[2287] config.c: No '=' (equal sign) in line 14 of /etc/asterisk/cli.conf
[2024-11-05 16:38:02.606] VERBOSE[2287] asterisk.c: Asterisk Ready.
[2024-11-05 16:38:02.629] NOTICE[2330] app_rpt.c: Normal Repeater Init  53011
[2024-11-05 16:38:02.634] NOTICE[2336] chan_simpleusb.c: Channel 53011: Set option TONE VERIFY, mode: OFF(0).
[2024-11-05 16:38:02.656] NOTICE[2337] chan_simpleusb.c: Channel 53011: Automatically assigned USB device 2-1:1.0 to SimpleUSB channel

Allan-N, 2024-11-05:
Note: if add a ".txt" extension to any log or configuratioin file then they can easily be uploaded to GitHub without losing any of the reformatting that occurs with copy/paste

KB4MDD, 2024-11-05:
If debugging is enabled and we are logging debug messages, we should be seeing the PTT action recorded in messages.log.  I don't see any debug messages from simpleusb.

toyotahead, 2024-11-05:
> If debugging is enabled and we are logging debug messages, we should be seeing the PTT action recorded in messages.log. I don't see any debug messages from simpleusb.

Since I did the instructions you mentioned earlier I have:

- executed the Interface Tune CLI option F (which keys the transmitter 3x
- there have been regular telemetry events
- ive connected to a remote node which has been keying up the transmitter seemingly normally without issue

Yet I do not see any of those events either?

Also I am having a thought...  what happens to the cm108's GPIO during an asterisk restart? Is there any initialization etc taking place? perhaps in this direction might be of interest?

Attached log again of since your instructions:
[messages.log](https://github.com/user-attachments/files/17638663/messages.log)

toyotahead, 2024-11-05:
> If debugging is enabled and we are logging debug messages, we should be seeing the PTT action recorded in messages.log. I don't see any debug messages from simpleusb.

I've confirmed the files have been edited as per your instructions, please 
[config files.txt](https://github.com/user-attachments/files/17638711/config.files.txt)
see attached.

KB4MDD, 2024-11-05:
@toyotahead any chance you can give me access to the machine?

I will note that we have not tested on a vm with usb pass through.

toyotahead, 2024-11-06:
> @toyotahead any chance you can give me access to the machine?
> 
> I will note that we have not tested on a vm with usb pass through.

I will send you an email to your address listed on your github profile.

KB4MDD, 2024-11-06:
Additional logging information:

```
[2024-11-06 09:41:42.150] NOTICE[7522] app_rpt.c: Normal Repeater Init  53011                                 
[2024-11-06 09:41:42.150] DEBUG[7529] app_rpt.c: Already have a config for 53011, skipping
[2024-11-06 09:41:42.151] DEBUG[7529] channel_internal_api.c:  <initializing>: Formats: (none)                
[2024-11-06 09:41:42.151] DEBUG[7529] channel_internal_api.c:  Channel is being initialized or destroyed      
[2024-11-06 09:41:42.151] DEBUG[7529] stasis.c: Creating topic. name: channel:1730896902.0, detail:           
[2024-11-06 09:41:42.151] DEBUG[7529] stasis.c: Topic 'channel:1730896902.0': 0x7fb02c004c80 created          
[2024-11-06 09:41:42.151] DEBUG[7529] channel.c: Channel 0x7fb02c001cd0 'SimpleUSB/53011' allocated           
[2024-11-06 09:41:42.151] DEBUG[7529] channel_internal_api.c:  SimpleUSB/53011: Formats: (slin)               
[2024-11-06 09:41:42.151] DEBUG[7529] channel_internal_api.c:  New topology set                               
[2024-11-06 09:41:42.151] DEBUG[7529] app_rpt/rpt_call.c: Requested channel SimpleUSB/53011                   
[2024-11-06 09:41:42.151] DEBUG[7529] channel.c: Channel driver natively set channel SimpleUSB/53011 to read  format slin                                                                                                   [2024-11-06 09:41:42.151] DEBUG[7529] channel_internal_api.c:  SimpleUSB/53011: Formats: (slin)               
[2024-11-06 09:41:42.151] DEBUG[7529] channel_internal_api.c:  New topology set                               
[2024-11-06 09:41:42.151] DEBUG[7529] channel.c: Channel driver natively set channel SimpleUSB/53011 to write format slin                                                                                                   [2024-11-06 09:41:42.151] DEBUG[7529] channel_internal_api.c:  SimpleUSB/53011: Formats: (slin)
[2024-11-06 09:41:42.151] DEBUG[7529] channel_internal_api.c:  New topology set
[2024-11-06 09:41:42.151] DEBUG[7529] app_rpt/rpt_call.c: No CDR present on SimpleUSB/53011
[2024-11-06 09:41:42.151] DEBUG[7529] app_rpt/rpt_call.c: rpt ((Repeater Rx)) initiating call to SimpleUSB/   53011 on SimpleUSB/53011
[2024-11-06 09:41:42.151] DEBUG[7529] chan_simpleusb.c: Channel 53011: ACRK TX ON.
[2024-11-06 09:41:42.151] DEBUG[7529] chan_simpleusb.c: Channel 53011: ACRUK TX OFF.
[2024-11-06 09:41:42.159] NOTICE[7529] chan_simpleusb.c: Channel 53011: Set option TONE VERIFY, mode: OFF(0).
[2024-11-06 09:41:42.226] NOTICE[7530] chan_simpleusb.c: Channel 53011: Automatically assigned USB device 2-1:1.0 to SimpleUSB channel
[2024-11-06 09:41:42.321] DEBUG[7530] chan_simpleusb.c: Channel 53011: Starting normally.                     
[2024-11-06 09:41:42.321] DEBUG[7530] chan_simpleusb.c: Channel 53011: Attached to usb device 2-1:1.0.
[2024-11-06 09:41:42.389] DEBUG[7530] chan_simpleusb.c: Channel 53011: Update rxhidsq = 1                     
[2024-11-06 09:41:42.389] DEBUG[7530] chan_simpleusb.c: Channel 53011: Update rxhidctcss = 1.
[2024-11-06 09:41:42.393] DEBUG[7529] chan_simpleusb.c: Channel 53011: fragment total 30, size 4096, available 30, bytes 122880

```
Results of Flash command
```
[2024-11-06 09:54:07.046] DEBUG[7602] chan_simpleusb.c: Channel 53011: update PTT = 1 on channel.             
[2024-11-06 09:54:07.049] DEBUG[7602] chan_simpleusb.c: Channel 53011: tx set to 1
[2024-11-06 09:54:08.096] DEBUG[7602] chan_simpleusb.c: Channel 53011: update PTT = 0.                        
[2024-11-06 09:54:08.096] DEBUG[7602] chan_simpleusb.c: Channel 53011: tx set to 0
[2024-11-06 09:54:09.046] DEBUG[7602] chan_simpleusb.c: Channel 53011: update PTT = 1 on channel.             
[2024-11-06 09:54:09.049] DEBUG[7602] chan_simpleusb.c: Channel 53011: tx set to 1
[2024-11-06 09:54:10.057] DEBUG[7602] chan_simpleusb.c: Channel 53011: update PTT = 0.                        
[2024-11-06 09:54:10.060] DEBUG[7602] chan_simpleusb.c: Channel 53011: tx set to 0
[2024-11-06 09:54:11.066] DEBUG[7602] chan_simpleusb.c: Channel 53011: update PTT = 1 on channel.             
[2024-11-06 09:54:11.069] DEBUG[7602] chan_simpleusb.c: Channel 53011: tx set to 1
[2024-11-06 09:54:12.084] DEBUG[7602] chan_simpleusb.c: Channel 53011: update PTT = 0.                        
[2024-11-06 09:54:12.087] DEBUG[7602] chan_simpleusb.c: Channel 53011: tx set to 0

```

toyotahead, 2024-11-06:
Hello KB4MDD,

Does this mean you found something wrong, or was i just not looking in the correct file earlier? Would you like me to reboot and see now?  I will leave it as is until you green light me to do so.

Best regards.

KB4MDD, 2024-11-06:
@toyotahead  You are clear to use your node.  I have data that I can now review to see what I can find.  I have not made any changes at this point.

toyotahead, 2024-11-17:
Hello KB4MDD,

I just did an apt update && apt upgrade followed by a reboot. 
Unfortunately upon reboot the stuck PTT remains.

I did order a couple more CM108 adapters. Perhaps I should try building one with a hardwired PTT inverted output, and of course the cosponsoring settings in the config? (the transistor installed)

KB4MDD, 2024-11-18:
Did you switch to the development build?  asl-repo-switch -l devel

Allan-N, 2024-11-18:
Minor update.  @KB4MDD has some changes that "may" help but those changes are not yet built into a package.  Stay tuned.

toyotahead, 2024-11-18:
Oh i missed that detail... apologies

I assume that is done in the apt sources?

That said, I'm not totally sure how to switch to the developement build...
Can you elaborate how to do this?

Thanks again!

Jared

Allan-N, 2024-11-18:
Info on switching between the main, beta, and devel package streams can be found @ [Beta Testing](https://allstarlink.github.io/user-guide/betatest).

PR #426 (2024-12-04): app_rpt:  Channel initialization for PTT
app_rpt is not waiting long enough in rpt_setup_channels for chan_simpleusb to initialize the usb device before asserting PTT on/off.

This change adds a 250ms delay before asserting PTT on and another 50ms delay before asserting PTT off.

I believe the intention of the original code was to insure that the PTT was toggled immediately after the channel driver was created by app_rpt.

Testing by the effected users is needed. This may close #421 and #424.
closed 2024-12-04 by Allan-N