AllStarLink/app_rpt #422: ASL 3 Crash on app_rpt/rpt_channel.c

mkmer, 2024-11-06:
**Describe the bug**
Repeater appears to "crash" randomly.

**To Reproduce**
Steps to reproduce the behavior:
Not sure what steps cause it to crash - maybe time?
At the "failure" the repeater was NOT active.

**Expected behavior**
Recovery (which it does appear to do fine)

**Screenshots**
Logs:
```
it appears the extirpation of the newkey timer starts the crash (maybe).

[2024-11-03 08:00:12.039] WARNING[604240][C-000001aa] app_rpt/rpt_channel.c: Failed to send text IAX2/68.58.182.228:4569-3685 on !NEWKEY1!
[2024-11-03 08:00:12.041] WARNING[421560][C-000001aa] channel.c: PBX may not have been terminated properly on 'IAX2/68.58.182.228:4569-3685'
[2024-11-03 08:00:12.059] ERROR[604240][C-000001aa] app_rpt/rpt_channel.c: FRACK!, Failed assertion bad magic number 0x0 for object 0x7fb718003b30 (0)
[2024-11-03 08:00:12.059] ERROR[604240][C-000001aa] : Got 11 backtrace records
# 0: /usr/sbin/asterisk(+0x6dc4d) [0x55832e4e0c4d]
# 1: /usr/sbin/asterisk(__ao2_lock+0xac) [0x55832e4e0d5c]
# 2: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x2357c) [0x7fb73962357c]
# 3: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x137f2) [0x7fb7396137f2]
# 4: /usr/sbin/asterisk(pbx_exec+0xce) [0x55832e5a373e]
# 5: /usr/sbin/asterisk(+0x124f75) [0x55832e597f75]
# 6: /usr/sbin/asterisk(+0x126d4c) [0x55832e599d4c]
# 7: /usr/sbin/asterisk(+0x1281eb) [0x55832e59b1eb]
# 8: /usr/sbin/asterisk(+0x1afd4d) [0x55832e622d4d]
# 9: /lib/x86_64-linux-gnu/libc.so.6(+0x89144) [0x7fb73d2f9144]
#10: /lib/x86_64-linux-gnu/libc.so.6(+0x1097dc) [0x7fb73d3797dc]

[2024-11-03 08:00:12.059] ERROR[604240][C-000001aa] channel.c: FRACK!, Failed assertion bad magic number 0x0 for object 0x7fb718003b30 (0)
[2024-11-03 08:00:12.060] ERROR[604240][C-000001aa] : Got 13 backtrace records
# 0: /usr/sbin/asterisk(+0x6dc4d) [0x55832e4e0c4d]
# 1: /usr/sbin/asterisk(__ao2_lock+0xac) [0x55832e4e0d5c]
# 2: /usr/sbin/asterisk(ast_sendtext_data+0x75) [0x55832e519385]
# 3: /usr/sbin/asterisk(ast_sendtext+0x75) [0x55832e519965]
# 4: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x23587) [0x7fb739623587]
# 5: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x137f2) [0x7fb7396137f2]
# 6: /usr/sbin/asterisk(pbx_exec+0xce) [0x55832e5a373e]
# 7: /usr/sbin/asterisk(+0x124f75) [0x55832e597f75]
# 8: /usr/sbin/asterisk(+0x126d4c) [0x55832e599d4c]
# 9: /usr/sbin/asterisk(+0x1281eb) [0x55832e59b1eb]
#10: /usr/sbin/asterisk(+0x1afd4d) [0x55832e622d4d]
#11: /lib/x86_64-linux-gnu/libc.so.6(+0x89144) [0x7fb73d2f9144]
#12: /lib/x86_64-linux-gnu/libc.so.6(+0x1097dc) [0x7fb73d3797dc]

[2024-11-03 08:00:12.060] ERROR[604240][C-000001aa] channel.c: FRACK!, Failed assertion bad magic number 0x0 for object 0x7fb718003b30 (0
)
[2024-11-03 08:00:12.060] ERROR[604240][C-000001aa] : Got 13 backtrace records
# 0: /usr/sbin/asterisk(+0x6dc4d) [0x55832e4e0c4d]
# 1: /usr/sbin/asterisk(__ao2_unlock+0x8a) [0x55832e4e0e1a]
# 2: /usr/sbin/asterisk(ast_sendtext_data+0x5bf) [0x55832e5198cf]
# 3: /usr/sbin/asterisk(ast_sendtext+0x75) [0x55832e519965]
# 4: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x23587) [0x7fb739623587]
# 5: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x137f2) [0x7fb7396137f2]
# 6: /usr/sbin/asterisk(pbx_exec+0xce) [0x55832e5a373e]
# 7: /usr/sbin/asterisk(+0x124f75) [0x55832e597f75]
# 8: /usr/sbin/asterisk(+0x126d4c) [0x55832e599d4c]
# 9: /usr/sbin/asterisk(+0x1281eb) [0x55832e59b1eb]
#10: /usr/sbin/asterisk(+0x1afd4d) [0x55832e622d4d]
#11: /lib/x86_64-linux-gnu/libc.so.6(+0x89144) [0x7fb73d2f9144]
#12: /lib/x86_64-linux-gnu/libc.so.6(+0x1097dc) [0x7fb73d3797dc]

[2024-11-03 08:00:12.060] WARNING[604240][C-000001aa] app_rpt/rpt_channel.c: Failed to send text  on !NEWKEY1!
[2024-11-03 08:00:12.060] ERROR[604240][C-000001aa] app_rpt/rpt_channel.c: FRACK!, Failed assertion bad magic number 0x0 for object 0x7fb718003b30 (0)
[2024-11-03 08:00:12.061] ERROR[604240][C-000001aa] : Got 10 backtrace records
# 0: /usr/sbin/asterisk(+0x6dc4d) [0x55832e4e0c4d]
# 1: /usr/sbin/asterisk(__ao2_unlock+0x8a) [0x55832e4e0e1a]
# 2: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x137f2) [0x7fb7396137f2]
# 3: /usr/sbin/asterisk(pbx_exec+0xce) [0x55832e5a373e]
# 4: /usr/sbin/asterisk(+0x124f75) [0x55832e597f75]
# 5: /usr/sbin/asterisk(+0x126d4c) [0x55832e599d4c]
# 6: /usr/sbin/asterisk(+0x1281eb) [0x55832e59b1eb]
# 7: /usr/sbin/asterisk(+0x1afd4d) [0x55832e622d4d]
# 8: /lib/x86_64-linux-gnu/libc.so.6(+0x89144) [0x7fb73d2f9144]
# 9: /lib/x86_64-linux-gnu/libc.so.6(+0x1097dc) [0x7fb73d3797dc]

[2024-11-03 08:00:17.610] Asterisk 20.9.3+asl3-3.0.6-1.deb12 built by builder @ allstarlink.org on a x86_64 running Linux on 2024-10-05 13:19:13 UTC
```
**Software versions (listed in asl-menu, option 4)**
 - asl3: 3.0.6-1
 - asl3-asterisk: Asterisk 20.9.3+asl3-3.0.6-1.deb12 

**Have you run a software update and rebooted?**
 - YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - :Platform
  - Debian 6.1.112-1 (2024-09-30) x86_64 GNU/Linux

**Additional context**
Add any other context about the problem here.
I may have truncated a bit more "important" info:

[2024-11-05 21:43:26.318] WARNING[1082641][C-00000009] app_rpt/rpt_channel.c: Failed to send text IAX2/68.58.182.228:4569-11436 on !NEWKEY1!
[2024-11-05 21:43:26.339] WARNING[1082641][C-00000009] app_rpt/rpt_channel.c: Failed to send text IAX2/68.58.182.228:4569-11436 on !NEWKEY1!
[2024-11-05 21:43:28.219] WARNING[1082629] app_rpt.c: 0x7f3c74009c20 newkeytimer expired on connected node, setting newkey from 2 to 0.
[2024-11-05 21:43:57.359] WARNING[1082673][C-0000000a] app_rpt/rpt_channel.c: Failed to send text IAX2/68.58.182.228:4569-5324 on !NEWKEY1!
[2024-11-05 21:43:57.359] WARNING[1082629][C-0000000a] channel.c: PBX may not have been terminated properly on 'IAX2/68.58.182.228:4569-5324'
[2024-11-05 21:43:57.379] ERROR[1082673][C-0000000a] app_rpt/rpt_channel.c: FRACK!, Failed assertion bad magic number 0x0 for object 0x7f3c700016d0 (0)

Allan-N, 2024-11-06:
Can you install the asl3-asterisk-dbgsym and asl3-asterisk-modules-dbgsym packages to your system.  This will augment any crash reports with some helpful information.

```
sudo apt install -y  asl3-asterisk-dbgsym asl3-asterisk-modules-dbgsym
```

KB4MDD, 2024-11-06:
After installing the symbols, please run `/var/lib/asterisk/scripts/ast_coredumper /core` 

The results are recorded in `/tmp`. Please post them to the issue. This may give us additional information.

mkmer, 2024-11-06:
Looks like the core is dumped here:
```var/lib/asterisk/scripts/ast_coredumper /var/lib/asterisk/core```
[core-asterisk-2024-11-06T12-00-17Z-full.txt](https://github.com/user-attachments/files/17649122/core-asterisk-2024-11-06T12-00-17Z-full.txt)
[core-asterisk-2024-11-06T12-00-17Z-info.txt](https://github.com/user-attachments/files/17649124/core-asterisk-2024-11-06T12-00-17Z-info.txt)
[core-asterisk-2024-11-06T12-00-17Z-locks.txt](https://github.com/user-attachments/files/17649126/core-asterisk-2024-11-06T12-00-17Z-locks.txt)
[core-asterisk-2024-11-06T12-00-17Z-thread1.txt](https://github.com/user-attachments/files/17649127/core-asterisk-2024-11-06T12-00-17Z-thread1.txt)
[core-asterisk-2024-01-16T23-04-58Z-full.txt](https://github.com/user-attachments/files/17649128/core-asterisk-2024-01-16T23-04-58Z-full.txt)
[core-asterisk-2024-11-06T12-00-17Z-brief.txt](https://github.com/user-attachments/files/17649129/core-asterisk-2024-11-06T12-00-17Z-brief.txt)

InterLinked1, 2024-11-06:
I just looked at this for a second, but the direct cause of the crash seems to be `l->chan` being NULL here: https://github.com/AllStarLink/app_rpt/blob/master/apps/app_rpt.c#L6761

I don't think it should be possible for it to be NULL there, so in that case there is likely some other bug upstream of that here.

mkmer, 2024-11-07:
I think the thing crashing is indicated here:
```
[2024-11-06 19:31:41.574] WARNING[1240639] app_rpt.c: 0x7f4ebc0098d0 newkeytimer expired on connected node, setting newkey from 2 to 0.
[2024-11-06 19:32:00.676] WARNING[1240723][C-0000000a] app_rpt/rpt_channel.c: Failed to send text IAX2/68.58.182.228:1031-3252 on !NEWKEY1!
```
when newkey "failed" the next event is "improper termination":
```
[2024-11-03 08:00:12.041] WARNING[421560][C-000001aa] channel.c: PBX may not have been terminated properly on 'IAX2/68.58.182.228:4569-3685'
[2024-11-03 08:00:12.059] ERROR[604240][C-000001aa] app_rpt/rpt_channel.c: FRACK!, Failed assertion bad magic number 0x0 for object 0x7fb718003b30 (0)
```

InterLinked1, 2024-11-07:
> I think the thing crashing is indicated here:
>
> ```
> [2024-11-06 19:31:41.574] WARNING[1240639] app_rpt.c: 0x7f4ebc0098d0 newkeytimer expired on connected node, setting newkey from 2 to 0.
> [2024-11-06 [...]

The improper termination event is expected, actually, as some work needs to be done still to get rid of that. The FRACK message is indicative of the problem and in the backtrace lines up with the line number mentioned.

The newkeytimer warning message is also an issue, but I don't recall it always causing a crash in the past, not exactly sure now. They probably are related in some way and we'll take a look into that.

mkmer, 2024-11-12:
I believe this is also related to #385.  
After a bit of testing, it appears any long term links will cause the above crash.  Additionally, it appears that after some amount of "ASL uptime", any inbound connection will cause an immediate crash/restart.

rob297, 2024-12-12:
I am having the same trouble.  It only takes about 5 to 6 days and we start getting the errors on new connections to the hub.  This started happening on the Raspberry Pi we loaded from the image about 6 months ago. The Pi working fine for several weeks and after an update around 3 months ago we started having the issue.   We are now running Debian 12 on a 64bit platform and installed asl3 using the repo.  We have roughly 25 to 40 nodes connected at a time.  If i disable the hub node 28528, the few connections on the other hubs seem to work fine.  However when I enable 28528 (main) and all the connections start trying to establish and its snowballs.  Only a reboot will fix the issue.

mkmer, 2024-12-12:
It may also be a useful idea to check the system memory (`free` command) when the events occur.  It's on my list to check next time it starts acting up.  I'm leaning towards a memory leak of some type that requires a full reboot system reboot to fully "recover".

rob297, 2024-12-12:
> @rob297 Please look above for the instructions for installing the symbols and obtaining crash dump information.

Have it ready for next crash,  Will post.  Thank you.

Allan-N, 2024-12-12:
>Have it ready for next crash, Will post. Thank you.

If you added the symbol packages then you should be able to run `ast_coredumper` on the core file that is already on your system.

rob297, 2024-12-12:
We have a net tonight so I will probably have it the morning.  It seems to happen when there are a lot of inbound connections at one time.   I disabled our main hub 28528 and with about 10 connections it seemed to be ok.  But when I enabled 28528 and the 27 connections came in it crashed.

rob297, 2024-12-13:
Here are the output file from the core dump.  I included the output log as well.  

[core-asterisk-2024-12-12T15-00-50Z-info.txt](https://github.com/user-attachments/files/18129031/core-asterisk-2024-12-12T15-00-50Z-info.txt)
[core-asterisk-2024-12-12T15-00-50Z-locks.txt](https://github.com/user-attachments/files/18129032/core-asterisk-2024-12-12T15-00-50Z-locks.txt)
[core-asterisk-2024-12-12T15-00-50Z-full.txt](https://github.com/user-attachments/files/18129033/core-asterisk-2024-12-12T15-00-50Z-full.txt)
[core-asterisk-2024-12-12T15-00-50Z-brief.txt](https://github.com/user-attachments/files/18129034/core-asterisk-2024-12-12T15-00-50Z-brief.txt)
[core-asterisk-2024-12-12T15-00-50Z-thread1.txt](https://github.com/user-attachments/files/18129035/core-asterisk-2024-12-12T15-00-50Z-thread1.txt)
[output.log](https://github.com/user-attachments/files/18129036/output.log)

rob297, 2024-12-26:
Update,
I was able to play around with the number of incoming connections to the system.  We currently have 17 repeaters on a VPN along with users from the internet.  Around 34 repeaters will connect on a restart.  I was able to configure IAX to only allow 1 connection from the subnet of our VPN.   All of the internet connections are able to establish.  There are some IAX errors but eventually they are able to connect.  I update the IAX  [callnumberlimits] from 1 to 5 and do an iax reload.  Repeaters connect without any issue.  I continue a reload letting 5 connect until they are all connected.  The system is stable.   I now have a way to safely restart the server.  Only other option is to connect outbound from the server to the VPN nodes. 

Added to iax.conf

[callnumberlimits]
192.168.11.0/255.255.255.0 = 5

Mason10198, 2025-01-06:
I can confirm that this issue is present on all three of the ASL3 servers that I host. As far as I can tell, @mkmer is correct about any connection events causing the crash after a certain amount of Asterisk uptime. My servers/nodes are:

- **55199**: Debian 12 VM, VOTER Simulcast system
  [core-asterisk-2025-01-03T14-11-53Z-brief.txt](https://github.com/user-attachments/files/18319962/core-asterisk-2025-01-03T14-11-53Z-brief.txt)
  [core-asterisk-2025-01-03T14-11-53Z-full.txt](https://github.com/user-attachments/files/18319967/core-asterisk-2025-01-03T14-11-53Z-full.txt)
  [core-asterisk-2025-01-03T14-11-53Z-info.txt](https://github.com/user-attachments/files/18319973/core-asterisk-2025-01-03T14-11-53Z-info.txt)
  [core-asterisk-2025-01-03T14-11-53Z-locks.txt](https://github.com/user-attachments/files/18319975/core-asterisk-2025-01-03T14-11-53Z-locks.txt)
  [core-asterisk-2025-01-03T14-11-53Z-thread1.txt](https://github.com/user-attachments/files/18319977/core-asterisk-2025-01-03T14-11-53Z-thread1.txt)
  ```plaintext
  [2024-12-27 17:30:44.225] WARNING[701248] channel.c: Exceptionally long voice queue length (97 voice / 97 total) queuing to IAX2/159.250.148.233:4568-10751
  [2024-12-27 17:30:45.504] WARNING[701246] channel.c: Exceptionally long voice queue length (97 voice / 97 total) queuing to IAX2/159.250.148.233:4568-10751
  [2024-12-27 17:30:46.784] WARNING[701247] channel.c: Exceptionally long voice queue length (97 voice / 97 total) queuing to IAX2/159.250.148.233:4568-10751
  [2024-12-27 17:30:52.293] WARNING[701301][C-0000000c] app_rpt/rpt_channel.c: Failed to send text IAX2/159.250.148.233:4568-10751 on !NEWKEY1!
  [2024-12-27 17:30:52.297] WARNING[701261][C-0000000c] channel.c: PBX may not have been terminated properly on 'IAX2/159.250.148.233:4568-10751'
  [2024-12-27 17:30:52.297] WARNING[701301][C-0000000c] app_rpt/rpt_channel.c: Failed to send text  on !NEWKEY1!
  [2024-12-27 17:30:52.297] ERROR[701301][C-0000000c] app_rpt/rpt_channel.c: FRACK!, Failed assertion bad magic number 0x0 for object 0x7f00d0001760 (0)
  [2024-12-27 17:30:52.297] ERROR[701301][C-0000000c] : Got 10 backtrace records
  # 0: /usr/sbin/asterisk(+0x6dcdd) [0x55a4297e9cdd]
  # 1: /usr/sbin/asterisk(__ao2_unlock+0x8a) [0x55a4297e9eaa]
  # 2: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x1357b) [0x7f014021357b]
  # 3: /usr/sbin/asterisk(pbx_exec+0xce) [0x55a4298ace3e]
  # 4: /usr/sbin/asterisk(+0x125675) [0x55a4298a1675]
  # 5: /usr/sbin/asterisk(+0x12744c) [0x55a4298a344c]
  # 6: /usr/sbin/asterisk(+0x1288eb) [0x55a4298a48eb]
  # 7: /usr/sbin/asterisk(+0x1b083d) [0x55a42992c83d]
  # 8: /lib/x86_64-linux-gnu/libc.so.6(+0x891c4) [0x7f0143ebf1c4]
  # 9: /lib/x86_64-linux-gnu/libc.so.6(+0x10985c) [0x7f0143f3f85c]

  [2024-12-27 17:30:57.995] Asterisk 20.10.0+asl3-3.1.0-1.deb12 built by builder @ allstarlink.org on a x86_64 running Linux on 2024-11-13 21:20:01 UTC
  [2024-12-27 17:30:58.007] NOTICE[701307] loader.c: 79 modules will be loaded.
  [2024-12-27 17:30:58.027] NOTICE[701307] cdr.c: CDR simple logging enabled.
  [2024-12-27 17:30:58.028] NOTICE[701307] dnsmgr.c: Managed DNS entries will be refreshed every 300 seconds.
  [2024-12-27 17:30:58.033] NOTICE[701307] indications.c: Default country for indication tones: us
  [2024-12-27 17:30:58.033] NOTICE[701307] indications.c: Setting default indication country to 'us'
  [2024-12-27 17:30:58.080] NOTICE[701307] res_smdi.c: No SMDI interfaces are available to listen on, not starting SMDI listener.
  [2024-12-27 17:30:58.082] WARNING[701307] chan_dahdi.c: Only FXO signalled channels may belong to a call group
  [2024-12-27 17:30:58.082] WARNING[701307] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
  [2024-12-27 17:30:58.082] NOTICE[701307] chan_dahdi.c: Ignoring any changes to 'userbase' (on reload) at line 23.
  [2024-12-27 17:30:58.082] NOTICE[701307] chan_dahdi.c: Ignoring any changes to 'vmsecret' (on reload) at line 31.
  [2024-12-27 17:30:58.082] NOTICE[701307] chan_dahdi.c: Ignoring any changes to 'hassip' (on reload) at line 35.
  [2024-12-27 17:30:58.082] NOTICE[701307] chan_dahdi.c: Ignoring any changes to 'hasiax' (on reload) at line 39.
  [2024-12-27 17:30:58.082] NOTICE[701307] chan_dahdi.c: Ignoring any changes to 'hasmanager' (on reload) at line 47.
  [2024-12-27 17:30:58.082] WARNING[701307] chan_dahdi.c: Only FXO signalled channels may belong to a call group
  [2024-12-27 17:30:58.082] WARNING[701307] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
  [2024-12-27 17:30:58.137] WARNING[701307] loader.c: Module 'res_adsi' has been loaded but may be removed in a future release.
  [2024-12-27 17:30:58.346] NOTICE[701350] app_rpt.c: Normal Repeater Init  55199
  ```

- **40010**: Debian 12 VM, Radioless hub node
  [core-asterisk-2025-01-06T03-08-05Z-brief.txt](https://github.com/user-attachments/files/18319992/core-asterisk-2025-01-06T03-08-05Z-brief.txt)
  [core-asterisk-2025-01-06T03-08-05Z-full.txt](https://github.com/user-attachments/files/18319993/core-asterisk-2025-01-06T03-08-05Z-full.txt)
  [core-asterisk-2025-01-06T03-08-05Z-info.txt](https://github.com/user-attachments/files/18319994/core-asterisk-2025-01-06T03-08-05Z-info.txt)
  [core-asterisk-2025-01-06T03-08-05Z-locks.txt](https://github.com/user-attachments/files/18319995/core-asterisk-2025-01-06T03-08-05Z-locks.txt)
[... cut ...]
  [2024-11-14 20:18:35.009] NOTICE[785993] chan_dahdi.c: Ignoring any changes to 'hasmanager' (on reload) at line 47.
  [2024-11-14 20:18:35.009] WARNING[785993] chan_dahdi.c: Only FXO signalled channels may belong to a call group
  [2024-11-14 20:18:35.009] WARNING[785993] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
  [2024-11-14 20:18:35.636] WARNING[785993] loader.c: Module 'res_adsi' has been loaded but may be removed in a future release.
  [2024-11-14 20:18:35.825] NOTICE[786036] app_rpt.c: Normal Repeater Init  454130
  ```

mkmer, 2025-01-06:
@Mason10198  - If your still have not rebooted, check memory availability (it's on my list for the next crash).  I'm suspicious of a memory leak as Asterisk crashes, restarts and still continues to have connection issues (in my case anyway). 

It also may just be related to the number of connect/disconnect events  - we have an automatic connect/disconnect each day which after 2 months would reach a certain number of events making it appear to be time based :(  

I think this is going to be a bit challenging to find unless we can figure out how to force it to happen :(

InterLinked1, 2025-01-06:
> @Mason10198 - If your still have not rebooted, check memory availability (it's on my list for the next crash). I'm suspicious of a memory leak as Asterisk crashes, restarts and still continues to have connection issues (in my case anyway).

This isn't a memory leak, it is a bug in `app_rpt` that is triggered by some combination of events.

If you really suspect a memory issue, you will need to compile in developer mode with `MALLOC_DEBUG` and you will have some CLI commands e.g. `memory show allocations` and `memory show summary` that you can use to confirm further.

mkmer, 2025-01-06:
I'm thinking the key point is here:
```[2025-01-05 21:08:05.322] WARNING[568][C-00000175] channel.c: PBX may not have been terminated properly on 'IAX2/74.197.213.129:4567-440'```
For whatever reason the channel code decided to close down the channel and report it as such, then
```[2025-01-05 21:08:05.344] ERROR[4171][C-00000175] app_rpt/rpt_channel.c: FRACK!, Failed assertion bad magic number 0x0 for object 0x555d7e5ca2d0 (0)```
Tried to send the message a second time on a cleared channel - must be rpt_channel.c is hanging onto the closed channel pointer and it complains. 
I'm no expert here.  I'll try anything to help figure it out.

InterLinked1, 2025-01-07:
> Appears this was also reported here: #85 without a resolution.
> 
> I have a few questions for the more experienced folks -> https://github.com/mkmer/app_rpt/blob/7c4d46e67cfd3f8d4ba04cebec5f28e1cd7d07b4/apps/app_rpt.c#L6760
> 
> Should this be `send_newkey(l->chan);` (Not sure it makes a difference, but everything around it uses `l->chan`) My understanding on the locking is very weak, probably makes no difference.... Same here: https://github.com/mkmer/app_rpt/blob/7c4d46e67cfd3f8d4ba04cebec5f28e1cd7d07b4/apps/app_rpt.c#L6747

Looking at the git blame, that appears to be Jim's original code, so I don't think we introduced a regression, though it's possible Jim made a mistake.

`chan` appears in several places and given the multi-channel interactions, I'm not convinced that it's a mistake.

> Based on these comments: https://github.com/mkmer/app_rpt/blob/7c4d46e67cfd3f8d4ba04cebec5f28e1cd7d07b4/apps/app_rpt.c#L3057-L3090
> 
> and the errors shared, the channel is hung up and app_rpt tries to send the second `send_newkey()` to a channel that is NULL (or not lockable anyway)?
>
> Maybe the short term answer is to check the channel is still active before trying to lock it and/or unlock it here: https://github.com/mkmer/app_rpt/blob/7c4d46e67cfd3f8d4ba04cebec5f28e1cd7d07b4/apps/app_rpt/rpt_channel.c#L467 and here: https://github.com/mkmer/app_rpt/blob/7c4d46e67cfd3f8d4ba04cebec5f28e1cd7d07b4/apps/app_rpt/rpt_channel.c#L471

I think we've seen there are a few cases where the channel is definitely NULL and that is definitely wrong. So we could certainly add an assertion prior to that. That would still result in a crash though as it doesn't fix the bug, just makes it slightly more obvious that it's happening.

> I'm thinking the key point is here:
> `[2025-01-05 21:08:05.322] WARNING[568][C-00000175] channel.c: PBX may not have been terminated properly on 'IAX2/74.197.213.129:4567-440'`

This actually appears for every call, see https://github.com/AllStarLink/app_rpt/issues/15
It is a problem that should be resolved but shouldn't generally cause any issue, I don't think it's related...

> For whatever reason the channel code decided to close down the channel and report it as such, then
> `[2025-01-05 21:08:05.344] ERROR[4171][C-00000175] app_rpt/rpt_channel.c: FRACK!, Failed assertion bad magic number 0x0 for object 0x555d7e5ca2d0 (0)`

Yup, so in this case it looks like we're trying to use a NULL channel somewhere, and an assertion could be added for that. It's still a bug though, while skipping if the channel is NULL would prevent the crash, this may be reflective of something else having gone wrong previously.

@mkmer If you are able to reproduce this consistently, I could prepare a patch for you that tries out a mitigation, and maybe you could let me know if it improves things for you? I'm not personally able to test the ramifications of such changes so that would be really helpful!

mkmer, 2025-01-07:
I was thinking the channel.c code decided to close the channel, yet the repeater code is "mid logic" trying to talk the the closed channel (due to threading issues) - If we checked it and didn't crash, the next time the repeater code started a pass it would "know" the channel is closed...  Figuring out why channel.c is closing the channel (maybe as simple as a remote disconnect / network problem?) may be important too.  (My understanding of the system here is VERY limited, just getting my feet a bit wet)

I could test (and would be happy to), but I am only seeing the problem every month or two.  Maybe @rob297 could give it a try as well as he seems to have a configuration that more consistently "fails".

On reading @rob297's comments, maybe the NEWKEYTIME is a bit too short for long delays when the system is starting up or many connections hit in a short period causing a disconnect "thrashing" event? 

Thanks for looking at this further!

InterLinked1, 2025-01-07:
I wish I could answer all your questions :)
Truthfully, I don't know for sure. The guy who wrote all this is sadly gone so the rest of us are just trying to figure out how it works. In my background, I have a strong background with both C and Asterisk, but no actual background in radio repeaters or use of the AllStarLink system, so at times it's a bit hard for me to piece together how it's supposed to work from a user perspective. So sometimes we have to experiment and see if it helps.

That being said, I think this points to a NULL pointer being passed in at this point where there's an assertion that it shouldn't be NULL:
`app_rpt/rpt_channel.c: FRACK!, Failed assertion bad magic number 0x0 for object 0x7f0184016080 (0)`

Looks like one we may have added in the past.

mkmer, 2025-01-07:
Ah - so I'm barking up the wrong pointer :)

One more observation here: It appears in all cases here that a channel is in the RING state - maybe helpful...
```IAX2/74.197.213.129:4567-15787 radio-secure", ' ' <repeats 67 times> 40010", ' ' <repeats 74 times>         12            AST_STATE_RING                  Rpt     ```

```IAX2/68.58.182.228:4569-11738  radio-secure", ' ' <repeats 67 times> 29972", ' ' <repeats 74 times>          1            AST_STATE_RING                  Rpt     ```

Thanks for straightening me out. Maybe I'll be a bit more helpful some day soon.

mkmer, 2025-01-07:
Had some time to play around with rapid connect/disconnect/connect -
1) If the node disconnects quickly, we see this "normally" with no crash:
```
[2025-01-07 16:50:35.819] WARNING[3588755][C-0000054f]: app_rpt/rpt_channel.c:469 send_newkey: Failed to send text IAX2/149.154.11.243:24287-10914 on !NEWKEY1!
    -- Remote UNIX connection
    -- Remote UNIX connection disconnected
[2025-01-07 16:50:35.840] WARNING[3588755][C-0000054f]: app_rpt/rpt_channel.c:469 send_newkey: Failed to send text IAX2/149.154.11.243:24287-10914 on !NEWKEY1!
    -- Remote UNIX connection
    -- Remote UNIX connection disconnected
[2025-01-07 16:50:37.720] WARNING[2480089]: app_rpt.c:3102 periodic_process_links: 0x5652ef0773b0 newkeytimer expired on connected node, setting newkey from 2 to 0.
```

After multiple connect/disconnect tries, I did get it to crash - so I think I can force it to crash.

[core-asterisk-2025-01-07T21-51-36Z-brief.txt](https://github.com/user-attachments/files/18338910/core-asterisk-2025-01-07T21-51-36Z-brief.txt)
[core-asterisk-2025-01-07T21-51-36Z-info.txt](https://github.com/user-attachments/files/18338913/core-asterisk-2025-01-07T21-51-36Z-info.txt)
[core-asterisk-2025-01-07T21-51-36Z-locks.txt](https://github.com/user-attachments/files/18338916/core-asterisk-2025-01-07T21-51-36Z-locks.txt)
[core-asterisk-2025-01-07T21-51-36Z-full.txt](https://github.com/user-attachments/files/18338921/core-asterisk-2025-01-07T21-51-36Z-full.txt)
[core-asterisk-2025-01-07T21-51-36Z-thread1.txt](https://github.com/user-attachments/files/18338948/core-asterisk-2025-01-07T21-51-36Z-thread1.txt)

and this is the console text if at all useful:
```
[2025-01-07 16:51:26.610] WARNING[2480089]: app_rpt.c:3102 periodic_process_links: 0x7f9dcc011750 newkeytimer expired on connected node, setting newkey fro
    -- Accepting UNAUTHENTICATED call from 149.154.11.243:24287:
    --        > requested format = ulaw,
    --        > requested prefs = (ulaw|adpcm|gsm),
    --        > actual format = ulaw,
    --        > host prefs = (slin|ulaw|gsm),
    --        > priority = mine
    -- Executing [29972@radio-secure:1] Rpt("IAX2/149.154.11.243:24287-11254", "29972") in new stack
    -- Hungup 'IAX2/149.154.11.243:24287-14156'
    -- Hungup 'DAHDI/pseudo-649184076'
[2025-01-07 16:51:36.581] WARNING[3589107][C-00000555]: app_rpt/rpt_channel.c:469 send_newkey: Failed to send text IAX2/149.154.11.243:24287-11254 on !NEWK
    -- Hungup 'IAX2/149.154.11.243:24287-11254'
[2025-01-07 16:51:36.581] WARNING[2480089][C-00000555]: channel.c:2291 ast_channel_destructor: PBX may not have been terminated properly on 'IAX2/149.154.1
[2025-01-07 16:51:36.590] ERROR[3589107][C-00000555]: app_rpt/rpt_channel.c:467 send_newkey: FRACK!, Failed assertion bad magic number 0x0 for object 0x7f9
[2025-01-07 16:51:36.590] ERROR[3589107][C-00000555]:   Got 11 backtrace records
# 0: /usr/sbin/asterisk(+0x6ecdd) [0x5652ee8abcdd]
# 1: /usr/sbin/asterisk(__ao2_lock+0xac) [0x5652ee8abdec]
# 2: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x2355c) [0x7f9ded62355c]
# 3: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x1350b) [0x7f9ded61350b]
# 4: /usr/sbin/asterisk(pbx_exec+0xce) [0x5652ee96efae]
# 5: /usr/sbin/asterisk(+0x1267e5) [0x5652ee9637e5]
# 6: /usr/sbin/asterisk(+0x1285bc) [0x5652ee9655bc]
# 7: /usr/sbin/asterisk(+0x129a5b) [0x5652ee966a5b]
# 8: /usr/sbin/asterisk(+0x1b1d4d) [0x5652ee9eed4d]
# 9: /lib/x86_64-linux-gnu/libc.so.6(+0x891c4) [0x7f9df12f91c4]
#10: /lib/x86_64-linux-gnu/libc.so.6(+0x10985c) [0x7f9df137985c]

[2025-01-07 16:51:36.590] ERROR[3589107][C-00000555]: channel.c:4818 ast_sendtext_data: FRACK!, Failed assertion bad magic number 0x0 for object 0x7f9de801
[2025-01-07 16:51:36.596] ERROR[3589107][C-00000555]:   Got 13 backtrace records
# 0: /usr/sbin/asterisk(+0x6ecdd) [0x5652ee8abcdd]
# 1: /usr/sbin/asterisk(__ao2_lock+0xac) [0x5652ee8abdec]
# 2: /usr/sbin/asterisk(ast_sendtext_data+0x75) [0x5652ee8e45e5]
# 3: /usr/sbin/asterisk(ast_sendtext+0x75) [0x5652ee8e4bc5]
# 4: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x23567) [0x7f9ded623567]
# 5: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x1350b) [0x7f9ded61350b]
# 6: /usr/sbin/asterisk(pbx_exec+0xce) [0x5652ee96efae]
# 7: /usr/sbin/asterisk(+0x1267e5) [0x5652ee9637e5]
# 8: /usr/sbin/asterisk(+0x1285bc) [0x5652ee9655bc]
# 9: /usr/sbin/asterisk(+0x129a5b) [0x5652ee966a5b]
#10: /usr/sbin/asterisk(+0x1b1d4d) [0x5652ee9eed4d]
#11: /lib/x86_64-linux-gnu/libc.so.6(+0x891c4) [0x7f9df12f91c4]
#12: /lib/x86_64-linux-gnu/libc.so.6(+0x10985c) [0x7f9df137985c]

[2025-01-07 16:51:36.596] ERROR[3589107][C-00000555]: channel.c:4821 ast_sendtext_data: FRACK!, Failed assertion bad magic number 0x0 for object 0x7f9de801
[2025-01-07 16:51:36.597] ERROR[3589107][C-00000555]:   Got 13 backtrace records
# 0: /usr/sbin/asterisk(+0x6ecdd) [0x5652ee8abcdd]
# 1: /usr/sbin/asterisk(__ao2_unlock+0x8a) [0x5652ee8abeaa]
# 2: /usr/sbin/asterisk(ast_sendtext_data+0x5bf) [0x5652ee8e4b2f]
# 3: /usr/sbin/asterisk(ast_sendtext+0x75) [0x5652ee8e4bc5]
# 4: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x23567) [0x7f9ded623567]
# 5: /usr/lib/x86_64-linux-gnu/asterisk/modules/app_rpt.so(+0x1350b) [0x7f9ded61350b]
# 6: /usr/sbin/asterisk(pbx_exec+0xce) [0x5652ee96efae]
# 7: /usr/sbin/asterisk(+0x1267e5) [0x5652ee9637e5]
# 8: /usr/sbin/asterisk(+0x1285bc) [0x5652ee9655bc]
# 9: /usr/sbin/asterisk(+0x129a5b) [0x5652ee966a5b]
#10: /usr/sbin/asterisk(+0x1b1d4d) [0x5652ee9eed4d]
#11: /lib/x86_64-linux-gnu/libc.so.6(+0x891c4) [0x7f9df12f91c4]
#12: /lib/x86_64-linux-gnu/libc.so.6(+0x10985c) [0x7f9df137985c]

[2025-01-07 16:51:36.597] WARNING[3589107][C-00000555]: app_rpt/rpt_channel.c:469 send_newkey: Failed to send text  on !NEWKEY1!
[2025-01-07 16:51:36.597] ERROR[3589107][C-00000555]: app_rpt/rpt_channel.c:471 send_newkey: FRACK!, Failed assertion bad magic number 0x0 for object 0x7f9
```

InterLinked1, 2025-01-07:
> After multiple connect/disconnect tries, I did get it to crash - so I think I can force it to crash.

This is great news! A consistently reproducing bug is a developer's best friend :)

> Thanks for straightening me out. Maybe I'll be a bit more helpful some day soon.

You're already being plenty helpful, thanks for helping to test these issues out.

I'll look into this a little bit more and see if I can put something together from that information.

mkmer, 2025-01-08:
One more tid bit here:  It appears to crash on the next connection following the 2 to 0 "fix":

```[2025-01-07 19:21:53.083] WARNING[3611528]: app_rpt.c:3102 periodic_process_links: 0x7f1a60009200 newkeytimer expired on connected node, setting newkey from 2 to 0.```

If we rapidly connect/disconnect, it "usually" fails 2 !NEWKEY1! but does not always trigger the periodic_process_links logic.  It appears that once this is triggered, the very next connection will crash.

InterLinked1, 2025-01-08:
> Asterisk claims we don't need to lock the channel to sendtext...

It isn't required, but channels can be recursively locked in Asterisk.
Whether or not it is locked, if the channel is NULL or invalid, an issue is inevitable.

> The here key is `[2025-01-07 16:51:36.581] WARNING[2480089][C-00000555]: channel.c:2291 ast_channel_destructor: PBX may not have been terminated properly on 'IAX2/149.154.1`

That happens every time the `Rpt()` application is called. It will require core Asterisk changes to work around that more elegantly, this is just a workaround which is why that message appears.

> I suppose we need to understand what triggers `ast_channel_destructor` to get a better handle on this... so far it's path to execution is evading me. It looks like we pass it to the setup function here: https://github.com/asterisk/asterisk/blob/18573fa36e1ab1a39a79cf924cc2a3845a874214/main/channel.c#L755

Channels are reference counted, the destructor is only called when there are no remaining references to it (everyone has called `ast_channel_unref` on it that previously had a reference to it).

Allan-N, 2025-01-08:
There are a few places in app_rpt.c where the l->chan is set to 0 (NULL).

In periodic_process_links() :
https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L3226-L3229

In remote_hangup_helper() :
https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4150-L4181

Since you are appear to be able to trigger the crashes I wonder if adding some logging to (a) note when l->chan is NULL'd out and (b) when we find l->chan==0 when we're expecting otherwise.  The logs will hopefully shed some more light into what's the sequence of events.

mkmer, 2025-01-08:
@Allan-N  - Thanks for the ideas.  

Adding debug pointed to the ```hangup_helper``` https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4143 
acting on a hangup message (which appears to be running in ```process_link_channels``` https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4495-L4497
This is happening while ```rpt_exec``` is still mid setup on the connection. https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L6154

I see there has been a TON of debug on this code, but this solution appears to fix the race condition by allowing only letting process link channels send the newkey.  It has solved crashing problem and appears to keep all link logic still working as expected.  Maybe there are other ways to solve it.  I think we need more folks to test out this proposal as it seems like there are other situations that it may break.

mkmer, 2025-01-08:
I guess this will take smarter guys then me :)  Hopefully figuring out what the issue is will allow you to propose an answer that works.

mkmer, 2025-01-09:
> That happens every time the `Rpt()` application is called. It will require core Asterisk changes to work around that more elegantly, this is just a workaround which is why that message appears.

You have said this a couple of times, but I am only seeing this message (in my instance anyway) when quickly disconnecting.  Normal (longer connections that ) disconnect do not appear generate it.

mkmer, 2025-01-09:
I was thinking we could check ```l->thisconnected``` before sending messages in  ```rpt_exec```.
Then I noticed https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4143 is a bit odd here: https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4169-L4180

We are setting ```l->hasconnected = 1``` even though it had to be 1 in the if statement.  One wonders if the test should be ```l->thisconnected```

It's also a bit strange that we are locking (and not unlocking) before we return. 
https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4155

https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4170

Looks a bit fishy as both calls to this function lock it again.

InterLinked1, 2025-01-10:
> I was thinking we could check `l->thisconnected` before sending messages in `rpt_exec`. Then I noticed [...]

But `l->thisconnected` is set to 0 in that same code block...

> It's also a bit strange that we are locking (and not unlocking) before we return.

This is similar to the other find from earlier, looks like it's supposed to return locked (although I agree that's confusing).

> https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4155
> 
> https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4170
> 
> Looks a bit fishy as both calls to this function lock it again.

The lock is expected to not be held when calling that function, but is expected to be held when returning from it.
All paths that I looked at seem to follow that, are you seeing a deviation from that?

InterLinked1, 2025-01-10:
> @Allan-N - Thanks for the ideas.
>
> Adding debug pointed to the `hangup_helper`
>
> https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4143
>
>
> acting on a [...]

Hmm, I guess I could see that causing issues potentially.

A more robust solution to this sort of thing in general is probably to explicitly increase the reference count of the channel by calling `ast_channel_ref` (and `unref` later) explicitly. This will ensure the channel doesn't get destroyed until we're ready. I'm not sure if `app_rpt` originally predated reference-counted channels in Asterisk, but given the multithreading here and various functions handling the channel, it would probably make sense to add that. That could possibly resolve any issues related to channels disappearing unexpectedly.

InterLinked1, 2025-01-10:
> You have said this a couple of times, but I am only seeing this message (in my instance anyway) when quickly disconnecting. Normal (longer connections that ) disconnect do not appear generate it.

Apologies, I might have been thinking of a slightly different warning here. I recall there's a warning related to the unclean return from `rpt_exec`. I see it's during channel cleanup so maybe that's different. Not sure if it's related but I'll see about resolving the related issue just so we can get that out of the way as being a factor on this.

mkmer, 2025-01-10:
Adding the ```break;``` from https://github.com/AllStarLink/app_rpt/pull/448 appears to fix the "crash" exit, but now Asterisk is "cleanly exiting" in the rapid connect/disconnect scenario, yet there is a core:

Some log from the console:
```
[2025-01-10 06:56:02.060] WARNING[1577243]: app_rpt.c:3102 periodic_process_links: 0x7f97bc089280 newkeytimer expired on connected node, setting newkey from 2 to 0.
  == Connect attempt from '44.15.66.2' unable to authenticate
    -- Accepting UNAUTHENTICATED call from 149.154.11.243:24499:
    --        > requested format = ulaw,
    --        > requested prefs = (ulaw|adpcm|gsm),
    --        > actual format = ulaw,
    --        > host prefs = (ulaw|g726aal2|gsm),
    --        > priority = mine
    -- Executing [287893@radio-secure:1] Rpt("IAX2/149.154.11.243:24499-5395", "287893") in new stack
    -- Hungup 'DAHDI/pseudo-2036423997'
<< [ TYPE: Control (4) SUBCLASS: Private Cause Code (33) ] [IAX2/149.154.11.243:24499-5395]
<< [ HANGUP (NULL) ] [IAX2/149.154.11.243:24499-5395]
    -- Hungup 'IAX2/149.154.11.243:24499-5395'
[2025-01-10 06:56:06.838] WARNING[1577243][C-00000027]: channel.c:2272 ast_channel_destructor: PBX may not have been terminated properly on 'IAX2/149.154.11.243:24499-5395'
Hub01*CLI>
Disconnected from Asterisk server
Asterisk cleanly ending (0).
Executing last minute cleanups
```
[core-asterisk-2025-01-10T11-56-06Z-full.txt](https://github.com/user-attachments/files/18376530/core-asterisk-2025-01-10T11-56-06Z-full.txt)
[core-asterisk-2025-01-10T11-56-06Z-brief.txt](https://github.com/user-attachments/files/18376531/core-asterisk-2025-01-10T11-56-06Z-brief.txt)
[core-asterisk-2025-01-10T11-56-06Z-locks.txt](https://github.com/user-attachments/files/18376532/core-asterisk-2025-01-10T11-56-06Z-locks.txt)
[core-asterisk-2025-01-10T11-56-06Z-info.txt](https://github.com/user-attachments/files/18376533/core-asterisk-2025-01-10T11-56-06Z-info.txt)

mkmer, 2025-01-10:
This debug trace may help a bit more too.  I added a few debug messages at the "Answer the call", "Send key #1" and "Send key #2" )
I think it shows us that the channel is being killed in between key 1 and key 2 with 2 refs to the channel.  I think that's why we get the error: channel.c is freeing the channel knowing there is still a reference to it.  Since rpt_app.c hasn't reached the end, it hasn't deferenced the channel yet.

```
[2025-01-10 08:37:03.248] DEBUG[1595708][C-00000025] app_rpt.c: rpt_exec has started
[2025-01-10 08:37:03.248] DEBUG[1595708][C-00000025] channel.c: Channel IAX2/149.154.11.243:24229-5246 setting read format path: ulaw -> slin
[2025-01-10 08:37:03.248] DEBUG[1595708][C-00000025] channel.c: Channel IAX2/149.154.11.243:24229-5246 setting write format path: slin -> ulaw
[2025-01-10 08:37:03.376] DEBUG[1595068] channel.c: Channel 0x7f31ac001320 'IAX2/149.154.11.243:24229-8360' hanging up.  Refs: 2
[2025-01-10 08:37:03.376] DEBUG[1595068] chan_iax2.c: We're hanging up IAX2/149.154.11.243:24229-8360 now...
[2025-01-10 08:37:03.376] DEBUG[1595068] chan_iax2.c: Really destroying IAX2/149.154.11.243:24229-8360 now...
[2025-01-10 08:37:03.376] DEBUG[1595068] chan_iax2.c: schedule decrement of callno used for 149.154.11.243 in 60 seconds
[2025-01-10 08:37:03.376] DEBUG[1595068] channel.c: Channel 0x7f31ac001320 'IAX2/149.154.11.243:24229-8360' destroying
[2025-01-10 08:37:03.376] DEBUG[1595068] stasis.c: Topic 'channel:1736516217.38': 0x7f31ac005720 destroyed
[2025-01-10 08:37:03.376] DEBUG[1595004] cdr.c: Finalized CDR for IAX2/149.154.11.243:24229-8360 - start 1736516217.560611 answer 1736516217.652473 end 1736516223.376767 dur 5.816 bill 5.724 dispo ANSWERED
[2025-01-10 08:37:03.376] DEBUG[1595068] channel.c: Channel 0x7f316800ff40 'DAHDI/pseudo-1103084718' hanging up.  Refs: 2
[2025-01-10 08:37:03.376] DEBUG[1595068] chan_dahdi.c: dahdi_hangup(DAHDI/pseudo-1103084718)
[2025-01-10 08:37:03.377] DEBUG[1595068] chan_dahdi.c: Hangup: channel: -2 index = 0, normal = 109, callwait = -1, thirdcall = -1
[2025-01-10 08:37:03.377] DEBUG[1595068] chan_dahdi.c: Set option TDD MODE, value: OFF(0) on DAHDI/pseudo-1103084718
[2025-01-10 08:37:03.377] DEBUG[1595068] chan_dahdi.c: Updated conferencing on -2, with 0 conference users
[2025-01-10 08:37:03.377] DEBUG[1595068] channel.c: Channel 0x7f316800ff40 'DAHDI/pseudo-1103084718' destroying
[2025-01-10 08:37:03.377] DEBUG[1595004] cdr.c: Finalized CDR for DAHDI/pseudo-1103084718 - start 1736516217.649075 answer 0.000000 end 1736516223.377261 dur 5.728 bill 1736516223.377 dispo FAILED
[2025-01-10 08:37:03.377] DEBUG[1595068] stasis.c: Topic 'channel:1736516217.39': 0x7f3168011910 destroyed
[2025-01-10 08:37:03.377] DEBUG[1595004] cdr.c: Skipping CDR for DAHDI/pseudo-1103084718 since we weren't answered
[2025-01-10 08:37:03.765] DEBUG[1595043][C-00000025] chan_iax2.c: Immediately destroying 5246, having received hangup
[2025-01-10 08:37:03.866] DEBUG[1595708][C-00000025] app_rpt.c: newkey: 2
[2025-01-10 08:37:03.866] DEBUG[1595708][C-00000025] chan_dahdi.c: Using channel -2
[2025-01-10 08:37:03.866] DEBUG[1595708][C-00000025] stasis.c: Topic 'channel:1736516223.41': 0x7f317c008f60 created
[2025-01-10 08:37:03.866] DEBUG[1595708][C-00000025] channel.c: Channel 0x7f317c00db60 'DAHDI/pseudo-1690207560' allocated
[2025-01-10 08:37:03.867] DEBUG[1595708][C-00000025] app_rpt/rpt_bridging.c: Requested channel DAHDI/pseudo-1690207560
[2025-01-10 08:37:03.867] DEBUG[1595708][C-00000025] channel.c: Channel DAHDI/pseudo-1690207560 setting read format path: ulaw -> slin
[2025-01-10 08:37:03.867] DEBUG[1595708][C-00000025] channel.c: Channel DAHDI/pseudo-1690207560 setting write format path: slin -> ulaw
[2025-01-10 08:37:03.867] DEBUG[1595708][C-00000025] chan_dahdi.c: Requested indication -1 on channel DAHDI/pseudo-1690207560
[2025-01-10 08:37:03.877] DEBUG[1595708][C-00000025] app_rpt.c: Ansering the call #1
[2025-01-10 08:37:03.877] DEBUG[1595708][C-00000025] app_rpt.c: Sending key #1
[2025-01-10 08:37:03.877] DEBUG[1595068] channel.c: Channel 0x7f31a40043b0 'IAX2/149.154.11.243:24229-5246' hanging up.  Refs: 2
[2025-01-10 08:37:03.877] DEBUG[1595068] chan_iax2.c: We're hanging up IAX2/149.154.11.243:24229-5246 now...
[2025-01-10 08:37:03.877] DEBUG[1595068] chan_iax2.c: Really destroying IAX2/149.154.11.243:24229-5246 now...
[2025-01-10 08:37:03.877] DEBUG[1595068] chan_iax2.c: schedule decrement of callno used for 149.154.11.243 in 60 seconds
[2025-01-10 08:37:03.877] DEBUG[1595068] channel.c: Channel 0x7f31a40043b0 'IAX2/149.154.11.243:24229-5246' destroying
[2025-01-10 08:37:03.878] DEBUG[1595068] stasis.c: Topic 'channel:1736516223.40': 0x7f31a40065f0 destroyed
[2025-01-10 08:37:03.878] DEBUG[1595004] cdr.c: Finalized CDR for IAX2/149.154.11.243:24229-5246 - start 1736516223.248311 answer 0.000000 end 1736516223.877900 dur 0.629 bill 1736516223.877 dispo FAILED
[2025-01-10 08:37:03.878] DEBUG[1595004] cdr.c: Skipping CDR for IAX2/149.154.11.243:24229-5246 since we weren't answered
[2025-01-10 08:37:03.880] DEBUG[1595708][C-00000025] app_rpt.c: Sending key #2
[2025-01-10 08:37:09.602] Asterisk 20.7.0+asl3-1.2-1.deb12 built by builder @ allstarlink.org on a x86_64 running Linux on 2025-01-08 15:52:26 UTC
```

InterLinked1, 2025-01-10:
> Adding the `break;` from #448 appears to fix the "crash" exit, but now Asterisk is "cleanly exiting" in the rapid connect/disconnect scenario, yet there is a core:
>
> Some log from the console:
>
> ``` [...]

The crash is on line 6764 of app_rpt.c, what is that on your system currently? I think it's slightly different from what I'm looking at.

InterLinked1, 2025-01-10:
> The crash is on line 6764 of app_rpt.c, what is that on your system currently? I think it's slightly different from what I'm looking at.

That said, this crash is definitely because `l->chan` is NULL at that particular point (that can be seen from the backtrace).
It seems like it's expected to be non NULL, and another thread may have set it to NULL before it had a chance to finish.

In your other comment, how are calculating "refs" to the channel? As far as I know, we don't do that at all currently.

InterLinked1, 2025-01-10:
Some more things are falling into place here. You mentioned you only saw the error about PBX being terminated incorrectly in this scenario.

Now I see why that is. We have the call `ast_channel_pbx_set(l->chan, NULL);` right after `l->chan` is NULL in the backtrace you provided. Running that line will suppress the warning you saw. So in the earlier cases, that line of code never had the chance to run, even though we expect it always will.

There are possibly 2 separate but related issues here:
1. `l->chan` is being set NULL by another part of `app_rpt` before `rpt_exec` has a chance to finish executing. Maybe there should be additional locking, maybe not, I haven't looked further than this yet.
2. The channel itself is possibly being destroyed because its reference count has reached 0 while `app_rpt` is still using it. The solution to this is increasing its reference count.

I'm still not convinced that 2 is possible, though it could be. 1 should be addressed first before we even look at 2.

You mentioned you added debug statements at all the places l->chan is set to NULL, is it consistently in the same place for you? That would help narrow down what's going on here.

As far as the newkeytimer, I think that is just exposing the bug here. It sounds like it creates a timing condition where things end up exiting faster than they usually do, triggering the crash. For now, I'd like to leave the newkeytimer alone and make the code robust against that sort of thing happening.

mkmer, 2025-01-10:
I was working out the code and testing while you were replying. 
Looking through the debug trace I think we can see channel.c's thread takes over for app_rpt right after the first key is sent:
```
[2025-01-10 08:37:03.877] DEBUG[1595708][C-00000025] app_rpt.c: Sending key #1
[2025-01-10 08:37:03.877] DEBUG[1595068] channel.c: Channel 0x7f31a40043b0 'IAX2/149.154.11.243:24229-5246' hanging up.  Refs: 2
[2025-01-10 08:37:03.877] DEBUG[1595068] chan_iax2.c: We're hanging up IAX2/149.154.11.243:24229-5246 now...
```

Because app_rpt did NOT succeed at answering the call, the channel is not really locked in, so the other thread is destroying it.
The latest PR is based around some other apps in the asterisk base code that checks to see if answer succeeds.  The failure code is my interpretation of what I think we need to do.
It 100% eliminated my test case.  
Others should test too!

Mason10198, 2025-01-10:
I don't know if this is helpful at all, so I apologize for cluttering the thread if not. It seems to me that the people reporting this crash are seeing it on any/all of the ASL3 systems that they host - myself included. I host 3x different ASL3 servers that all experience the crash at least once per week. Usually it's a few / several times per week. Two of them see a few connect/disconnect events per day, but one of them does not get connected to/disconnected from very often. Again, all 3 systems seem to crash and restart with similar frequency.

A friend of mine hosts a system of 7x ASL3 servers, all of which are connected together 24/7. He has some servers with additional private nodes for echolink, DMR, etc... it is a very busy system. I noticed that his connection uptimes between all of his nodes are 250+ hours. His servers are either not crashing at all, or are crashing far less frequently than mine, even though he gets much more traffic and connect/disconnects than I do.

I have been unable so far to find any clues as to why I would be crashing while he is not. All of each of our servers are running the latest ASL3.

mkmer, 2025-01-10:
@Mason10198  -> I'm suspicious of a race between answering a call and disconnecting the call.  It's possible a faster or slower computer may not see this condition. Maybe better network connections mask it as well.   

If you could enable debug at level 1 (core set debug 1) and send debug to a file via the logger.conf we can compare crash events.  I have ```debug.log => debug``` in my logger.conf.

At your next crash, filter the debug.log file around the time of the crash and share it (along with the core dump stuff). 
This will at least help confirm you and I have the same series of events causing the crash.

InterLinked1, 2025-01-10:
I just commented on the other issue, but answering a channel is not required. It probably just happens to be getting hung up around that time in the code.

That doesn't mean I don't think this is helpful, it gives us a datapoint about the scenario, which is information.

I'll take a closer look at this issue next and see if I can come up with a solution.

mkmer, 2025-01-10:
Here are a few I checked out...

https://github.com/asterisk/asterisk/blob/48df8b63759995e2cddc2331ec344b79/apps/app_privacy.c#L108-L110

https://github.com/asterisk/asterisk/blob/4ef68fbce8df8b63759995e2cddc2331ec344b79/apps/app_read.c#L248-L253

https://github.com/asterisk/asterisk/blob/4ef68fbce8df8b63759995e2cddc2331ec344b79/apps/app_talkdetect.c#L125-L131

I really don't have a very good understanding of what is "right".  Just trying a few things that look like they might be right.

InterLinked1, 2025-01-10:
> Here are a few I checked out...
>
> https://github.com/asterisk/asterisk/blob/48df8b63759995e2cddc2331ec344b79/apps/app_privacy.c#L108-L110 [...]

Blame the poor design of Asterisk for that. A lot of this was written by people without a strong telco-background, that thought channels should answer for no good reason, "just cause". It's a constant pain in my side because I think it instills bad habits in users and it's also more annoying to work around.

Fundamentally, there is no requirement that channels be answered. In the telco world, the only thing an answer indicates is that billing should start. For example, test numbers should never return answer supervision. Sometimes, media won't flow bidirectionally without supervision on the PSTN, but this is usually intentional as an anti-fraud mechanism.

Because Asterisk is designed to be "easy" for people to work with, a lot of stuff answers automatically, such as the examples you linked there, but you'll notice that they all have options to disable the auto-answer behavior (typically `n`), for people who don't want this stupid behavior.

If I were to have written Asterisk myself, nothing would answer automatically, it would always require explicitly calling "Answer" (or relaying an Answer through dial), but that ship sailed a long time ago and making that behavior change now is untenable.

So unfortunately, no, this is not the solution.

InterLinked1, 2025-01-11:
@mkmer Based on a few ideas I had as I looked through some of the parts of the code, I put this patch together: https://github.com/AllStarLink/app_rpt/pull/451

I don't think it will fix all of the problems here on its own, but it'd be helpful if you could give it a try and report any difference and the logging output when this issue happens - thanks!

mkmer, 2025-01-12:
OK - I think we have it licked :)
Here is how I believe this all shook out:
1st problem was a "lock" leak - after a disconnect an extra lock was generated.
2nd problem was channel disconnect de-referencing a pointer that was still in use.

It looked like it was a time base problem (at one point), but it was actually a node disconnect problem.  After a certain number of disconnects, `rpt_exec` would have a problem or long delay getting a lock which forced the "fast connect/disconnect" to showing up in a much slower disconnect time.  The lock delay probably gets long enough that the caller would just hang up waiting for NEWKEY1 message...

It would be great if we could get some testers that are seeing this crash more often.

Mason10198, 2025-01-16:
> If you could enable debug at level 1 (core set debug 1) and send debug to a file via the logger.conf we can compare crash events. I have `debug.log => debug` in my logger.conf.
>
> At your next crash, [...]

I set debugging as requested, and it took a couple of days to crash out.
Here is the log:

```
[2025-01-15 07:42:12.868] DEBUG[82107] chan_iax2.c: ip callno count incremented to 1 for 173.185.108.179
[2025-01-15 07:42:12.911] DEBUG[82099][C-00000c52] stasis.c: Topic 'channel:1736948532.3181': 0x55afa431dc90 created
[2025-01-15 07:42:12.911] DEBUG[82099][C-00000c52] channel.c: Channel 0x55afa43175d0 'IAX2/173.185.108.179:4569-9796' allocated
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] pbx.c: Launching 'NoOp'
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] pbx.c: Launching 'NoOp'
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] pbx.c: Launching 'NoOp'
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] pbx_variables.c: Expression result is '0'
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] pbx.c: Launching 'GotoIf'
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] pbx_builtins.c: Not taking any branch
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] pbx_variables.c: Expression result is '1'
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] pbx.c: Launching 'GotoIf'
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] db.c: Unable to find key '40010/62777' in family 'denylist'
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] pbx.c: Launching 'GotoIf'
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] pbx.c: Launching 'Rpt'
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] channel.c: Channel IAX2/173.185.108.179:4569-9796 setting read format path: ulaw -> slin
[2025-01-15 07:42:12.911] DEBUG[230776][C-00000c52] channel.c: Channel IAX2/173.185.108.179:4569-9796 setting write format path: slin -> ulaw
[2025-01-15 07:42:17.989] DEBUG[82099][C-00000c52] chan_iax2.c: Immediately destroying 9796, having received hangup
[2025-01-15 07:42:18.095] DEBUG[230776][C-00000c52] chan_dahdi.c: Using channel -2
[2025-01-15 07:42:18.095] DEBUG[230776][C-00000c52] stasis.c: Topic 'channel:1736948538.3182': 0x7f14d40371c0 created
[2025-01-15 07:42:18.095] DEBUG[230776][C-00000c52] channel.c: Channel 0x7f14d40e0480 'DAHDI/pseudo-1968566556' allocated
[2025-01-15 07:42:18.095] DEBUG[230776][C-00000c52] app_rpt/rpt_bridging.c: Requested channel DAHDI/pseudo-1968566556
[2025-01-15 07:42:18.095] DEBUG[230776][C-00000c52] channel.c: Channel DAHDI/pseudo-1968566556 setting read format path: ulaw -> slin
[2025-01-15 07:42:18.095] DEBUG[230776][C-00000c52] channel.c: Channel DAHDI/pseudo-1968566556 setting write format path: slin -> ulaw
[2025-01-15 07:42:18.095] DEBUG[230776][C-00000c52] chan_dahdi.c: Requested indication -1 on channel DAHDI/pseudo-1968566556
[2025-01-15 07:42:18.109] DEBUG[82113] channel.c: Channel 0x55afa43175d0 'IAX2/173.185.108.179:4569-9796' hanging up.  Refs: 2
[2025-01-15 07:42:18.109] DEBUG[82113] chan_iax2.c: We're hanging up IAX2/173.185.108.179:4569-9796 now...
[2025-01-15 07:42:18.109] DEBUG[82113] chan_iax2.c: Really destroying IAX2/173.185.108.179:4569-9796 now...
[2025-01-15 07:42:18.110] DEBUG[82113] chan_iax2.c: schedule decrement of callno used for 173.185.108.179 in 60 seconds
[2025-01-15 07:42:18.110] DEBUG[82113] channel.c: Channel 0x55afa43175d0 'IAX2/173.185.108.179:4569-9796' destroying
[2025-01-15 07:42:18.110] DEBUG[82090] cdr.c: Finalized CDR for IAX2/173.185.108.179:4569-9796 - start 1736948532.911391 answer 0.000000 end 1736948538.110157 dur 5.198 bill 1736948538.110 dispo FAILED
[2025-01-15 07:42:18.110] DEBUG[82113] stasis.c: Topic 'channel:1736948532.3181': 0x55afa431dc90 destroyed
[2025-01-15 07:42:18.110] DEBUG[82090] cdr.c: Skipping CDR for IAX2/173.185.108.179:4569-9796 since we weren't answered
[2025-01-15 07:43:55.837] Asterisk 20.11.0+asl3-3.2.0-2.deb12 built by builder @ allstarlink.org on a x86_64 running Linux on 2024-12-20 20:18:06 UTC
```

Here is the core dump stuff. It looks like it crashed 2 more times since the one shown in the above log before I was able to get in and get the dump. I'm not sure if the dump only has info from the last crash? If so, then the log and the dump will be different crashes. I only have a debug log from one of the crashes since the logging gets reset when Asterisk restarts.

[core-asterisk-2025-01-15T22-09-57Z-brief.txt](https://github.com/user-attachments/files/18440367/core-asterisk-2025-01-15T22-09-57Z-brief.txt)
[core-asterisk-2025-01-15T22-09-57Z-full.txt](https://github.com/user-attachments/files/18440365/core-asterisk-2025-01-15T22-09-57Z-full.txt)
[core-asterisk-2025-01-15T22-09-57Z-info.txt](https://github.com/user-attachments/files/18440366/core-asterisk-2025-01-15T22-09-57Z-info.txt)
[core-asterisk-2025-01-15T22-09-57Z-locks.txt](https://github.com/user-attachments/files/18440364/core-asterisk-2025-01-15T22-09-57Z-locks.txt)
[core-asterisk-2025-01-15T22-09-57Z-thread1.txt](https://github.com/user-attachments/files/18440363/core-asterisk-2025-01-15T22-09-57Z-thread1.txt)

mkmer, 2025-01-16:
Awsome!  This appears to be the same situation I was seeing on our system and also triggering in my test case.  If you are capable of compiling, this should fix your problem
PR https://github.com/AllStarLink/app_rpt/pull/451

closed 2025-01-17 by Allan-N

Mason10198, 2025-02-17:
Is this resolved? I see the issue is closed and there is a referenced PR that is merged. Did this make it into the beta or prod release of the ASL3 packages yet? It does not look like it's fixed in 3.2.0.

Allan-N, 2025-02-17:
>Is this resolved? I see the issue is closed and there is a referenced PR that is merged. Did this make it into the beta or prod release of the ASL3 packages yet? It does not look like it's fixed in 3.2.0.

A fix has been identified and the changes have been merged.  But, the change has not yet been built into a package.  We're looking to fix one more crasher and then we'll start to turn the crank.

commit 75e909be (2025-01-11): app_rpt: Don't hard hangup channels currently executing PBX.

Resolves: #422
commit 68cf227a (2025-01-11): app_rpt: Don't hard hangup channels currently executing PBX.

Resolves: #422
commit 90c1223e (2025-01-11): app_rpt: Don't hard hangup channels currently executing PBX.

Resolves: #422
commit e82718a3 (2025-01-11): app_rpt: Don't hard hangup channels currently executing PBX.

Resolves: #422