AllStarLink/app_rpt #424: SimpleUSB When configured PTT active low CM108 FOBs PTT line asserted low 'permanently' on asterisk startup - restart

MichiganBroadband, 2024-11-11:
**Describe the bug**
A clear and concise description of what the bug is.

SimpleUSB: When configured for PTT active low - FOBs are asserting PTT low *forever* when asterisk is restarted.
This (when hooked to a radio that is keyed active low) is causing radio to be stuck transmitting.

**To Reproduce**
Steps to reproduce the behavior:
1. Install + Configure ASL3 on Debian12 (up to date) 
2. Configure node to be PTT active low using asl-menu
Configure: H) Toggle PTT mode (currently 'open')
Simplex node config type "1".

**Expected behavior**
A clear and concise description of what you expected to happen.

Expect that PTT should be high (idle state) upon asterisk restart unless the radio being keyed is called for by software to pulls this line low.

**Screenshots**
If applicable, add screenshots to help explain your problem.
While in this stuck state (PTT is held low) but software shows that it is not.  
Using ALS3 Menu Status display shows proper idle status and transmitter not keyed:
Receiver/Transmitter Status Display:
  COS   | CTCSS  | COS   | PTT
 Input  | Input  | Out   | Out
 Clear  | Off    | Clear | Clear

But PTT is stuck keyed per voltmeter connected to FOB or radio is keyed as a result if radio is hooked up.

This condition cannot be cleared by executing various commands to key the radio
From the asterisk CLI or from the menu such as "T" 
T) Toggle Transmit Test Tone/Keying (currently 'disabled')
Or by COS detect that would also normally key the radio.
PTT remains stuck asserted low on the FOB.
However executing the "F) Flash (Toggle PTT and Tone output several times)
Clears this stuck condition and then everything works properly as it should,
until asterisk is restarted.
Any time asterisk is restarted or started this stuck condition exists.  

**Software versions (listed in asl-menu, option 4)**
 
OS            : Debian GNU/Linux 12 (bookworm)
                      │ OS Kernel     : 6.1.0-27-amd64
                      │
                      │ Asterisk      : 20.9.3+asl3-3.0.6-1.deb12
                      │ ASL [app_rpt] : 3.0.6
                      │
                      │ Installed ASL packages :
                      │
                      │   Package                         Version
                      │   ==============================  ==============================
                      │   asl3                            3.5.0-1.deb
                      │   asl3-asterisk                   2:20.9.3+asl3-3.0.6-1.deb12
                      │   asl3-asterisk-config            2:20.9.3+asl3-3.0.6-1.deb12
                      │   asl3-asterisk-modules           2:20.9.3+asl3-3.0.6-1.deb12
                      │   asl3-menu                       1.9-1.deb12
                      │   dahdi                           1:3.1.0-2

**Have you run a software update and rebooted?**
  YES system is current and up to date

**What is the platform
  :ASL3 Desktop PC Debian 12 Intel 64bit MBR/BIOS boot

**Additional context**
Additional info: During this stuck condition (if a radio is hooked up) 
Transmitter audio is not passed from the FOB.. 
Do not hear ID telemetry etc. or RX audio.  

I also have a detailed post on the public forum as I work on this at: 
Please have a look at that and comment if appropriate.
https://community.allstarlink.org/t/asl3-simplex-node-transmit-stuck-in-transmit-when-asterisk-is-restarted-or-rebooted/22221

Thanks!

Allan-N, 2024-11-11:
Moved to app_rpt

MichiganBroadband, 2024-11-11:
Not sure if this is needed or helpful.  
Regarding the USB device in case it is helpful, 
I do not have any other variant to try handy. 
I might be able to dig up a much older one soon and try on that.

Bus 001 Device 006: ID 0d8c:000c C-Media Electronics, Inc. Audio Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0d8c C-Media Electronics, Inc.
  idProduct          0x000c Audio Adapter
  bcdDevice            1.00
  iManufacturer           0
  iProduct                1 C-Media USB Headphone Set
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00fd
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0
      iInterface              0
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0064
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0
        iTerminal               0
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             2
        wTerminalType      0x0201 Microphone
[... 209 lines cut ...]
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      50
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              32
Device Status:     0x0000
  (Bus Powered)

Allan-N, 2024-11-11:
Can you share any more details of the USB device?
Was this purchased online?  If so, can you share a link?  or any other product information?
Was the device modified?  If so, can you share a link to the instructions you followed?  and any additional changes that you made?

MichiganBroadband, 2024-11-12:
I can share anything you like and do any testing you like.  
Just let me know what more info you are looking for.  
I'll get a link, as you know these can change, but this was about 1-month ago when they were purchased.  
In general they seem to work great and build quality is excellent.  
We have also tried some different FOBS, I have an old CM108 I purchased and modified in 2009 and was in service for years.  
It exhibits the same exact problem.
https://www.amazon.com/gp/product/B0BKPSDNZL/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
I modified these myself.  
General modification instructions are at: https://allstarsetup.com/how-to-modify-a-cm108-for-allstar/  
For testing and simplification purposes concerning the issue, any testing or reference done at this point has the FOB with only the PTT and COS lines exposed and hooked up for testing.  
This allows the problem/testing to be done without the complications of involving a radio or near RF fields interference floating signals and anything like that. 
At this point until this can get resolved I am only testing the logic at the FOB, to keep things simple and clear.  
The problem is very much here with this test and it's very easy to duplicate with minimal variables involved.

MichiganBroadband, 2024-11-12:
Was that helpful at all?  
Is there anything else I can be doing on my end to help?  
The problem is very easy to replicate/reproduce and at this point I do not think the FOB hardware version matters.
I have also tried different PC hardware. This is looking like a global issue.  
It does this on two completely different FOBs I am now testing.  
One is very old has has the actual CM108 on it.  
I have also narrowed this testing down to "FOB only" with nothing else connected for simplicity.
It's easy to test right at the FOB itself.   
0V at the PTT pin (GPIO3) and 3.3V when idle. (normal operation). 
The chip is rated to source/sink 8mA of current.  
The chip also provides pullup on it's own when in the 3.3V state.
And can drive up to 8mA per spec.  
This is perfect for testing and is all that is needed to show the bug is there or not there.  
And the bug is always there when asterisk is restarted. in every test and hardware combination I have tried so far.  
I have NOT yet tested on a Raspberry Pi.
PC hardware only so far AMD/Intel.

Allan-N, 2024-11-12:
>Was that helpful at all?

The information you provided helps in that we know what device(s) we're working with.

>Is there anything else I can be doing on my end to help?

At this point I think we just need to do some digging.  If we have questions or need help we'll reach out.

MichiganBroadband, 2024-11-12:
Thanks,  I do not want to confuse matters by adding more "things" to this mix.  
This is a pretty basic easy to replicate problem and is very repeatable in my tests with different hardware.  
I was considering trying a Raspberry Pi build just to see if the issue is also present there or not.
Do you think I should try this? or just hold off and stick to PC hardware and the very simple test configuration where I am seeing this problem?

MichiganBroadband, 2024-11-12:
Datapoint: Problem is very much there plain as day on up to date RPi build. :)

Allan-N, 2024-11-12:
This should not make any difference but does your `simpleusb.conf` file include :
```
clipledgpio = 1
```
If so, can you change the value to zero (0) and recheck the behavior?

MichiganBroadband, 2024-11-13:
Yes, clipledgpio = 1 is included in simpleusb.conf
Same behavior after changing clipledgpio = 1 to clipledgpio = 0
tried both restarting asterisk (problem comes back)  
And full poweroff + reboot.  
Problem comes back as soon as asterisk starts.

KB4MDD, 2024-11-13:
@MichiganBroadband We are having discussions on this problem now.

MichiganBroadband, 2024-11-13:
Excellent!  
I'm happy to help/get involved if wanted.  
And learn new things..  
If I have to build/compile stuff as well.   
Might need a little help on that.

jxmx, 2024-11-14:
I'm confused by some of the troubleshooting being done here. @MichiganBroadband  states above "0V at the PTT pin (GPIO3) and 3.3V when idle. (normal operation)." It should be the case that `invertptt=no` is ground to transmit and `invertptt=yes` is float/pull-high to transmit. Assuming the attached device is also float/pull-high to transmit, why would a measured voltage of 0v with PTT be triggering transmit? If `invertptt=yes` then I would expect the measured voltage on the PTT line to be always 0v until something needs to transmit.

jxmx, 2024-11-14:
It should also be noted that the instructions at https://allstarsetup.com/how-to-modify-a-cm108-for-allstar/ are specifically contemplating a PTT-high/float situation. So if it's modified as exactly in that way, then PTT is expected to be high/float.

jxmx, 2024-11-14:
Looking again back through all of this, this also stuck out to me:
[image]

To be clear, setting PTT to "open" in the menu is `invertptt=yes` which is transmit on high/float.

MichiganBroadband, 2024-11-15:
Hi, I'm not sure where this conversation going or why.  
The bug remains and it's a very serious bug that hold the transmitter on whenever asterisk is restarted.  
Can you please confirm this has been identified and confirmed as I have stated.  
Also if there is anything I can do to help.  
I'd hate to think that progress on this is stalled for any reason on my end and this new what appears to be a side-conversation wrestling with software terms is confusing the matter or potentially confusing it.
Please let me know if any information is missing or unclear for any reason.  
  
Thanks.  
-SteveG

jxmx, 2024-11-15:
It's not confusing nor a side-conversation about software terms. The conversation is trying to get you to answer some questions. Your description in the bug information is conflicting. You are describing your setup/configuration for an active=high PTT as well as sharing a link to a fob modification that's for an active=high PTT configuration. However your testing narrative is explaining a problem in that you believe the voltage should be floating when the system is idle. 

I have three nodes here at the house, all with active-low PTT configurations using different sound chips for testing, and all of them work fine and I cannot reproduce your problem.

The "software terms" in the configuration matters here - is the PTT pin configured to be pulled low (default, invertptt=no) or pushed high (invertptt=yes).

As I've asked before here and in the forum: what is `invertptt` set to in `/etc/asterisk/simpleusb.conf`?

MichiganBroadband, 2024-11-15:
invertptt = no  
; no  - ground to transmit
; yes - open to transmit
This is my setting and it is sticking at ground (Transmit) when asterisk is restarted.

jxmx, 2024-11-15:
Okay, thanks. Also, do you happen to know which fob or, even better, which type of chip is on the fob if you can see the print on the sound chip? There was someone else a few weeks ago who had a fob off Amazon that had some odd behaviors that seemed to be some sort of voltage over-saturation problem.

MichiganBroadband, 2024-11-15:
Actually sorry,   I was wrong.   (I was looking at global/general settings and not my setting under the node stanza).
invertptt=yes   
And this results in having it work the way I want with exception to the stuck low on asterisk start.
Again just to clarify low (0V) for PTT to be asserted at the fob chip itself.  
Please having a transistor present as most do would actually invert this.
And I do not know if the included comments are referring to at the chip or at the transistor inverter/buffer (if it were installed).
This may be a source of additional confusion!  
Any and all testing that I am doing is at the FOB chip itself.  
I am seeing the same problem/result with the two chips that I have.  
I will get you that information as well.  
One is newer and one is really old from 2007ish (CM108).

MichiganBroadband, 2024-11-15:
I also changed the setting to invertptt=yes in what I refer to as the global settings [general] section  
to be sure.  
And get same result.

MichiganBroadband, 2024-11-15:
Older Chip:
[image]
 
CM108
MM4JP
UYD1PC-GS
0618

MichiganBroadband, 2024-11-15:
Newer Chip: 

CM108
MQ2SY
UYD1PC-GS
0818
[image]

jxmx, 2024-11-15:
So this goes back to where I'm confused. With invertptt=yes which means it's expecting a radio to float/high to transmit. It should be the case, then, that when app_rpt is idle that the pin is pulled to ground (i.e. not transmitting). To me, that means that the configuration is causing the expected electrical behavior. If app_rpt is idle and the channel is "Clear", then what else would you be expecting on the PTT line?

So I'm still not following the problem you're having or what a transistor you keep mentioning has anything to do with this? The intention of the audio fob should be connected directly to a PTT control pin. If the PTT of the radio was active high, then you just set invertptt=yes. Some people may use a comparator or a schmitt trigger is the transition range is broad for the radio.

MichiganBroadband, 2024-11-15:
Also not that it even really matters at this point but in my forum post I answered you 
with my correct settings when you asked: 
So I am confused by your statement while I am doing my best to provide any information that you have asked for.
Unedited and one post up from the bottom on 11-10-2024: 
-SteveG
[image]

MichiganBroadband, 2024-11-15:
Let's boil this down to be a simple as possible.  
I am testing with ONLY a FOB.   
Please drop the thought of any transistor. 
I have only mentioned it because the terminology in simpleusb.conf specifies:
 
invertptt = yes/no
; no - ground to transmit
; yes - open to transmit
Outside of this bug and during normal operation (which happens as soon and you run the Menu flash command).
It indeed does the exact opposite when testing as to what the included comments state.  
If set to no, the pin goes to high (3.3V) when transmitting
If set to yes, the pin pulls low to 0V when transmitting.  

These comments might assume a transistor is installed and what is "seen" at the radio in that case.

MichiganBroadband, 2024-11-15:
Yes... THE RADIO.  
We are testing this bug without a RADIO.   
I am only speaking in terms of what the pin does at the FOB with nothing connected other than a meter/logic probe.  
Because that is all that matters at this point and for fixing/troubleshooting the bug.
Trying to make this as simple and direct as possible.
There is no radio, there is no transistor.   
The FOB (when configured for PTT to be low when PTT is supposed to be asserted) is asserting low (0V) when asterisk is restarted. 
In my configuration this pin should be high when the transmitter is not supposed to be keyed.
And it is being pulled low when Asterisk is restarted and stays that way forever until the Flash command is ran from the menu.  
Then everything works as it should.  
If you restart asterisk it is stuck again with the pin pulled low and it should not be doing this.  
Are you still confused or do we have solid common ground to work from?  
I'll do anything I can to help!

jxmx, 2024-11-15:
So, right now I have the meter on my CM108B-based device with the leads soldered into place. The radio is the SA818 module which is an active-low PTT but I have the probe tacked onto the CM108B output.

In `/etc/asterisk/simpleusb.conf` I have `invertptt=no` (tune menu H is "ground") which is the **default and is for active-low PTT**. The multimeter is reading just shy of 3.3v. When I key the node to transmit, the measured voltage drops 0v for the duration of the transmission. This is what I would expect to be occurring. 

Now, when I change `invertptt=yes` (or in the tune menu change H to "open"), **which would be for an active high/float situation**, I also experience the expected behavior. When I key the node to transmit, the measured voltage on PTT rises to 3.3v. When the transmission is complete, the GND is re-asserted on the PTT pin and it goes back to 0v.

In both modes of `invertptt` I am seeing the behavior I would expect.

MichiganBroadband, 2024-11-16:
Hi Jason,  
Work from your second configuration (this is what I am using) invertptt=yes in the config file  
You are describing the exact opposite logic results of what I am seeing on mine.  
Are you 100% sure you are measuring at/from the CM108 pin 13 and not possibly on some other point (which could possibly be inverted)? 
While not in the "bug"/stuck condition (after running the flash TX menu command) Mine is for certain is high when not transmitting (~3.3V) and low (0V) when it is transmitting.
H menu setting is Toggle PTT mode (currently 'open')
And invertptt=yes in simpleusb.conf
-Steve

jxmx, 2024-11-16:
I am 100% certain there is no other inversion in the middle on my test. Either you have a sneak config in there somewhere or the fobs you have behave oddly. Unfortunately not all CM1xx devices have the same quality.

jxmx, 2024-11-16:
This sounds like an initialization error for that hardware maybe. Fixing the title again. @KB4MDD has some cleanups to maybe address this.

MichiganBroadband, 2024-11-16:
What do you think our next steps should be-
I think this might include a lot of chips that are out there.
I also tested a much older Fob/chip that was purchased in 2008 and may be older than that date and am seeing exact same behavior.
What is a "sneak config in there somewhere" that was referred to?
Had same exact behavior on a fresh install Rpi two days ago no sneak configs unless that is included in the build.
Can you refer me to an exact FOB/chip to purchase and try out that would be known to not have this problem?  
 I might be getting ahead of us on some of my questions here and am just thinking/typing out loud.
 The complete reverse logic exhibited from your bench test (aside from the stuck logic) is also puzzling.
 All that said I'm having a lot of fun here on this.  
 Hope to be helpful!  
 Also just as a datapoint and sanity check, reversing the settings to make transmit on high (3.3V) Instead of transmit on low (0V) works as expected and does not hang pin stuck state on startup.
 -sg

closed 2024-12-04 by Allan-N