AllStarLink/app_rpt #427: Latest version 11/23 wont allow private nodes to connect to public

wspivak, 2024-11-24:
I get this:
[2024-11-23 12:43:05.090] NOTICE[165081]: chan_iax2.c:4790 __auto_congest: Auto-congesting call due to slow response
– Hungup ‘IAX2/127.0.0.1:4568-13805’
– Hungup ‘DAHDI/pseudo-132052208’

Public to Private connections work.

jxmx, 2024-11-24:
Nothing should have changed with this. Based on the above, the public node and private node are on the same server?

To confirm, the public node's section of the template still has the private node's section in the right location? Can you attach your `rpt.conf`?

jxmx, 2024-11-24:
If you've changed from the default port of 4569 I believe the issue is you're missing the non-standard IAX port for the other private nodes here:

[image]

wspivak, 2024-11-24:
Ahhhh. Thank you. When I get back I’ll fix/test and report back.

wspivak, 2024-11-24:
That was the issue.  

THANK YOU