AllStarLink/app_rpt #435: DTMF via IAX VOIP connection fails

wspivak, 2024-12-19:
Using HOIP on an Asterisk 20.10.0+asl3-3.1.0-1.deb12, DTMF fails after connection to the private node.

The HOIP softphone connects in monitor mode, but I am unable to get node to decode *99.  Gescio Alpuro, WH6AV of HOIP tried to get it to work on my node without success.  Before latest update, it WAS operational.

Currently: DTMF works via CLI and/or RF connected to node.

Tried setting jitterbuffer to "NO"  (via Jason N8EI) and it refuses to maintain a connection to HOIP.

wspivak, 2024-12-21:
HOIP says this is a recurring, occasional error on my node:

`[2024-12-18 18:26:43.763] DEBUG[1383]: chan_dahdi.c:9262 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2`

MuevCFO, 2025-01-29:
Update...   I just added HamShack to my node, and that too doesn't activate *99/transmit.

If up migrated that node to Beta, would it fix the issue?

KB4MDD, 2025-01-29:
It would be helpful to see what asterisk is reporting on the cli.  Can you log into asterisk, connect and attempt to transmit and capture the output?

MuevCFO, 2025-01-29:
-- Hungup 'DAHDI/pseudo-1531960377'

    -- Hungup 'DAHDI/pseudo-1000347902'

    -- Hungup 'DAHDI/pseudo-614103087'

    -- Accepting AUTHENTICATED call from 144.202.54.216:4569:

    --        > requested auth methods = (MD5),

    --        > actual auth method = MD5,

    --        > encrypted = no,

    --        > requested format = ulaw,

    --        > requested prefs = (ulaw|alaw|gsm),

    --        > actual format = ulaw,

    --        > host prefs = (ulaw),

    --        > priority = mine

    -- Executing ***@***.***:1] Answer("IAX2/hh-KC2NJV-9324", "") in new stack

    -- Executing ***@***.***:2] Playback("IAX2/hh-KC2NJV-9324", "rpt/node") in new stack

    -- <IAX2/hh-KC2NJV-9324> Playing 'rpt/node.gsm' (language 'en')

    -- Hungup 'DAHDI/pseudo-1925437669'

    -- Executing ***@***.***:3] SayAlpha("IAX2/hh-KC2NJV-9324", "62499") in new stack

    -- <IAX2/hh-KC2NJV-9324> Playing 'digits/6.ulaw' (language 'en')

    -- <IAX2/hh-KC2NJV-9324> Playing 'digits/2.ulaw' (language 'en')

    -- <IAX2/hh-KC2NJV-9324> Playing 'digits/4.ulaw' (language 'en')

    -- <IAX2/hh-KC2NJV-9324> Playing 'digits/9.ulaw' (language 'en')

    -- <IAX2/hh-KC2NJV-9324> Playing 'digits/9.ulaw' (language 'en')

    -- Executing ***@***.***:4] Playback("IAX2/hh-KC2NJV-9324", "rpt/connected") in new stack

    -- <IAX2/hh-KC2NJV-9324> Playing 'rpt/connected.gsm' (language 'en')

    -- Executing ***@***.***:5] Rpt("IAX2/hh-KC2NJV-9324", "62499|P") in new stack

 

Then I try *99 and see NOTHING.

wspivak, 2025-01-29:
BTW, same issue with HOIP.  I monitor w/o issue, no DTMF tones recorded.

If I try DMTF tones from my handheld, they are recorded, per keystroke.

wspivak, 2025-01-29:
I corrected per the instructions above, but no joy.

mkmer, 2025-01-29:
I'm not an expert on dial plans.  I'm not sure this is OK...
`exten => 1500,n,rpt(1500|P|${CALLSIGN})`
Either way I don't think you want two of them (one with and one without the callsign).
The fact that DTMF is not passing through the channel make me lean toward some sort of configuration problem (like it's not in phone mode?).

My dialplan (which works for us) looks like this:
```
exten => ${NODE1},1,Ringing()
 same => n,Wait(5)
 same => n,Answer()
 same => n,Playback(silence/1)
 same => n,Playback(hello)
 same => n,Playback(/etc/asterisk/local/speech/sipanswer)
 same => n,Playback(silence/2)
 same => n,rpt(${NODE1}|P)
 same => n,Hangup()
```

KB4MDD, 2025-01-29:
Let's turn up the debugging.  Make sure logger.conf has debug enabled for the console.
`console => notice,warning,error,dtmf,debug`
If this was not set, you will have to restart asterisk.

In the cli enter these commands.

core set verbose 7
core set debug 7 app_rpt

connect and try *99 again.  Send over the captures.

wspivak, 2025-01-29:
The first attempt was using HamShack, didn't work.
The next was using HOIP, again didn't work[

[debug.log](https://github.com/user-attachments/files/18593791/debug.log)

](url)

KB4MDD, 2025-01-29:
The debug.log did not contain any of the verbose entries.

wspivak, 2025-01-29:
Logger.conf:

debug.log => debug
;console => notice,warning,error,dtmf
console => notice,warning,error,debug,verbose
;messages.log => notice,warning,error,dtmf,verbose
messages.log => notice,warning,error,dtmf

KB4MDD, 2025-01-29:
You can change debug.log => notice,warning,error,debug,verbose

restart and reproduce.

wspivak, 2025-01-29:
Here you go...  HamShack w/2 tries at *99, followed by HOIP...

[debug.log](https://github.com/user-attachments/files/18594081/debug.log)

KB4MDD, 2025-01-29:
It seems that we are not hearing the "*".

```
[2025-01-29 16:05:27.569] DEBUG[110389] app_rpt.c: digits=9  source=3
[2025-01-29 16:05:27.767] DEBUG[110389] app_rpt.c: digits=99  source=3

[2025-01-29 16:05:33.666] DEBUG[110389] app_rpt.c: digits=9  source=3
[2025-01-29 16:05:33.840] DEBUG[110389] app_rpt.c: digits=99  source=3

[2025-01-29 16:06:25.846] DEBUG[110393] app_rpt.c: digits=9  source=3
[2025-01-29 16:06:26.097] DEBUG[110393] app_rpt.c: digits=99  source=3

```
Do we know that the "*" key is working?

wspivak, 2025-01-29:
Yes...  on HamShack I sent *97 and *43 successfully...

wspivak, 2025-01-29:
I checked and funcchar is not specifically set in any conf file in asterisk.

What if we set it to "*"?

KB4MDD, 2025-01-29:
From what I can tell, we get the '99'.  However, we cannot find a 99 command.  If the command was found, it would be in the log file.  I believe this is due to the fact that you have:

In extensions.conf:
same => n,rpt(1500|P)

This should direct to a node 1500 in rpt.conf.  I find that node 1500 is disabled.

I would like for you to update rpt.conf with the following minimum.  You have a ` ;functions = functions-1500` under 1500 now.  Don't enable that line as there is no functions-1500 defined.

```
[1500](node-main)
rxchannel = dahdi/pseudo
idrecording = |iKC2NJV
idtalkover = |iKC2NJV
duplex = 0
holdofftelem=1
telemdefault=0 ; was 2
telemdynamic = 0
linktolink = no
startup_macro = *6962499
```
I don't know if that startup macro is correct.

Please update and restart.  Delete the old debug.log and run another test.  Remember to enable verbose and debugging as before.

wspivak, 2025-01-29:
Added exactly what you quoted.  Startup macro is correct, but isn't working for 1500, but does for 1806 & 1807.

[debug.log](https://github.com/user-attachments/files/18595266/debug.log)

KB4MDD, 2025-01-30:
Can you attach the current extensions and rpt conf

wspivak, 2025-01-30:
Named ...log so I could upload...

[iax.conf.log](https://github.com/user-attachments/files/18595842/iax.conf.log)
[extensions.conf.log](https://github.com/user-attachments/files/18595841/extensions.conf.log)

mkmer, 2025-01-30:
I don't think you want this - make them more like hamshackhotline.  Does Hamshack hotline work?
```
exten => 1500,n,rpt(1500|P|${CALLSIGN})
exten => 1500,n,rpt(1500|P)
```
Just this:
```
exten => 1500,n,rpt(1500|P)
```

wspivak, 2025-01-30:
None transmit.  Regards, WayneWayne SpivakSBA.NET.WEBA div of SBA * Consulting LTD Tel LI: +1 (516) 221-3306   NY Tel: +1 (212) 487-5085  Tel CT: +1-860-760-0250Fax: +1 (516) ***@***.***://www.sbaconsulting.comLinkedIn: http://LinkedIn.com/in/WayneSpivakTwitter: @SBAConsult  Skype: SBAConsult  Sent from my iPhone; typos expected and endorsed by Apple

KB4MDD, 2025-01-30:
Can you attach `rpt.conf`

Also run cli command `rpt localnodes` and post the results.

KB4MDD, 2025-01-30:
It would be helpful to see cli command `rpt nodes 62499`

Also do you have the file n9oig-rpt.conf in /etc/asterisk?

wspivak, 2025-01-30:
Good morning...

No, I don't have a file n9oig-rpt.conf

```
 rpt nodes 62499

************************* CONNECTED NODES *************************

T1500, T1806, T1807, T3355800, T420677, R426095, R43570, T509520
R509522, T62679, T63668
```

```
Allstar*CLI> rpt localnodes

Node
----
62499
1806
1807
1500

```

Thanks for your help!

[rpt.conf.log](https://github.com/user-attachments/files/18601600/rpt.conf.log)

mkmer, 2025-01-30:
I didn't notice this before, but it looks like your missing the context for 1500;
```
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
;;;;;;;;; 1500 Ham Over IP ;;;;;;;;;
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
rxchannel = dahdi/pseudo
idrecording = |iKC2NJV
idtalkover = |iKC2NJV
duplex = 0
holdofftelem=1
telemdefault=0 ; was 2
telemdynamic = 0
linktolink = no
startup_macro = *6962499
```

At the xxxxxx I think you need `[1500]`

wspivak, 2025-01-30:
"At the xxxxxx I think you need [1500]"

Your right, I missed that when I updated the file earlier.  Let fix and try again...

wspivak, 2025-01-30:
Retested, still not working... i.e., monitors, doesn't transmit.

KB4MDD, 2025-02-01:
@wspivak allowed me to access the node.   Any valid DTMF sent over the phone was not being recognized as a valid DTMF sequence.  It was not just *99.  The problem has something to do with [functions] stanza.  It is not being loaded properly. 

To test this, I added a new [functions_phone] to rpt.conf.  It has 99 = cop,6.  I updated phone_functions = functions_phone and restart.  I ran tests and we appear to be transmitting.  I will wait for an update from @wspivak as he tests.

I suspect the [functions] section needs to be deleted and reentered.  There is something tripping up the asterisk conf reader.

wspivak, 2025-02-01:
I tested with both HOIP and HH, both systems permitted me to transmit.

THANK YOU DANNY, KB4MDD !!!!!!!!!

closed 2025-02-01 by wspivak

KB4MDD, 2025-02-01:
No problem.  They you for allowing me to access the system.  You can close the port now.Danny Lloyd

Allan-N, 2025-02-01:
I just took another peek at the `rpt.conf.log` file @wspivak attached above ... with a focus on the `[functions]` section.  Found this line :
```
80 = cmd,asterisk -rx “rpt cmd 62499 status 1 62499”
```
I don't know if this line was part of the problem but if you look very closely you will see that you have two different quotation characters.  Don't ya just love text editors that have both an open quote and a close quote?

wspivak, 2025-02-01:
Eagle eyes Allan.

 

Thanks

 

Subject: Re: [AllStarLink/app_rpt] DTMF via IAX VOIP connection fails (Issue #435)

 

I just took another peek at the rpt.conf.log file @wspivak <https://github.com/wspivak>  attached above ... with a focus on the [functions] section. Found this line :

80 = cmd,asterisk -rx “rpt cmd 62499 status 1 62499”

I don't know if this line was part of the problem but if you look very closely you will see that you have two different quotation characters. Don't