AllStarLink/app_rpt #437: ASL3 Warning From chan_usrp.c

wspivak, 2024-12-21:
James wrote in a post on Dec 16, the following.  I too any seeing the same error (different ports) on a friends node.

```
Using a RPi5 I installed DVSwitch on the same instance as ASL3 and linked it to ASL3 using chan_usrp. The following message is repeated for each transmission received from DVSwitch in the asterisk log:

WARNING[265959] chan_usrp.c: Channel usrp/127.0.0.1:34001:32001: Receive queue exceeds the threshold of 320

I tried to determine the origin of this message in the source code but have been unsuccessful.

I also notice that received audio from DVSwitch is mangled at the very beginning of a received transmission but soon becomes normal. Possibly a result of the same issue.

Is there a known resolution to this issue? Can it be resolved in configuration, or a change to the source code?
```

KB4MDD, 2024-12-21:
The error message is displayed when we have queued up (received) more than 25 voice packets.  We should never get that far behind.  

I am using analog_bridge with ASL3.  I have not seen this problem; can you share your configuration?

wspivak, 2024-12-21:
[extensions.conf.62679.txt](https://github.com/user-attachments/files/18218738/extensions.conf.62679.txt)
[iax.conf.62679.txt](https://github.com/user-attachments/files/18218739/iax.conf.62679.txt)
[rpt.conf.62679.txt](https://github.com/user-attachments/files/18218740/rpt.conf.62679.txt)

wspivak, 2024-12-21:
BTW, 

This is a representative sample of MMDVM log

```
M: 2024-12-21 13:59:47.928 YSF, GPS Position from K0EHT-RON  on radio=0x34 is lat=39.208000 long=-94.679665
M: 2024-12-21 14:00:11.734 YSF, received network end of transmission, 24.7 seconds, 0% packet loss, BER: 0.0%
M: 2024-12-21 14:00:17.009 YSF, received network data from KE8EFQ     to ALL        at KU0S/GW
I: 2024-12-21 14:00:17.009 YSF, Lookup call KE8EFQ returned id 0 -> 1234567
M: 2024-12-21 14:00:18.499 YSF, GPS Position from KE8EFQ     on radio=FTM-100D is lat=41.188000 long=-81.976997
M: 2024-12-21 14:00:25.305 YSF, received network end of transmission, 8.4 seconds, 0% packet loss, BER: 0.0%
M: 2024-12-21 14:00:31.876 YSF, received network data from K0EHT-RON  to ALL        at KU0S/GW
I: 2024-12-21 14:00:31.876 YSF, Lookup call K0EHT returned id 3129461 -> 3129461
M: 2024-12-21 14:00:32.665 YSF, GPS Position from K0EHT-RON  on radio=0x34 is lat=39.208000 long=-94.679665
M: 2024-12-21 14:00:36.566 YSF, received network end of transmission, 4.8 seconds, 0% packet loss, BER: 0.0%
M: 2024-12-21 14:03:14.379 YSF, received network data from W1KN       to ALL        at BM3104
I: 2024-12-21 14:03:14.379 YSF, Lookup call W1KN returned id 3154382 -> 3154382
M: 2024-12-21 14:03:14.849 YSF, received network end of transmission, 0.9 seconds, 0% packet loss, BER: 0.0%
M: 2024-12-21 14:03:21.339 YSF, received network data from W1KN       to ALL        at BM3104
I: 2024-12-21 14:03:21.340 YSF, Lookup call W1KN returned id 3154382 -> 3154382
M: 2024-12-21 14:03:24.460 YSF, received network end of transmission, 3.7 seconds, 0% packet loss, BER: 0.0%
```
This is a representative sample of Analog log:

```
I: 2024-12-21 14:18:27.121 Begin TX: src=3119338 rpt=319639911 dst=9 slot=2 cc=0 call=N0EHQ
I: 2024-12-21 14:19:01.064 Begin TX: src=1234567 rpt=319639911 dst=9 slot=2 cc=0 call=N0OMX
I: 2024-12-21 14:19:47.769 Begin TX: src=3119338 rpt=319639911 dst=9 slot=2 cc=0 call=N0EHQ
I: 2024-12-21 14:20:16.067 Begin TX: src=1234567 rpt=319639911 dst=9 slot=2 cc=0 call=N0OMX
I: 2024-12-21 14:21:01.926 Begin TX: src=3119338 rpt=319639911 dst=9 slot=2 cc=0 call=N0EHQ
I: 2024-12-21 14:21:49.294 Begin TX: src=1234567 rpt=319639911 dst=9 slot=2 cc=0 call=N0OMX
I: 2024-12-21 14:22:01.050 Begin TX: src=3119338 rpt=319639911 dst=9 slot=2 cc=0 call=N0EHQ
I: 2024-12-21 14:22:27.360 Begin TX: src=1234567 rpt=319639911 dst=9 slot=2 cc=0 call=N0OMX
```
No errors “E:” in those files or YSFGateway (where the switch is tuned).

mkmer, 2025-01-03:
I have also noticed this error with an Analog_Bridge to DMR.  Any transmission > 3 seconds generates 1 log entry (1 time for each transmission on the DMR side)
Some folks have been chatting about it here: https://community.allstarlink.org/t/asl3-warning-from-chan-usrp-c/22493

Our configs: (Maybe they are helpful?)
[Analog_Bridge.ini.txt](https://github.com/user-attachments/files/18300952/Analog_Bridge.ini.txt)
[DVSwitch.ini.txt](https://github.com/user-attachments/files/18300953/DVSwitch.ini.txt)
[MMDVM_Bridge.ini.txt](https://github.com/user-attachments/files/18300954/MMDVM_Bridge.ini.txt)

Running on a x86 64 arch in our case:

```Linux Hub01 6.1.0-28-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.119-1 (2024-11-22) x86_64 GNU/Linux```

wspivak, 2025-01-03:
I looked at your ini files and made 2 adjustments, in MMDVM.ini to Jitter and in DVSwitch, hang time.  Otherwise they were essentially similar to mine.

We'll see if this fixes the issue.

Thanks and Happy New Year..

wspivak, 2025-01-03:
Still happening:

```
[2025-01-03 14:46:10.376] WARNING[1111]: chan_usrp.c:598 usrp_xwrite: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-03 14:46:51.236] WARNING[1111]: chan_usrp.c:598 usrp_xwrite: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
erliham*CLI>

```

mkmer, 2025-01-03:
It may be interesting (and/or odd) that your warning contains a bit more information than mine (includes line and function).  All be it the same "problem".  Maybe that console shows more info than the log file.
```
warning.log.1:[2024-12-28 18:04:51.785] WARNING[1142] chan_usrp.c: Channel usrp/127.0.0.1:34001:32001: Receive queue exceeds the threshold of 320
warning.log.1:[2024-12-28 18:05:27.944] WARNING[1142] chan_usrp.c: Channel usrp/127.0.0.1:34001:32001: Receive queue exceeds the threshold of 320
```

mkmer, 2025-01-03:
So the console does show different output..
```[2025-01-03 15:17:52.281] WARNING[657736]: chan_usrp.c:598 usrp_xwrite: Channel usrp/127.0.0.1:34001:32001: Receive queue exceeds the threshold of 320```

I guess not so interesting.

wspivak, 2025-01-04:
I added this line to CLI:

core set debug 7 chan_usrp.c

And only  saw rx notices (no body on the node did any transmitting) in debug.log

```
[2025-01-04 06:38:34.405] DEBUG[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: RX OFF
[2025-01-04 06:38:58.025] DEBUG[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: RX ON
[2025-01-04 06:38:58.086] DEBUG[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: RX OFF
[2025-01-04 06:38:58.525] DEBUG[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: RX ON
[2025-01-04 06:38:58.585] DEBUG[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: RX OFF
```

in messages.log:

```
[2025-01-04 06:02:05.886] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:02:31.306] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:02:39.923] NOTICE[201433] app_gps.c: sent packet(general): W2JJT-13>APRS,qAR,W2JJT-VS:=4046.38N/07328.55WrPHG04030/AllStar 62679, A node on the Emergency Radio Long Island net.
[2025-01-04 06:03:13.745] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:12:39.929] NOTICE[201433] app_gps.c: sent packet(general): W2JJT-13>APRS,qAR,W2JJT-VS:=4046.38N/07328.55WrPHG04030/AllStar 62679, A node on the Emergency Radio Long Island net.
[2025-01-04 06:13:29.605] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:15:21.366] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:15:47.226] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:18:13.125] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:20:12.625] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:20:30.185] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:20:50.425] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:22:39.936] NOTICE[201433] app_gps.c: sent packet(general): W2JJT-13>APRS,qAR,W2JJT-VS:=4046.38N/07328.55WrPHG04030/AllStar 62679, A node on the Emergency Radio Long Island net.
[2025-01-04 06:22:52.245] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:23:33.465] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:23:47.485] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:26:08.126] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:31:42.346] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320
[2025-01-04 06:32:39.942] NOTICE[201433] app_gps.c: sent packet(general): W2JJT-13>APRS,qAR,W2JJT-VS:=4046.38N/07328.55WrPHG04030/AllStar 62679, A node on the Emergency Radio Long Island net.
[2025-01-04 06:37:21.025] WARNING[201439] chan_usrp.c: Channel usrp/127.0.0.1:32001:31001: Receive queue exceeds the threshold of 320

```

PR #454 (2025-01-15): chan_usrp: Change queue overflow to debug message
This changes chan_usrp to report the queue overflow message as a debug message instead of a warning.  This message was not included in the original code.  It was added to aid in debugging future problems.

We are getting a burst of packets from analog_bridge that overflows the queue.  Analysis of this issue would require further knowledge of analog_bridge.  These packets do not cause a problem with the operation of the channel driver.

I have made it a debug level 1 message should there be a need to see this in the future.

The overflow message [... cut]
commit e6596b62 (2025-01-15): chan_usrp: Change queue overflow to debug message (#454)

This changes chan_usrp to report the queue overflow message as a debug
message instead of a warning.  This message was not included in the original
code.  It was added to aid in debugging future problems.

We are getting a burst of packets from analog_bridge that overflows
the queue.  Analysis of this issue would require further knowledge
of analog_bridge.  These packets do not cause a problem with the
operation of the channel driver.

I have made it a debug level 1 message should there be a need to see
this in the future.

The overflow [... cut]
closed 2025-01-15 by Allan-N