AllStarLink/app_rpt #441: Feature Request - Audio Compressor

mk2driver, 2024-12-31:
Hi all,

I appreciate the subject of audio compression in ASL will create some heated debate.  I fully understand the argument to ensure people fix their bad audio at source rather than trying to implement workarounds to make bad audio sound better but that said, in the world of vocal audio a certain level of compression, even on "good" audio can yield positive results.

My limited understanding of ASL / Asterisk is that the USB channel driver communicates with the sound card directly at a hardware level so using ALSA LADSPA plugins is not an option.  If that's the case, the only way of implementing any kind of dynamic range compression would have to be done in the simpleusb / usbradio channel drivers themselves.

Just throwing the idea out there really.  If enough people think this would be a useful feature, maybe it could be added to the road-map?

Regards.
Chris.

jxmx, 2025-01-03:
Marking as enhancement. Likely this will require someone with interest in the topic + the C skills to write it.

mk2driver, 2025-01-07:
I thought I'd have a go at this myself by cloning the app_rpt repo and then using the rpt_install.sh script to compile everything.  I could then edit the chan_usbradio.c file and recompile using the script however for some reason the chan_usbradio.so file created by the rpt_install.sh script is a much larger file compared to the one on a working system (626kB vs 146kB).  The working system was installed using the apt repo.

Both the development PC and the working ASL system are x86.

If I copy the 626kB chan_usbradio.so file over to the working ASL system and restart Asterisk, it just crashes during startup.

I'm obviously doing something stupid.  Any ideas?

Allan-N, 2025-01-07:
Have a look at the asl3-asterisk repo, the build-asl3 script, and the build-asl3.1.md doc.

mk2driver, 2025-01-07:
Thanks Allan.  I'll check that out!

mk2driver, 2025-01-07:
Hi Allan, that worked!  Thanks.

As a quick test I amended the "legacy audio scaling" code around line 2155 in chan_usbradio.c.  I removed the if condition and then reduced the clamping values right down to +/-1000.  This of course caused terrible clipped audio on the TX radio but it proved I was adjusting the correct audio data and proved the workflow.

The next job is to write some functional compression code.  Luckily the LADSPA compressor plugins are all written in C so I have a place to start.

jxmx, 2025-01-07:
Is this part of usbradio only or can it be used in both usbradio and simpleusb?

mk2driver, 2025-01-07:
I'm testing it in usbradio at the moment just because that's what I use on my repeaters.  Assuming I can get it working well enough there's no reason why it couldn't be added to simpleusb too.  The code that deals with the TX audio stream is basically the same in both drivers.

jxmx, 2025-01-07:
That's fine, my only point was make sure it ends up on the shared code/include between  both modules not in the usbradio module....

mk2driver, 2025-01-15:
Just a quick update on this.  I managed to implement a basic compressor in usbradio.so and whilst it demonstrates some improvement on stations with quiet audio, it also has the effect of amplifying background noise (obviously).  This wouldn't be an issue if the majority of users had fully-quiet audio but this is often not the case.

Mobile stations naturally have a certain level of road noise which the compressor happily amplifies.  Some base station operators seem happy to use a powered desk mike with the gain turned to max and then sit half way across the room from it.  All of the above causes poor S/N ratio which sadly, a compressor only highlights.

So I'm not going any further with this one guys.  I'll leave this thread open for a few days in case there's any other comments or suggestions and then I'll close it as "not planned".

closed 2025-01-20 by mk2driver