AllStarLink/app_rpt #447: Double frame free in rpt_app.c?

mkmer, 2025-01-09:
Something I noticed while digging into #422   - looks like it could be a mistake.

https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4247

We get a frame:
https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4312

Then free it multiple times AND still reference it (at least after the first free):
https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4377

https://github.com/AllStarLink/app_rpt/blob/5101b1bb1
338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4496

https://github.com/AllStarLink/app_rpt/blob/5101b1bb1338144ecd2ea00f7d4c7ffd18104a25/apps/app_rpt.c#L4500

It's most likely not OK to reference it after we free it (right?)
Is it "ok" to free the same frame up to 3 times?

mkmer, 2025-01-09:
OK - I missed the reassignment of f on the first example.  We are not referencing a freed frame... just freeing it twice in the last two examples.

Allan-N, 2025-01-09:
To my eyes, it looks like line 4496 should be removed.

mkmer, 2025-01-09:
I think a ```break;``` was dropped in the conversion here:
https://github.com/AllStarLink/app_rpt/commit/3834be01031a4098a14b2c36f99f99a26223f396#diff-e9c04babdfa10caa4961bbcc4d26b4d2b254adcd13d5c6d3a40dfa342f455653L4727 at line 4727

It may be fine to just remove the extra free as written, or just put the break back in...

InterLinked1, 2025-01-09:
Yes, you're right - the `break` was removed accidentally and should be restored, especially since that function locks the channel again so it would be locking twice if just the frfree were removed, as you also observed. I'll put that up in a PR tonight if nobody else has done it by then.

commit 96ad6dfc (2025-01-11): app_rpt: Readd missing break statement. (#448)

Restore a break statement that was accidentally removed during the refactor in
commit 3834be01031a4098a14b2c36f99f99a26223f396.

Fixes: #447
closed 2025-01-11 by InterLinked1