AllStarLink/app_rpt #456: ASL3 Debian12 app_rpt interrupted/distorted USB device transmit audio on current and recent kernel updates

MichiganBroadband, 2025-01-15:
Experiencing this problem on ALL newly installed Debian12 ASL3 systems. 
And any that have been updated recently (apt update apt upgrade).  
Is happening across three recent kernels I have updated to and tried: 
Linux ASL3 6.1.0-30-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.124-1 (2025-01-12) x86_64 GNU/Linux
previously tried 6.1.0.28, and 6.1.0.29 over the past couple of days.  
Easy to reproduce:  
From the Asterisk console issue a  rpt fun <nodenumber> *904
You will need to enable this command in rpt.conf  
---> 904 = cop,4                        ; Test tone on/off (toggle)

What is expected is a clean uninterrupted test tone from the USB audio device.  
What is actually happening since this problem has developed is distorted/interrupted "noisy" transmit audio from the USB device.  
Systems that have not been updated recently are not experiencing this problem.
Any system that is updated or freshly installed have this new problem.  
This problem is affecting all transmit audio and is especially noticeable when telemetry tones etc are being transmitted.
It varies in severity and pattern of what sounds like "software interference" or audio buffers not being serviced properly.  
This new problem is great in severity and appears to be across the board/global for Debian/12 on Intel PC hardware.  
We are seeing it on multiple installs/updated systems of different Intel CPU hardware, and more reports are starting to come in on the Community forum.  
I am ready to provide more details and testing as needed.   
What has been so far is being discussed on the Community Forum at:  
https://community.allstarlink.org/t/after-kernel-update-debian12-asl3-distortion-choppy-audio-on-simpleusb-tx-audio-especially-telemetry/22684

I have also posted a video where this can be seen and heard at: https://youtu.be/1NFvY0KV7-I

davidgsd, 2025-01-15:
I can confirm this issue is related to apt upgrade being done.  I built a node yesterday from a clonezilla image, did an apt upgrade and this issue started happening. I then reloaded the clonezilla image on the node (which is a few weeks old) but did not then do the apt update, and everything works perfectly. This is on a Beelink T5 miniPC with Deb12 and KDE.

InterLinked1, 2025-01-15:
> I can confirm this issue is related to apt upgrade being done. I built a node yesterday from a clonezilla image, did an apt upgrade and this issue started happening. I then reloaded the clonezilla image (which is a few weeks old) onto the node but did not then do the apt update, and everything works perfectly. This is on a Beelink T5 miniPC with Deb12 and KDE.

I think this is implicit, but just to confirm, with the *exact same* Asterisk source (including `app_rpt`), it works before the system upgrade but not after?

In other words, you only upgraded the kernel, but not Asterisk?

Allan-N, 2025-01-15:
@davidgsd Can you attach the output from `sudo asl-show-version` when exec'd from a non-upgraded system/node (i.e. load up a system from your clonezilla image and capture what [ASL3] package versions are installed) ?

Allan-N, 2025-01-16:
On community, W6NIJ shared the `asl-show-version` output from one of his nodes that was NOT experiencing the distortion.  I compared this to one of my own, recently updated, and nodes.  The difference between the two was :
```
4c4
< OS Kernel     : 6.1.0-28-amd64
---
> OS Kernel     : 6.1.0-30-amd64
21,22c21,23
<   dahdi-dkms                     1:3.4.0-4+asl
<   dahdi-linux                    1:3.4.0-4+asl
---
>   dahdi-dkms                     1:3.4.0-5+asl
>   dahdi-linux                    1:3.4.0-5+asl
>   dahdi-source                   1:3.4.0-5+asl

Note: my node is essentially HUB node (no USB adapters in use)

Allan-N, 2025-01-16:
And comparing W6NIJ "working" system to a newly built "not working" system the differences were :
```
4c4
< OS Kernel     : 6.1.0-28-amd64
---
> OS Kernel     : 6.1.0-30-amd64
```

davidgsd, 2025-01-16:
From a properly working node:

```
allscan@t5:~$ sudo asl-show-version
dpkg-query: no packages found matching cockpit*
********** AllStarLink [ASL] Version Info **********

OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.1.0-28-amd64

Asterisk      : 20.10.0+asl3-3.1.0-1.deb12
ASL [app_rpt] : 3.1.0

Installed ASL packages :

  Package                         Version
  ==============================  ==============================
  allmon3                         1.4.1-1.deb12
  asl3                            3.6.0-1.deb
  asl3-asterisk                   2:20.10.0+asl3-3.1.0-1.deb12
  asl3-asterisk-config            2:20.10.0+asl3-3.1.0-1.deb12
  asl3-asterisk-modules           2:20.10.0+asl3-3.1.0-1.deb12
  asl3-menu                       1.10-1.deb12
  asl3-tts                        1.0.1-1.deb12
  asl3-update-nodelist            1.5.1-1.deb12
  dahdi                           1:3.1.0-2
  dahdi-dkms                      1:3.4.0-4+asl
  dahdi-linux                     1:3.4.0-4+asl
```

This is the same node that had the issue after an apt upgrade. After going back to a previous disk image it works fine.

Allan-N, 2025-01-16:
@davidgsd Thank you for the info.  It looks like your update/upgrade would include a new kernel and new ASL3 packages.

MichiganBroadband, 2025-01-16:
If there's anything I can do on my end please let me know. 
Put me to work if I can do anything.  
I have a good high level understanding of how things work but I would really like to learn more.  
I'm not a programmer developer but I keep seeing that I think I need to dive into it and start learning new things.  
I'd like to better understand app_rpt and the audio hardware and how the audio/buffering and all that works.  
I don't thing there is any really good "block level diagram" how all the software works how interrupts are serviced and all that.  
I also have no idea how to revert to previous kernels other than Debian's default single previous kernel.  
I tried reverting to -25 kernel on one of our boxes and this resulted in the USB audio device not working at all and not being able to use it "no matter what I tried" and I had to put it back to -29.
There are obviously some kind of dependencies that do not let this work, and I can;t just go back to a previous kernel willy-nilly.  
-25 and -29 were my only choices in the grub boot menu (as detailed on the forum and no response yet).
I also realize this is not really the place for these type of questions.  
But just want to take this opportunity to put forth I want to learn and work on this stuff!  
ANd this problem seems like a great opportunity for me to start learning mroe and get into the areas of work I am actually very interested in..  lower level hardware stuff.   
-SteveG

jxmx, 2025-01-22:
Moving to ASL3 for tracking but this is NOT an ASL bug.