AllStarLink/app_rpt #475: CM119 not recognised after update from allstar3-arm64-3.0.12.img.xz

plantjes1, 2025-01-30:
Hi, I have several CM119 sound fobs in my repeater.
After updating my raspberry the fobs are not working anymore..

lsusb shows  here a connected cm119a and a cm119
Bus 001 Device 004: ID 0d8c:013a C-Media Electronics, Inc. USB PnP Sound Device
Bus 001 Device 003: ID 0d8c:000e C-Media Electronics, Inc. Audio Adapter (Planet UP-100, Genius G-Talk)

asl-find-sound shows
1-1.1:1.0       -->     0d8c:000e Generic USB Audio Device
1-1.4:1.0       -->     0d8c:013a C-Media Electronics Inc.

This was also the case in 3.0.12 but it still works fine.
in 3.0.13 and 3.0.16 is shows the same but does not work anymore..

Best regards
Benno

Allan-N, 2025-01-30:
In the `/etc/asterisk/simpleusb.conf` file, check the `devstr =`  strings for each of your nodes. Do they match?

Allan-N, 2025-04-27:
@plantjes1 Did you get your sound fobs working?

Allan-N, 2025-04-27:
Great!

closed 2025-04-27 by Allan-N