AllStarLink/app_rpt #483: status 15 command: (full system status all nodes) does not appear to do anything or work as expected

MichiganBroadband, 2025-02-04:
Raspberry Pi ASL3 image install fully up to date.  
Also have noticed this over the past few months that it has never worked.  

enabled in rpt.conf 

;;;;; Link Commands ;;;;;
; 15 - Full system status (all nodes)

715 = status,15

While other nodes are connected - expected that when sending command *715 It should report  Full system status (all nodes)
Or executing- 
Pi5*CLI> rpt fun <nodenumber> *715
Pi5*CLI>
"nothing happens"
And nothing shows in the console with verbosity turned up. 
asterisk -rvvvvvvvv

I have noticed this on other nodes I have tested it on running different hardware.  
This is also discussed on the community forum at: 
  https://community.allstarlink.org/t/status-15-full-system-status-all-nodes-does-not-appear-to-do-anything-or-work-as-expected/22844

I was asked to file a bug report here.  

 OS            : Debian GNU/Linux 12 (bookworm)
                       │ OS Kernel     : 6.6.74+rpt-rpi-2712
                       │
                       │ Asterisk      : 20.11.0+asl3-3.2.0-2.deb12
                       │ ASL [app_rpt] : 3.2.0
                       │
                       │ Installed ASL packages :
                       │
                       │   Package                         Version
                       │   ==============================  ==============================
                       │   allmon3                         1.4.2-1.deb12
                       │   asl3                            3.7.1-1.deb
                       │   asl3-asterisk                   2:20.11.0+asl3-3.2.0-2.deb12
                       │   asl3-asterisk-config            2:20.11.0+asl3-3.2.0-2.deb12
                       │   asl3-asterisk-modules           2:20.11.0+asl3-3.2.0-2.deb12
                       │   asl3-menu                       1.11-1.deb12

MichiganBroadband, 2025-02-04:
duplex = 1 in the case of the attached rpt.conf (had to rename to .txt

[rpt.conf.txt](https://github.com/user-attachments/files/18660981/rpt.conf.txt)

 to allow attaching file).
I also have another system that is duplex = 2  (actual repeater).  
And am getting the same results.  
Also tried with a very basic out of the box "nearly default" simplex node config.

Mason10198, 2025-02-04:
I see your problem. The command `status,15` does not exist. What you are looking for is `ilink,15`. See below for available commands reference.

---

# Status Commands

Status commands (functional class 'status') provide general information about the node.

## Sample:
```
712 = status,12   ; Give Time of Day (local only)
```

| Status | Description                        |
|--------|------------------------------------|
| 1      | Force ID (global)                 |
| 2      | Give Time of Day (global)         |
| 3      | Give software Version (global)    |
| 4      | Give GPS location info            |
| 5      | Speak the last (dtmf) user        |
| 11     | Force ID (local only)             |
| 12     | Give Time of Day (local only)     |

# Link Commands

Link commands (functional class 'link') connect to, disconnect from, monitor (Rx only) other nodes and give link status.

## Sample:
```
3 = ilink,3   ; Connect specified link -- transceive
```

| ilink | Description                                                       |
|-------|-------------------------------------------------------------------|
| 1     | Disconnect specified link                                         |
| 2     | Connect specified link -- monitor only                            |
| 3     | Connect specified link -- transceive                              |
| 4     | Enter command mode on specified link                              |
| 5     | System status                                                     |
| 6     | Disconnect all links                                              |
| 7     | Last Node to Key Up                                               |
| 8     | Connect specified link -- local monitor only                      |
| 9     | Send Text Message (9,<destnodeno or 0 (for all)>,Message Text, etc. |
| 10    | Disconnect all RANGER links (except permalinks)                   |
| 11    | Disconnect a previously permanently connected link                |
| 12    | Permanently connect specified link -- monitor only                |
| 13    | Permanently connect specified link -- transceive                   |
| 15    | Full system status (all nodes)                                    |
| 16    | Reconnect links disconnected with "disconnect all links"          |
| 17    | MDC test (for diag purposes)                                      |
| 18    | Permanently Connect specified link -- local monitor only          |

MichiganBroadband, 2025-02-04:
The problem is user error and I am very embarrassed!
Between the layout/spacing on my screen and the worked "status" in the description I made a quick assumption that it was a status command!  
Very sorry for wasting your time I should have looked at it a lot closer.  
And of course it works fine!  
Thank You!  URGH!

closed 2025-02-04 by Allan-N