AllStarLink/app_rpt #486: PTT drops when executing Playback or SayAlpha functions

davidgsd, 2025-02-12:
I have been seeing this issue on most nodes I build.  In the past several months I have built about 20 nodes, using several different MiniPC models (Dell Wyse 4040, Beelink T5, Nucbox 5), a wide range of USB interfaces, and a wide range of SimpleUSB and USBRadio configurations. This has occurred with many ASL3 versions as I have updated various nodes over time always staying up-to-date with the latest ASL at the time of each node build.

To reproduce:
Add following to rpt.conf under autopatch commands:
`69 = autopatchup,context=my-ip,noct=1,farenddisconnect=1,dialtime=9000,quiet=1`

Add following to extensions.conf:
```
[my-ip]
; *698: Playback Test
exten => 8,1,Wait(2)
exten => 8,n,SayAlpha(abcdefghijklmnopqrstuvwxyz0123456789)
exten => 8,n,Hangup
```

Now if you do a *698 DTMF command it will play back the above text. After approximately 15 seconds however, PTT drops for about 1 second.  When this happens the following message shows in the asterisk log:

`[2025-02-12 13:32:19.473] WARNING[1923] app_rpt/rpt_bridging.c: Cannot start tone on DAHDI/pseudo-117647281`

I have not been able to find any additional debug/log info about why PTT is dropping.  There is nothing unusual about my ASL configuration.

The following video shows the issue: https://youtu.be/1_tgKNkqCx4

Thanks for any help to further diagnose and fix this issue.

KB4MDD, 2025-02-12:
Can you set verbose to 7 and debugging to 7 and post the results?

davidgsd, 2025-02-12:
I set debug to 7 for every module I could think of (asterisk -vvvvvvvr, core set debug 7, core set debug 7 [susb, dahdi, dialplan, rpt, bridge]), and there are no messages shown when PTT goes off.  I see just the following normal messages: 

```
debian*CLI> rpt fun 571573 *698
[2025-02-12 15:17:16.921] WARNING[6197]: app_rpt/rpt_bridging.c:567 rpt_play_tone: Cannot start tone on DAHDI/pseudo-172961059
    -- Executing [8@my-ip:1] Wait("DAHDI/pseudo-2145247319", "2") in new stack
    -- Executing [8@my-ip:2] SayAlpha("DAHDI/pseudo-2145247319", "abcdefghijklmnopqrstuvwxyz0123456789") in new stack
    -- <DAHDI/pseudo-2145247319> Playing 'letters/a.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/b.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/c.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/d.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/e.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/f.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/g.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/h.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/i.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/j.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/k.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/l.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/m.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/n.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/o.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/p.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/q.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/r.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/s.ulaw' (language 'en')  **around here is where PTT drops**
    -- <DAHDI/pseudo-2145247319> Playing 'letters/t.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/u.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/v.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/w.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/x.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/y.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'letters/z.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'digits/0.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'digits/1.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'digits/2.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'digits/3.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'digits/4.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'digits/5.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'digits/6.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'digits/7.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'digits/8.ulaw' (language 'en')
    -- <DAHDI/pseudo-2145247319> Playing 'digits/9.ulaw' (language 'en')
    -- Executing [8@my-ip:3] Hangup("DAHDI/pseudo-2145247319", "") in new stack
  == Spawn extension (my-ip, 8, 3) exited non-zero on 'DAHDI/pseudo-2145247319'
    -- Hungup 'DAHDI/pseudo-2145247319'
[2025-02-12 15:17:45.502] WARNING[6197]: app_rpt/rpt_bridging.c:567 rpt_play_tone: Cannot start tone on DAHDI/pseudo-172961059
    -- Hungup 'DAHDI/pseudo-172961059'
debian*CLI>
```

KB4MDD, 2025-02-12:
```
core set verbose 7
core set debug 7
core set debug 7 chan_simpleusb.so
```

davidgsd, 2025-02-12:
Those commands did not result in any extra log messages shown. Same log output as before.

InterLinked1, 2025-02-13:
> ```
> core set verbose 7
> core set debug 7
> core set debug 7 chan_simpleusb.so
> ```

The last command is redundant here, core set debug 7 enables level 7 debug on *everything* with no filter.

> Those commands did not result in any extra log messages shown. Same log output as before.

You probably need to modify your `/etc/asterisk/logger.conf` to enable logging to the console. It may already be going to a log file, depending on the configuration.

davidgsd, 2025-02-13:
I set `console => notice,warning,error,dtmf,debug,verbose` in /etc/asterisk/logger.conf, restarted Asterisk, and there are still no other additional log messages shown in the console during the PTT drop.

mkmer, 2025-02-13:
Maybe easier to add something like this to the `logger.conf`
`debug.log => error,warning,notice,debug,verbose`
Then attach the `debug.log` file here.  (You still need to activate the debug/verbose levels in the CLI after reloading asterisk)
Don't forget to turn it back off, `debug.log` will get quite large.

mkmer, 2025-02-14:
Looks like VOX timeout is too short for the letters... or there is a bug in VOX (or both).
Are these default or is there something in the config:
```
[YourNode]
voxtimeout = 10000     ; VOX timeout time in ms
voxrecover = 2000      ; VOX recover time in ms
```
Maybe try adjusting these to "prove" it is VOX causing the PTT to turn off....

mkmer, 2025-02-14:
I guess I'm going to school here too:
https://community.allstarlink.org/t/autopatch-vox-settings-cant-find-them/19700/2
https://wiki.allstarlink.org/wiki/Setup_Autopatch
If I understand this correctly, looks like your in simplex mode and VOX max is 10 seconds. 

Need full duplex or change the values to keep TX up longer.

davidgsd, 2025-02-14:
Adding `voxtimeout = 20000` and `voxrecover = 4000` in rpt.conf under the node# stanza seems to fix the issue.  Seems that there is a bug with what autopatch is doing here though because this audio is from Asterisk itself, not a telephone call. But maybe app-rpt has no easy way to differentiate that.

Another question is how can DTMF commands be set up to trigger dialplan functions without using autopatch? I'm not trying to make a phone call here so using 'autopatch' seems like a hack. Is there a better / simpler / more direct way to have a DTMF command be able to execute dialplan functions eg, Wait(), AGI(), SayAlpha(), Playback()...?

mkmer, 2025-02-16:
> Another question is how can DTMF commands be set up to trigger dialplan functions without using autopatch? I'm not trying to make a phone call, so use of 'autopatch' seems like a hack. Is there a better / simpler / more direct way to have a DTMF command be able to execute dialplan functions eg, Wait(), AGI(), SayAlpha(), Playback()...?

That's a good question for the forums (see what other folks are doing).

closed 2025-02-16 by mkmer

mkmer, 2025-03-02:
@davidgsd I was looking at my radioless node and wonder why you are not configuring yours with `duplex=3`
I believe this is the "way" for radioless nodes plus it automatically fixes the 1/2 duplex patch issue.  I can't recall where I found the examples as I set mine up a long time ago.

davidgsd, 2025-03-02:
@mkmer yes most of my nodes are full-duplex w/duplex=3. I only recently started building & selling half-duplex nodes (such as the one in the above video which uses a Retevis RT85 HT and URI150).