AllStarLink/app_rpt #490: The perform_statpost function needs better explanation of errors

jxmx, 2025-02-14:
For example:

```
[2025-02-14 14:46:37.060] WARNING[1333]: app_rpt.c:1009 perform_statpost: statpost to URL ‘http://stats.allstarlink.org/uhandler?node=46287&time=1739508396&seqno=3&nodes=T3428425&apprptvers=3.2.0&apprptuptime=14&totalkerchunks=0&totalkeyups=0&totaltxtime=0&timeouts=0&totalexecdcommands=0’ failed with code 401
```

should say something like "failed with HTTP Code 401 Unauthorized: Wrong password?"

InterLinked1, 2025-02-14:
> For example:
>
> ```
> [2025-02-14 14:46:37.060] WARNING[1333]: app_rpt.c:1009 perform_statpost: statpost to URL [...]

It does say "failed with code 401" at the end.

jxmx, 2025-02-14:
Yeah, but 99% of users have no idea what "failed with code 401" means. We answer A LOT of questions about that in the user forum.

closed 2025-02-20 by Allan-N