AllStarLink/app_rpt #491: rpt_install.sh: Needs to be updated for oss low latency

KB4MDD, 2025-02-15:
rpt_install.sh needs to be updated with changes for  the oss lowlatency option.  Users that build from source needs this setting for proper operation.  It needs to accomodate for the difference on rpi5.

InterLinked1, 2025-02-15:
I don't know anything about this option, but is there any possible downside of enabling this on every system out there?

For example, I use `rpt_install.sh` to install `app_rpt` and friends on non-ASL systems for development. Is there any potential for something to go wrong there? I tried looking this option up but couldn't find documentation for it.

Allan-N, 2025-02-15:
You can read more about the low latency option @ [AllStarLink v3 Critial Software Update Released - 2025-02-01](https://community.allstarlink.org/t/allstarlink-v3-critial-software-update-released-2025-02-01/22838)

Also, while I see the need to set the option on affected systems I'm not sure `rpt_install.sh` is the right place for making the change.

KB4MDD, 2025-02-15:
I feel that we need something, if it is no more than an echo in the script and a note in the readme.

mkmer, 2025-04-14:
Once the fixes are in, we should not need the low latency option.

okcrum, 2025-08-02:
Verified @mkmer has merged both the above PRs; closing issue as no complaints have surfaced since May.

closed 2025-08-02 by okcrum