AllStarLink/app_rpt #495: app_rpt should compile with NO warnings/errors

Allan-N, 2025-02-17:
Looking at the build output for app_rpt shows a number of compilation warnings.  While some may argue that a few known/understood warnings are OK we run into problems when we are not looking for "new" warnings.  The simple solution is to banish ALL warnings from the project and then treat any warnings as errors.

InterLinked1, 2025-02-17:
> Looking at the build output for app_rpt shows a number of compilation warnings. While some may argue that a few known/understood warnings are OK we run into problems when we are not looking for [...]

What warnings are you seeing? The CI builds in dev mode and any warning should cause the build to fail.

Allan-N, 2025-02-17:
We just merged two fixes for warnings : https://github.com/AllStarLink/app_rpt/pull/494, https://github.com/AllStarLink/app_rpt/pull/471

On arm64, we have all of these :
```
#warning sys.io is not available on this architecture and some functionality will be disabled
```

and @mkmer noticed that in `rpt_install.sh` we have :
```
make -j$(nproc) apps
make -j$(nproc) channels
make -j$(nproc) res

# Compilation failed
if [ $? -ne 0 ]; then
        exit $?
fi
```
so the builds will only fail for `res` (and not `apps` or `channels`)