AllStarLink/app_rpt #503: CI takes too long / rebuilds too much.

mkmer, 2025-02-21:
Putting this one out there for discussion.   

There really is no reason to download and rebuild Asterisk EVERY time we do CI.
I did a bit of testing, and we can create a repo that builds Asterisk one time (and we can rebuild that image as we move to different Asterisk versions)
My test repo: 
   https://github.com/mkmer/Asterisk -> builds and publishes asterisk container (probably need a different name and should be in AllStarLink repos)

Then we modify the app_rpt repo `main.yaml` to this (or some variant):
```
jobs:
  container:
    runs-on: ubuntu-latest
    container:
      image: ghcr.io/mkmer/asl3-asterisk-ci

    steps:
      - name: Checkout
        uses: actions/checkout@v2
      - name: Build app_rpt
        run: |
          pwd
          ls -la ..
          ls -la
          cp -r ../app_rpt /usr/src
          ls -la /usr/src
          cd /usr/src/app_rpt
          ./rpt_install.sh
```
My test run reduced the run time from  6 minutes 11 seconds to 1 minute 15 seconds. This will save a lot of compute for each PR and push and still give us the same results.

InterLinked1, 2025-02-21:
> My test run reduced the run time from 6 minutes 11 seconds to 1 minute 15 seconds. This will save a lot of compute for each PR and push and still give us the same results.

Personally, I'd rather make the build robust and up to date than cache it for a little time savings. This is a public repo, so we aren't metered on CI minutes at all. A 5 minute build is actually pretty fast to me. The PhreakScript CI typically takes over half an hour, for comparison. The Asterisk CI is also longer than 5 minutes.

> we can create a repo that builds Asterisk one time

`app_rpt` is part of Asterisk, they aren't really separable in this context. Separating them introduces issues and complexity.

Additionally, although it's not present, at some point I would like to add the test suite to the CI. When we do that, there will be more stuff that we're leveraging in order to run the tests. Having the latest source for tests would also be important.

> Q? our asl3-asterisk repo is used to build our packages. Could we capture that image and use it for the app_rpt CI?

The builds need to use the development options, which are important. If the build options differ, they will fail to load. I'm not sure what the existing packages use but a separate set might be needed for those.

Additionally, even if this were a viable option, what would the mechanism be to keep such packages up to date? Would that add more work to somebody's plate?

Allan-N, 2025-02-21:
>The builds need to use the development options, which are important. If the build options differ, they will fail to load. I'm not sure what the existing packages use but a separate set might be needed for those.

This raises interesting questions.  What options might one want to use when building for development (CI)? for testing?  and for packaging/deployment?

>Additionally, even if this were a viable option, what would the mechanism be to keep such packages up to date? Would that add more work to somebody's plate?

I was thinking that we would keep the last asl3-asterisk release as the starting point for the CI builds.  Here, when we build a release we specify the Asterisk version (tag) and an app_rpt version (tag). The CI would keep the already download/compiled version of Asterisk, update the app_rpt content, with compile only what's needed.

Just something to think about.

InterLinked1, 2025-02-21:
> This raises interesting questions. What options might one want to use when building for development (CI)? for testing? and for packaging/deployment?

The packages I imagine are geared towards production usage. For example, they likely don't need any of the test-related stuff enabled, e.g. TEST_FRAMEWORK, test suite, etc.

For the CI we also want to build in developer mode and enable some of the developer options. User-facing packages don't need to be built in developer mode. I always build Asterisk with developer mode enabled, since I find I usually need that anyways, but the average user probably doesn't.

> I was thinking that we would keep the last asl3-asterisk release as the starting point for the CI builds. Here, when we build a release we specify the Asterisk version (tag) and an app_rpt version [...]

So when a new minor release of Asterisk comes out, the package would be outdated until a packager manually repackages a new asl3-asterisk package?

It might work, I'm not entirely sure what all the implications would be. But I'm pretty sure the existing packages would not be suitable. There would need to be a separate one with all the relevant test/development compiler options.

An alternate approach this reminded me of is there is a way to cache source compiles to some extent, rather than using packages. I think someone on the Asterisk dev team mentioned `ccache` for this purpose here. Only the parts that have changed then get rebuilt. I don't really have any experience myself with caching partial source builds like that, but thought I'd mention it.

Allan-N, 2025-02-21:
>The packages I imagine are geared towards production usage. For example, they likely don't need any of the test-related stuff enabled, e.g. TEST_FRAMEWORK, test suite, etc.

For the TEST_FRAMEWORK, test suite, etc ... is this just packaging?  or are we needing/using different compilation options?

>For the CI we also want to build in developer mode and enable some of the developer options. User-facing packages don't need to be built in developer mode. I always build Asterisk with developer mode enabled, since I find I usually need that anyways, but the average user probably doesn't.

I'm not convinced that we "need" to build the CI in developer mode.  It's not like we are (currently) using the CI bits for any actual testing.  Now, once we start having the CI actually performing some tests with the builds then, yes, we may enable certain options that make sense during testing.

And, yes, for those developing code we probably should be using the dev mode options.

InterLinked1, 2025-02-21:
> > The packages I imagine are geared towards production usage. For example, they likely don't need any of the test-related stuff enabled, e.g. TEST_FRAMEWORK, test suite, etc.
> 
> For the TEST_FRAMEWORK, test suite, etc ... is this just packaging? or are we needing/using different compilation options?

The test suite is an additional component entirely that needs to be compiled/installed (though this could probably be cached somehow)
TEST_FRAMEWORK is a compilation option as are most of the other options

> I'm not convinced that we "need" to build the CI in developer mode. It's not like we are (currently) using the CI bits for any actual testing. Now, once we start having the CI actually performing [...]

Building in developer mode makes the builds "stricter". Without building developer mode, a number of things that would cause build failures in dev mode would compile fine. This isn't even for testing, just for normal compilation of any changes.

I've seen this happen *many* times where some dev makes changes on their system, they "compile fine", and then they submit them in an Asterisk PR (which builds in dev mode) and it all comes crashing down. This is one reason I always build in dev mode myself, to adhere to the strictest standards.

I'm hoping we can start adding tests sooner rather than later so I think we should treat that as something that would happen in the near future, rather than set it up one way and have to change it later this year.

closed 2025-04-30 by mkmer