AllStarLink/app_rpt #508: chan_echolink may report APRS location incorrectly

wd5m, 2025-02-22:
EchoLink has a feature that will report the location of the node station via APRS.  Allstar includes this information in the echolink.conf settings.  There is an obscure, intermittent, bug in chan_echolink.c where the converted longitude or latitude value may be incorrectly formatted when sent to the status servers. Reviewing this issue with Jonathan K1RFD a while back, it appears the code always rounds numbers up when converting longitude and latitude to integers. Depending on the configured set value, this may result in the longitude/latitude string value length being longer than EchoLink and APRS accepts.  

For example, this status string below was logged on the EchoLink status server as an invalid value for the longitude. The longitude value has an extra digit. For a longitude setting of -98.6, the string "09835.100W" was sent, but the expected format is "09835.10W". 

> EL-  wd5m!3515.06NE09835.100W0PHG3330/145110/000=WX5FWD at 2043

If the EchoLink configuration setting is changed to say -98.601, the conversion and APRS string value sent does not include the extra digit. 

Reference chan_echolink.c starting about line 3503. Jonathan's initial suggestion was to remove the " +0.5" being added in the conversion. Someone else may have a better solution or conversion method.  
```
				latc = (mylat >= 0.0) ? 'N' : 'S';
				lonc = (mylon >= 0.0) ? 'E' : 'W';
				lata = fabs(mylat);
				lona = fabs(mylon);
				latb = (lata - floor(lata)) * 60;
				latd = (latb - floor(latb)) * 100 + 0.5;
				lonb = (lona - floor(lona)) * 60;
				lond = (lonb - floor(lonb)) * 100 + 0.5;
```
I briefly looked into the history of this code on the previous Allstar version.  I think this conversion method in Allstar may have come from Skip's theLinkbox, which has the same issue. Some history of changes to the floor function may also have affected this. I removed the "+0.5" in theLinkbox source code and it now works as expected.

David McAnally, WD5M

jxmx, 2025-02-22:
@wd5m - Wouldn't the proper longitude in APRS format for -98.6 be 09836.00W? I think this whole function is slightly off.

jxmx, 2025-02-22:
I'm working on a fix for this.

KB4MDD, 2025-02-22:
The adding of .5 also occurs in app_gps, if that is found a problem.

Allan-N, 2025-02-22:
Does the .5 rounding need to be added for positive locations and subtracted for negative locations?

jxmx, 2025-02-22:
That whole function/formula is wrong for APRS. Pull coming. It should be more like:

```
    mylat = 31.9;
    mylon = -98.6;

    /* lat conversion */
    latc = ( mylat >= 0 ) ? 'N' : 'S';
    latdeg = (int)fabs(mylat);
    latmin = ( fabs(mylat) - latdeg ) * 60.0;

    /* lon conversion */
    lonc = ( mylon >= 0 ) ? 'E' : 'W';
    londeg = (int)fabs(mylon);
    lonmin = ( fabs(mylon) - londeg ) * 60.0;

    printf("%02d%05.2f%c\n%03d%05.2f%c\n", latdeg, latmin, latc, londeg, lonmin, lonc);
``

jxmx, 2025-02-22:
> The adding of .5 also occurs in app_gps, if that is found a problem.

Yes. pull #509 fixes it too.

commit efd8132a (2025-02-22): chan_echolink: Replace function for conversion of decimal coordinates to APRS format

fixes #508

fix whitespace space/tab mixup

fix whitespace space/tap

app_gps: Fix functions for conversion of decimal coordinates to APRS format

fix styling to match asterisk conventions
closed 2025-02-22 by Allan-N