AllStarLink/app_rpt #515: Link Activity Timer not working

Allan-N, 2025-02-25:
From Community : [Lnkacttimerwarn in ASL3](https://community.allstarlink.org/t/lnkacttimerwarn-in-asl3/22951)

The activity timer logic in ASL3 does not appear to be working as expected.

Allan-N, 2025-02-25:
Minimally, it looks like the code to pickup the "lnkacttimerwarn" variable was not carried over from ASL2 to ASL3.

mkmer, 2025-02-27:
Confirmed from source that the macro does work, only the audio file is not working.  That should mean something is broken right here:
https://github.com/AllStarLink/app_rpt/blob/7b926425281203552b48840f9c5092af669263e9/apps/app_rpt.c#L2575-L2579
Maybe the file path is not accessible.  

"30seconds" is not an audio file we include with ASL (that I could find anyway) -> should we find it and include, or change the sample file?

Does it make sense that a link activity timer plays the "warning" on LOCALPLAY? 
https://github.com/AllStarLink/app_rpt/blob/7b926425281203552b48840f9c5092af669263e9/apps/app_rpt.c#L2573-L2586

I "think" this is supposed to operate on the connecting (local) node -> so yes, LOCALPLAY would make sense as the local node connected somewhere, there has been no activity for `lnkacttime`, then run a macro (which in theory is a disconnect, but could be anything).

During a bit of poking around, I noticed the timing seems to be a bit off.  With `lnkacttime` set to 60 seconds, it takes more than 60 seconds to trigger the macro.  Maybe https://github.com/AllStarLink/app_rpt/blob/7b926425281203552b48840f9c5092af669263e9/apps/app_rpt.c#L2545 is not running every second?

Allan-N, 2025-02-27:
>"30seconds" is not an audio file we include with ASL (that I could find anyway) -> should we find it and include, or change the sample file?

We have `rpt/act-timeout-warning` available but that file doesn't include the time remaining (30 seconds).  Can we concatenate that with `digits/30` and `seconds`?