AllStarLink/app_rpt #526: Crash when rxctcssfreqs / txctcssfreqs list counts do not match

Allan-N, 2025-03-02:
From [Community](https://community.allstarlink.org/t/asl3-asterisk-failing-to-start-code-dumped-status-11-segv/22965) :

Asterisk will crash if the size of the `rxctcssfreqs` and `txctcssfreqs` do not have the same number of tones.

InterLinked1, 2025-03-02:
Since this seems to be config-related, however we decide to fix it, this is a perfect candidate for adding a test for this as well.

mkmer, 2025-03-26:
This is probably not good:  
https://github.com/AllStarLink/app_rpt/blob/b542ccd42c7d106c824ac402345c0dda115c7420/channels/chan_usbradio.c#L4731-L4734
Only checking one size but looking at both....

mkmer, 2025-03-26:
Does it make sense to have 2 tx ctcss carriers?  Seems a bit odd

InterLinked1, 2025-03-26:
Looks like that entire function is also written wrong it seems - it's called from a CLI command, but instead of printing to that file descriptor, it uses printf, which is wrong.

closed 2025-04-22 by mkmer