AllStarLink/app_rpt #527: rpt_link.c: Attempt to calloc 0 bytes in rpt_update_links

InterLinked1, 2025-03-02:
This code will occasionally attempt to allocate 0 bytes: https://github.com/AllStarLink/app_rpt/blob/master/apps/app_rpt/rpt_link.c#L582

That, in turn, triggers a memory warning in Asterisk, which could cause it to crash:

```
WARNING[432247]: asterisk.asterisk:249 errReceived: Asterisk 127.0.0.1 received error: WARNING: High fence violation of 0x7ff61c029dc0 allocated at app_rpt/rpt_link.c rpt_update_links() line 582
```

I'm not sure if attempts to allocate 0 bytes are actually illegal and that's the direct cause of this, there may be another issue in play somewhere here. However, other allocations don't trigger this in testing, just those with an argument of 0.

mkmer, 2025-03-02:
That's a good find.  I'm surprised I didn't see this error on either of my machines. 
FWIW, I'm looking at moving this whole thing to `ast_str* ` - maybe by the April release.  This should simplify everything.

mkmer, 2025-03-05:
535 fixes this

closed 2025-03-05 by mkmer