AllStarLink/app_rpt #534: New config file template format breaks standard config parsers

SigmaScott, 2025-03-02:
```The new config file format with the template functionality breaks standard configuration parsers.

The extra information put in the ini style config file rpt.conf for some headers

[node-main](!)

Gets ignored when using python config parser to update the file and gets re-written as 
[node-main]

This happens on the child sections also.

[1990](node-main)

Becomes
[1990]

This behavior then breaks asterisk/rpt once the new file is loaded.

Would it be possible to maybe have the section heads be the standard [header] style and then put a include parameter in the child

[some-template]
value1=123
value2=456
[some-child]
inherit=some-template

Allan-N, 2025-03-02:
The template configuration support is provided by Asterisk and is unlikely to change.
 
 Asterisk : [Using Templates](https://docs.asterisk.org/Fundamentals/Asterisk-Configuration/Asterisk-Configuration-Files/Templates/Using-Templates)

What might be a better would be to find/extend a configuration parser that supports the Asterisk template format.

What "python config parser" are you using now?

Allan-N, 2025-03-02:
Also, have you looked at using Asterisk's AMI [GetConfig](https://docs.asterisk.org/Latest_API/API_Documentation/AMI_Actions/GetConfig/?h=getconfig), [UpdateConfig](https://docs.asterisk.org/Latest_API/API_Documentation/AMI_Actions/UpdateConfig/?h=updateconfig), etc?

InterLinked1, 2025-03-02:
Also worth adding, using templates in your configs is entirely optional, if you don't want to. But it certainly makes management much easier.

SigmaScott, 2025-03-03:
I am just using the config parser built into python.

https://docs.python.org/3/whatsnew/3.13.html#configparser

I realized I could expand the config and just go back the full style configs.

I will take look at the AMI stuff.

closed 2025-03-03 by SigmaScott