AllStarLink/app_rpt #536: app_rpt: Channel already being serviced when calling send_newkey/ast_sendtext

InterLinked1, 2025-03-03:
Assertion from a test execution yesterday:

I seem to recall seeing this before occasionally as well. Probably only happens under certain conditions.
This is unrelated to #252 

```
[2025-03-01 23:13:23.930] ERROR[387445][C-00000013] : Got 14 backtrace records
# 0: [0x55f367b531a6] asterisk utils.c:2835 __ast_assert_failed()
# 1: [0x55f3679d463a] asterisk utils.h:737 _ast_assert()
# 2: [0x55f3679e3502] asterisk channel.c:4786 ast_sendtext_data()
# 3: [0x55f3679e3b26] asterisk channel.c:4851 ast_sendtext()
# 4: [0x7fe4a722c394] app_rpt.so rpt_channel.c:473 send_newkey()
# 5: [0x7fe4a72242d3] app_rpt.so app_rpt.c:6822 rpt_exec()
# 6: [0x55f367a9d3aa] asterisk pbx_app.c:492 pbx_exec()
# 7: [0x55f367a86b79] asterisk pbx.c:2980 pbx_extension_helper()
# 8: [0x55f367a8aea9] asterisk pbx.c:4235 ast_spawn_extension()
# 9: [0x55f367a8bc96] asterisk pbx.c:4409 __ast_pbx_run()
#10: [0x55f367a8d733] asterisk pbx.c:4734 pbx_thread()
#11: [0x55f367b4ff48] asterisk utils.c:1607 dummy_start()
#12: [0x7fe4e70a81c4] libc.so.6 pthread_create.c:442 start_thread()
#13: [0x7fe4e712885c] libc.so.6 clone3.S:83 clone3()
```

mkmer, 2025-03-03:
I'm curious if this too can be caused by adding the link to the linked list, allowing waitfor_n or other link processing to begin a bit early... Have to keep an eye out now the code is changed a bit.
ref PR https://github.com/AllStarLink/app_rpt/pull/525

mkmer, 2025-03-05:
Could this be the result of calling `send_newkey()` at the same time (or at least overlapping in some part)?
We call it a couple of times when we receive a new call (from dialplan):
https://github.com/AllStarLink/app_rpt/blob/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4/apps/app_rpt.c#L6791-L6796
and here
https://github.com/AllStarLink/app_rpt/blob/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4/apps/app_rpt.c#L6801-L6804

Then we also send it when we receive and answer frame: (In the `process_link_channels()` function which is run periodically tied to the 20ms `waitfor_n` loops)
https://github.com/AllStarLink/app_rpt/blob/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4/apps/app_rpt.c#L4493-L4506
And again here:
https://github.com/AllStarLink/app_rpt/blob/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4/apps/app_rpt.c#L6198-L6204

InterLinked1, 2025-03-05:
> Could this be the result of calling `send_newkey()` at the same time (or at least overlapping in some part)? We call it a couple of times when we receive a new call (from dialplan):
> 
> [app_rpt/apps/app_rpt.c](https://github.com/AllStarLink/app_rpt/blob/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4/apps/app_rpt.c#L6791-L6796)
> 
> Lines 6791 to 6796 in [9db0ce2](/AllStarLink/app_rpt/commit/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4)
> 
>  	} else { 
>  		if (!phone_mode) { 
>  			send_newkey(chan); 
>  		} 
>  	} 
>  } 
> 
> and here
> [app_rpt/apps/app_rpt.c](https://github.com/AllStarLink/app_rpt/blob/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4/apps/app_rpt.c#L6801-L6804)
> 
> Lines 6801 to 6804 in [9db0ce2](/AllStarLink/app_rpt/commit/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4)
> 
>  doconpgm(myrpt, l->name); 
>  if ((!phone_mode) && (l->name[0] <= '9')) { 
>  	send_newkey(chan); 
>  } 
> Then we also send it when we receive and answer frame: (In the `process_link_channels()` function which is run periodically tied to the 20ms `waitfor_n` loops)
> 
> [app_rpt/apps/app_rpt.c](https://github.com/AllStarLink/app_rpt/blob/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4/apps/app_rpt.c#L4493-L4506)
> 
> Lines 4493 to 4506 in [9db0ce2](/AllStarLink/app_rpt/commit/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4)
> 
>  } else if (f->frametype == AST_FRAME_CONTROL) { 
>  	if (f->subclass.integer == AST_CONTROL_ANSWER) { 
>  		char lconnected = l->connected; 
>   
>  		__kickshort(myrpt); 
>  		myrpt->rxlingertimer = ((myrpt->iaxkey) ? RX_LINGER_TIME_IAXKEY : RX_LINGER_TIME); 
>  		l->connected = 1; 
>  		l->hasconnected = 1; 
>  		l->thisconnected = 1; 
>  		l->elaptime = -1; 
>  		if (!l->phonemode) { 
>  			rpt_mutex_lock(&myrpt->blocklock); 
>  			send_newkey(l->chan); 
>  			rpt_mutex_unlock(&myrpt->blocklock); 
> 
> And again here:
> [app_rpt/apps/app_rpt.c](https://github.com/AllStarLink/app_rpt/blob/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4/apps/app_rpt.c#L6198-L6204)
> 
> Lines 6198 to 6204 in [9db0ce2](/AllStarLink/app_rpt/commit/9db0ce2f3ca6a6edd73c1b313ec2ddfe3d62d8b4)
> 
>  static void answer_newkey_helper(struct rpt *myrpt, struct ast_channel *chan, int phone_mode) 
>  { 
>  	rpt_mutex_lock(&myrpt->blocklock); 
>  	if (ast_channel_state(chan) != AST_STATE_UP) { 
>  		ast_answer(chan); 
>  		if (!phone_mode) { 
>  			send_newkey(chan);

Possibly... unfortunately I don't have a full backtrace from that particular recurrence of it, but this issue is one I've seen here and there *many* times now so probably not difficult to reproduce. It only happened once I think though with that particular test.

mkmer, 2025-03-21:
Should we be locking the channel for ANY `ast_write()` to the same channel?
Few examples:
https://github.com/AllStarLink/app_rpt/blob/e554ae424e653f28ce468571faf05ab1e953f318/apps/app_rpt.c#L3045-L3048

https://github.com/AllStarLink/app_rpt/blob/e554ae424e653f28ce468571faf05ab1e953f318/apps/app_rpt.c#L4341-L4353

Link channels have both voice and text messages flowing async... sending status messages and such.

InterLinked1, 2025-03-21:
> Should we be locking the channel for ANY `ast_write()` to the same channel? Few examples:
> 
> [app_rpt/apps/app_rpt.c](https://github.com/AllStarLink/app_rpt/blob/e554ae424e653f28ce468571faf05ab1e953f318/apps/app_rpt.c#L3045-L3048)
> 
> Lines 3045 to 3048 in [e554ae4](/AllStarLink/app_rpt/commit/e554ae424e653f28ce468571faf05ab1e953f318)
> 
>  if (l->chan && l->thisconnected && !AST_LIST_EMPTY(&l->textq)) { 
>  	f = AST_LIST_REMOVE_HEAD(&l->textq, frame_list); 
>  	ast_write(l->chan, f); 
>  	ast_frfree(f); 
> [app_rpt/apps/app_rpt.c](https://github.com/AllStarLink/app_rpt/blob/e554ae424e653f28ce468571faf05ab1e953f318/apps/app_rpt.c#L4341-L4353)
> 
> Lines 4341 to 4353 in [e554ae4](/AllStarLink/app_rpt/commit/e554ae424e653f28ce468571faf05ab1e953f318)
> 
>  		f1 = rpt_frame_helper(myrpt, f, ismuted); 
>  		if (f1) { 
>  			ast_write(l->pchan, f1); 
>  			ast_frfree(f1); 
>  		} 
>  	} else { 
>  		/* if a voting rx link and not the winner, mute audio */ 
>  		ismuted = (myrpt->p.votertype == 1) && l->voterlink && (myrpt->voted_link != l); 
>  		if (!l->lastrx || ismuted) 
>  			RPT_MUTE_FRAME(f); 
>  		ast_write(l->pchan, f); 
>  	} 
>  } else if (f->frametype == AST_FRAME_DTMF_BEGIN) { 
> Link channels have both voice and text messages flowing async... sending status messages and such.
> 
> We lock the channel here when sending a text:
> 
> [app_rpt/apps/app_rpt/rpt_channel.c](https://github.com/AllStarLink/app_rpt/blob/e554ae424e653f28ce468571faf05ab1e953f318/apps/app_rpt/rpt_channel.c#L454-L463)
> 
> Lines 454 to 463 in [e554ae4](/AllStarLink/app_rpt/commit/e554ae424e653f28ce468571faf05ab1e953f318)
> 
>  void send_newkey(struct ast_channel *chan) 
>  { 
>  	ast_assert(chan != NULL); 
>  	/* app_sendtext locks the channel before calling ast_sendtext, 
>  	 * do this to prevent simultaneous channel servicing which can cause an assertion. */ 
>  	ast_channel_lock(chan); 
>  	if (ast_sendtext(chan, NEWKEY1STR)) { 
>  		ast_log(LOG_WARNING, "Failed to send text %s on %s\n", NEWKEY1STR, ast_channel_name(chan)); 
>  	} 
>  	ast_channel_unlock(chan); 
> So shouldn't we be checking locks at the other ast_write locations?

It's a bit weird, since Asterisk enables recursive locking (which is atypical).
`ast_write` will lock the channel as part of its deadlock avoidance code, so in theory, I don't think locking outside of that is explicitly necessary. Because recursive locking is enabled, it shouldn't hurt anything either though.

You may want to take a look at `ast_write_stream()`. At the top you can see the trylock. The `CHECK_BLOCKING` macro for certain frames is what triggers an assertion if there is simultaneous servicing.

mkmer, 2025-03-22:
It looks like we have text frames that need to go to multiple links (distribute to all links) -> We put those messages on a queue and "peal" them off one at a time here:
https://github.com/AllStarLink/app_rpt/blob/e554ae424e653f28ce468571faf05ab1e953f318/apps/app_rpt.c#L3045-L3048

Everything in the `rpt()` thread is (I think) single threaded so we should be able to write to channels without multiple access...
The `rpt_exec()` function IS a new thread, but `rpt()` shouldn't know about it until we add it to the linked list... 

I'm 99.99% sure this has been addressed in this commit [78845176f3c725fb4da5fc5a8d5ab1bf67dbc768](https://github.com/AllStarLink/app_rpt/commit/78845176f3c725fb4da5fc5a8d5ab1bf67dbc768) by moving `rpt_link_add()` to a location after we `send_newkey()`.

InterLinked1, 2025-03-22:
> It looks like we have text frames that need to go to multiple links (distribute to all links) -> We put those messages on a queue and "peal" them off one at a time here:
> 
> [app_rpt/apps/app_rpt.c](https://github.com/AllStarLink/app_rpt/blob/e554ae424e653f28ce468571faf05ab1e953f318/apps/app_rpt.c#L3045-L3048)
> 
> Lines 3045 to 3048 in [e554ae4](/AllStarLink/app_rpt/commit/e554ae424e653f28ce468571faf05ab1e953f318)
> 
>  if (l->chan && l->thisconnected && !AST_LIST_EMPTY(&l->textq)) { 
>  	f = AST_LIST_REMOVE_HEAD(&l->textq, frame_list); 
>  	ast_write(l->chan, f); 
>  	ast_frfree(f); 
> Everything in the `rpt()` thread is (I think) single threaded so we should be able to write to channels without multiple access... The `rpt_exec()` function IS a new thread, but `rpt()` shouldn't know about it until we add it to the linked list...
> 
> I'm 99.99% sure this has been addressed in this commit [78845176f3c725fb4da5fc5a8d5ab1bf67dbc768](https://github.com/AllStarLink/app_rpt/commit/78845176f3c725fb4da5fc5a8d5ab1bf67dbc768) by moving `rpt_link_add()` to a location after we `send_newkey()`.

Since you've done some analysis and it makes sense, I think we could try this out and see what happens. If assertions crop up again, we can look at them on a case by case basis.

The key thing is that everyone testing should make sure to build in developer mode with DO_CRASH so that if this happens it will cause a crash and draw attention to it.