AllStarLink/app_rpt #553: Add a new feature that behaves like the existing Link Activity timer

ttierney, 2025-03-09:
The Link Activity Timer functionality doesn’t provide the the feature set I would like to see. The timer starts when there has been a change to the connection state. This change could be establishing a new connection or performing a disconnection. If we set the lnkacttime to 180 (3 mins) the timer starts counting up after any change to the connection state. If there has been no local RF activity, the timer will reach 180 and then the macro defined by lnkactmacro will run. If there is local RF activity at say 120 seconds, the timer is reset to 0 and starts counting up again.

The existing functionality is useful and therefore should not be deprecated.

I would like to have a separate new feature created that is very similar to the existing Activity timer but instead of waiting for a local RF signal to reset the timer, any (local RF & internet) activity on the link causes the timer to be reset to 0. This would be useful to identify any periods of inactivity and then ‘do something’ by running a macro. So for example, if there has been no activity on the current connection, the macro could then perform a disconnect/reconnect to a busier/default connection. Or a repeater/gateway keeper might decide to leave the node in a disconnected state if there has been no activity for a given period on the current connection.

The existing Link Activity functionality may need to be renamed to Local RF Activity Timer.

localrfactenable=
localrfacttime=
localrfactmacro=
localrfacttimerwarn=

Then the new function could use the following naming standard. Internet Node Activity Timer.

intnodeactenable=
intnodeacttime=
intnodeactmacro=
intnodeacttimerwarn=

There maybe existing code that can be re-used to provide this new requirement. 

/usr/sbin/asterisk -rx 'core stop when convenient'

mkmer, 2025-03-12:
Wonder if we need both modes active at the same time, or could a "switch" be used to set how the timeout timer works:
lnkactmode = x
Maybe 3 modes for resetting the timer:
Mode 1 = local activity, 
Mode 2 = internet activity, 
Mode 3 = both local and internet activity

Allan-N, 2025-03-12:
Here's a thought :

Maybe we get out of the business of having to keep timers for every possible scenario and just have a set of events with simple triggers.  As an example, for a local repeater we needed to turn a cooling fan off/on based on when the repeater was keyed.  So, I added a set of events :
```
[events]
touch /tmp/RPT_TXKEYED = s|t|RPT_TXKEYED
rm -f /tmp/RPT_TXKEYED = s|f|RPT_TXKEYED
```
and then I used an external long-running script that looked for the presence (absence) of this file to control whether we turned the fan on/off ... including a delay to ensure the fan stayed running for 60s after last keyed.

ttierney, 2025-03-12:
I like the idea of the 3 modes. The original use case described for this functionality was that of a local mobile RF user who had moved the node to another connection but then went out of RF range before they could issue DTMF commands to return to the original (default) connection. I never really liked this example because it ignored any ongoing internet traffic on the connection and it just fired a macro. The macro presumably doing a disconnect and reconnect to elsewhere. A valid use case might be if a repeater had been disconnected from other nodes and was being used in local repeater mode for local RF stations. Then after X mins of no local activity a macro reconnects it to a default connection such as HubNet. Option 3 should sort this. Regarding using the /tmp directory to avoid separate timers. Would this involve physical writes to the SD card or would it be all in ram?

Allan-N, 2025-03-12:
>Regarding using the /tmp directory to avoid separate timers. Would this involve physical writes to the SD card or would it be all in ram?

That would depend on how your system has been setup.  Our Raspberry Pi "appliance" systems have the "/tmp" directory (and a few others) in ram.

InterLinked1, 2025-03-12:
> Maybe we get out of the business of having to keep timers for every possible scenario and just have a set of events with simple triggers. As an example, for a local repeater we needed to turn a cooling fan off/on based on when the repeater was keyed. So, I added a set of events :

I agree. But specifically, this seems like a good case for AMI. This is exactly what AMI events are for. The program logic for these events is then entirely elsewhere, and up to the user's imagination. We just need to make sure we are providing enough events to make all that possible.

Allan-N, 2025-03-12:
>I agree. But specifically, this seems like a good case for AMI.

I like this idea.  We just need to provide a few "well documented" examples that others could use / copy / modify / extend.

mkmer, 2025-03-13:
FWIW, we were using Allan's original case
```
[events]
touch /tmp/RPT_TXKEYED = s|t|RPT_TXKEYED
rm -f /tmp/RPT_TXKEYED = s|f|RPT_TXKEYED
```
And switch to using AMI with one of the multitude of Python AMI libraries.  I think there are good use cases for BOTH access methods - 
One example: having an "system" cronjob that "says something" on the repeater and we don't want it to talk on top of people.  Checking a file in the script is "easy".

InterLinked1, 2025-03-13:
> Are you implying that the Asterisk team is going to remove system calls from dialplans?

No. But using system interfaces from either dialplan or within a module (which is what we are talking about here) are generally frowned upon if there are better ways of doing things. So even though the current way "works", moving away from that would streamline things, reduce code complexity, and adopt more of the best practices