AllStarLink/app_rpt #561: Hub has audio burst digital sound at end of transmission

redcrosscommunications, 2025-03-15:
I have a cloud hub running on vultr and works nice with around 30-50 connections. about 75% of the time from no particular incoming node audio, there will be a digital sounding burst. This is new as we come from a old ham voip image on a PI to now runing ASL3.  Are there any settings i can change to chop off say 50ms of audio in the Sudo Hub setting?

jxmx, 2025-03-16:
This is the same as https://github.com/AllStarLink/app_rpt/issues/404

mkmer, 2025-03-16:
I'm not so sure this is the same as #404.  404 is discussing squelch bursts, this one sounds more like the noise Tim has been asking about.  There is an R2D2 sound on the end of some transmissions.

jxmx, 2025-03-16:
The sound might be different, but is all a manifestation of the same problem. We need a way to have a transmit delay so that when the squelch closes, it cuts off the audio stream before it broadcast whatever garbage noise is in between those two signals.

mkmer, 2025-03-16:
Tim is sure it is new behavior since ASL3 and ASL2 did not exhibit the end of transmission noise.  I believe the squelch tail and simpleusb code is exactly the same (aka no functionality was intentionally removed or changed), so I'm gonna say they are not the same issue (at least based on what Tim has been saying).

jxmx, 2025-03-16:
I've re-opened this for discussion but I still think this is all the same issue.

Regardless of what it sounds like - digital noise, "rubber band snapping", metal "twang" - it's likely all a manifestation of the same issue. This is a universal problem across all radio systems. There's always a timing gap between the end of an audio stream from a transmitter and the time the receiver detects the closure of squelch and/or loss of CTCSS. Even reverse tone doesn't completely solve the issue. This is almost always implemented by either a complex "tail close" process or by implementing an audio delay.

The audio delay is the easiest. You set delay where the audio is buffered for N ms. Then when the squelch detection is closed, you cut the audiostream right away. Once tuned, it chops off all "not audio" - digital hash, static, electronic noise, whatever - that occurs between the previous audio cut and the COS/CTCSS drop.

This didn't happen on the old systems likely because there's been massive improvements in the efficiency and code-cleanups. It's been an issue since early betas. However, there's currently no plans to make any major improvements to `chan_simpleusb` or `chan_usbradio`. As you know, the work is going into the forthcoming ASLA/PipeAudio-based channel that will closely mimic the behaviors of `chan_usbradio`.

I'll leave this open, but I still believe this is fixed by whatever is the long-term resolution to #404.

KB4MDD, 2025-03-16:
I hear this audio artifact on my production switchvox (asterisk) system when a caller hangs up the call. It is not consistent. It sounds identical to what i hear on occasion with asl when someone unkeys over a link.

It may be related to the newer version of asterisk. 

Some one needs to record the audio on both sides of a link using the simpleusb record option.

Danny

tsawyer, 2025-03-16:
> This is almost always implemented by either a complex "tail close" process or by implementing an audio delay.

There is also another possibility. I've heard the twang in the middle of a transmission as well. This leads me to think this could be an audio artifact created from loud or high energy audio bursts. A squelch tail could conceivably artifact the audio if it's hitting the audio too hard at times.

mkmer, 2025-03-16:
We recently found a memory leak around a mistake in the frame delay logic.  It essentially was 1 frame less than it used to be.
There is a good chance this will fix the noise sneaking in: https://github.com/AllStarLink/app_rpt/pull/538

mkmer, 2025-03-30:
This is most likely the solution:
PR# https://github.com/AllStarLink/app_rpt/pull/596