AllStarLink/app_rpt #564: HUB audio gate burst isue

redcrosscommunications, 2025-03-16:
Please don't delete this post. Its not the same as issue #404........ running ASL3 as a HUB has a new audio gate burst of noise about 75% of the time. Even a radio-less node that has rx audio delay set perfectly with no sound what so ever will have this issue at the end of transmission. For reference, the Peak Radio Association in Oregon has been running a hub for about 6 years using allstar hamvoip image. no issues. Changed to ASL3 to run our hub and that's when this began. All the updates and other older issues being fixed is why we changed and look forward to an update that might address this issue.

mkmer, 2025-03-16:
A few questions about the noise:
It sounds like you have determined it's "random" - are you sure it's not coming from specific node types? (Say radioless node, simplex nodes, HAMVoip, ANYTHING that might help us)?
Can you share your configuration files? Make sure to *** any passwords or other secrets before attaching them.

This is going to be  a difficult one to diagnose, any details that may help us reliably reproduce the problem will be valuable!

redcrosscommunications, 2025-03-16:
Yes, its completely random. radio less nodes, repeater nodes with RF connected. some ham voip and even a ASL3 running on a pi. It sounds like a R2D2 burst of something. Its like there is a issue with the SUDO packet repeater in the hub. it delays sometimes after a incoming packet stream ends. The sound is very quick. id say less then 50ms but its a loud burst.

tsawyer, 2025-03-16:
I want to follow this issue so I've added myself as an assignee, not that I can code or need to approve code. I might be able to do some testing as time allows.

tsawyer, 2025-03-16:
> Its like there is a issue with the SUDO packet repeater in the hub

@redcrosscommunications Please confirm that the only ASL3 in a QSO is the hub, that both ends are not ASL3.

redcrosscommunications, 2025-03-17:
Correct, ASL3 is the hub and ALL connections (about 30 connections) are hamvoip. 

Did I see a post that we need to add/change some code from mkmer?

tsawyer, 2025-03-17:
Here are some recordings which @KB4MDD has been requesting. 

- This is two nodes connected to each other. Both current ASL3. No hub involved.
- One node is a Quantar and RTCM running vultr.com host.
- The other node is a Hotspot Radios duplex usb on a RPi5.  
- The recordings without PL hum are the HST receiving.
- The recordings with PL hum are the Quantar receiving.
- All recordings are performed on the Quantar side with archivedir.

[twangRecord.tgz](https://github.com/user-attachments/files/19292900/twangRecord.tgz)

tsawyer, 2025-03-17:
@KB4MDD Here are some PCM files recorded on the RPi5 with the same setup above. 

[2506_pcm.tgz](https://github.com/user-attachments/files/19293467/2506_pcm.tgz)

redcrosscommunications, 2025-03-17:
How can we try this?

InterLinked1, 2025-03-18:
> How can we try this? I looked at # 538 but so much info its confusing. may post directions here, if so, I'll try it right away and report!

Assuming you have compiled Asterisk from source, you can download the patch into your source directory: `wget https://patch-diff.githubusercontent.com/raw/AllStarLink/app_rpt/pull/538.patch`

Then, just rerun `make install` and restart Asterisk.

Allan-N, 2025-03-18:
or hang tight for a couple of weeks until we release our next "beta"

redcrosscommunications, 2025-03-18:
make install command didn't run.  **no rule to make target install.  stop

redcrosscommunications, 2025-03-18:
Best to wait. Understood. Where do the Beta release notices originate?

mkmer, 2025-03-25:
@tsawyer What is the CT delay on the test files?
Any other timers set in the config would be helpful too.

tsawyer, 2025-03-30:
@mkmer Standard wait-times, if that's what you're asking.

```
[wait-times]
; This section allows wait times for telemetry events to be adjusted
; A section for wait times can be defined for every repeater
telemwait = 2000                    ; Time to wait before sending most telemetry
idwait = 500                        ; Time to wait before starting ID
unkeywait = 2000                    ; Time to wait after unkey before sending CT's and link telemetry
calltermwait = 2000                 ; Time to wait before announcing "call terminated"
```

mkmer, 2025-03-30:
This is possibly the answer:
https://github.com/AllStarLink/app_rpt/pull/596